wildmountainfarms/solarthing #2: Feature Request: Grafana Dashboard JSON

davidandrewcope, 2019-12-19:
I really like the dashboard you have for Grafana and would like to setup something similar. Could you share the Grafana json file for the dashboard you show in the screenshot?

retrodaredevil, 2019-12-23:
Sorry I was a little late getting back to you. (I just realized I wasn't "watching" this repo)

I just committed the json file: https://github.com/wildmountainfarms/solarthing/blob/master/solar/solar_grafana_example.json

That JSON file probably won't work as is with your set up. I change the retention policy based on how quick I want the data to update and how long I want it to stick around. I'm sure there's other stuff that might need to be tweaked as well.

Thanks for taking an interest!

davidandrewcope, 2019-12-23:
Thanks, I appreciate that. It should be a good starting point. Also, thanks for this project! I hope to have this running soon, and will let you know if I have any issues.

closed 2019-12-23 by davidandrewcope

retrodaredevil, 2019-12-30:
Hey I was wondering if you got Grafana working. If you are using the SolarThing program I'd be interested to know if you got that set up easily. For the past month I've been slowly adding documentation on how to get this set up so more people can use it in the future. 

Any feedback for things to improve on is welcome!

Also note that some of the graphs in the Grafana JSON are "stacked". Hopefully you've realized that if you are using a significant chunk of that.

CapnJackOff, 2020-12-18:
Here's my dashboard in case anyone wants it. 

[image]

And the json:
`{
  "annotations": {
    "list": [
      {
        "builtIn": 1,
        "datasource": "-- Grafana --",
        "enable": true,
        "hide": true,
        "iconColor": "rgba(0, 211, 255, 1)",
        "name": "Annotations & Alerts",
        "type": "dashboard"
      }
    ]
  },
  "editable": true,
  "gnetId": null,
  "graphTooltip": 0,
  "id": 2,
  "links": [],
  "panels": [
    {
      "cacheTimeout": null,
      "colorBackground": false,
      "colorValue": false,
      "colors": [
        "#299c46",
        "rgba(237, 129, 40, 0.89)",
        "#d44a3a"
      ],
      "datasource": "InfluxDB",
      "format": "none",
      "gauge": {
        "maxValue": 100,
        "minValue": 0,
        "show": false,
        "thresholdLabels": false,
        "thresholdMarkers": true
      },
      "gridPos": {
        "h": 4,
        "w": 3,
        "x": 0,
        "y": 0
      },
      "id": 22,
      "interval": null,
      "links": [],
      "mappingType": 1,
      "mappingTypes": [
        {
          "name": "value to text",
          "value": 1
        },
        {
          "name": "range to text",
          "value": 2
        }
      ],
      "maxDataPoints": 100,
      "nullPointMode": "connected",
      "nullText": null,
      "options": {},
      "pluginVersion": "6.6.1",
      "postfix": "",
      "postfixFontSize": "50%",
[... 945 lines cut ...]
    "list": []
  },
  "time": {
    "from": "now-12h",
    "to": "now"
  },
  "timepicker": {
    "refresh_intervals": [
      "5s",
      "10s",
      "30s",
      "1m",
      "5m",
      "15m",
      "30m",
      "1h",
      "2h",
      "1d"
    ]
  },
  "timezone": "browser",
  "title": "MySolar",
  "uid": "",
  "version": 31
}`