wildmountainfarms/solarthing #10: Battery Type change to 'User' on Rover caused Solarthing to fail

damienheiser, 2020-06-05:
I am using a Renogy Rover 60A Li.  Solarthing was working fantastic while I had the Rover setup with the Battery type as SLA.

I changed the battery type to USR yesterday, and found that Solarthing started to fail at that point.  The log also reveals some sort of Immutable Battery Type. 

This might be due to the Rover 60A Li being a little newer and having another enumeration type for the battery within the specification, since it also handles Lithium Ion, which wasn't on the original 20A and 40A.

I'm going to to my best to dig through the code to see if I can provide a pull request, however my Java skills aren't really there.  

Thank you very much for an incredible project, I'd love to buy you a beer or EANAB (equally appealing non-alcoholic beverage)

config/base.json:
`{
  "type": "rover",
  "source": "default",
  "analytics_enabled": false,
  "fragment": 2,
  "unique": 30,
  "databases": [
    "config/influxdb.json"
  ],
  "io": "config/default_linux_serial.json",
  "modbus": 1,
  "bulk_request": true,
  "extra_option_flags": [ "rpi_cpu_temp" ]
}`

Log:
`Using base config = config/base.json
2020-06-05 07:06:01.703 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main
[stdout] Beginning main
2020-06-05 07:06:02.328 [main] INFO  me.retrodaredevil.solarthing.program.RoverMain - Beginning rover program
2020-06-05 07:06:02.616 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is disabled
2020-06-05 07:06:03.522 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57378 didn't have a cached value.
2020-06-05 07:06:03.605 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57379 didn't have a cached value.
2020-06-05 07:06:03.688 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57380 didn't have a cached value.
2020-06-05 07:06:03.771 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57381 didn't have a cached value.
2020-06-05 07:06:03.858 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57382 didn't have a cached value.
2020-06-05 07:06:03.946 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57383 didn't have a cached value.
2020-06-05 07:06:04.030 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57384 didn't have a cached value.
2020-06-05 07:06:04.110 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57385 didn't have a cached value.
2020-06-05 07:06:04.194 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57386 didn't have a cached value.
2020-06-05 07:06:04.274 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57387 didn't have a cached value.
2020-06-05 07:06:04.358 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57388 didn't have a cached value.
2020-06-05 07:06:04.443 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57389 didn't have a cached value.
2020-06-05 07:06:04.563 [main] DEBUG me.retrodaredevil.solarthing.program.RoverMain - Debugging special power control values: (Will debug all packets later)
charging mode controlled by voltage: false
	special power control: disabled
	each night on: disabled
	no charging below 0C: enabled
	charging method: PWM
intelligent power: disabled
	each night on: disabled
	battery type: lead-acid
	charging method: PWM
	no charging below 0C: disabled
	system voltage: 12V
2020-06-05 07:06:04.568 [main] DEBUG me.retrodaredevil.solarthing.program.RoverMain - took 1665ms to read from Rover
2020-06-05 07:06:04.688 [main] DEBUG me.retrodaredevil.solarthing.program.RoverMain - Debugging all packets
2020-06-05 07:06:05.121 [main] DEBUG me.retrodaredevil.solarthing.program.RoverMain - Never mind about that...
com.fasterxml.jackson.databind.JsonMappingException: valueCode: 0 not found in enum: class me.retrodaredevil.solarthing.solar.renogy.BatteryType (through reference chain: java.util.ArrayList[0]->me.retrodaredevil.solarthing.solar.renogy.rover.ImmutableRoverStatusPacket["batteryTypeName"])
	at com.fasterxml.jackson.databind.JsonMappingException.wrapWithPath(JsonMappingException.java:397) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.JsonMappingException.wrapWithPath(JsonMappingException.java:356) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.std.StdSerializer.wrapAndThrow(StdSerializer.java:316) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.std.BeanSerializerBase.serializeFields(BeanSerializerBase.java:731) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.BeanSerializer.serialize(BeanSerializer.java:166) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.impl.IndexedListSerializer.serializeContents(IndexedListSerializer.java:119) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.impl.IndexedListSerializer.serialize(IndexedListSerializer.java:79) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.impl.IndexedListSerializer.serialize(IndexedListSerializer.java:18) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.DefaultSerializerProvider._serialize(DefaultSerializerProvider.java:480) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.DefaultSerializerProvider.serializeValue(DefaultSerializerProvider.java:319) ~[solarthing-2020.2.1.jar:?]
[... 49 lines cut ...]
	at com.fasterxml.jackson.databind.ser.std.BeanSerializerBase.serializeWithType(BeanSerializerBase.java:608) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.impl.TypeWrappedSerializer.serialize(TypeWrappedSerializer.java:32) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.DefaultSerializerProvider._serialize(DefaultSerializerProvider.java:480) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.DefaultSerializerProvider.serializeValue(DefaultSerializerProvider.java:319) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper.writeValue(ObjectMapper.java:3024) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper.valueToTree(ObjectMapper.java:3193) ~[solarthing-2020.2.1.jar:?]
	... 12 more
Caused by: java.util.NoSuchElementException: valueCode: 0 not found in enum: class me.retrodaredevil.solarthing.solar.renogy.BatteryType
	at me.retrodaredevil.solarthing.packets.Modes.getActiveMode(Modes.java:56) ~[solarthing-2020.2.1.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.RoverReadTable.getBatteryType(RoverReadTable.java:344) ~[solarthing-2020.2.1.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.RoverReadTable.getBatteryTypeName(RoverReadTable.java:346) ~[solarthing-2020.2.1.jar:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at com.fasterxml.jackson.databind.ser.BeanPropertyWriter.serializeAsField(BeanPropertyWriter.java:689) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.std.BeanSerializerBase.serializeFields(BeanSerializerBase.java:723) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.std.BeanSerializerBase.serializeWithType(BeanSerializerBase.java:608) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.impl.TypeWrappedSerializer.serialize(TypeWrappedSerializer.java:32) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.DefaultSerializerProvider._serialize(DefaultSerializerProvider.java:480) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ser.DefaultSerializerProvider.serializeValue(DefaultSerializerProvider.java:319) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper.writeValue(ObjectMapper.java:3024) ~[solarthing-2020.2.1.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper.valueToTree(ObjectMapper.java:3193) ~[solarthing-2020.2.1.jar:?]
	... 12 more
`

retrodaredevil, 2020-06-05:
Thank you for reporting this. I believe I have now fixed it. It's weird that the documentation doesn't say 0 represents the user battery type. I thought 5 represented that type.

> Thank you very much for an incredible project, I'd love to buy you a beer or EANAB (equally appealing non-alcoholic beverage)

That means a lot. Thanks. I'll consider setting up a way to receive donations. I'm not old enough to drink yet, so an EANAB sounds good. Could go for some lemonade right about now.

To update run a git pull and then run `program/download_solarthing.sh`.

If this fixes your issue, feel free to close this issue.

retrodaredevil, 2020-06-09:
I'm closing this for now. If you come across this error again, feel free to re-open this.

Thanks for using SolarThing!

closed 2020-06-09 by retrodaredevil

retrodaredevil, 2020-06-12:
So apparently this isn't just a thing for newer rover models. I tried manually changing the battery type on my rover itself, and then the battery type was 0. So the "user" battery type actually corresponds to 0 and 5. 0 is basically unlocked, and 5 is locked.

There's some more info about that here: https://support.renogy.com/en/support/discussions/topics/35000016803