wildmountainfarms/solarthing #11: Modbus exception

flammagenitus, 2020-11-20:
Hello,
I've got a Renogy Rover Li and an RS-232 to USB cable. I'm running Ubuntu 20.04 and connected the Rover to a USB port. Running `dmesg` shows that the FTDI serial device converter is attached to ttyUSB0. My base.json file contains 
```
{
  "type": "rover",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "databases": [
    "config/influxdb.json"
  ],
  "io": "config/default_linux_serial.json",
  "modbus": 1
}
```
and config/default_linux_serial.json contains
```
{
  "type": "serial",
  "port": "/dev/ttyUSB0",
  "serial_config": null
}
```

However when I execute run.sh I see 
2020-11-19 21:05:40.845 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception

I'm sure this isn't a problem with solarthing but I'd really like to get this working- it looks like it would do exactly what I'd like to track my solar data. Thanks for any ideas you might have!

retrodaredevil, 2020-11-20:
When the `Modbus exception` line is printed, there should be a bunch of gibberish text below that that is also part of the error message. Could you give me the rest of the error message?

flammagenitus, 2020-11-20:
Here's the rest of the terminal output:
```
2020-11-19 22:16:35.050 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.ModbusTimeoutException: Timed out! startTime=1605845793047 currentTime=1605845795049 initialTimeout=2000
	at me.retrodaredevil.io.modbus.RtuDataEncoder.readBytes(RtuDataEncoder.java:113) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.io.modbus.RtuDataEncoder.readMessage(RtuDataEncoder.java:61) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.io.modbus.IOModbusSlaveBus.sendRequestMessage(IOModbusSlaveBus.java:26) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.io.modbus.ImmutableAddressModbusSlave.sendRequestMessage(ImmutableAddressModbusSlave.java:14) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:24) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:28) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheRangeInclusive(ModbusCacheSlave.java:36) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$3(RoverMain.java:89) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRoverProgram$6(RoverMain.java:147) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:40) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:112) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:83) ~[solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:153) [solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:54) [solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:112) [solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:198) [solarthing-2020.4.3.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:218) [solarthing-2020.4.3.jar:?]
2020-11-19 22:16:35.057 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 2987ms
```
I've got both the Bluetooth BT-1 and BT-2 modules and I plugged one in but I've never tried it past that. Once I saw that the Renogy app required precise GPS location and other privacy-infringing information I uninstalled the app. 

Unfortunately the cable I have is just a cheap Chinese one from Amazon ([https://www.amazon.com/DSD-TECH-SH-RJ12A-RS232-Serial/dp/B07R2L22GW]).

retrodaredevil, 2020-11-20:
So since you plugged in the BT modules, I don't think SolarThing will work with your Renogy Rover, or any other SolarThing alternative such as solarshed for that matter. You can see some details about someone who had the same problem here: https://support.renogy.com/en/support/discussions/topics/35000015051/page/3?url_locale=en

I wish I knew how to help, but I don't know how to reverse whatever the BT modules do to the Rover. I don't even know what the BT modules do to the Rover. Maybe update the firmware to not work with custom solutions? I'm not sure.

That error only happens for two reasons: What I just mentioned above, and if the cable isn't wired correctly. Since you plugged in the BT module, I can only assume the cable is wired correctly and the problem is with something the BT module did.

flammagenitus, 2020-11-20:
OK, thanks for all the help. That's very disappointing about the BT modules. I was hoping to buy a Rover Elite originally but it was sold out at the time. If I can't get SolarThing working on my Rover Li with a custom cable then I might just have to buy an Elite so I have a clean slate.

flammagenitus, 2020-11-20:
I found my old DB9/RS232-USB cable and ordered a 6-pin RJ12 cable and DB9 breakout box like you used. I should be able to test this out next week hopefully.

retrodaredevil, 2020-11-20:
Awesome. Also, you mentioned earlier that Renogy's app requires precise GPS location. I really hate where Android is going with this, but I doubt the Renogy app actually uses GPS. That permission is required to use bluetooth scanning, which is likely what it is for. If you want a technical source on this, look here: https://developer.android.com/guide/topics/connectivity/bluetooth#Permissions

It's really dumb, because there's no in between permission for this kind of thing. Or there was, but then depending on what version you target in your code you have to use a different permission. I'm not sure. It's very dumb. You have to do the same thing if you want to scan for WiFi networks or even get the current network the device is connected to.

I'm gonna close this now, but feel free to reopen it or create a separate issue once you get a chance to test this out again.

closed 2020-11-20 by retrodaredevil

flammagenitus, 2020-12-02:
I just received my DB9 breakout and an RJ12 cable this evening. I wired it up according to your diagram (thanks for documenting everything so well) and it appears to be working. I'll need to finish getting everything set up with a database to confirm but so far so good I think.
Re. Android: I wish there was a truly open-source alternative to Android. I'm mostly Google-free now but I can't avoid it on my phone.