wildmountainfarms/solarthing #14: Changing reporting frequency?

CapnJackOff, 2020-12-21:
Hello,
is there an option I can put in base.json to change the reporting frequency? It seems to be set to around 5s now and I would like to change it to something like 30s or 60s.

retrodaredevil, 2020-12-21:
You can define the `period` field to be something other than 5 seconds in the base.json, or you can set each database's throttle_factor for packet_upload to be something other than 1. If the throttle factor is 6, it will upload every 6th time, which for a 5 second period would be 30 seconds.

CapnJackOff, 2020-12-21:
Well the throttle factor worked well since it was already in the post db file,
but do I need to add the whole thing with settings key in the influxdb file?
```
{
  "type": "influxdb",
  "config": {
    "url": "http://localhost:8086",
    "username": "root",
    "password": "root",
    "database": "default_database",
    "measurement": null,
"settings": {
    "packet_upload": {
      "throttle_factor": 6,
      "initial_skip": 0
    },
    "status_retention_policies": [
      {
        "frequency": 29,
        "name": "medium_duration_retention",
        "duration": "4w",
        "auto_alter": true,
        "set_as_default": true
      },
      {
        "frequency": 7,
        "name": "normal_forever_retention",
        "duration": "INF",
        "replication": null,
        "shard_duration": null,
        "set_as_default": false,
        "auto_alter": false,
        "ignore_unsuccessful_create": false
      },
      {
        "frequency": null,
        "name": "normal_short_retention",
        "duration": "1h"
      }
    ],

    "event_retention_policy": {
      "name": "normal_forever_retention"
    },

    "retry_on_connection_failure": true,
    "call_timeout": 0.0,
    "connection_timeout": 10.0,
    "read_timeout": 10.0,
    "write_timeout": 10.0,
    "ping_interval": 0.0
  }
}
```

retrodaredevil, 2020-12-21:
You don't have to add the entire settings field, just the part with `packet_upload`. You can add the other stuff if you want to change that stuff too.

CapnJackOff, 2020-12-22:
I have no idea where that config file above for influx came from, but I did find out the surest way to make Solarthing go quiet is to mess it up, which I did! I went back to the simpler version. As far as I can tell, this does not work either.

```
{
  "type": "influxdb",

"packet_upload": {
      "throttle_factor": 6,
      "initial_skip": 0
    },

  "config": {
    "url": "http://localhost:8086",
    "username": "root",
    "password": "root",
    "database": "default_database",
    "measurement": null,

    "status_retention_policies": [
      {
        "frequency": 120,
        "name": "autogen"
      }
    ],

    "event_retention_policy": {
      "name": "autogen"
    }
  }
}
```

retrodaredevil, 2020-12-22:
If you want to use the simple version, it looks like this:

```
{
  "type": "influxdb",
  "settings": {
    "packet_upload": {
      "throttle_factor": 6,
      "initial_skip": 0
    },
  },
  "config": {
    "url": "http://localhost:8086",
    "username": "root",
    "password": "root",
    "database": "default_database",
    "measurement": null,

    "status_retention_policies": [
      {
        "frequency": 120,
        "name": "autogen"
      }
    ],

    "event_retention_policy": {
      "name": "autogen"
    }
  }
}
```

Sorry if I confused you in my last post. You do need to add the settings part, it's just I thought you were referring to some more advanced configuration, which is also possible in the `settings` field.

closed 2020-12-22 by CapnJackOff