wildmountainfarms/solarthing #16: Throwing Modbus Error with Rover Elite

sabarnes3, 2021-01-04:
Hi Josh--I have installed SolarThing on my Rover Elite with a Raspberry Pi. I had no issues getting set up and accessing real time data using your Rover-Setup program and base.json. I moved it over to the standard Rover base.json, connected to CouchDB. I'm now throwing a repeating ModBus exception, and wanted to share it with you and see if you have any thoughts, or if I am just missing something small in a config file.

Thanks,

Sam

`2021-01-03 23:54:09.214 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57378 didn't have a cached value.
2021-01-03 23:54:09.273 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.handling.FunctionCodeException: Expected to get 3 as the function code, but actually got 131
	at me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:62) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:13) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead.get(RoverModbusSlaveRead.java:40) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead.oneRegister(RoverModbusSlaveRead.java:43) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead.getWorkingHoursRaw(RoverModbusSlaveRead.java:377) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.RoverReadTable.getSensingBundle(RoverReadTable.java:472) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.RoverStatusPackets.createFromReadTable(RoverStatusPackets.java:46) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:44) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:112) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:83) ~[solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:153) [solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:54) [solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:112) [solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:195) [solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:253) [solarthing-2020.5.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:257) [solarthing-2020.5.1.jar:?]
2021-01-03 23:54:09.278 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4487ms
`

retrodaredevil, 2021-01-04:
That's a weird error, and I don't think it's your fault through any configuration. I believe it's an inconsistency with the Rover Elite versus other Rover models.

I'll do my best to try to create a fix for this and I'll probably end up having you compile SolarThing yourself to try out a new version, or I can compile it for you if you don't think you can do that. It might take some trial and error, but I think after I give you a new version to try a couple of times we can get rid of that error.

I'll probably have a new version for you to try sometime between 20 minutes from now and the next 24 hours. I'm not really sure.

retrodaredevil, 2021-01-04:
So I think the reason this was happening is because the Rover Elite doesn't have a load on it. Thank you for reporting this error. I'm surprised it wasn't reported sooner.

Now, I made a change that I think will fix it. I'd like you to run a `git pull` then compile it by running `./compile_and_move.sh`. Once you do that, the next time you run it will have the new version with the unreleased changes.

I think my changes will fix your problem, but if they don't, just give me the new error. 

Also, I added a debug message that should be printed out when you run SolarThing. It should look like `Error code exception message: ...`. If you could give me that message, that would be helpful to me.

Also if possible, could you copy paste your successful packet that's also debugged. It should look like: 
```
{
            "packetType": "RENOGY_ROVER_STATUS",
......... a bunch of stuff ........

            "loadWorkingModeName": "MANUAL
}
```
Note the only sensitive information in that packet is the `productSerialNumber`, so you may want to remove that.

I just want to better understand the data the Rover Elite gives so anything you can give me once you get it working successfully would be helpful to me.

sabarnes3, 2021-01-04:
Thank you very much for the quick response. I had no issues pulling and compiling/running the new code. 

I am still getting a looping error, the error code exception is: 'Expected to get 3 as the function code, but actually got 131. With exception code 2'

Here is the full printout combined with the packet:

`2021-01-04 08:56:34.692 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57378 didn't have a cached value.
2021-01-04 08:56:34.750 [main] DEBUG me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead - Error code exception message: Expected to get 3 as the function code, but actually got 131. With exception code 2
2021-01-04 08:56:34.751 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57379 didn't have a cached value.
2021-01-04 08:56:34.809 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57380 didn't have a cached value.
2021-01-04 08:56:34.867 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57381 didn't have a cached value.
2021-01-04 08:56:34.917 [main] DEBUG me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead - Error code exception message: Expected to get 3 as the function code, but actually got 131. With exception code 2
2021-01-04 08:56:34.918 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57382 didn't have a cached value.
2021-01-04 08:56:34.980 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57383 didn't have a cached value.
2021-01-04 08:56:35.039 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57384 didn't have a cached value.
2021-01-04 08:56:35.097 [main] DEBUG me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead - Error code exception message: Expected to get 3 as the function code, but actually got 131. With exception code 2
2021-01-04 08:56:35.098 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57385 didn't have a cached value.
2021-01-04 08:56:35.148 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57386 didn't have a cached value.
2021-01-04 08:56:35.206 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57387 didn't have a cached value.
2021-01-04 08:56:35.261 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57388 didn't have a cached value.
2021-01-04 08:56:35.315 [main] DEBUG me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave - Register: 57389 didn't have a cached value.
2021-01-04 08:56:35.370 [main] DEBUG me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Debugging special power control values: (Will debug all packets later)
charging mode controlled by voltage: false
	special power control: disabled
	each night on: disabled
	no charging below 0C: disabled
	charging method: Direct
2021-01-04 08:56:35.371 [main] DEBUG me.retrodaredevil.solarthing.program.RoverPacketListUpdater - took 1124ms to read from Rover
2021-01-04 08:56:35.371 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Debugging all packets
2021-01-04 08:56:35.374 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - [ {
  "packetType" : "RENOGY_ROVER_STATUS",
  "maxVoltage" : 24,
  "ratedChargingCurrent" : 20,
  "ratedDischargingCurrent" : 20,
  "productType" : 0,
  "productModelEncoded" : "UkNDMjBSVlJFLUcxICAgIA==",
  "softwareVersion" : 0,
  "hardwareVersionString" : "V00.00.00",
  "productSerialNumber" : 0,
  "controllerDeviceAddress" : 1,
  "batteryCapacitySOC" : 100,
  "batteryVoltage" : 13.8,
  "chargingCurrent" : 0.3,
  "controllerTemperatureRaw" : 26,
  "batteryTemperatureRaw" : 20,
  "loadVoltage" : 0.0,
  "loadCurrent" : 0.0,
  "loadPower" : 0,
  "inputVoltage" : 18.5,
  "pvCurrent" : 0.21,
  "chargingPower" : 4,
  "dailyMinBatteryVoltage" : 12.5,
  "dailyMaxBatteryVoltage" : 14.3,
  "dailyMaxChargingCurrent" : 3.6,
  "dailyMaxDischargingCurrent" : 0.0,
  "dailyMaxChargingPower" : 40,
  "dailyMaxDischargingPower" : 0,
  "dailyAH" : 10,
  "dailyAHDischarging" : 0,
  "dailyKWH" : 0.131,
  "dailyKWHConsumption" : 0.0,
  "operatingDaysCount" : 7,
  "batteryOverDischargesCount" : 0,
  "batteryFullChargesCount" : 0,
  "chargingAmpHoursOfBatteryCount" : 18,
  "dischargingAmpHoursOfBatteryCount" : 0,
  "cumulativeKWH" : 0.219,
  "cumulativeKWHConsumption" : 0.0,
  "streetLightValue" : 0,
  "chargingState" : 5,
[... 41 lines cut ...]
  "ledLoadCurrentSettingRaw" : 0,
  "specialPowerControlE021Raw" : 0,
  "sensed1" : null,
  "sensed2" : null,
  "sensed3" : null,
  "sensingTimeDelayRaw" : null,
  "ledLoadCurrentRaw" : null,
  "specialPowerControlE02DRaw" : null,
  "productModelString" : "RCC20RVRE-G1",
  "softwareVersionString" : "V00.00.00",
  "hardwareVersion" : 0,
  "streetLightBrightness" : 0,
  "streetLightOn" : false,
  "chargingStateName" : "Float",
  "errors" : "",
  "batteryTypeName" : "gel",
  "loadWorkingModeName" : "LIGHT_CONTROL"
}, {
  "packetType" : "SOURCE",
  "sourceId" : "default"
}, {
  "packetType" : "FRAGMENT_INDICATOR",
  "fragmentId" : 2
} ]
2021-01-04 08:56:35.382 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 3865ms`

sabarnes3, 2021-01-04:
As a side note--it is pushing data to CouchDB without issue.

retrodaredevil, 2021-01-04:
Thank you for that information. That error that is being debugged is intentional and I will remove it next commit. You can recompile the code next time I commit if you'd like or you can ignore the error. I was just curious to see what the exception code was. It turns out it's 2. I'll probably create an official release soon, which you can get by pulling, then running `./download_solarthing.sh` in the program directory once the new version is released. 

So with the output I see, everything should be working fine. If you run into any more problems, feel free to create another issue.

closed 2021-01-04 by retrodaredevil