wildmountainfarms/solarthing #17: Great Job

SkyMoCo, 2021-01-27:
You have done an amazing amount of work.   SolarThing pretty much worked "out of the box" for me on my Pi.   Little bit of confusion on my part, but overall very very well done.

Quick question.   I can see my data in couchdb and installed the graphql and basic config file.

solarthing@pisolar:/opt/solarthing/program/rover$ more config/application.properties
solarthing.config.database=../config/couchdb.json

Is there anything else I need to do?    Setting up the datasource in grafana gives me a "HTTP Response Undefined" message and I don't see anything listening on port 8080 like in your example.

retrodaredevil, 2021-01-27:
Thanks. I'm glad to hear you got it running well.

The graphql config should be put in /opt/solarthing/program/graphql/config. You should also install the solarthing-graphql service by running `/opt/solarthing/other/systemd/install.sh graphql`. If you want to run it and see debug output, `cd /opt/solarthing/program/graphql` and run `./run.sh`. That will download a separate jar file than the other programs use. If you get an error from running that, you can post the output here.

The documentation for GraphQL is probably the poorest of everything in SolarThing. I will eventually make it better.

To keep GraphQL running, you can start the service with `sudo systemctl start solarthing-graphql` and of course enable it on startup `sudo systemctl enable solarthing-graphql`, just like you would do with the rover program.

SkyMoCo, 2021-01-27:
I was hoping you would answer...

I think we travelled down the exact same route, but you did all the hard
work and I was just able to follow along a year later.   I have better
pictures of my cable setup, like you I had to make my own.

Do you have a sample query for grafana, I have the data source working
thanks to your help, but I am used to the dropdown working and guiding me
through the setup  (its been a while and I just used graphite).  I think
you could just send me a copy of your dashboard in json format if I
remember correctly.

Any plans on sending the raw data json via MQTT?    The reason I ask is
that my panels and pi are at two very remote locations that just happen to
have great internet due to having a mine nearby.   Would like to have each
one storing the data locally, and perhaps send the data to central system
so I could show both.

retrodaredevil, 2021-01-27:
Here's the raw json for a GraphQL dashboard I made a while back. This hasn't been updated in a while and I currently don't remote access to the system I help monitor, so I can't update it. https://github.com/wildmountainfarms/solarthing/blob/master/other/grafana/dashboards/solar_graphql_wildmountainfarms.json

Since you're just using a Rover, you'll probably want to remove a good chunk of that dashboard, as it won't apply to the rover.

I might think about sending the raw data via MQTT, it's just I don't see a huge advantage of it over CouchDB since CouchDB can sync to multiple databases if configured correct (it's a bit tricky to set up, but that's what I have going on right now).

The advantage of something like the GraphQL program is that it can do simple or complex queries and sum, or even integrate the data for useful calculations that wouldn't be possible with simple queries. If I did end up making a way to send MQTT data, when you query it on Grafana, you'd be messing with raw data rather than data that goes through the SolarThing GraphQL middleman, which is what happens now if you decide to use InfluxDB.

Also keep in mind that if you don't want to set up replication for CouchDB, you can always make SolarThing upload to two different databases at once.

Although I don't think MQTT support can be better than CouchDB support, I'll definitely think about supporting it because of how common it's becoming. I don't think it would be too difficult to implement anyway.

SkyMoCo, 2021-01-27:
I was thinking of just doing the couchdb replication.   It doesn't seem
that hard, but I already have a MQTT infrastructure.

That json (from looking at it) is just for influxdb.     Do you have a
sample query that works with GraphQL?    I understand you cannot get to the
actual system right now but I am at a bit of loss where to go.   I am used
to dropdowns working in grafana.  I think I have some more setup to do, but
not sure where to start.

SkyMoCo, 2021-01-27:
I think grafana is telling me that it cannot show me the data schema if
that's a thing with couchdb...

retrodaredevil, 2021-01-27:
Some of the queries are for InfluxDB, but most of the non-annotation queries are for GraphQL. Take the battery voltage query for example: https://github.com/wildmountainfarms/solarthing/blob/85dc5fd3e84daa697290e6b6703c98c747ce1666/other/grafana/dashboards/solar_graphql_wildmountainfarms.json#L147

You can read some of that JSON to see what you should make groupBy, dataPath, etc. With only one rover, groupBy isn't that important, but you need to make sure to get dataPath correct.

I don't recommend copying that dashboard entirely into your Grafana. Just copy the parts you want from it. Once I get remote access back, I'll try and create a snapshot of the dashboard and update the JSON. I think the snapshot would be useful to actually see what everything is for.

> I think grafana is telling me that it cannot show me the data schema if that's a thing with couchdb...

That's probably something to do with GraphQL. I'm not really sure what that error message is. If you'd like you can show me a screenshot of where you're seeing that message.

SkyMoCo, 2021-01-28:
Another thought.

Instead of just standing up my own influxdb, they do have a free cloud
offering that would work for a dashboard, although it only keeps 30 days,
but it would be accessible from anywhere.

But it wants tokens..   Is that a big change?

public class InfluxDB2Example { public static void main(final String[]
args) { // You can generate a Token from the "Tokens Tab" in the UI String
token =
"IXk2BKaOKmpNzsbZs1Ll__vKps5y_Ch75RB7IUBfozhKo8FubGR8ANd0XuzPsUOFpbwgDFkeVl2Dd-0Szrb0rA==";
String bucket = "ksmolkowski's Bucket"; String org = "ksmolkowski@gmail.com";
InfluxDBClient client = InfluxDBClientFactory.create("
https://us-west-2-1.aws.cloud2.influxdata.com", token.toCharArray()); } }

retrodaredevil, 2021-01-28:
It's not that the token is a big change, I think that's pretty easy to use, but I think the cloud service you're talking about is using InfluxDB 2.0 instead of InfluxDB 1.X.

Someone else asked about InfluxDB 2.0 a little while back and their solution was to just use InfluxDB 1.X: https://github.com/wildmountainfarms/solarthing/issues/12#issuecomment-737413912, which isn't ideal, but supporting InfluxDB 2.0 isn't a huge priority for me right now.

SkyMoCo, 2021-01-29:
> solution was to just use InfluxDB 1.X: #12 (comment)

I understand.   I just setup the pvoutput and that pretty much does what I
want.  I can modify things later,  however...

Something very odd is happening and even being a pretty experienced SA, I
am not sure why.   The rover program cannot open ttyUSB0.  It was working,
then tried to start it as a service and now.

solarthing@pisolar:~$ /opt/solarthing/program/rover/run.sh
Using base config = config/base.json
2021-01-29 08:57:58.043 [main] INFO
 me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main.
Jar: solarthing-2021.1.1.jar Java version: 11.0.9.1
[stdout] Beginning main. Jar: solarthing-2021.1.1.jar Java version: 11.0.9.1
[stderr] Beginning main. Jar: solarthing-2021.1.1.jar Java version: 11.0.9.1
Using legacy arguments! Please use --base instead! (If you are running this
using ./run.sh, this will be automatically fixed in a future update)
(ignore this).
2021-01-29 08:57:58.892 [main] INFO
 me.retrodaredevil.solarthing.program.SolarMain - Using base configuration
file: config/base.json
2021-01-29 08:58:00.220 [main] INFO
 me.retrodaredevil.solarthing.program.RoverMain - Beginning rover program
2021-01-29 08:58:00.239 [main] INFO
 me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics
is ENABLED!
2021-01-29 08:58:00.476 [main] DEBUG
me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics
UUID: 091d2416-f095-4274-863f-8808781fd4dd
2021-01-29 08:58:04.268 [main] INFO
 me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program
type to Google Analytics
2021-01-29 08:58:04.622 [main] ERROR
me.retrodaredevil.solarthing.program.RoverMain - (Fatal)Got exception!
java.lang.RuntimeException:
me.retrodaredevil.io.serial.SerialPortException: Was unsuccessful while
trying to open port: ttyUSB0 descriptive name: User-Specified Port
description: User-Specified Port
        at
me.retrodaredevil.solarthing.program.SolarMain.createIOBundle(SolarMain.java:153)
~[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:129)
[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:50)
[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:102)
[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:200)
[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:264)
[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:268)
[solarthing-2021.1.1.jar:?]
Caused by: me.retrodaredevil.io.serial.SerialPortException: Was
unsuccessful while trying to open port: ttyUSB0 descriptive name:
User-Specified Port description: User-Specified Port
        at
me.retrodaredevil.io.serial.JSerialIOBundle.<init>(JSerialIOBundle.java:22)
~[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.config.io.SerialIOConfig.createIOBundle(SerialIOConfig.java:46)
~[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.createIOBundle(SolarMain.java:151)
~[solarthing-2021.1.1.jar:?]
        ... 6 more

I checked groups, permissions on /dev/ttyUSB  tried to run it as root,
nothing works.   I didn't change any of the configuration files between it
working and not.  I can open the port and read the rover via a python
script.

   if you happen to know where to look, let me know, otherwise I might just
start over and rebuild it.

retrodaredevil, 2021-01-29:
That's really weird. I'll list some things for you to try first:
* Make sure the systemd service is NOT running with (systemctl status solarthing-rover)
  * If it is, then stop it and try running it manually (you don't want two instances of SolarThing running)
* Restart your pi (hopefully this fixes it)
* Unplug your USB then plug it back in (even try another port)
* Run `sudo /opt/solarthing/other/linux/create_user.sh` and try again by running as solarthing user
* Try running as sudo again I guess

If none of these work, then maybe you could try starting over, but it's weird you can get it to work in Python, but not with SolarThing. If you still can't get it to work, I might go back to how I had it a few versions back where you could choose which serial implementation SolarThing used. I didn't think that was still necessary, but it might be useful to test out if you're getting an error like that.

SkyMoCo, 2021-01-29:
Well, it's not solarthing.  I reloaded the Pi (I wanted to get rid of the
gui stuff anyway).   And copied back the whole solarthing config and it
worked just fine.   Odd...

retrodaredevil, 2021-01-29:
It sounds to me like you were running two instances of SolarThing. If you get an error like that again, it's probably because another program is using the serial port. Only one program can use the serial port at one time. However, it's very weird you were able to use a Python program to interact with the serial port, so maybe it isn't the case you were running two instances of SolarThing.

retrodaredevil, 2021-02-01:
Thought I'd give an update that versions 2021.1.2 and onward will support InfluxDB 2.0, but you can't migrate from InfluxDB 1.X, so if you're using 1.X, you don't have to upgrade.

Is SolarThing working well for you now? If it is feel free to close this issue.

SkyMoCo, 2021-02-02:
I thought I would try out the influxdb2 options and I did a git pull and
added the config file, but I got this.   Is there something else I need to
do to install the latest version?

Caused by: com.fasterxml.jackson.databind.exc.InvalidTypeIdException: Could
not resolve type id 'influxdb2' as a subtype of
`me.retrodaredevil.solarthing.config.databases.DatabaseSettings`: known
type ids = [couc
hdb, influxdb, latest, post] (for POJO property 'config')
 at [Source: (FileInputStream); line: 7, column: 5]
        at
com.fasterxml.jackson.databind.exc.InvalidTypeIdException.from(InvalidTypeIdException.java:43)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.DeserializationContext.invalidTypeIdException(DeserializationContext.java:1935)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.DeserializationContext.handleUnknownTypeId(DeserializationContext.java:1426)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.jsontype.impl.TypeDeserializerBase._handleUnknownTypeId(TypeDeserializerBase.java:298)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.jsontype.impl.TypeDeserializerBase._findDeserializer(TypeDeserializerBase.java:165)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.jsontype.impl.AsArrayTypeDeserializer._deserialize(AsArrayTypeDeserializer.java:97)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.jsontype.impl.AsArrayTypeDeserializer.deserializeTypedFromObject(AsArrayTypeDeserializer.java:61)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.AbstractDeserializer.deserializeWithType(AbstractDeserializer.java:263)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.impl.FieldProperty.deserializeAndSet(FieldProperty.java:147)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.impl.ExternalTypeHandler._deserializeAndSet(ExternalTypeHandler.java:382)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.impl.ExternalTypeHandler.handlePropertyValue(ExternalTypeHandler.java:191)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeWithExternalTypeId(BeanDeserializer.java:979)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeWithExternalTypeId(BeanDeserializer.java:943)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:360)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:195)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:322)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4591)
~[solarthing-2021.1.1.jar:?]
        at
com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3583)
~[solarthing-2021.1.1.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.getDatabaseConfig(SolarMain.java:126)
~[solarthing-2021.1.1.jar:?]

retrodaredevil, 2021-02-02:
Yes. You need to run `program/download_solarthing.sh`. https://github.com/wildmountainfarms/solarthing/blob/master/other/docs/updating.md

SkyMoCo, 2021-02-02:
I figured it out.  I manually downloaded the latest version from .download
and am running that.   Works great.  My data is flowing into a local
influx2 database that I setup.   Haven't tried the cloud yet.

I'd be happy to help with the documentation if you like.   Maybe offer a
different perspective?    Less programmer, more system admin?

retrodaredevil, 2021-02-02:
Yeah. If you'd like to help on the documentation that'd be great. A different perspective would be really helpful.

Feel free to create a pull request and I'll review it. Just make sure to keep all the docs in the `other/docs` folder if you end up adding new files.

retrodaredevil, 2021-02-08:
I'm closing this for now. Feel free to open another issue if you need more help or re-open this one if you don't feel there needs to be another issue.

SkyMoCo, 2021-02-18:
solarthing is just cruising along, working great.    I am thinking of
adding a couple of these RS485 modules you can get cheaply from AliExpress
or EBay.   They all seem to be made by PeaceFair and do all sorts of
measurements..   Here is the PAEM-003 current sensor.
https://www.aliexpress.com/item/32912901621.html.   I bought mine of EBay
and it seems to work pretty well.

I tried digging through your code to see where you read the packets but I
wasn't able to grasp how you did it.   Are you leveraging ModBus or just
reading it directly?

If I bought you a couple of these, would you add them to solarthing?   I'd
be happy to do that, easier than reverse engineering it.  :)

retrodaredevil, 2021-02-18:
I might consider adding support for PAEM-003. It looks very simple to setup and read from. It looks like there's also a pdf file for it here: https://github.com/SN-8023/PZEM-017-003-Change-Address/blob/master/PZEM-003_017.pdf

If you want me to try and add something to SolarThing for the PAEM-003, I can do it since it doesn't look like much work and you'll be the one testing it. I might get around to adding it sometime in the next week. There are also plenty of other repos on GitHub which can do this, but my guess is that none of them integrate too nicely with Grafana. Maybe there's one that uploads to InfluxDB.

I am using Modbus to read from the Rover. [here](https://github.com/wildmountainfarms/solarthing/blob/master/core/src/main/java/me/retrodaredevil/solarthing/solar/renogy/rover/modbus/RoverModbusSlaveRead.java) is where I read and [here](https://github.com/wildmountainfarms/solarthing/blob/master/client/src/main/java/me/retrodaredevil/solarthing/program/RoverPacketListUpdater.java) is where I request and add rover packets to the packet list.

Also, I'd recommend waiting for me to actually write the code that will request from these and once I get around to doing that, then maybe only buy one to test with, just to make sure it works as you'd expect before going all in.

SkyMoCo, 2021-02-18:
I'd love it if you could add it.  Since you already do modbus and I would
think it would not be too hard.   Part of my logic is that why not have one
easy to use product that does it all and I don't have dozen things running
and talking to my devices.

retrodaredevil, 2021-02-19:
Alright it took me less time than I thought. I was pretty bored yesterday.

You can add this to your rover program's config:
```
  "request": [
    {
      "type": "pzem",
      "io": "../../config_templates/io/standard_io.json",
      "data_id": 1,
      "modbus": 1
    }
  ],

```
And of course replace io with the configuration to your serial port file.

SkyMoCo, 2021-02-19:
I love it when you are bored.   Will test it out this evening!

Once I "comprehended" modbus, it makes things easy.   I ended up writing a
small test harness and that really helped see what was happening.   That
and looking at your code..

I am still pondering updates to the docs.   Was thinking of the quickstart
and how it's kinda confusing as there are a lot of options.    People want
to make it work first, then explore all  the features.    The quickstart
guide could just be some defaults to make this work quick and easy, then
here are all the other cool things.

retrodaredevil, 2021-02-19:
Once you confirm it works as expected, I can add it to the GraphQL program.

> Was thinking of the quickstart and how it's kinda confusing as there are a lot of options

That's very true. It probably only needs a reference to CouchDB. And I can probably remove some of the program options that are for after you set up the rover or mate program. Also I have yet to have someone actually use the `mate` program, so maybe I'll remove that too.

SkyMoCo, 2021-02-19:
I can see it talking to the PZEM and it responding.   I get this error.

2021-02-19 10:16:53.803 [main] ERROR
me.retrodaredevil.solarthing.program.PzemShuntPacketListUpdater - Modbus
exception
me.retrodaredevil.io.modbus.handling.ErrorCodeException: Expected to get 3
as the function code, but actually got 131. With exception code 2
        at
me.retrodaredevil.io.modbus.handling.HandleResponseHelper.checkResponse(HandleResponseHelper.java:17)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:61)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:13)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.solar.pzem.modbus.PzemShuntModbusSlaveRead.getEnergyValueRaw(PzemShuntModbusSlaveRead.java:41)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.solar.pzem.ImmutablePzemShuntStatusPacket.createFromReadTable(ImmutablePzemShuntStatusPacket.java:36)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.PzemShuntPacketListUpdater.receive(PzemShuntPacketListUpdater.java:31)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:111)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:83)
~[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:127)
[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:51)
[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:103)
[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:204)
[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:244)
[solarthing-2021.3.0.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:272)
[solarthing-2021.3.0.jar:?]

My config is below.   The PZEM is at address 0x02 and the data starts at
0.    The command to read the 8 registers is 0x04.   I should just look at
the code and see what maps where but I have to get back to work. :)

I assume we are looking at the same datasheet.
http://myosuploads3.banggood.com/products/20190723/20190723213410PZEM-003017UserManual.pdf
Section 2.3

{
    "type": "rover",
    "source": "default",
    "fragment": 2,
    "unique": 30,
    "databases": [
        "config/couchdb.json",
        "config/influxdb2.json"
    ],
    "request": [
        {
            "type": "pzem",
            "io": "config/pzem.io.json",
            "data_id": 0,
            "modbus": 2
        }
    ],
    "dummy": "test/dummy_rover.json",
    "modbus": 1

}

retrodaredevil, 2021-02-19:
That's really weird that there's an "illegal address" error when trying to get energy data. I'll have to think of what the best option is here...

retrodaredevil, 2021-02-19:
Go ahead and run `git pull` and then compile solarthing with `./compile_and_move.sh`. Then try running that.

I don't want to make another release for this since I don't know if it works yet.

SkyMoCo, 2021-02-19:
Took a bit to compile.

root@pisolar2:/opt/solarthing/program/rover# ./run.sh
Using base config = config/base.json
2021-02-19 13:07:58.478 [main] INFO
 me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main.
Jar: solarthing-SNAPSHOT.jar Java version: 11.0.9.1
[stdout] Beginning main. Jar: solarthing-SNAPSHOT.jar Java version: 11.0.9.1
[stderr] Beginning main. Jar: solarthing-SNAPSHOT.jar Java version: 11.0.9.1
2021-02-19 13:07:59.404 [main] INFO
 me.retrodaredevil.solarthing.program.SolarMain - Using base configuration
file: config/base.json
2021-02-19 13:08:00.880 [main] INFO
 me.retrodaredevil.solarthing.program.RoverMain - Beginning rover program
2021-02-19 13:08:00.899 [main] INFO
 me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics
is ENABLED!
2021-02-19 13:08:00.965 [main] DEBUG
me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics
UUID: d0173f1c-5087-4eb7-9a78-8363488e33a4
2021-02-19 13:08:05.552 [main] INFO
 me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program
type to Google Analytics
2021-02-19 13:08:06.560 [main] DEBUG
me.retrodaredevil.solarthing.program.RoverMain - We will not attempt to
send rover analytic data because dummy file is active.
2021-02-19 13:08:11.283 [main] ERROR
me.retrodaredevil.solarthing.program.PzemShuntPacketListUpdater - Modbus
exception
me.retrodaredevil.io.modbus.handling.ErrorCodeException: Expected to get 3
as the function code, but actually got 131. With exception code 2
        at
me.retrodaredevil.io.modbus.handling.HandleResponseHelper.checkResponse(HandleResponseHelper.java:17)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:61)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:13)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.solar.pzem.modbus.PzemShuntModbusSlaveRead.getHighVoltageAlarmStatus(PzemShuntModbusSlaveRead.java:60)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.solar.pzem.ImmutablePzemShuntStatusPacket.createFromReadTable(ImmutablePzemShuntStatusPacket.java:40)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.PzemShuntPacketListUpdater.receive(PzemShuntPacketListUpdater.java:31)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:111)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:83)
~[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:127)
[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:51)
[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:103)
[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:204)
[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:244)
[solarthing-SNAPSHOT.jar:?]
        at
me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:272)
[solarthing-SNAPSHOT.jar:?]
2021-02-19 13:08:11.370 [main] DEBUG
me.retrodaredevil.solarthing.program.RoverPacketListUpdater - took 37ms to
read from Rover

SkyMoCo, 2021-02-19:
Doesn't seem to matter at all what I set data id to.   Same error.   Just
FYI.

    "request": [
        {
            "type": "pzem",
            "io": "config/pzem.io.json",
            "data_id": 4,
            "modbus": 2
        }

SkyMoCo, 2021-02-19:
The data coming back looks like.

1174,55,64,0,70,0,0,0

Raw packets looks like
04 96 00 37 00 40 00 00 00 46 00 00 00 00 00 00

retrodaredevil, 2021-02-20:
The `data_id` is just for SolarThing to keep track of what "sensor" is what.

Try recompiling it now after pulling and see if it works. It looks like the PZEM-003 doesn't support everything that the 017 does.

SkyMoCo, 2021-02-20:
It's getting the data, but it's off a bit.

  "packetType" : "PZEM_SHUNT",
  "dataId" : 4,
  "voltageValueRaw" : 30000,
  "currentValueRaw" : 700,
  "powerValueRaw" : 2,
  "energyValueRaw" : null,
  "highVoltageAlarmStatus" : null,
  "lowVoltageAlarmStatus" : null,
  "modbusAddress" : 2

More like 9.68Volts

retrodaredevil, 2021-02-20:
Huh. For some reason it's reading 300V at .7A with a power value of 0.2 watts. Something is not right...

Honestly I have no idea. Maybe it needs to be calibrated? But then again, 300*0.7 does not equal 0.2, so I'm not sure.

SkyMoCo, 2021-02-20:
I think the problem is
that ./core/src/main/java/me/retrodaredevil/solarthing/solar/pzem/modbus/PzemShuntModbusSlaveRead.java
needs to be read the input registers and not the holding registers.   I
tried changing the import and ReadHoldingRegsters to ReadInputRegisters but
perhaps you didn't implement that.   I tried to find the io class file but
that's where I got stumped.

retrodaredevil, 2021-02-20:
You are absolutely right. I have to implement that in my io-lib library. Good catch.

retrodaredevil, 2021-02-20:
Go ahead and pull then recompile. Maybe this time it will work.

SkyMoCo, 2021-02-20:
Back to this issue.  I can help debug it in the am, am pretty sure what to
do but it's late...

2021-02-19 22:38:07.289 [main] ERROR
me.retrodaredevil.solarthing.program.PzemShuntPacketListUpdater - Modbus
exception
me.retrodaredevil.io.modbus.handling.ErrorCodeException: Expected to get 3
as the function code, but actually got 131. With exception code 2
at
me.retrodaredevil.io.modbus.handling.HandleResponseHelper.checkResponse(HandleResponseHelper.java:17)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.io.modbus.handling.BaseReadRegisters.handleResponse(BaseReadRegisters.java:52)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.io.modbus.handling.BaseReadRegisters.handleResponse(BaseReadRegisters.java:11)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.solar.pzem.modbus.PzemShuntModbusSlaveRead.getEnergyValueRaw(PzemShuntModbusSlaveRead.java:46)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.solar.pzem.ImmutablePzemShuntStatusPacket.createFromReadTable(ImmutablePzemShuntStatusPacket.java:39)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.program.PzemShuntPacketListUpdater.receive(PzemShuntPacketListUpdater.java:31)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:111)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:83)
~[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:127)
[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:51)
[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:103)
[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:204)
[solarthing-SNAPSHOT.jar:?]
at
me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:244)
[solarthing-SNAPSHOT.jar:?]
at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:272)
[solarthing-SNAPSHOT.jar:?]

retrodaredevil, 2021-02-20:
Once again. My bad. The changes I made didn't actually change anything. Try it again.

SkyMoCo, 2021-02-20:
Success!!!!   I've been playing around and yes, I drew down my battery to
7.25 volts....

  "packetType" : "PZEM_SHUNT",
  "dataId" : 4,
  "voltageValueRaw" : 725,
  "currentValueRaw" : 18,
  "powerValueRaw" : 13,
  "energyValueRaw" : 152,
  "highVoltageAlarmStatus" : 0,
  "lowVoltageAlarmStatus" : 0,
  "modbusAddress" : 2

retrodaredevil, 2021-03-09:
Thought I'd let you know that I updated the readme and quickstarts a bit to not throw as much information at new users.

Also, the SolarThing GraphQL now has a query for pzem shunt data, but I haven't made a release yet. If you want, you can compile the GraphQL program yourself using `./graphql_compile_and_move.sh`. I might make a release soon, I might not.

SkyMoCo, 2021-03-09:
I will give it a read.   Solarthing just chugs along at the ranch w/o any
issues.  :)

closed 2021-03-09 by retrodaredevil

SkyMoCo, 2021-03-09:
Just for grins.  Here is where I have solarthing running...

Nothing for miles and miles....   I do have reasonable internet, thanks to
a mine on a nearby mountain.

retrodaredevil, 2021-03-09:
By the way if you uploaded a picture to the email, I don't think it showed up in GitHub. I think if you want to upload a picture you have to do it on github.com itself.