wildmountainfarms/solarthing #18: Renogy 30A Dual-Input DC-to-DC MPPT On-Board Battery Charger

gecko951, 2021-01-30:
Hi,
I'm interested in trying solarthing with my Renogy 30A Dual-Input DC-to-DC MPPT On-Board Battery Charger.  The device uses an RS485 port and I was curious if solarthing has been tested to work in this configuration?  Sounds like the other monitoring option is the Renogy BT-2 but many reviews seem to be disappointed in the quality of the app.  I was going to purchase the Renogy rs485-to-usb-serial-cable but their site says its only for updating the FW on their Lithium batteries so I'm also checking with their support.

retrodaredevil, 2021-01-30:
I'd say that since it works with the BT-2 module, it's probably very similar to most models of the Renogy Rover, especially the Rover Elite. So it probably does work with SolarThing. However, there's a good chance that it won't work with SolarThing without changes to SolarThing, similar to how https://github.com/wildmountainfarms/solarthing/issues/16 had to be fixed to make SolarThing work with Rover Elites. If SolarThing doesn't end up working with the Duel-Input DC-to-DC, I'd be happy to take the error produced (if there is one), and make SolarThing work with it.

TDLR: There's a good chance it works with SolarThing, I might just have to make a few changes to make sure it works.

As for the cable, I didn't know that existed. If that works, it sounds like a much better option than creating your own cable (for RS485 devices anyway). I'd say it probably works.

I think you're one of the first people asking these questions because of how new their, well new, products are. Sorry I don't have any definite answers. My answer is probably to both of your questions.

gecko951, 2021-01-30:
OK, thanks for the quick feedback! I have gone ahead and purchased the RS485-to-serial cable and will be using a Lenovo ThinkCentre M90n IoT as the compute device.  I will try to update you again in a couple of weeks once all the hardware arrives. :)

riker65, 2021-02-02:
Hi,
Renogy support told me that it is not compatible with the old MT-50 meter they are having which is working with the ROVER. So I think there is a problem.

retrodaredevil, 2021-02-02:
@riker65 I'm curious, which rover do you have? And which MT-50 do you have? Do you have a link to the MT-50 you're using?

From what I understand the MT-50 doesn't work with any of Renogy's new products. I didn't think it worked with the Rover over RS232 or RS485, but maybe I'm wrong.

riker65, 2021-02-02:
Hi do not have a rover, sorry for confusion.

response from Renogy Support 

We have already contacted our technician and are told that this product is not compatible with DC-DC Mppt charge controllers. And it is expected that the special monitor for DC-DC MPPT charge controllers will be in the warehouse at the end of June. If you are still interested, ​we can tell you per email then.
If you have other questions, please do not hesitate to contact us.
Best regards

    
    Hi

    https://www.renogy.com/template/files/Manuals/TrcrMtr-MT-50.pdf

    I am wondering that the MT-50 is working for other devices from renogy.

    thanks for checking

retrodaredevil, 2021-02-02:
Just to clarify, you asked them about the RS485-to-serial cable specifically and they said it was not compatible? Or did they say that the MT-50 was not compatible?

If they said the MT-50 is not compatible, I don't think that necessarily means that the RS485-to-serial cable is not compatible.

riker65, 2021-02-02:
Hi, this is also  unclear to me. Cable should work hopefully. I am struggling with the wiring.

just asked this in another threat #20 

thanks for the support

did you ever try this with an ESP8266 and tasmota maybe?

riker65, 2021-02-02:
> Just to clarify, you asked them about the RS485-to-serial cable specifically and they said it was not compatible? Or did they say that the MT-50 was not compatible?
> 
> If they said the MT-50 is not compatible, I don't think that necessarily means that the RS485-to-serial cable is not compatible.

I expected - probably naive - that the MT-50 is a more standardized product, also available from others. EPever for example.

retrodaredevil, 2021-02-02:
> did you ever try this with an arduino and/or ESP8266 and tasmota maybe?

I have not tried any communication between Renogy products with an Arduino or ESP8266. Since SolarThing uses Java, SolarThing will never work on these devices.

> I expected - probably naive - that the MT-50 is a more standardized product

I definitely agree with this, however I believe the last renogy product that supported this was the Commander. Ever since then Renogy has been using their own proprietary protocol based off of SRNE products.

riker65, 2021-02-02:
thanks, will now try the RPI route.

gecko951, 2021-02-07:
Hi,
I now have the Renogy USB to RS-485 cable and I am working on setting up SolarThing on a Lenovo NanoIOT device running Ubuntu Server 20.0.4

gecko951, 2021-02-07:
Hi Joshua,
I went through the config but I'm encountering the following error message starting solarthing:

` sudo ./run.sh
Using base config = config/base.json
2021-02-07 21:24:13.642 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: solarthing-2021.1.2.jar Java version: 1.8.0_275
[stdout] Beginning main. Jar: solarthing-2021.1.2.jar Java version: 1.8.0_275
[stderr] Beginning main. Jar: solarthing-2021.1.2.jar Java version: 1.8.0_275
Using legacy arguments! Please use --base instead! (If you are running this using ./run.sh, this will be automatically fixed in a future update) (ignore this).
2021-02-07 21:24:13.698 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2021-02-07 21:24:13.747 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Error while parsing ProgramOptions.
com.fasterxml.jackson.core.JsonParseException: Unexpected character ('"' (code 34)): was expecting comma to separate Object entries
 at [Source: (FileReader); line: 3, column: 4]
        at com.fasterxml.jackson.core.JsonParser._constructError(JsonParser.java:1922) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.core.base.ParserMinimalBase._reportError(ParserMinimalBase.java:710) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.core.base.ParserMinimalBase._reportUnexpectedChar(ParserMinimalBase.java:635) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.core.json.ReaderBasedJsonParser._skipComma(ReaderBasedJsonParser.java:2329) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.nextToken(ReaderBasedJsonParser.java:695) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer.deserializeTypedFromObject(AsPropertyTypeDeserializer.java:100) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.databind.deser.AbstractDeserializer.deserializeWithType(AbstractDeserializer.java:263) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.databind.deser.impl.TypeWrappedDeserializer.deserialize(TypeWrappedDeserializer.java:74) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:322) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4591) ~[solarthing-2021.1.2.jar:?]
        at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3559) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:183) [solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:262) [solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:266) [solarthing-2021.1.2.jar:?]
`

retrodaredevil, 2021-02-07:
Please post your base.json. Something is wrong with the way it is formatted.

gecko951, 2021-02-07:
base.json

`{
  "analytics_enabled": false
  "type": "rover",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "databases": [
    "config/couchdb.json"
  ],
  "io": "config/default_linux_serial.json",
  "modbus": 1
}`

retrodaredevil, 2021-02-07:
After false on the analytics enabled thing, you need a comma.

gecko951, 2021-02-07:
Let's try again... :)

[base.json.txt](https://github.com/wildmountainfarms/solarthing/files/5940088/base.json.txt)

gecko951, 2021-02-07:
Looks like you are correct and now it is just looking for the serial connection on startup! 👍

gecko951, 2021-02-07:
SolarThing-Rover service is running after attaching the USB-RS-485 cable.

`Feb 07 21:54:38 nanoiot systemd[1]: Started The service for SolarThing Rover.
lines 1-12/12 (END)
● solarthing-rover.service - The service for SolarThing Rover
     Loaded: loaded (/etc/systemd/system/solarthing-rover.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2021-02-07 21:54:38 UTC; 4min 3s ago
   Main PID: 661 (sh)
      Tasks: 18 (limit: 4455)
     Memory: 137.5M
     CGroup: /system.slice/solarthing-rover.service
             ├─661 sh /opt/solarthing/program/rover/run.sh
             ├─731 sh ../.scripts/solarthing.sh
             └─788 java -jar ../.scripts/../solarthing.jar config/base.json`

retrodaredevil, 2021-02-07:
Awesome! Is it uploaded data to CouchDB as expected?

gecko951, 2021-02-07:
I ran the command to populate the couchdb databases and restarted the solarthing-rover service but I'm only seeing ? in the Solarthing Web interface.  How would I see if data is flowing to CouchDB?

retrodaredevil, 2021-02-07:
The easiest way to check to see if there's data going into CouchDB is to check the logs in `/opt/solarthing/program/rover/logs/log_debug.log'. If you see no errors and real data, then it's working. Note that the log_debug.log file may be cut off at the end, which is fine. Just scroll up until you see text that isn't cut off. 

As for solarthing web, that hasn't been updated in a while and you need to manually change the ip address in the source code so it points to your couchdb database.

gecko951, 2021-02-08:
OK, I am seeing data in the log_debug.log:  and I did try .  Would you suggest trying Grafana to easily display the stats?

`"packetType" : "RENOGY_ROVER_STATUS",
  "maxVoltage" : 12,
  "ratedChargingCurrent" : 30,
  "ratedDischargingCurrent" : 0,
  "productType" : 0,
  "productModelEncoded" : "ICBSQkMzMEQxUy1HMSAgIA==",
  "softwareVersion" : 256,
  "hardwareVersionString" : "V00.01.00",
  "productSerialNumber" : 1,
  "controllerDeviceAddress" : 1,
  "batteryCapacitySOC" : 59,
  "batteryVoltage" : 12.4,
  "chargingCurrent" : 0.0,
  "controllerTemperatureRaw" : 24,
  "batteryTemperatureRaw" : 21,
  "loadVoltage" : 12.2,
  "loadCurrent" : 0.0,
  "loadPower" : 0,
  "inputVoltage" : 0.0,
  "pvCurrent" : 0.0,
  "chargingPower" : 0,
  "dailyMinBatteryVoltage" : 11.7,
  "dailyMaxBatteryVoltage" : 15.1,
  "dailyMaxChargingCurrent" : 14.2,
  "dailyMaxDischargingCurrent" : 0.0,
  "dailyMaxChargingPower" : 203,
  "dailyMaxDischargingPower" : 0,
  "dailyAH" : 18,
  "dailyAHDischarging" : 0,
  "dailyKWH" : 0.253,
  "dailyKWHConsumption" : 0.0,
  "operatingDaysCount" : 1,
  "batteryOverDischargesCount" : 0,
  "batteryFullChargesCount" : 73,
  "chargingAmpHoursOfBatteryCount" : 14,
  "dischargingAmpHoursOfBatteryCount" : 0,
  "cumulativeKWH" : 0.197,
  "cumulativeKWHConsumption" : 0.0,
  "streetLightValue" : 0,
  "chargingState" : 0,
  "errorMode" : 0,
  "nominalBatteryCapacity" : 100,
  "systemVoltageSetting" : 12,
  "recognizedVoltage" : 12,
  "batteryType" : 1,
  "overVoltageThresholdRaw" : 160,
  "chargingVoltageLimitRaw" : 155,
  "equalizingChargingVoltageRaw" : 148,
  "boostChargingVoltageRaw" : 146,
  "floatingChargingVoltageRaw" : 138,
  "boostChargingRecoveryVoltageRaw" : 132,`

Also I have changed the index.js for Solarthing Web to:

`const DATABASE_URL = DATABASE_URL_PARAM != null ? DATABASE_URL_PARAM : (window.location.protocol === "file:" ?
        "http://127.0.0.1:5984" :
        window.location.protocol + "//" +  window.location.hostname + ":5984");`

Would you suggest trying Grafana to see if I can get some of these stats to display?

retrodaredevil, 2021-02-08:
I would suggest using grafana, but since you have solarthing web mostly setup, there's a few more things you can try. First, are you accessing it on the computer that is hosting the database? If you are accessing it on your phone, the database url needs to be something like 192.169.x.x.

gecko951, 2021-02-08:
I have the NanoIOT host device setup as a wireless AP with the IP address 192.168.251.1 so I am joining that network from my web device (PC, phone, tablet) and then accessing the URL: http://192.168.251.1/solarthing/index.html

I have changed the database URL as you suggested and now I see the following ERROR in solarthing web
[image]
[image]

gecko951, 2021-02-08:
I tried to get graphql running and I get the following error:

`Error starting ApplicationContext. To display the conditions report re-run your application with 'debug' enabled.
2021-02-08 03:06:21.578 ERROR 1392 --- [           main] o.s.boot.SpringApplication               : Application run failed

org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'graphQLProvider': Injection of autowired dependencies failed; nested exc                             eption is java.lang.IllegalArgumentException: Could not resolve placeholder 'solarthing.config.database' in value "${solarthing.config.database}"`

`cat application.properties
solarthing.config.database=/opt/solarthing/program/rover/config/couchdb.json
gecko951@nanoiot:/opt/solarthing/program/rover/config$ cat /opt/solarthing/program/rover/config/couchdb.json
{
  "type": "couchdb",
  "settings": {
    "packet_upload": {
      "throttle_factor": 3,
      "initial_skip": 3
    },
    "command_download": {
      "throttle_factor": 3,
      "initial_skip": 4
    }
  },
  "config": {
    "protocol": "http",
    "host": "localhost",
    "port": 5984,
    "username": "uploader",
    "password": "***********",
    "connection_timeout": 1.5,
    "socket_timeout": 10
  }
}`

Grafana is installed and accessible.

gecko951, 2021-02-08:
There was some pathing confusion but its sorted and graphql will run and I am able to use it as a data source in Grafana. :)

[image]

retrodaredevil, 2021-02-08:
Awesome. Good to hear you got GraphQL working. IF you want me to help more with solarthing-web, you can give me errors from the web console (press f12). If you're happy with Grafana, you can close the issue once you've got everything working.

gecko951, 2021-02-08:
Really appreciate the help; awesome to see that its possible to monitor the DC-DC charger without the bluetooth dongle!

The errors from SolarThing Web are attached.

[192.168.251.1-1612765235383.log](https://github.com/wildmountainfarms/solarthing/files/5941629/192.168.251.1-1612765235383.log)

I used the Solarthing graphql json file for the Grafana dashboard and only Battery Voltages, Battery Voltage and Temperature (F) are displaying data.  The other panels all display 'no data' and when the page is loaded the following error message is repeated: 

Annotation Query Failed
Validation error of type FieldUndefined: Field 'submissions' in type 'Query' is undefined @ 'submissions'

retrodaredevil, 2021-02-08:
One of the example dashboards has some panels in it that only use data from Outback FX inverter devices, so it will display no data with only data from a Renogy Rover.

I'm gonna need more info for that annotation query. SolarThing GraphQL doesn't have a `sumissions` query, so I don't know where that's coming from. It might be something random in the GraphQL datasource itself.

As for why solarthing-web isn't working, you need to enable CORS on your CouchDB in the settings. I also removed a few lines in the main() function to make it start working again. You can get those changes from solarthing-web, then change your database url if necessary.

gecko951, 2021-02-09:
OK that advice really helped and after enabling CORS for couchdb and updating solarthing-web I now see Solarthing displaying some of the stats! :)

I'm not getting anything for Load W or Generator Status.  I guess generator status specifically isn't applicable to my use case but it would be nice to show when th
[192.168.251.1-1612894811282.log](https://github.com/wildmountainfarms/solarthing/files/5953487/192.168.251.1-1612894811282.log)
e battery is getting charged from the vehicle alternator and the rate of charge.

[image]

gecko951, 2021-02-09:
Do I need to try and get the full schema of graphql so I can figure out how to adjust the queries?

The errors I am seeing are like...

{
  "request": {
    "url": "api/datasources/proxy/1",
    "method": "POST",
    "data": {
      "query": "{\n        data:queryStatus(from:\"1612900052788\", to:\"1612900352788\", sourceId:\"default\") {\n            pvWattage {\n                Time:dateMillis\n                packet {\n                    pvWattage\n                    identifier { representation }\n                    identityInfo { displayName }\n                }\n            }\n        }\n    }"
    },
    "hideFromInspector": false
  },
  "response": {
    "errors": [
      {
        "message": "Validation error of type FieldUndefined: Field 'pvWattage' in type 'SolarThingStatusQuery' is undefined @ 'queryStatus/pvWattage'",
        "locations": [
          {
            "line": 3,
            "column": 13
          }
        ],
        "extensions": {
          "classification": "ValidationError"
        }
      }
    ]
  }
}

retrodaredevil, 2021-02-09:
If you want to play around with the schema and queries I don't have documented, you can use [GraphQL Playground](https://github.com/graphql/graphql-playground). It has nice autocomplete and you can easily view everything you're able to query. We talked about getting autocomplete into the GraphQL Datasource, but I think I just lacked the motivation to try and figure that out. Maybe one day I'll look into that again.

For pvWattage, you can use this:
```graphql
{
  data: queryStatus(sourceId: "default", from: "$__from", to: "$__to") {
    data: chargeController {
      Time: dateMillis
      packet {
        pvWattage
      	identifier {
          representation
        }
        identityInfo {
          displayName
        }
      }
    }
  }
}
```
Also, since you only have one charge controller, the stuff with the identifier is optional, but it's good to include in case you ever device to add another charge controller, then SolarThing will just magically start displaying data from two charge controllers assuming you have Group By set correctly.

I think the power being charged by the alternator AND the solar can be retrieved using `chargingPower`. I'm looking at the documentation for DCC charge controllers now, and they're slightly different than other Renogy CCs. It looks like SolarThing is reporting the alternator charging power in what is usually used to represent the load power being consumed.

I'll have to think of a way to separate the data based on what type of charge controller it is. I'll try to do this in a future update, but for now you can use the `loadPower` to ONLY get the power being charged by the alternator. To use that, you will have to query for `roverStatus` instead of `chargeController`.

I could try to update solarthing-web, but honestly that's mostly abandon at this point. I don't really want to program the logic for determining if the `loadPower` actually represents a load or an alternator charging the battery.

gecko951, 2021-02-10:
I tried the query for pvWattage and it was giving an error saying it cannot find the data source.

After guess/checking the query for DailyKWH is working with the following...

`{
  data:queryStatus(from:"$timeFrom", to:"$timeTo", sourceId:"default") {
    roverStatus {
      Time:dateMillis
      packet {
        dailyKWH
        identifier {representation}
        identityInfo { displayName }
      }
    }
  }
}`

So then I tried something similar for pvWattage and it seems to work as well!

`{
  data:queryStatus(from:"$timeFrom", to:"$timeTo", sourceId:"default") {
    roverStatus {
      Time:dateMillis
      packet {
        pvWattage
        identifier {representation}
        identityInfo { displayName }
      }
    }
  }
}`

Also I have attached a zip of the debug log for you to peruse. 👍 
[Uploading log_debug.zip…]()

retrodaredevil, 2021-02-10:
By the way the log didn't upload correctly.

I'm glad you got it working with your guess checking lol. For anything else I seriously recommend GraphQL Playground. It's autocomplete is soo nice.

Your daily kWh query works as expected, but I actually like to use this for my daily kWh query:

```graphql
{
  data: queryFullDay(sourceId: "default", from: $__from, to: $__to) {
    data: dailyKWH {
      dailyKWH: data
      identifiable {
        identifier {
          representation
        }
        identityInfo {
          displayName
        }
      }
    }
  }
}
```
The reason I use the above query is because it actually does some additional calculations to make sure daily kWh resets at midnight. And, if you for some reason unplug your charge controller and plug it back in during the day, with the above query it **will not** reset.

If you still want to do guessing and checking, most of the fields in log_debug.log actually correspond to stuff like pvWattage and dailyKWH, so most of the time you can find fields there that will \*just work\* in the `roverStatus` query.

gecko951, 2021-02-10:
Ok the debug log upload is fixed in my previous comment.

Is there another way to copy pasta the code correctly for Grafana?  The below gives me the error that the data source doesn't exist. :(

[image]

retrodaredevil, 2021-02-10:
The data path is going to be `data.data`. The reason for that is first you see the `data: queryFullDay...`, next the `data: dailyKWH {`. So the first one is data so you'll have `data.` and the second one is also data, so it becomes `data.data`. Everything inside the curly braces for `data: dailyKWH {}` is what the GraphQL datasource does stuff with. You'll also need a `Time: dateMillis` inside it, which I'll edit to include in my previous comment. Naming it `Time` is necessary because that's how the datasource detects it.

Also thank you for the log_debug.log. There's actually a few values in there I wasn't expecting. So hopefully this will help me differentiate between dual input CCs in the SolarThing code.

gecko951, 2021-02-10:
The DailyKWH and PVwattage panels are now working with your suggested queries and I'm going to have a play with GraphQL Playground to see if I can get some other interesting stats to display. 

Thanks so much for your patience; if there is another channel you would rather use to continue this discussion just let me know as I think its moved beyond a bug report.

retrodaredevil, 2021-02-10:
Glad you got it working. Yeah, I currently don't have another public channel  to continue this discussion, but I might make one sometime. I know there are a few chat applications designed for programmers and users of open source software, so I'll have to look into them.

riker65, 2021-02-13:
Hi 
just want to confirm the wiring.
You just use the USB to RS485 serial and only connecting A and B wire?

retrodaredevil, 2021-02-13:
@riker65 if you have an RS485 adapter, that you need to manually wire, you will end up connecting either A to TX and B to RX on your adapter, or possibly A to RX and B to TX on your adapter (I don't know which is which, but you can try both). And you will also have to connect ground to ground.

However, it sounds like the easier solution is to buy Renogy's RS485 to USB cable. If you look in SolarThing's documentation, there are a few references to creating your own cable, but that's because Renogy doesn't sell their RS232 to USB cable anymore and you actually have to create your own, but for Renogy's new products that support RS485 (all the ones that support the BT-2 module), it's much easier to just buy Renogy's cable.

gecko951, 2021-02-14:
Do you have a suggested query to get the CPU temperature from my Intel based machine?

lm-sensors reports:

`nanoiot:/opt/solarthing/program/rover/logs$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +27.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +27.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +26.0°C  (high = +100.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +33.0°C

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +23.0°C`

The Grafana query seems to be looking for a Rasberry Pi CPU temperature value:

[image]

gecko951, 2021-02-14:
Did you happen to see a value in the debug log that grafana could use to show the load or power usage in Watts?  The current panel for Power Usage seems like it was monitoring an inverter (not present in my configuration).  

My NanoIOT device killed the leisure battery completely a few days ago because dnsmasq was misconfigured and causing the CPU to run at 100% load. After fixing the dnsmasq configuration the device now only pulls 2W maximum and has been idling at less than 1W.

retrodaredevil, 2021-02-14:
Currently there is no query to get CPU temperature on anything other than a Raspberry Pi. I may consider adding one in the future. Maybe it will store all the different cores' temperatures or maybe just the highest core. There's also the question of whether or not I include the temperature of other sensors.

As far as I'm aware from your debug log, there was nothing in there that could tell me how much power you were using. Since the DCDC Charger doesn't have load terminals, there's no way to connect a load to the controller, so the controller will never know how much power you're drawing. How do you have your NanoIOT device hooked up? Do you have it hooked up through an inverter? Being powered directly from the battery?

gecko951, 2021-02-14:
The NanoIOT is being powered directly from the battery so I was just hoping to see how quickly the battery is being discharged in grafana.  On the NanoIOT I can run powertop at the CLI and see how many watts its using but ultimately I will have a fridge and air compressor connected and it would be nice to know about how long the battery will last. :)

[image]

retrodaredevil, 2021-02-14:
Yeah, I can't really help you there since your NanoIOT and your appliances aren't connected to the charge controller. Even if they were able to be connected to the charge controller, I wouldn't recommend you use the load for something that could draw a lot of current.

I don't know the best way to monitor how much power those devices use. And even if you do find something that could help monitor that, I probably won't add it in to SolarThing any time soon.

I think the best way to know how long the battery will last is really get familiar with how your battery voltage acts. There's no magic calculation that I can do to give you a perfect state of charge value of your battery.

gecko951, 2021-02-14:
OK, I understand; seems like I would need a battery monitoring shunt to get that information.  The next panel I wanted to figure out was a graph like PV Wattage that would show charging rate when the vehicle is running.

retrodaredevil, 2021-02-14:
This only works temporarily until I update SolarThing (hopefully soon):

```graphql
{
  data: queryStatus(sourceId: "default", from: "$__from", to: "$__to") {
    data: roverStatus {
      Time: dateMillis
      packet {
        pvWattage # pv wattage
        loadPower # load power, or until I update SolarThing, the alternator power on DCDC CCs
        chargingPower
      	identifier {
          representation
        }
        identityInfo {
          displayName
        }
      }
    }
  }
}
```
Once I update SolarThing:

```graphql
{
  data: queryStatus(sourceId: "default", from: "$__from", to: "$__to") {
    data: roverStatus {
      Time: dateMillis
      packet {
        pvWattage # pv wattage
        loadPower # load power, gives 0 on DCDC CCs
        generatorPower # alternator charging power, gives 0 on non DCDC CCs
        chargingPower # should be equal to pvWattage + generatorPower (I think)
      	identifier {
          representation
        }
        identityInfo {
          displayName
        }
      }
    }
  }
}
```

gecko951, 2021-02-14:
Ok I will try the above query.

When the vehicle is off the "chargingCurrent" seems to show 0 and when the vehicle is running (alternator charging) the value looks like "chargingCurrent" : 15.16, so I created a panel with this query....

{
  data: queryStatus(sourceId: "default", from: "$__from", to: "$__to") {
    data: chargeController {
      Time: dateMillis
      packet {
        chargingCurrent
      	identifier {
          representation
        }
        identityInfo {
          displayName
        }
      }
    }
  }
}

gecko951, 2021-02-14:
I tried the temporary query above and get the following error:

Validation error of type FieldUndefined: Field 'chargingPower' in type 'RoverStatusPacket' is undefined @ 'queryStatus/roverStatus/packet/chargingPower'

[image]

retrodaredevil, 2021-02-14:
Ah that's my bad. I think the chargingPower query will work if you replace `roverStatus` with `chargeController`, which means you'd have to do a separate query.

I'll fix this in the next update.

retrodaredevil, 2021-02-14:
> When the vehicle is off the "chargingCurrent" seems to show 0 and when the vehicle is running (alternator charging) the value looks like "chargingCurrent" : 15.16, so I created a panel with this query....

This happens when the solar is also charging? As far as I know `chargingPower` should be the sum of `pvWattage` and `generatorPower` (or until I update SolarThing `loadPower`).

gecko951, 2021-02-15:
I checked the debug log and found an example of ON ALTERNATOR and ON SOLAR.  Tomorrow I will check for output that includes both types of charging at the same time.
  
  "packetType" : "RENOGY_ROVER_STATUS",
  "maxVoltage" : 12,
  "ratedChargingCurrent" : 30,
  "ratedDischargingCurrent" : 0,
  "productType" : 0,
  "productModelEncoded" : "ICBSQkMzMEQxUy1HMSAgIA==",
  "softwareVersion" : 256,
  "hardwareVersionString" : "V00.01.00",
  "productSerialNumber" : 1,
  "controllerDeviceAddress" : 1,
  "batteryCapacitySOC" : 100,
  "batteryVoltage" : 14.8,
  "chargingCurrent" : 25.3,
  "controllerTemperatureRaw" : 23,
  "batteryTemperatureRaw" : 15,
  "loadVoltage" : 13.1,
  "loadCurrent" : 30.14,
  "loadPower" : 395,
  "inputVoltage" : 0.0,
  "pvCurrent" : 0.0,
  "chargingPower" : 374,
  
  ON SOLAR
  
  "packetType" : "RENOGY_ROVER_STATUS",
  "maxVoltage" : 12,
  "ratedChargingCurrent" : 30,
  "ratedDischargingCurrent" : 0,
  "productType" : 0,
  "productModelEncoded" : "ICBSQkMzMEQxUy1HMSAgIA==",
  "softwareVersion" : 256,
  "hardwareVersionString" : "V00.01.00",
  "productSerialNumber" : 1,
  "controllerDeviceAddress" : 1,
  "batteryCapacitySOC" : 49,
  "batteryVoltage" : 12.1,
  "chargingCurrent" : 0.59,
  "controllerTemperatureRaw" : 21,
  "batteryTemperatureRaw" : 16,
  "loadVoltage" : 12.2,
  "loadCurrent" : 0.0,
  "loadPower" : 0,
  "inputVoltage" : 17.3,
  "pvCurrent" : 0.41,
  "chargingPower" : 7,

retrodaredevil, 2021-02-15:
Yup, that looks correct to me. In the JSON data, load or alternator current/voltage/power will be represented by load***. However, with the new SolarThing update (I released < 5 minutes ago), in GraphQL you can query `generatorPower` and should get your desired result.

Now if you want charging power (pv + alternator), use `chargingPower`. If you just want pv, use `pvWattage`. If you just want alternator, use `generatorPower`. Also, if you query for `isDcdc`, it should give you true. If it doesn't, then I need to fix something.

gecko951, 2021-02-16:
Sorry if this an obvious question but to update solarthing in-place do I just need to...

curl https://raw.githubusercontent.com/wildmountainfarms/solarthing/master/other/linux/clone_install.sh | sudo bash

retrodaredevil, 2021-02-16:
You can look at [updating](https://github.com/wildmountainfarms/solarthing/blob/master/other/docs/updating.md).

You'll also have to update GraphQL separately (or only update GraphQL as that's what has the changes you want).

```shell
program/graphql_download_solarthing.sh
```

gecko951, 2021-02-16:
Also here is another debug log which has some solar and some alternator charging at the end of the day.

[nanoiot_log_debug.zip](https://github.com/wildmountainfarms/solarthing/files/5984865/nanoiot_log_debug.zip)

retrodaredevil, 2021-02-16:
> Also here is another debug log

Looks like that's from today UTC, so that only has 17 minutes worth of log info. If you want me to look at Feb 15's logs (UTC), you can give me `/opt/solarthing/program/rover/logs/log_debug_2021.02.15.log`

gecko951, 2021-02-16:
Looks like it pulled in updates for everything and I have restarted the solarthing-rover service.

nanoiot:/opt/solarthing$ git pull
remote: Enumerating objects: 463, done.
remote: Counting objects: 100% (463/463), done.
remote: Compressing objects: 100% (145/145), done.
remote: Total 349 (delta 210), reused 288 (delta 152), pack-reused 0
Receiving objects: 100% (349/349), 183.33 KiB | 972.00 KiB/s, done.
Resolving deltas: 100% (210/210), completed with 88 local objects.
From https://github.com/wildmountainfarms/solarthing
   ef744a1..760bb2c  master     -> origin/master
 * [new tag]         v2021.2.0  -> v2021.2.0
Updating ef744a1..760bb2c
Fast-forward
 .editorconfig                                                  |   2 +-
 .idea/dictionaries/josh.xml                                    |   1 +
 README.md                                                      |  31 +-
 client/build.gradle                                            |   2 +-
 .../me/retrodaredevil/solarthing/config/io/SerialIOConfig.java |   1 +
 .../solarthing/config/options/PacketHandlingOption.java        |   1 -
 .../solarthing/config/options/PacketHandlingOptionBase.java    |  16 +-
 .../solarthing/io/NotInitializedIOException.java               |  20 +
 .../me/retrodaredevil/solarthing/io/ReloadableIOBundle.java    | 136 ++++
 .../solarthing/message/event/LowACInputEvent.java              |   2 +-
 .../me/retrodaredevil/solarthing/program/OutbackMateMain.java  |   4 -
 .../java/me/retrodaredevil/solarthing/program/RequestMain.java |   1 -
 .../java/me/retrodaredevil/solarthing/program/RoverMain.java   |  27 +-
 .../solarthing/program/RoverPacketListUpdater.java             |  25 +-
 .../retrodaredevil/solarthing/program/RoverSetupProgram.java   |  10 +-
 .../java/me/retrodaredevil/solarthing/program/SolarMain.java   |   4 +
 .../solarthing/program/pvoutput/PVOutputHandler.java           |   2 +-
 config_templates/docker/couchdb_docker-compose.yml             |   2 +-
 config_templates/docker/grafana/README.md                      |  30 +
 config_templates/docker/grafana/docker-compose.yml             |  16 +
 config_templates/docker/grafana/grafana.ini                    | 892 +++++++++++++++++++++++++++
 config_templates/docker/grafana/setup.sh                       |  14 +
 .../docker-compose.yml => influxdb2_docker-compose.yml}        |   0
 config_templates/docker/syslog-ng/docker-compose.yml           |   2 +-
 .../solarthing/misc/common/DataIdentifiable.java               |   1 -
 .../solarthing/misc/error/ExceptionErrorPacket.java            |   1 -
 .../main/java/me/retrodaredevil/solarthing/packets/Mode.java   |   5 +
 .../solarthing/packets/identification/Identifier.java          |   5 +-
 .../solarthing/packets/identification/IdentityInfo.java        |   2 +
 .../packets/identification/SupplementaryIdentifiable.java      |   2 -
 .../java/me/retrodaredevil/solarthing/solar/PowerUtil.java     |   2 +-
 .../solarthing/solar/common/BasicChargeController.java         |   3 +
 .../retrodaredevil/solarthing/solar/common/ErrorReporter.java  |   5 +
 .../solarthing/solar/common/PVCurrentAndVoltage.java           |   7 +-
 .../solarthing/solar/common/RecordBatteryVoltage.java          |   6 +
 .../solarthing/solar/outback/OutbackStatusPacket.java          |   1 -
 .../solarthing/solar/outback/SupplementaryOutbackPacket.java   |   2 -
 .../solarthing/solar/outback/mx/ImmutableMXStatusPacket.java   |   2 +-
 .../solarthing/solar/outback/mx/MXStatusPacket.java            |   6 +-
 .../me/retrodaredevil/solarthing/solar/renogy/BatteryType.java |   6 +-
 .../solarthing/solar/renogy/ProductModelUtil.java              |  13 +-
 .../me/retrodaredevil/solarthing/solar/renogy/Version.java     |   2 +
 .../retrodaredevil/solarthing/solar/renogy/package-info.java   |   5 +-
 .../solarthing/solar/renogy/rover/ChargingState.java           |   2 +-
 .../solarthing/solar/renogy/rover/DcdcErrorMode.java           |  46 ++
 .../solarthing/solar/renogy/rover/DummyRoverReadWrite.java     |   8 +-
 .../solar/renogy/rover/ImmutableRoverStatusPacket.java         |   8 +-
 .../solarthing/solar/renogy/rover/KnownRoverController.java    |  73 +++
 .../solarthing/solar/renogy/rover/RoverErrorMode.java          |   5 +-
 .../solarthing/solar/renogy/rover/RoverReadTable.java          | 213 ++++++-
 .../solarthing/solar/renogy/rover/RoverStatusPacket.java       |   2 +
 .../solarthing/solar/renogy/rover/RoverStatusPackets.java      |   4 +-
 .../solarthing/solar/renogy/rover/SimpleRoverErrorMode.java    |   6 +
 .../solarthing/solar/renogy/rover/StreetLight.java             |   8 +-
 .../solarthing/solar/renogy/rover/annotations/DcdcOnly.java    |  11 +
 .../solar/renogy/rover/annotations/ResetEvening.java           |  14 +
[... 8 lines cut ...]
 .../solarthing/graphql/JacksonResolverBuilder.java             |   7 +-
 .../solarthing/graphql/SolarThingResolverBuilder.java          |  10 +-
 .../solarthing/graphql/SolarThingTypeInfoGenerator.java        |  41 ++
 .../solarthing/graphql/packets/IdentifierFilter.java           |  15 +-
 .../retrodaredevil/solarthing/graphql/packets/PacketUtil.java  |   4 +
 .../solarthing/graphql/packets/SimplePacketNode.java           |   3 +
 .../solarthing/graphql/service/PacketGetter.java               |   4 +
 .../solarthing/graphql/service/SchemaConstants.java            |  12 +
 .../graphql/service/SolarThingGraphQLDailyService.java         |  12 +-
[... cut ...]
 other/grafana/datasource/vanilla_grafana_graphql.png           | Bin 0 -> 66696 bytes
 other/grafana/graphql_queries.md                               |  64 ++
 other/ideas/query.txt                                          |   6 -
 other/solar/README.md                                          |  15 +-
 program/.scripts/export_version.sh                             |   2 +-
 program/.scripts/solarthing.sh                                 |   6 +-
 program/graphql/run.sh                                         |   6 +-

retrodaredevil, 2021-02-16:
Did you also run the commands to update the GraphQL program? I think that's where the changes you want are. You'll end up restarting solarthing-graphql. You can leave the rover program running how it is if you'd like.

gecko951, 2021-02-16:
OK, yes graphql is updated and I'm going to shutdown and move the nanoiot back to my vehicle. :)

nanoiot:/opt/solarthing$ cd program/
gecko951@nanoiot:/opt/solarthing/program$ ./graphql_download_solarthing.sh
Going to download version: 2021.2.0 GraphQL
--2021-02-16 00:55:15--  https://github.com/wildmountainfarms/solarthing/releases/download/v2021.2.0/solarthing-graphql-2021.2.0.jar
Resolving github.com (github.com)... 140.82.112.4
Connecting to github.com (github.com)|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-releases.githubusercontent.com/195026020/78f47500-6efd-11eb-9b75-94997c6b598b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210216%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210216T005515Z&X-Amz-Expires=300&X-Amz-Signature=703f06530c745d9c798e65f9338dea1974e97b0973a3ae9a163d2bad9fe22290&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=195026020&response-content-disposition=attachment%3B%20filename%3Dsolarthing-graphql-2021.2.0.jar&response-content-type=application%2Foctet-stream [following]
--2021-02-16 00:55:15--  https://github-releases.githubusercontent.com/195026020/78f47500-6efd-11eb-9b75-94997c6b598b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210216%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210216T005515Z&X-Amz-Expires=300&X-Amz-Signature=703f06530c745d9c798e65f9338dea1974e97b0973a3ae9a163d2bad9fe22290&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=195026020&response-content-disposition=attachment%3B%20filename%3Dsolarthing-graphql-2021.2.0.jar&response-content-type=application%2Foctet-stream
Resolving github-releases.githubusercontent.com (github-releases.githubusercontent.com)... 185.199.108.154, 185.199.109.154, 185.199.110.154, ...
Connecting to github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.108.154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 22870237 (22M) [application/octet-stream]
Saving to: ‘solarthing-graphql-2021.2.0.jar’

solarthing-graphql-2021. 100%[================================>]  21.81M  7.11MB/s    in 3.1s

2021-02-16 00:55:19 (7.11 MB/s) - ‘solarthing-graphql-2021.2.0.jar’ saved [22870237/22870237]

Using SolarThing GraphQL Version: 2021.2.0

[2 more replies]

closed 2021-02-20 by retrodaredevil