wildmountainfarms/solarthing #20: wiring with max485

riker65, 2021-02-02:
Hi,
wondering if this is working with the max485 adapter.

Do you have a decripton of the wiring you are using in general?

Did not find it .

Thanks for guiding.

BR T

retrodaredevil, 2021-02-02:
For Rovers that use the RS232 port (not RS485), I have [this](https://github.com/wildmountainfarms/solarthing/blob/master/other/solar/resources/renogy_rover_rs232_pinout.png) (which I think is slightly incorrect, you need to hook TX to RX and RX to TX.

For Rovers using the RS485 communication, you can look at [this](https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/35092136259/original/DCC%20Charger%20Modbus%20RS485%20V1.7.pdf?response-content-type=application%2Fpdf&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAS6FNSMY2WD6T3JNC%2F20210202%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210202T073453Z&X-Amz-Expires=300&X-Amz-SignedHeaders=host&X-Amz-Signature=f94a16005d446b512d851d17a03f0c52cab4d766049fe34f54752de6b1247a1d). I might put that in the solarthing repository soon. I found it [here](https://support.renogy.com/en/support/discussions/topics/35000017634).

riker65, 2021-02-02:
Hi thanks a lot.
also found the renogy forum and answered as well there.

will check and dig into it.

Do you have any experience with MAX485 [https://arduinoinfo.mywikis.net/wiki/RS485-Modules](url)

retrodaredevil, 2021-02-02:
Personally I don't have any experience with MAX485. The best I have is the documentation for the pinout above.

closed 2021-02-02 by riker65