wildmountainfarms/solarthing #21: Getting data into Grafana? Documentation seems confusing

wcrice, 2021-02-04:
Hi! What a great project!

For starters, thank you. I am having trouble getting data into Grafana but I seem to be putting data into the databases and I seem to have Grafana and the GraphQL service up and running correctly. I'm missing something but I don't know what. At this point I have done the following (not necessarily in this order):

I am planning to use a Rover 30A but right now I'm using the dummy test file.

Install Java:

`sudo apt install openjdk-11-jre-headless`

Downloaded Solarthing:
`
curl https://raw.githubusercontent.com/wildmountainfarms/solarthing/master/other/linux/clone_install.sh | sudo bash
sudo usermod -a -G solarthing,dialout,tty,video $USER`

Configure Solarthing for Rover:

`cd /opt/solarthing/program/rover
cp ../../config_templates/io/default_linux_serial.json config/
cp ../../config_templates/base/rover_template.json config/base.json
`

Edit this bit of the base.json:
`{
  //...
  "dummy": "test/dummy_rover.json"
}`

Install (and configure) both couchdb and influxdb with these commands:

`curl -L https://couchdb.apache.org/repo/bintray-pubkey.asc | sudo apt-key add -
echo "deb https://apache.bintray.com/couchdb-deb bionic main" | sudo tee -a /etc/apt/sources.list
sudo apt update
sudo apt install couchdb
`
I made a password during installation that I used for the configuration later on Solarthing.

Install influxdb:
`sudo apt install influxdb`

Easy way to make a first user on influxdb:
` curl "http://localhost:8086/query" --data-urlencode "q=CREATE USER username WITH PASSWORD 'password' WITH ALL PRIVILEGES"
`

Configure couchdb and influxdb for my rover setup:

`cp ../../config_templates/databases/couchdb_template.json config/couchdb.json
cp ../../config_templates/databases/influxdb_template.json config/influxdb.json`

I edited both of those files to include the username and passwords I made above.

Now when I test run the rover file I get . what I think if a positive output:

`2021-02-04 17:24:06.543 [main] DEBUG me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver - Already had a design document for packets on database=solarthing`

`2021-02-04 17:24:06.591 [main] INFO  org.influxdb - --> POST http://localhost:8086/query?q=CREATE+DATABASE+default_database (0-byte body)`

`2021-02-04 17:24:06.594 [main] INFO  org.influxdb - <-- 200 OK http://localhost:8086/query?q=CREATE+DATABASE+default_database (2ms, unknown-length body)`

`2021-02-04 17:24:06.597 [main] INFO  org.influxdb - --> POST http://localhost:8086/write?db=default_database&precision=n&consistency=all (2870-byte body)`

`2021-02-04 17:24:06.601 [main] INFO  org.influxdb - <-- 204 No Content http://localhost:8086/write?db=default_database&precision=n&consistency=all (4ms, 0-byte body)`

`2021-02-04 17:24:06.602 [main] DEBUG me.retrodaredevil.solarthing.influxdb.influxdb1.InfluxDbPacketSaver - Wrote 1 packets to InfluxDB! database=default_database retention policy=null`

`2021-02-04 17:24:06.602 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4879ms`

Cool!

Install Grafana on Ubuntu 18.04:
[... 28 lines cut ...]
`

I made application.properties say:
`solarthing.config.database=../rover/config/couchdb.json`

And now I can enable the service and check on it:

`
sudo systemctl enable solarthing-graphql
sudo systemctl start solarthing-graphql
sudo systemctl status solarthing-graphql
`
It runs! It says active!

When I add a datasource to graphql (http://localhost:8080/graphql) I get the following:
`HTTP Response undefined: undefined
`

When I try connecting to influxdb with my database, username, and password from the influxdb.json file I get a webpage starting with:

`Cannot create property 'executedQueryString' on string '<!DOCTYPE html> <html lang="en"> <head> <script> !(function() { if ('PerformanceLongTaskTiming' in window) { var g = (window.__tti = { e: [] }); g.o = new PerformanceObserver(function(l) { g.e = g.e.concat(l.getEntries()); }); g.o.observe({ entryTypes: ['longtask'] }); } })(); </script> <meta charset="utf-8" /> <meta http-equiv="X-UA-Compatible" content="IE=edge,chrome=1" /> <meta name="viewport" content="width=device-width" /> <meta name="theme-color" content="#000" /> <title>Grafana</title> <base href="/" /> <link rel="preload" href="public/fonts/roboto/RxZJdnzeo3R5zSexge8UUVtXRa8TVwTICgirnJhmVJw.woff2" as="font" crossorigin /> <link rel="icon" type="image/png" href="public/img/fav32.png" /> <link rel="apple-touch-icon" sizes="180x180" href="public/img/apple-touch-icon.png" /> <link rel="mask-icon" href="public/img/grafana_mask_icon.svg" color="#F05A28" /> <link rel="stylesheet" href="public/build/grafana.dark.ddf7518003180e1ca7bc.css" /> <script> performance.mark('css done blocking'); </script> <meta name="apple-mobile-web-app-capable" content="yes" /> <meta name="apple-mobile-web-app-status-bar-style" content="black" /> <meta name="msapplication-TileColor" content="#2b5797" /> <meta name="msapplication-config" content="public/img/browserconfig.xml" /> </head> <body class="theme-dark app-grafana"> <style> .preloader { height: 100%; flex-direction: column; display: flex; justify-content: center; align-items: center; } .preloader__enter { opacity: 0; animation-name: preloader-fade-in; animation-iteration-count: 1; animation-duration: 0.9s; animation-delay: 1.35s; animation-fill-mode: forwards; } .preloader__bounce { text-align: center; animation-name: preloader-bounce; animation-duration: 0.9s; animation-iteration-count: infinite; } .preloader__logo { display: inline-block; animation-name: preloader-squash; animation-duration: 0.9s; animation-iteration-count: infinite; width: 60px; height: 60px; background-repeat: no-repeat; background-size: contain; background-image: url("data:image/svg+xml,%3csvg version='1.1' id='Layer_1' xmlns='http://www.w3.org/2000/svg' xmlns:xlink='http://www.w3.org/1999/xlink' x='0px' y='0px' width='351px' height='365px' viewBox='0 0 351 365' style='enable-background:new 0 0 351 365%3b' xml:space='preserve'%3e %3cstyle type='text/css'%3e .st0%7bfill:url(%23SVGID_1_)%3b%7d %3c/style%3e %3cg id='Layer_1_1_'%3e %3c/g%3e %3clinearGradient id='SVGID_1_' gradientUnits='userSpaceOnUse' x1='175.5' y1='445.4948' x2='175.5' y2='114.0346'%3e %3cstop offset='0' style='stop-color:%23FFF100'/%3e %3cstop offset='1' style='stop-color:%23F05A28'/%3e %3c/linearGradient%3e %3cpath class='st0' d='M342%2c161.2c-0.6-6.1-1.6-13.1-3.6-20.9c-2-7.7-5-16.2-9.4-25c-4.4-8.8-10.1-17.9-17.5-26.8 c-2.9-3.5-6.1-6.9-9.5-10.2c5.1-20.3-6.2-37.9-6.2-37.9c-19.5-1.2-31.9%2c6.1-36.5%2c9.4c-0.8-0.3-1.5-0.7-2.3-1 c-3.3-1.3-6.7-2.6-10.3-3.7c-3.5-1.1-7.1-2.1-10.8-3c-3.7-0.9-7.4-1.6-11.2-2.2c-0.7-0.1-1.3-0.2-2-0.3 c-8.5-27.2-32.9-38.6-32.9-38.6c-27.3%2c17.3-32.4%2c41.5-32.4%2c41.5s-0.1%2c0.5-0.3%2c1.4c-1.5%2c0.4-3%2c0.9-4.5%2c1.3c-2.1%2c0.6-4.2%2c1.4-6.2%2c2.2 c-2.1%2c0.8-4.1%2c1.6-6.2%2c2.5c-4.1%2c1.8-8.2%2c3.8-12.2%2c6c-3.9%2c2.2-7.7%2c4.6-11.4%2c7.1c-0.5-0.2-1-0.4-1-0.4c-37.8-14.4-71.3%2c2.9-71.3%2c2.9 c-3.1%2c40.2%2c15.1%2c65.5%2c18.7%2c70.1c-0.9%2c2.5-1.7%2c5-2.5%2c7.5c-2.8%2c9.1-4.9%2c18.4-6.2%2c28.1c-0.2%2c1.4-0.4%2c2.8-0.5%2c4.2 C18.8%2c192.7%2c8.5%2c228%2c8.5%2c228c29.1%2c33.5%2c63.1%2c35.6%2c63.1%2c35.6c0%2c0%2c0.1-0.1%2c0.1-0.1c4.3%2c7.7%2c9.3%2c15%2c14.9%2c21.9c2.4%2c2.9%2c4.8%2c5.6%2c7.4%2c8.3 c-10.6%2c30.4%2c1.5%2c55.6%2c1.5%2c55.6c32.4%2c1.2%2c53.7-14.2%2c58.2-17.7c3.2%2c1.1%2c6.5%2c2.1%2c9.8%2c2.9c10%2c2.6%2c20.2%2c4.1%2c30.4%2c4.5 c2.5%2c0.1%2c5.1%2c0.2%2c7.6%2c0.1l1.2%2c0l0.8%2c0l1.6%2c0l1.6-0.1l0%2c0.1c15.3%2c21.8%2c42.1%2c24.9%2c42.1%2c24.9c19.1-20.1%2c20.2-40.1%2c20.2-44.4l0%2c0 c0%2c0%2c0-0.1%2c0-0.3c0-0.4%2c0-0.6%2c0-0.6l0%2c0c0-0.3%2c0-0.6%2c0-0.9c4-2.8%2c7.8-5.8%2c11.4-9.1c7.6-6.9%2c14.3-14.8%2c19.9-23.3 c0.5-0.8%2c1-1.6%2c1.5-2.4c21.6%2c1.2%2c36.9-13.4%2c36.9-13.4c-3.6-22.5-16.4-33.5-19.1-35.6l0%2c0c0%2c0-0.1-0.1-0.3-0.2 c-0.2-0.1-0.2-0.2-0.2-0.2c0%2c0%2c0%2c0%2c0%2c0c-0.1-0.1-0.3-0.2-0.5-0.3c0.1-1.4%2c0.2-2.7%2c0.3-4.1c0.2-2.4%2c0.2-4.9%2c0.2-7.3l0-1.8l0-0.9 l0-0.5c0-0.6%2c0-0.4%2c0-0.6l-0.1-1.5l-0.1-2c0-0.7-0.1-1.3-0.2-1.9c-0.1-0.6-0.1-1.3-0.2-1.9l-0.2-1.9l-0.3-1.9 c-0.4-2.5-0.8-4.9-1.4-7.4c-2.3-9.7-6.1-18.9-11-27.2c-5-8.3-11.2-15.6-18.3-21.8c-7-6.2-14.9-11.2-23.1-14.9 c-8.3-3.7-16.9-6.1-25.5-7.2c-4.3-0.6-8.6-0.8-12.9-0.7l-1.6%2c0l-0.4%2c0c-0.1%2c0-0.6%2c0-0.5%2c0l-0.7%2c0l-1.6%2c0.1c-0.6%2c0-1.2%2c0.1-1.7%2c0.1 c-2.2%2c0.2-4.4%2c0.5-6.5%2c0.9c-8.6%2c1.6-16.7%2c4.7-23.8%2c9c-7.1%2c4.3-13.3%2c9.6-18.3%2c15.6c-5%2c6-8.9%2c12.7-11.6%2c19.6c-2.7%2c6.9-4.2%2c14.1-4.6%2c21 c-0.1%2c1.7-0.1%2c3.5-0 [... cut ...] .ddf7518003180e1ca7bc.js" type="text/javascript"></script> <script src="public/build/vendors~app.ddf7518003180e1ca7bc.js" type="text/javascript"></script> <script> performance.mark('js done blocking'); </script> </body> </html> ' `

So I think I have two separate ways to potentially connect but neither of them are working. I've tried nothing and I'm all out of options. Okay, that is a joke. I've tried a lot and I'm all out of ideas. Can you suggest anything? I feel like I'm so close!

Thank you!

retrodaredevil, 2021-02-04:
Alright so I want to clear up any confusion first. You are trying to get data into Grafana in two different ways.

First, CouchDB and SolarThing GraphQL. It sounds like you have this set up correctly. I'm not really sure why its giving you an undefined response when you put that in for the GraphQL datasource in Grafana. What do you mean by you set it to "browser"? Is there another option you can use? It's been a while since I've set that up and I don't remember what this is supposed to be set to.

Second, InfluxDB. Again, looks like this is set up correctly in SolarThing. From the debug, I can tell you are using InfluxDB 1.X. Can you make sure you are using the InfluxQL and not Flux as the query language, as flux is only supported on InfluxDB 1.8 and 2.0. I've never seen that error you got for InfluxDB, so I'm not really sure why that's happening either.

I'll admit helping people figure out why their databases aren't working is not something I'm best at because I don't set up databases very frequently. Sometime in the next few days I'll try to install some of these on my own, and hopefully update the documentation (maybe even suggest using docker instead of installing directly.)

Sorry I can't help much right now. Hopefully I'll figure out how to reproduce whatever you're doing wrong and then maybe I can help.

wcrice, 2021-02-04:
"Browser" is the option Grafana uses when my database server isn't visible from the web. For security reasons my server is locked down. It pretty much says Grafana to connect locally. Maybe GraphQL doesn't support that option?

The database stuff seems pretty confusing! I am sure you get a little tired of this.

Does Solarthing offer the option to save output files to a flat csv or tab file, appended one row at a time? If so, I already have code on this server that can read that kind of file and process it into an rShiny app. I should admit that I have never touched Java, I'm rusty with Python, but I can do most things in R. I also reluctantly use CRBasic (Campbell Scientific implementation of Basic). I would rewrite the driver for the Rover on CRBasic but wrapping my head around the weird Modbus addresses is making me a little crazy.

retrodaredevil, 2021-02-04:
Does GraphQL have a Server access option? I'm pretty sure InfluxDB has that option. From what I understand the if you use the "browser" option it basically means that whatever browser you use have to be able to access the GraphQL server itself, so the browser would be making the request.

SolarThing doesn't support csv. It does support a "POST" request option, which makes SolarThing send its raw json data to a webserver of your choice. [post template here](https://github.com/wildmountainfarms/solarthing/blob/master/config_templates/databases/post_json_template.json). This might be a good option if you want to get data out of SolarThing and use your own script to do what you want with the data.

I will try and figure out what those errors you got mean, I just can't do that for a little while until I find the time to set up a Grafana instance and play around with it. Which could be today or a week from today.

Also, can I ask what version of InfluxDB you're running?

And for the GraphQL program, can you stop it with `sudo systemctl stop solarthing-graphql` and then run it manually using `cd /opt/solarthing/program/graphql; ./run.sh`. Once you've done that, can you again try to use Grafana to access it? I want to make sure there are no errors in the output of the GraphQL program. Running it manually is the easiest way to see output of the GraphQL program as the logs are different from the other SolarThing programs.

wcrice, 2021-02-04:
This is all running from on the same server and everything has permission for localhost port access. It potentially could be a problem with GraphQL accessing the database if it is trying to access the database using 127.0.0.1 it should work. It would only be an issue if it was trying to connect from someplace on the internet.

I currently have InfluxDB version 1.8.4-1 installed.

I tried running GraphQL from the terminal and it still has the same error from Grafana. It also gave me a warning that may help make sense of my problem. I don't know what it means but maybe you do?

`WARNING: An illegal reflective access operation has occurred`
`WARNING: Illegal reflective access by me.retrodaredevil.solarthing.graphql.GraphQLProvider (jar:file:/opt/solarthing/program                          /.downloads/solarthing-graphql-SNAPSHOT.jar!/BOOT-INF/classes!/) to field java.lang.reflect.Field.modifiers`
`WARNING: Please consider reporting this to the maintainers of me.retrodaredevil.solarthing.graphql.GraphQLProvider`
`WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations`
`WARNING: All illegal access operations will be denied in a future release
file: ../rover/config/couchdb.json absolute: /opt/solarthing/program/graphql/../rover/config/couchdb.json`
`No solcast config! Not using solcast!`
`Using defaultInstanceOptions=DefaultInstanceOptions(defaultSourceId='<UNUSED SOURCE ID THAT WILL NEVER BE IN A PACKET>', def                          aultFragmentId=0)`

Thank you!

retrodaredevil, 2021-02-04:
That warning can be ignored. I'll try to silence it in an update. Just to make sure, after that warning, you get a message like `Started SolarThingGraphQLApplication in 2.875 seconds (JVM running for 3.566)`, right?

Assuming you get that message I mentioned, then the GraphQL program is running perfectly. Do you get any messages in there if you try to access it from Grafana? Navigate to `localhost:8080` and make sure that *something* pops up in the logs. If it does, that's good. Now try to access it from Grafana. Again, you should get some more log messages in the output. If you don't, then something isn't allowing you to access it.

wcrice, 2021-02-04:
I should have specified. After that warning everything seems to run fine!

`2021-02-04 22:56:00.474  INFO 26950 --- [           main] o.s.s.concurrent.ThreadPoolTaskExecutor  : Initializing ExecutorService 'applicationTaskExecutor'`
`2021-02-04 22:56:00.709  INFO 26950 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat started on port(s): 8080 (http) with context path ''`
`2021-02-04 22:56:00.726  INFO 26950 --- [           main] m.r.s.g.SolarThingGraphQLApplication     : Started SolarThingGraphQLApplication in 4.3 seconds (JVM running for 4.983)`

And yeah, I just test. From an actual browser I get "Whitelabel Error Page" and a 404 error. From the command curl "http://localhost:8080" I get essentially a 404. If I close the process for GraphQL I get a "Failed to connect to localhost port 8080" with my curl command so I know it did something.

Maybe I should specify that this thing is running headless on an Ubuntu 18.04 server.

I'm still getting the same error connecting to Grafana. Maybe I'm just not setting up my data source correctly?

retrodaredevil, 2021-02-04:
When you attempt to connect to GraphQL in Grafana, do the logs say anything in SolarThing GrahpQL? Try relaunching the GraphQL program with ./run.sh, then look to see if anything pops up *after* `Started SolarThingGraphQLApplication`.

If anything pops up after that, can you send me those logs? If nothing pops up after that, then for some reason Grafana can't access it.

wcrice, 2021-02-04:
Good thought! It only seems to give me something when I use the access method "Server (default)" but doesn't do anything with Browser. It still gives an error in Grafana: `HTTP Response undefined: undefined`

Tis is what GraphQL gives:
`2021-02-04 23:25:53.396  INFO 28412 --- [           main] o.s.s.concurrent.ThreadPoolTaskExecutor  : Initializing ExecutorService 'applicationTaskExecutor'`
`2021-02-04 23:25:53.634  INFO 28412 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat started on port(s): 8080 (http) with context path ''`
`2021-02-04 23:25:53.652  INFO 28412 --- [           main] m.r.s.g.SolarThingGraphQLApplication     : Started SolarThingGraphQLApplication in 4.523 seconds (JVM running for 5.256)`
`2021-02-04 23:39:55.291  INFO 28412 --- [nio-8080-exec-1] o.a.c.c.C.[Tomcat].[localhost].[/]       : Initializing Spring DispatcherServlet 'dispatcherServlet'`
`2021-02-04 23:39:55.292  INFO 28412 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet        : Initializing Servlet 'dispatcherServlet'`
`2021-02-04 23:39:55.301  INFO 28412 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet        : Completed initialization in 9 ms`

And then that's it. I get the error in Grafana and GraphQL gives no further output.

retrodaredevil, 2021-02-05:
Ok, well it's good that Grafana can access SolarThing GraphQL. That means a bunch of things are going right. I was looking at the GraphQL datasource source code and it looks like the GraphQL datasource thinks it's getting some kind of error.

When you test the GraphQL Datasource in Grafana and it gives you `HTTP Response undefined: undefined`, can you go into the console of your browser (by pressing F12 usually) and give me the most recent log from it? It should look something like `Object {...}` If you could right click on that, click copy object, and send that to me here.

wcrice, 2021-02-05:
OH! I'm so sorry! I'm 100% sure that at some point I told it to go to the correct URL but during your process of helping me troubleshoot I must have 1) fixed the underlying problem and 2) screwed up the URL so instead of reading http://localhost:8080/graphql it only read http://localhost:8080

Uhg! I didn't notice it until I went back and did what you asked and got an error that made me go back and reconsider my URL. Thank you for all your help! Now I get to start tinkering.

retrodaredevil, 2021-02-05:
Ah yeah I should have stressed that the `/graphql` part is important. I did mention just navigating straight to `localhost:8080` in a browser earlier.

So when you test the connection do you still get the `HTTP Response undefined: undefined`?

Is the GraphQL datasource working for you now? Can this issue be closed? If you want me to help you get InfluxDB working you can keep this issue open. Otherwise feel free to close it if everything is working fine.

Sorry in advance that there aren't really any examples for GraphQL. Once I get remote access back to the system I monitor, I'll get the JSON for the Grafana dashboard. [This dashboard](https://github.com/wildmountainfarms/solarthing/blob/master/other/grafana/dashboards/solar_graphql_wildmountainfarms.json) has a combination of InfluxDB annotation queries and regular GraphQL queries that you can look at. I don't recommend copy-pasting it as that usually doesn't work very well.

wcrice, 2021-02-05:
Well, I think GraphQL is working because when I run:
{
  data:queryStatusLast(to:"$timeTo", sourceId:"default") {
    batteryVoltage {
      Time:dateMillis
      packet {
        batteryVoltage
      }
    }
  }
}

I get:

batteryVoltage.0.Time
1612486860967

batteryVoltage.0.packet.batteryVoltage
26

I still get diddly from InfluxDB and it seems to be running correctly. I'm still happy to keep troubleshooting! I wasn't trying to use you as free tech support for something that you made for free. I thought I caught a bug! Or at least I found a small flaw in the documentation. Considering the entire process I went through above was run from the terminal I wonder if a simple bash script asking for user responses could be written to fill in the handful of variables in a "default" installation like mine? Maybe there is no such thing as a default installation.

Thank you again!

wcrice, 2021-02-05:
And if you could write an example Grafana dashboard that uses GraphQL that would be AMAZING because I'm looking at all of this and... well, I'm sure it is very intuitive but my brain is full of code and jargon and I can't see it right now.

retrodaredevil, 2021-02-05:
I'm glad you got GraphQL working. Yeah I will get the documentation for GraphQL better soon. I've been using GraphQL for a while, but just recently did people start using it for SolarThing. CouchDB used to be pretty difficult to set up but now it's much easier.

How are you trying to access InfluxDB? I don't think InfluxDB 1.8 has a web interface, so you can only do things through a terminal I think.

wcrice, 2021-02-05:
I have only tried using the influx CLI through the terminal but I was able to make a username and password for myself using a version of the command:
`curl "http://localhost:8086/query" --data-urlencode "q=CREATE USER username WITH PASSWORD 'password' WITH ALL PRIVILEGES"`

When I try connecting to myself I follow something pretty similar to this: https://siytek.com/influxdb-grafana-setup/

But at the very last step instead of a friendly green "Data source is working" all I see accessing it via "server (default)" is:
`InfluxDB Error: Bad Gateway' 

And browser is the long gobbldygook I posted in my first message.

I may be on to something.. When I pulled up sudo systemctl status influxd.service I got this:

Feb 05 01:30:06 ip-172-31-21-135 influxd[20330]: [httpd] 127.0.0.1 - bill [05/Feb/2021:01:30:06 +0000] "POST /write?db=default_database&precision=n&consistency=all HTTP/1.1" 204 0 "-" "okhttp/4.9.0" ad2da05e-6751-11eb-868d-06c419357b84 3052
Feb 05 01:30:11 ip-172-31-21-135 influxd[20330]: ts=2021-02-05T01:30:11.038198Z lvl=info msg="Executing query" log_id=0S7_AxO0000 service=query query="CREATE DATABASE default_database"
Feb 05 01:30:11 ip-172-31-21-135 influxd[20330]: [httpd] 127.0.0.1 - bill [05/Feb/2021:01:30:11 +0000] "POST /query?q=CREATE+DATABASE+default_database HTTP/1.1" 200 57 "-" "okhttp/4.9.0" b0285dbe-6751-11eb-868e-06c419357b84 422
Feb 05 01:30:11 ip-172-31-21-135 influxd[20330]: [httpd] 127.0.0.1 - bill [05/Feb/2021:01:30:11 +0000] "POST /write?db=default_database&precision=n&consistency=all HTTP/1.1" 204 0 "-" "okhttp/4.9.0" b0288da6-6751-11eb-868f-06c419357b84 2826
Feb 05 01:30:16 ip-172-31-21-135 influxd[20330]: ts=2021-02-05T01:30:16.061093Z lvl=info msg="Executing query" log_id=0S7_AxO0000 service=query query="CREATE DATABASE default_database"
Feb 05 01:30:16 ip-172-31-21-135 influxd[20330]: [httpd] 127.0.0.1 - bill [05/Feb/2021:01:30:16 +0000] "POST /query?q=CREATE+DATABASE+default_database HTTP/1.1" 200 57 "-" "okhttp/4.9.0" b326cc97-6751-11eb-8690-06c419357b84 405
Feb 05 01:30:16 ip-172-31-21-135 influxd[20330]: [httpd] 127.0.0.1 - bill [05/Feb/2021:01:30:16 +0000] "POST /write?db=default_database&precision=n&consistency=all HTTP/1.1" 204 0 "-" "okhttp/4.9.0" b32714c2-6751-11eb-8691-06c419357b84 3072
Feb 05 01:30:21 ip-172-31-21-135 influxd[20330]: ts=2021-02-05T01:30:21.038027Z lvl=info msg="Executing query" log_id=0S7_AxO0000 service=query query="CREATE DATABASE default_database"
Feb 05 01:30:21 ip-172-31-21-135 influxd[20330]: [httpd] 127.0.0.1 - bill [05/Feb/2021:01:30:21 +0000] "POST /query?q=CREATE+DATABASE+default_database HTTP/1.1" 200 57 "-" "okhttp/4.9.0" b61e3828-6751-11eb-8692-06c419357b84 981

Which made me wonder if the server is just making a new database over and over and over again. I connected to influx via terminal and ran `show database` which listed default_database so I think it exists. Yep, I'm lost. I'm getting more familiar with how you have this setup and it is a wonderful piece of software! It sure is complicated and getting everything to play nicely is a challenge, but it seems pretty cool when it works!

retrodaredevil, 2021-02-05:
SolarThing will attempt to create the `default_database` over and over again, just to make sure it's there. This is normal.

Can you follow some of the commands on [this page](https://docs.influxdata.com/influxdb/v1.8/administration/authentication_and_authorization/) and make sure that your user you created is actually created? `curl -G http://localhost:8086/query -u username:password --data-urlencode "q=SHOW DATABASES"`. I expect that to work, but I just want to make sure.

wcrice, 2021-02-05:
Well, your query SHOW DATABASES via RFC 2617, query parameters, and using the request body all gave me:
{"results":[{"statement_id":0,"series":[{"name":"databases","columns":["name"],"values":[["_internal"],["default_database"]]}]}]}

OH, I just saw your mention about using `http://localhost:8086/` and that fixed it. Okay, weird. Cool. Well, that is neat and confusing. So for some reason my computer was stuck on that `/` and the GraphQL thing was probably because I needed to start and stop all the services a few times.... I think in the end that was all I did. Huh! Thank you SO MUCH!

retrodaredevil, 2021-02-05:
> OH, I just saw your mention about using `http://localhost:8086/`

That's kind of dumb that this was the reason it wasn't working... I'll be sure to stress that somewhere in the documentation when setting up InfluxDB. I'm sure I ran into this issue a while ago and if I didn't... then I'm just lucky. Honestly this should probably be fixed in the InfluxDB datasource.

closed 2021-02-11 by retrodaredevil