wildmountainfarms/solarthing #22: Solarthing not noticing when ttyP1 comes back after disconnect

wcrice, 2021-02-09:
I have a working system! Thank you for your awesome code to make this happen. I even made a dashboard to connect to my Rover. What I'm troubleshooting now is making the service come back to life after a disconnect on my tty. My tty is over TCP.

I make a connection as a system service using the following command controlled by a systemctl service:

`#!/usr/bin/env sh`
`socat TCP4:192.168.1.50:3001 PTY,link=/dev/ttyP1,wait-slave,raw`

Unfortunately, it occasionally drops a connection to the network and solarthing doesn't seem to resume properly. Solarthing is running as a service. If the connection is in place and I run the solarthing service it works fine but if the tty drops for a while the solarthing doesn't seem to notice that it came back. I think when the device I'm calling /dev/ttyP1 is recreated, solarthing doesn't have access to it anymore. Is there a way to make Solarthing error out the service and terminate if it fails to connect to the tty?

retrodaredevil, 2021-02-09:
I'm not familiar with having a tty go over a network, so this isn't something I've considered before. SolarThing is set up to just keep on spewing out errors whenever a serial port becomes disconnected, so I imagine that's what happens when your tty over TCP becomes disconnected.

I could try and make SolarThing recreate the "serial port" whenever it disconnects. Coincidentally, this should also help out people who unplug and plug their serial port adapter back in while SolarThing is running. I'll try updating SolarThing to make this happen in the next couple of days. I'll report back here once it's fixed. I might have you compile SolarThing yourself to test this out, unless you'd like me to do it. (It's very easy if your pi/computer running SolarThing has a stable internet connection).

retrodaredevil, 2021-02-10:
Does SolarThing straight up crash or does SolarThing try to continue? Can you send me what the error looks like? (`/opt/solarthing/program/rover/logs/log_info.log` should have the error(s)`.

wcrice, 2021-02-10:
I think SolarThing is behaving the way you intended it to in that it keeps running and gives an error. I just watched the log you pointed me to and simulated a disconnect by rebooting the cell modem remotely. The systemctl service running the socat tty and the systemctl service running SolarThing kept running without a problem. I got this error from SolarThing:

```
2021-02-10 16:17:31.156 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.ModbusTimeoutException: Timed out! startTime=1612973849154 currentTime=1612973851156 initialTimeout=2000
        at me.retrodaredevil.io.modbus.RtuDataEncoder.readBytes(RtuDataEncoder.java:111) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.io.modbus.RtuDataEncoder.readMessage(RtuDataEncoder.java:59) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.io.modbus.IOModbusSlaveBus.sendRequestMessage(IOModbusSlaveBus.java:26) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.io.modbus.ImmutableAddressModbusSlave.sendRequestMessage(ImmutableAddressModbusSlave.java:14) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:24) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:28) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheRangeInclusive(ModbusCacheSlave.java:36) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$3(RoverMain.java:79) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRoverProgram$6(RoverMain.java:137) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:61) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:112) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:77) ~[solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:143) [solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:50) [solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:102) [solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:198) [solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:262) [solarthing-2021.1.2.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:266) [solarthing-2021.1.2.jar:?]
2021-02-10 16:17:31.157 [main] INFO  me.retrodaredevil.solarthing.program.RoverPacketListUpdater -
```

Hey! We noticed you got a ModbusTimeoutException.
This is likely a problem with your cable. SolarThing is communicating fine with your RS232 adapter, but it cannot reach the Rover.
Make sure the cable you have has the correct pinout, and feel free to open an issue at https://github.com/wildmountainfarms/solarthing/issues if you need help.

```
2021-02-10 16:17:39.729 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending Rover status to Google Analytics. data=RNG-CTRL-RVR30,30 uptime hours=0
ubuntu@ip-172-31-21-135:~$ sudo systemctl status solarthing-rover.service
● solarthing-rover.service - The service for SolarThing Rover
   Loaded: loaded (/etc/systemd/system/solarthing-rover.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-02-10 16:16:32 UTC; 2min 4s ago
 Main PID: 7520 (sh)
    Tasks: 23 (limit: 4915)
   CGroup: /system.slice/solarthing-rover.service
           ├─7520 sh /opt/solarthing/program/rover/run.sh
           ├─7533 sh ../.scripts/solarthing.sh
           └─7541 java --add-opens=java.base/java.lang.invoke=ALL-UNNAMED -jar ../.scripts/../solarthing.jar config/base.json
```

But the socat connection is back up. I have it set to read out the data from the service:

```
ubuntu@ip-172-31-21-135:~$ sudo systemctl status rover.service
● rover.service - Connect Rover Charter RS232 to ttyP1
   Loaded: loaded (/etc/systemd/system/rover.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-02-10 16:16:28 UTC; 5min ago
 Main PID: 7501 (sh)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/rover.service
           ├─7501 /bin/sh -c /opt/solarthing/rover_rs232.sh
           ├─7504 sh /opt/solarthing/rover_rs232.sh
           └─7509 socat -d -d -d PTY,link=/dev/ttyP1,echo=0,raw,unlink-close=0 TCP4:192.168.1.50:3001,forever,reuseaddr,keepalive,keepidle=10,keepintvl=10,keepcnt=2
Feb 10 16:21:59 ip-172-31-21-135 sh[7501]: 2021/02/10 16:21:59 socat[7509] I transferred 8 bytes from 5 to 7
Feb 10 16:21:59 ip-172-31-21-135 sh[7501]: 2021/02/10 16:21:59 socat[7509] I transferred 7 bytes from 7 to 5
Feb 10 16:21:59 ip-172-31-21-135 sh[7501]: 2021/02/10 16:21:59 socat[7509] I transferred 8 bytes from 5 to 7
Feb 10 16:21:59 ip-172-31-21-135 sh[7501]: 2021/02/10 16:21:59 socat[7509] I transferred 7 bytes from 7 to 5
[... cut ...] and it just came right back! But yesterday it didn't... There were hours of missing data.

Is it possible to make a flag in the systemctl service so SolarThing reopens if there is a failure or tries to re-create a tty connection if there is an RS232 disconnect? I think that could have applications for people who aren't just in my strange situation. It seems to just take a service restart for the tty and Solarthing (in that order) to reliably fix it. For some reason today it fixed it on it's own.

retrodaredevil, 2021-02-10:
Dang, I was hoping it would be anything except that timeout exception. 

> Is it possible to make a flag in the systemctl service so SolarThing reopens if there is a failure or tries to re-create a tty connection if there is an RS232 disconnect?

Yes, very possible. There's no perfect way for SolarThing to detect the disconnection, but I could try and make it reinitialize if it has a few timeouts in a row (should be equivalent to restarting the solarthing-rover service).

wcrice, 2021-02-10:
Ah. I see. Is there anything I can do from my side about this a timeout in SolarThing? Would it be super complicated to make a future version apply a default setting and let us change the timeout in the base.json? Or make it run a user selectable command if it times out, like `systemctl restart rover_rs232.service && systemctl restart solarthing-rover.service`

I'm sorry I'm not really able to make substantial code for you. I don't know Java well enough to be helpful...

retrodaredevil, 2021-02-10:
I'm working on a fix right now. Basically if SolarThing can't communicate with the "serial port" after a few tries, it will try to reinitialize the serial port. Which hopefully is basically the same as restarting the service, without completely restarting it.

retrodaredevil, 2021-02-10:
Alright, I have a mostly untested fix for you to try. If you could compile it yourself, that would be helpful so I know it works before I do a release.

You can run this:
```shell
cd /opt/solarthing
git pull
./compile_and_move.sh
sudo systemctl restart solarthing-rover
```

> Ah. I see. Is there anything I can do from my side about this a timeout in SolarThing?

It's not actually a timeout. SolarThing just sees it as a timeout because the serial port (from SolarThing's view), isn't giving any data. If you need the ability to customize a command to run after enough timeouts after this fix, let me know. I hope this fix that I've done is enough to correctly reload the connection.

wcrice, 2021-02-10:
Weirdly, after pulling, compiling, and restarting (or running from command) I can't get it running again.

I get this:

```
ubuntu@ip-172-31-21-135:/opt/solarthing/program/rover$ sudo ./run.sh
Using base config = config/base.json
2021-02-10 19:59:35.694 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: solarthing-SNAPSHOT.jar Java version: 11.0.10
[stdout] Beginning main. Jar: solarthing-SNAPSHOT.jar Java version: 11.0.10
[stderr] Beginning main. Jar: solarthing-SNAPSHOT.jar Java version: 11.0.10
2021-02-10 19:59:35.765 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2021-02-10 19:59:35.881 [main] INFO  me.retrodaredevil.solarthing.program.RoverMain - Beginning rover program
2021-02-10 19:59:35.885 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2021-02-10 19:59:35.892 [main] DEBUG me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics UUID: a79db4cb-bcdc-4b98-a582-a2b9ec583bd3
2021-02-10 19:59:36.177 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2021-02-10 19:59:36.663 [main] DEBUG me.retrodaredevil.solarthing.io.ReloadableIOBundle - Successfully reloaded IOBundle
2021-02-10 19:59:36.769 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - You are using InfluxDB 1.X! It is recommended that you switch to 2.0, but is not required.
2021-02-10 19:59:38.785 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.ModbusTimeoutException: Timed out! startTime=1612987176782 currentTime=1612987178785 initialTimeout=2000
        at me.retrodaredevil.io.modbus.RtuDataEncoder.readBytes(RtuDataEncoder.java:111) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.io.modbus.RtuDataEncoder.readMessage(RtuDataEncoder.java:59) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.io.modbus.IOModbusSlaveBus.sendRequestMessage(IOModbusSlaveBus.java:26) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.io.modbus.ImmutableAddressModbusSlave.sendRequestMessage(ImmutableAddressModbusSlave.java:14) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:24) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:28) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheRangeInclusive(ModbusCacheSlave.java:36) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$3(RoverMain.java:83) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRoverProgram$8(RoverMain.java:141) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:66) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:111) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:81) ~[solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:147) [solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:51) [solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:106) [solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:202) [solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:242) [solarthing-SNAPSHOT.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:270) [solarthing-SNAPSHOT.jar:?]
2021-02-10 19:59:38.790 [main] INFO  me.retrodaredevil.solarthing.program.RoverPacketListUpdater -
Hey! We noticed you got a ModbusTimeoutException.
This is likely a problem with your cable. SolarThing is communicating fine with your serial adapter, but it cannot reach the Rover.
Make sure the cable you have has the correct pinout, and feel free to open an issue at https://github.com/wildmountainfarms/solarthing/issues if you need help.
```

But weirdly when SolarThing is running it _is_ talking to the port because I can watch it pass data:

```
Feb 10 20:00:23 ip-172-31-21-135 sh[13567]: 2021/02/10 20:00:23 socat[13580] I transferred 8 bytes from 5 to 7
Feb 10 20:00:24 ip-172-31-21-135 sh[13567]: 2021/02/10 20:00:24 socat[13580] I transferred 39 bytes from 7 to 5
Feb 10 20:01:59 ip-172-31-21-135 sh[13567]: 2021/02/10 20:01:59 socat[13580] I transferred 8 bytes from 5 to 7
Feb 10 20:01:59 ip-172-31-21-135 sh[13567]: 2021/02/10 20:01:59 socat[13580] I transferred 39 bytes from 7 to 5
Feb 10 20:02:04 ip-172-31-21-135 sh[13567]: 2021/02/10 20:02:04 socat[13580] I transferred 8 bytes from 5 to 7
Feb 10 20:02:04 ip-172-31-21-135 sh[13567]: 2021/02/10 20:02:04 socat[13580] I transferred 39 bytes from 7 to 5
Feb 10 20:02:09 ip-172-31-21-135 sh[13567]: 2021/02/10 20:02:09 socat[13580] I transferred 8 bytes from 5 to 7
Feb 10 20:02:09 ip-172-31-21-135 sh[13567]: 2021/02/10 20:02:09 socat[13580] I transferred 39 bytes from 7 to 5
Feb 10 20:02:14 ip-172-31-21-135 sh[13567]: 2021/02/10 20:02:14 socat[13580] I transferred 8 bytes from 5 to 7
Feb 10 20:02:14 ip-172-31-21-135 sh[13567]: 2021/02/10 20:02:14 socat[13580] I transferred 39 bytes from 7 to 5
```

Each time SolarThing talks to the port I see an update on this port...

retrodaredevil, 2021-02-10:
Oh, maybe the timeout is actually a timeout. Every time I see a timeout I usually doubt it because there was one time when I made the timeout infinite (with a connected serial port), and it still didn't work.

What's weird is that if this is actually a timeout, then it takes more than 2 seconds to get data over the network., which it doesn't look like that's the case since those log messages show data going there and back in less than a second.

It's possible that my small update messed something else up. I'll try to look into it and I'll get back to you if I have any ideas.

retrodaredevil, 2021-02-10:
Ok, I had a bug in the code to where the rover program would never actually work (I wasn't able to test it lol, that's what you're for).

Try recompiling with the commands I had above. It should work now.

wcrice, 2021-02-10:
That seems to have fixed it. I'll let you know if it does okay over the next few weeks. I'm sure I will have a disconnect on the modem eventually.

FYI, the "server" is an Amazon AWS that handles a bunch of central data aggregation for my research.

retrodaredevil, 2021-02-10:
Depending on how frequently it was disconnecting before, you can see if it reinitialized the IO by looked at the `/opt/solarthing/program/rover/logs/log_info.log` log file. If you see it timeout 5 times, then reinitialize, then the fix worked. Let me know how it goes.

retrodaredevil, 2021-02-15:
I'm gonna go ahead and close this on the assumption that this is working for you. Releases 2021.2.0 and onward will have this feature.

Let me know if this just doesn't work for you for some reason.

closed 2021-02-15 by retrodaredevil