wildmountainfarms/solarthing #23: Re-Read dummy file

SkyMoCo, 2021-02-11:
This is very much a "nice to have" but it seems that the rover dummy file only get's read once at startup.  I am working with various graph and alert settings and it would be nice to just change the dummy.json file and have it be re-read.   I guess that's a bit problematic with timing and all so it might not be worth it.   Just a thought.

retrodaredevil, 2021-02-11:
I might add this in the future, but I'm not sure how much code I'd have to change.

Currently, the file is read once, then if the program wants to change any parameters of the rover with set commands, it will "change" the corresponding value and print a debug message. This makes testing the `rover-setup` program easy and testing commands easy. However re-reading the file may not play nicely with this. It would definitely be possible, but it's not high on my priority to look into right now.

SkyMoCo, 2021-02-11:
It's easy enough for me to restart solarthing to see if my charts go red or not.   So don't worry about it.

closed 2021-02-11 by SkyMoCo