wildmountainfarms/solarthing #24: Future plans for inverter support

SkyMoCo, 2021-02-11:
I'm looking to add an inverter/charger to my mix.   Do you have any plans to add support for something like that.   I haven't bought one yet so if you have something in the works I'll go with that.

retrodaredevil, 2021-02-11:
SolarThing already supports outback FX inverters, assuming you have a ~$200 mate too, so that isn't the best choice budget-wise.

I currently don't have anything in the works, but if Renogy puts out an inverter with the ability to monitor it through their BT-2 module, then I might consider supporting it. There's a chance that [this model](https://www.renogy.com/1000w-12v-pure-sine-wave-inverter-with-power-saving-mode-new-edition/) supports it, but on that page there's 0 mention of being compatible with the BT-2 module even though there's a comm port. The manual does say it supports the BT-2 module, so maybe soon Renogy will start really advertising that model.

No guarantees that I'll end up supporting Renogy's New Edition inverter in SolarThing, but it looks like it has the possibility of being monitored. I'm only mentioning that because before I knew that existed I went on a search for cheap inverters with the ability to be monitored and I couldn't find any.

retrodaredevil, 2021-02-20:
I was able to get the Modbus document for the newer Renogy inverter. But the document is huge (similar size to the Rover document) and I'm not going to partially add stuff to SolarThing, so I'll probably never add it unless I actually get one, which is pretty unlikely.

closed 2021-02-20 by retrodaredevil

SkyMoCo, 2021-02-24:
Would you be able to share that document with me?   Just in case I end up
with one.

retrodaredevil, 2021-02-24:
Here it is:
[Inverter Modbus Protocol V1.6.zh-CN.en.pdf](https://github.com/wildmountainfarms/solarthing/files/6038868/Inverter.Modbus.Protocol.V1.6.zh-CN.en.pdf)

I just emailed Renogy asking for it, so I imagine they'll give it to anyone. I assume they also have documents for their other devices. I keep two modbus files in this repo, but I might avoid that in the future since git isn't the best way to store binary files.

retrodaredevil, 2021-10-20:
SolarThing currently doesn't support any Renogy inverters. I imagine it'll stay that way unless I get one.