wildmountainfarms/solarthing #25: install for renogy DCC50S

riker65, 2021-02-16:
Hi,
maybe I missed something but could you guide me how to install and config
for Renogy DCC50S?

in quick I only found reference to Rover

[**Rover Quickstart**](quickstart_rover.md) - Monitors Renogy Rover and other supported products

Thanks for your support

BR Thomas

retrodaredevil, 2021-02-16:
Follow the rover quickstart normally, and it will work perfectly for the Renogy DCC50S. Don't get confused whenever you see "rover". SolarThing works Rover, Rover Elite, Wanderer, Adventurer, and the dual input DCDC charcgers, such as the DCC50S. SolarThing calls all of these devices "rovers".

riker65, 2021-02-16:
ok thanks, will try

riker65, 2021-02-16:
Hi
I am confused in the couchdb setting section.

unclear which path I should be using:

in `java -jar solarthing.jar --couchdb-setup config/couchdb.json`
solarthing.jar is not present in ../prgram/rover or wenn in ../program config/couchdb.json not found 
should be rover/config/couchdb.json? 
 

```
## Required setup
SolarThing has a special program to help you set up CouchDB. We're going to run that now. You should have
created a couchdb.json file.

You will now run this command:
```shell
java -jar solarthing.jar --couchdb-setup config/couchdb.json
```
```

riker65, 2021-02-16:
another question:

what is the sequence installing things?

did setup rover
java
now a started couchdb config but got an error 

either waiting for enter? - when doing <enter>   error appears

`java -jar ../solarthing.jar --couchdb-setup config/couchdb.json
2021-02-16 20:14:43.115 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: solarthing-2021.2.0.jar Java version: 11.0.9.1
[stdout] Beginning main. Jar: solarthing-2021.2.0.jar Java version: 11.0.9.1
[stderr] Beginning main. Jar: solarthing-2021.2.0.jar Java version: 11.0.9.1
You will now setup your CouchDB instance! Some databases will be automatically created (enter)
`

Thanks

retrodaredevil, 2021-02-16:
Looks like that command should be `java -jar ../solarthing.jar --couchdb-setup config/couchdb.json` if you are in the `/opt/solarthing/program/rover` directory. You could run that from the `/opt/solarthing/program` directory, it doesn't really matter. You just have to give it the correct path to your CouchDB config.

> either waiting for enter? - when doing error appears

That's not an error. It tells you to press enter to continue the setup.

The sequence is something like install CouchDB, create your couchdb.json file, run the CouchDB setup, then start the rover program once you've configured it.

riker65, 2021-02-17:
> Looks like that command should be `java -jar ../solarthing.jar --couchdb-setup config/couchdb.json` if you are in the `/opt/solarthing/program/rover` directory. You could run that from the [...]

Unfortunately I am facing some error while creating the couchdb

couchdb is up and running.
create the admin user and edit in in config/couchdb.json

after pressing enter: first message was goo created solarthing, but after that the errors: 

thanks for hints.

BR T
`
Created solarthing
Exception in thread "main" org.ektorp.DbAccessException: java.net.SocketTimeoutException: Read timed out
        at org.ektorp.util.Exceptions.propagate(Exceptions.java:19)
        at org.ektorp.http.StdHttpClient.executeRequest(StdHttpClient.java:183)
        at org.ektorp.http.StdHttpClient.executeRequest(StdHttpClient.java:192)
        at org.ektorp.http.StdHttpClient.put(StdHttpClient.java:126)
        at org.ektorp.http.RestTemplate.put(RestTemplate.java:43)
        at org.ektorp.impl.StdCouchDbInstance.createDatabaseIfNotExists(StdCouchDbInstance.java:68)
        at org.ektorp.impl.StdCouchDbInstance.createDatabaseIfNotExists(StdCouchDbInstance.java:59)
        at me.retrodaredevil.solarthing.program.CouchDbSetupMain.createDatabase(CouchDbSetupMain.java:23)
        at me.retrodaredevil.solarthing.program.CouchDbSetupMain.doCouchDbSetupMain(CouchDbSetupMain.java:37)
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:258)
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:270)
Caused by: java.net.SocketTimeoutException: Read timed out
        at java.base/java.net.SocketInputStream.socketRead0(Native Method)
        at java.base/java.net.SocketInputStream.socketRead(SocketInputStream.java:115)
        at java.base/java.net.SocketInputStream.read(SocketInputStream.java:168)
        at java.base/java.net.SocketInputStream.read(SocketInputStream.java:140)
        at org.apache.http.impl.io.AbstractSessionInputBuffer.fillBuffer(AbstractSessionInputBuffer.java:158)
        at org.apache.http.impl.io.SocketInputBuffer.fillBuffer(SocketInputBuffer.java:82)
        at org.apache.http.impl.io.AbstractSessionInputBuffer.readLine(AbstractSessionInputBuffer.java:271)
        at org.apache.http.impl.conn.DefaultHttpResponseParser.parseHead(DefaultHttpResponseParser.java:138)
        at org.apache.http.impl.conn.DefaultHttpResponseParser.parseHead(DefaultHttpResponseParser.java:56)
        at org.apache.http.impl.io.AbstractMessageParser.parse(AbstractMessageParser.java:259)
        at org.apache.http.impl.AbstractHttpClientConnection.receiveResponseHeader(AbstractHttpClientConnection.java:281)
        at org.apache.http.impl.conn.DefaultClientConnection.receiveResponseHeader(DefaultClientConnection.java:257)
        at org.apache.http.impl.conn.ManagedClientConnectionImpl.receiveResponseHeader(ManagedClientConnectionImpl.java:207)
        at org.apache.http.protocol.HttpRequestExecutor.doReceiveResponse(HttpRequestExecutor.java:273)
        at org.apache.http.protocol.HttpRequestExecutor.execute(HttpRequestExecutor.java:125)
        at org.apache.http.impl.client.DefaultRequestDirector.tryExecute(DefaultRequestDirector.java:684)
        at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:486)
        at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:835)
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:72)
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:56)
        at org.apache.http.impl.client.DecompressingHttpClient.execute(DecompressingHttpClient.java:164)
        at org.apache.http.impl.client.cache.CachingHttpClient.callBackend(CachingHttpClient.java:747)
        at org.apache.http.impl.client.cache.CachingHttpClient.execute(CachingHttpClient.java:446)
        at org.apache.http.impl.client.cache.CachingHttpClient.execute(CachingHttpClient.java:339)
        at org.ektorp.http.StdHttpClient.executeRequest(StdHttpClient.java:177)
        ... 9 more
`

riker65, 2021-02-17:
update.
i tried again:
even more looked better. 

so how to verify or corret the status?

thanks 

`Already exists: solarthing
Already exists: solarthing_events
Created solarthing_closed
Exception in thread "main" org.ektorp.DbAccessException: java.net.SocketTimeoutException: Read timed out
        at org.ektorp.util.Exceptions.propagate(Exceptions.java:19)
        at org.ektorp.http.StdHttpClient.executeRequest(StdHttpClient.java:183)
        at org.ektorp.http.StdHttpClient.executeRequest(StdHttpClient.java:192)
        at org.ektorp.http.StdHttpClient.put(StdHttpClient.java:126)
        at org.ektorp.http.RestTemplate.put(RestTemplate.java:43)
        at org.ektorp.impl.StdCouchDbInstance.createDatabaseIfNotExists(StdCouchDbInstance.java:68)
        at org.ektorp.impl.StdCouchDbInstance.createDatabaseIfNotExists(StdCouchDbInstance.java:59)
        at me.retrodaredevil.solarthing.program.CouchDbSetupMain.createDatabase(CouchDbSetupMain.java:23)
        at me.retrodaredevil.solarthing.program.CouchDbSetupMain.doCouchDbSetupMain(CouchDbSetupMain.java:39)
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:258)
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:270)
Caused by: java.net.SocketTimeoutException: Read timed out
        at java.base/java.net.SocketInputStream.socketRead0(Native Method)
        at java.base/java.net.SocketInputStream.socketRead(SocketInputStream.java:115)
        at java.base/java.net.SocketInputStream.read(SocketInputStream.java:168)
        at java.base/java.net.SocketInputStream.read(SocketInputStream.java:140)
        at org.apache.http.impl.io.AbstractSessionInputBuffer.fillBuffer(AbstractSessionInputBuffer.java:158)
        at org.apache.http.impl.io.SocketInputBuffer.fillBuffer(SocketInputBuffer.java:82)
        at org.apache.http.impl.io.AbstractSessionInputBuffer.readLine(AbstractSessionInputBuffer.java:271)
        at org.apache.http.impl.conn.DefaultHttpResponseParser.parseHead(DefaultHttpResponseParser.java:138)
        at org.apache.http.impl.conn.DefaultHttpResponseParser.parseHead(DefaultHttpResponseParser.java:56)
        at org.apache.http.impl.io.AbstractMessageParser.parse(AbstractMessageParser.java:259)
        at org.apache.http.impl.AbstractHttpClientConnection.receiveResponseHeader(AbstractHttpClientConnection.java:281)
        at org.apache.http.impl.conn.DefaultClientConnection.receiveResponseHeader(DefaultClientConnection.java:257)
        at org.apache.http.impl.conn.ManagedClientConnectionImpl.receiveResponseHeader(ManagedClientConnectionImpl.java:207)
        at org.apache.http.protocol.HttpRequestExecutor.doReceiveResponse(HttpRequestExecutor.java:273)
        at org.apache.http.protocol.HttpRequestExecutor.execute(HttpRequestExecutor.java:125)
        at org.apache.http.impl.client.DefaultRequestDirector.tryExecute(DefaultRequestDirector.java:684)
        at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:486)
        at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:835)
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:72)
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:56)
        at org.apache.http.impl.client.DecompressingHttpClient.execute(DecompressingHttpClient.java:164)
        at org.apache.http.impl.client.cache.CachingHttpClient.callBackend(CachingHttpClient.java:747)
        at org.apache.http.impl.client.cache.CachingHttpClient.execute(CachingHttpClient.java:446)
        at org.apache.http.impl.client.cache.CachingHttpClient.execute(CachingHttpClient.java:339)
        at org.ektorp.http.StdHttpClient.executeRequest(StdHttpClient.java:177)
`

riker65, 2021-02-17:
update

looks like solarthing is loading into the db solarthing.

having 4 documents already....looks good. , but still a little unsure about status of installation for the other dbs.
`events, open, closed.`

T

retrodaredevil, 2021-02-17:
That's weird that you got that error while running the program for the first time. I'd recommend just rerunning it until you complete it without any errors. The program can be run multiple times.

The important database is `solarthing`, the other databases have some other nice features, but aren't required. After the setup program completes creating the 4 databases, it will put some additional security on the databases to make sure only people with permissions can write to them, while anyone can read them.

`solarthing_events` isn't currently used for `rover` programs, but that may change in the future. `solarthing_open` is used for commands, which is an advanced feature with lacking documentation. `solarthing_closed` is also used for commands and metadata, and lacks in documentation more than `solarthing_open`.

riker65, 2021-02-18:
thanks a lot 
...progressing with the installation and setup.

really perfect support and project.

...quick question:
for running request program, I will set up a service similar to solarthing-rover, right?

in [document:](# GraphQL Quickstart)
I assume to cd into program/<rover|mate|request> ? 
If i have request and rover running, i have to do this config for both?

```
Note that you must have set up either the `mate`, `rover`, or `request` program for this to be useful.
This program is used to expose CouchDB as a GraphQL API, which is commonly used with Grafana and [the GraphQL Datasource](https://github.com/fifemon/graphql-datasource).

Once everything is installed, you're ready to edit the configs. You will cd to the **`program/mate**` directory.
```
cd /opt/solarthing/program/graphql
```
```

retrodaredevil, 2021-02-18:
What are you trying to do? Are you trying to get the GraphQL program running? If so, you just need to go into the `/opt/solarthing/program/graphql` directory and edit config/applications.properties.

The GraphQL program is useful if you want to be able to query CouchDB from Grafana, which I assume is what you want.

riker65, 2021-02-19:
Hi 
yes thanks, I wanted 2 different topics:
1. to set up GraphQL and Grafana : I thought in the first place grafana is only working with influxdb.,
   Will check and follow your advise

>  If so, you just need to go into the /opt/solarthing/program/graphql directory and edit config/applications.properties

2. I wanted to add RPI_temperatur: I thought this is a own program, I it is required to extend the **base.json** from program, **rover**.

retrodaredevil, 2021-02-19:
> so I expected to be request to be a own program.

It can be it's own program, but it's better to do it with the rover program if you're already running it.

> is this base.json correct now?

That looks correct to me.

> Where best to place config/couchdb?

Personally, I usually put my couchdb config in `/opt/solarthing/program/config/couchdb.json`, but really you can put it anywhere you want. If you put it where I said above, you can reference it by using `../config/couchdb.json` in the rover config and graphql config. Or if you want to be absolutely sure, you can always just refer to it by `/opt/solarthing/program/config/couchdb.json` (its absolute path).

> when trying to clonse I do not have the privileges.

Yeah, I remember having this problem. I think when I had to mess with that directory I ended up using sudo a lot. It's not ideal, but it's what I ended up doing. There might be a way to mess with permissions, but I can't help you there. I now use the docker version of Grafana, so I don't have to deal with that, but the non-docker version is usually simpler to setup without docker knowledge.

I don't know why those commands to give you grafana permisison didn't work. There may need to be something else done to get it working (maybe change the permissions recursively on all the grafana files to allow group members to edit). I'd just keep doing what you're doing and do stuff with sudo. There's probably a better answer, but I don't know it.

retrodaredevil, 2021-02-19:
>     1. to set up GraphQL and Grafana : I thought in the first place grafana is only working with influxdb.,

Yeah there are two different ways to get data to Grafana. If you're using CouchDB right now and have that working. Go with the GraphQL way. If you want to try your own thing with InfluxDB, you can do that too. It's up to you.

I'm trying to push new users towards CouchDB + SolarThing GraphQL and I think it's going pretty well. There's more customization available using SolarThing GraphQL along with a few extra queries that would be very difficult if not impossible to do with InfluxDB alone.

retrodaredevil, 2021-02-19:
Also, SolarThing version 2021.2.0 has a bug in it that doesn't allow the `request` field to be filled in. I will fix that in SolarThing 2021.3.0.

riker65, 2021-02-19:
Hi 
thanks a lot for your answers.

just not following the couchdb route with grafana.

but when running ./run.sh in graphgl path I ended up with an error:

will reboot and retry. 
anything I made wrong? 

`cat application.properties
#solarthing.config.database=../config/couchdb/couchdb.json
solarthing.config.database=/opt/solarthing/program/config/couchdb.json
`

`
2021-02-19 08:38:27.593  INFO 9760 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet        : Initializing Servlet 'dispatcherServlet'
2021-02-19 08:38:27.770  INFO 9760 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet        : Completed initialization in 175 ms
2021-02-19 08:38:28.696  WARN 9760 --- [nio-8080-exec-1] .w.s.m.s.DefaultHandlerExceptionResolver : Resolved [org.springframework.web.bind.MissingServletRequestParameterException: Required String parameter 'query' is not present]
2021-02-19 08:42:24.668  WARN 9760 --- [nio-8080-exec-5] .w.s.m.s.DefaultHandlerExceptionResolver : Resolved [org.springframework.web.bind.MissingServletRequestParameterException: Required String parameter 'query' is not present]
2021-02-19 08:51:33.908  WARN 9760 --- [nio-8080-exec-4] .w.s.m.s.DefaultHandlerExceptionResolver : Resolved [org.springframework.web.bind.MissingServletRequestParameterException: Required String parameter 'query' is not present]
2021-02-19 09:00:52.104  WARN 9760 --- [nio-8080-exec-6] .w.s.m.s.DefaultHandlerExceptionResolver : Resolved [org.springframework.web.bind.MissingServletRequestParameterException: Required String parameter 'query' is not present]
`

riker65, 2021-02-19:
I am looking forward to get it running.
it is just such an amazing software.

riker65, 2021-02-19:
while running ./run.sh in program/requests -just for rpi temp to keep it simple- I ended up in an error:

were is my error?

base.json looks like
`
pi@pi1pixel:/opt/solarthing/program/request $ cat config/base.json
{
  "type": "request",
  "source": "default",
  "fragment": 3,
  "unique": 30,
  "databases": [
    "/opt/solarthing/program/config/couchdb.json"
 ],
  "request": [
    {
      "type": "rpi-cpu-temp"
    }
  ]
}
`

` solarthing-2021.2.0.jar Java version: 11.0.9.1
2021-02-19 09:22:47.115 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2021-02-19 09:23:05.905 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Error while parsing ProgramOptions.
com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "request" (class me.retrodaredevil.solarthing.config.options.RequestProgramOptions), not marked as ignorable (10 known properties: "dataRequesterList", "unique", "minimum_wait", "databases", "source", "databaseConfigurationFiles", "time_zone", "period", "analytics_enabled", "fragment"])
 at [Source: (FileReader); line: 9, column: 15] (through reference chain: me.retrodaredevil.solarthing.config.options.RequestProgramOptions["request"])
        at com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException.from(UnrecognizedPropertyException.java:61) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownProperty(DeserializationContext.java:989) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.std.StdDeserializer.handleUnknownProperty(StdDeserializer.java:1965) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownProperty(BeanDeserializerBase.java:1686) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownVanilla(BeanDeserializerBase.java:1664) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:330) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.BeanDeserializer._deserializeOther(BeanDeserializer.java:225) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:197) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer._deserializeTypedForId(AsPropertyTypeDeserializer.java:135) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer.deserializeTypedFromObject(AsPropertyTypeDeserializer.java:105) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.AbstractDeserializer.deserializeWithType(AbstractDeserializer.java:263) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.impl.TypeWrappedDeserializer.deserialize(TypeWrappedDeserializer.java:74) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:322) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4591) ~[solarthing-2021.2.0.jar:?]
        at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3559) ~[solarthing-2021.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:184) [solarthing-2021.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:242) [solarthing-2021.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:270) [solarthing-2021.2.0.jar:?]
`

retrodaredevil, 2021-02-19:
> were is my error?

No error there. That's my bad. I added that bug 7 days ago and fixed it today with the 2021.3.0 release. You can go to [updating](https://github.com/wildmountainfarms/solarthing/blob/master/other/docs/updating.md) to learn how to update. Basically, just git pull and run the scripts in the program directory to download the latest versions of SolarThing.

retrodaredevil, 2021-02-19:
> but when running ./run.sh in graphgl path I ended up with an error:

Can you give the entire output of that run? What you're showing is only a tiny portion of it.

riker65, 2021-02-19:
ok will upgrade and let you know
thanks a lot

riker65, 2021-02-19:
> > but when running ./run.sh in graphgl path I ended up with an error:
> 
> Can you give the entire output of that run? What you're showing is only a tiny portion of it.

Hi 
please see error and log message here.

`pi1pixel:/opt/solarthing/program/graphql $ ./run.sh

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.3.1.RELEASE)

2021-02-19 10:29:11.169  INFO 6296 --- [           main] m.r.s.g.SolarThingGraphQLApplication     : Starting SolarThingGraphQLApplication on pi1pixel with PID 6296 (/opt/solarthing/program/.downloads/solarthing-graphql-2021.3.0.jar started by pi in /opt/solarthing/program/graphql)
2021-02-19 10:29:11.239  INFO 6296 --- [           main] m.r.s.g.SolarThingGraphQLApplication     : No active profile set, falling back to default profiles: default
2021-02-19 10:29:47.590  INFO 6296 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat initialized with port(s): 8080 (http)
2021-02-19 10:29:47.889  INFO 6296 --- [           main] o.apache.catalina.core.StandardService   : Starting service [Tomcat]
2021-02-19 10:29:47.896  INFO 6296 --- [           main] org.apache.catalina.core.StandardEngine  : Starting Servlet engine: [Apache Tomcat/9.0.36]
2021-02-19 10:29:51.450  INFO 6296 --- [           main] o.a.c.c.C.[Tomcat].[localhost].[/]       : Initializing Spring embedded WebApplicationContext
2021-02-19 10:29:51.453  INFO 6296 --- [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 38622 ms
file: /opt/solarthing/program/config/couchdb.json absolute: /opt/solarthing/program/config/couchdb.json
No solcast config! Not using solcast!
Using defaultInstanceOptions=DefaultInstanceOptions(defaultSourceId='<UNUSED SOURCE ID THAT WILL NEVER BE IN A PACKET>', defaultFragmentId=0)
2021-02-19 10:30:26.750  INFO 6296 --- [           main] o.s.s.concurrent.ThreadPoolTaskExecutor  : Initializing ExecutorService 'applicationTaskExecutor'
2021-02-19 10:30:30.535  INFO 6296 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat started on port(s): 8080 (http) with context path ''
2021-02-19 10:30:30.816  INFO 6296 --- [           main] m.r.s.g.SolarThingGraphQLApplication     : Started SolarThingGraphQLApplication in 129.453 seconds (JVM running for 158.813)
2021-02-19 10:37:37.736  INFO 6296 --- [nio-8080-exec-1] o.a.c.c.C.[Tomcat].[localhost].[/]       : Initializing Spring DispatcherServlet 'dispatcherServlet'
2021-02-19 10:37:37.747  INFO 6296 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet        : Initializing Servlet 'dispatcherServlet'
2021-02-19 10:37:37.999  INFO 6296 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet        : Completed initialization in 249 ms
2021-02-19 10:37:39.251  WARN 6296 --- [nio-8080-exec-1] .w.s.m.s.DefaultHandlerExceptionResolver : Resolved [org.springframework.web.bind.MissingServletRequestParameterException: Required String parameter 'query' is not present]
`

riker65, 2021-02-19:
quick question.

is there a verbose mode? 

which log is the best to check for status? 

thanks

for example: tried pvoutput, process is running, but no uploads
how to monitor? similar to requests, running now but how to find them in couchdb?
`
{
  "type": "pvoutput-upload",
  "system_id": 100,
  "api_key": "mykey",
  "database": "../config/couchdb.json",
  "source": "default",
  "join_teams": true
}`

riker65, 2021-02-19:
sorry for keeping you busy

retrodaredevil, 2021-02-19:
> 2021-02-19 10:37:37.999 INFO 6296 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet : Completed initialization in 249 m

This means that you set up GraphQL correctly, but in Grafana I think your query is wrong. Can you show me what your Grafana query is?

> is there a verbose mode?

For all the programs except the GraphQL program, they each have their own log directory. For the rover program, you can look in `/opt/solarthing/program/rover/logs` and look for a `log_debug.log`. You can use the command `tail -f log_debug.log` to see the debug log live. However, note that in order to avoid SD card corruption, the debug log is not instantaneous. So if you want to only see errors instantly, tail log_info.log instead.

You can also stop the systemctl service and run it individually to instantly see the debug output.

> for example: tried pvoutput, process is running, but no uploads
> how to monitor? similar to requests, running now but how to find them in couchdb?

Best bet is to run the command `tail -f /opt/solarthing/program/pvoutput/logs/log_info.log`.

I also assume that you redacted your system_id in your example PVOutput configuration?

> sorry for keeping you busy

No problem. I enjoy having something to do.

riker65, 2021-02-19:
Hi,
thanks again.

where are you based?

following up on different streams now.
Unfortunately a lot of new this for me. 
new is:
- grafana 
- couchdb (I know sql and normal oracle I would say)
so almost everything. so struggling with some setups and first steps as well.

### Topic: request: just fo save and track RPI temp.

Service looks good to me.
```
pi@pi1pixel:/opt/solarthing/program/rover/config $ sudo service solarthing-request status -l
● solarthing-request.service - The service for SolarThing Request
   Loaded: loaded (/etc/systemd/system/solarthing-request.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-02-19 10:38:21 CET; 7h ago
 Main PID: 6953 (sh)
    Tasks: 16 (limit: 2062)
   CGroup: /system.slice/solarthing-request.service
           ├─6953 sh /opt/solarthing/program/request/run.sh
           ├─6956 sh ../.scripts/solarthing.sh
           └─6968 java --add-opens=java.base/java.lang.invoke=ALL-UNNAMED -jar ../.scripts/../solarthing.jar --base config/base.json

Feb 19 10:38:21 pi1pixel systemd[1]: Started The service for SolarThing Request.

```
Errors showing up. with tail.

`tail -f /opt/solarthing/program/request/logs/log_info.log
2021-02-19 17:58:39.455 [main] ERROR me.retrodaredevil.solarthing.packets.handling.PrintPacketHandleExceptionWrapper - Caught PacketHandleException from me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@17c8c7e. Message: Conflict while saving something to couchdb. id=2021,02,19,17,(30/30),[7a0cf67e] rev=2-4ff157ad01778da4c920d41b4850fd3b. This usually means we put a packet in the database, but we weren't able to cache its rev id.
2021-02-19 18:00:34.383 [main] ERROR me.retrodaredevil.solarthing.packets.handling.PrintPacketHandleExceptionWrapper - Caught PacketHandleException from me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@17c8c7e. Message: We got a DbAccessException probably meaning we couldn't reach the database.
2021-02-19 18:00:50.672 [main] INFO  me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver - We were able to get the actual Revision ID for id=2021,02,19,18,(01/30),[7a0cf67e] actual rev=2-d5af1c138daff9d093c634ad1c2d933f
2021-02-19 18:00:50.673 [main] ERROR me.retrodaredevil.solarthing.packets.handling.PrintPacketHandleExceptionWrapper - Caught PacketHandleException from me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@17c8c7e. Message: Conflict while saving something to couchdb. id=2021,02,19,18,(01/30),[7a0cf67e] rev=2-d5af1c138daff9d093c634ad1c2d933f. This usually means we put a packet in the database, but we weren't able to cache its rev id.
2021-02-19 18:05:57.565 [main] ERROR me.retrodaredevil.solarthing.packets.handling.PrintPacketHandleExceptionWrapper - Caught PacketHandleException from me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@17c8c7e. Message: We got a DbAccessException probably meaning we couldn't reach the database.
2021-02-19 18:07:29.078 [main] WARN  me.retrodaredevil.solarthing.misc.device.RaspberryPiCpuTemperatureListUpdater - vcgencmd must have taken longer than 300ms!
2021-02-19 18:11:48.106 [main] ERROR me.retrodaredevil.solarthing.packets.handling.PrintPacketHandleExceptionWrapper - Caught PacketHandleException from me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@17c8c7e. Message: We got a DbAccessException probably meaning we couldn't reach the database.
2021-02-19 18:13:15.948 [main] ERROR me.retrodaredevil.solarthing.packets.handling.PrintPacketHandleExceptionWrapper - Caught PacketHandleException from me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@17c8c7e. Message: We got a DbAccessException probably meaning we couldn't reach the database.
2021-02-19 18:13:30.458 [main] INFO  me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver - We were able to get the actual Revision ID for id=2021,02,19,18,(07/30),[7a0cf67e] actual rev=5-31f53143bdd1e969912ac8971d04b7ef
2021-02-19 18:13:30.460 [main] ERROR me.retrodaredevil.solarthing.packets.handling.PrintPacketHandleExceptionWrapper - Caught PacketHandleException from me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@17c8c7e. Message: Conflict while saving something to couchdb. id=2021,02,19,18,(07/30),[7a0cf67e] rev=5-31f53143bdd1e969912ac8971d04b7ef. This usually means we put a packet in the database, but we weren't able to cache its rev id.
`

running request/runsh.
output looks ggod to me?

`
} ]
2021-02-19 18:33:28.613 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 1000ms
2021-02-19 18:33:29.635 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Debugging all packets
2021-02-19 18:33:29.639 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - [ {
  "packetType" : "DEVICE_CPU_TEMPERATURE",
  "cpuTemperatureCelsius" : 42.2,
  "cpuTemperatureFahrenheit" : 107.96
}, {
  "packetType" : "SOURCE",
  "sourceId" : "default"
}, {
  "packetType" : "FRAGMENT_INDICATOR",
  "fragmentId" : 3
} ]
2021-02-19 18:33:29.644 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4971ms
2021-02-19 18:33:34.640 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Debugging all packets
2021-02-19 18:33:34.646 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - [ {
  "packetType" : "DEVICE_CPU_TEMPERATURE",
  "cpuTemperatureCelsius" : 42.2,
  "cpuTemperatureFahrenheit" : 107.96
}, {
[... 48 lines cut ...]
[... cut ...]
2021-02-19 18:38:58.072 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - --> POST https://pvoutput.org/service/r2/addstatus.jsp (38-byte body)
2021-02-19 18:38:59.809 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - <-- 401 Unauthorized https://pvoutput.org/service/r2/addstatus.jsp (1733ms, unknown-length body)
2021-02-19 18:38:59.829 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Unsuccessful. Message: Unauthorized code: 401
2021-02-19 18:38:59.831 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Going to sleep now
`
I keep digging

thanks

riker65, 2021-02-19:
Ah found one issue with pvout and your team.

 - Unknown error joining SolarThing team! Response: Unauthorized 401: Missing, invalid or inactive api key information (X-Pvoutput-Apikey)

digging for base.json team extention

retrodaredevil, 2021-02-19:
> Errors showing up. with tail.

Those errors are fine. Every once in a while it's going to fail to reach your database, which isn't perfect, but it's fine. I believe your request program is working perfectly.

> Also how to easy see temp data in couchdb?

On the couchdb web interface, (navigate to http://your_url:5984/_utils), go to the solarthing database and go to the `millis` view, and query the data from newest to oldest (descending). Then you can see uploaded packets. It's not the easiest thing to do, but it's just clicking around.

> I did not set up a query, I expected that the run.sh is pickung up a predefined query?

Hmm. Then I don't know why it's doing that. The GraphQL program does stay running, correct? If it does stay running, just ignore that error message until you start trying to get Grafana working.

>     * Unknown error joining SolarThing team! Response: Unauthorized 401: Missing, invalid or inactive api key information (X-Pvoutput-Apikey)

> I added my API-key into the json. Anything else to be defined for the upload?

Yes, you need your system id too. Make sure both system id and api key are correct. Your system id will be something like `72206` and your api key will be something like `e68051c.......` where the ... is a bunch of more random characters.

riker65, 2021-02-19:
Hi

how to add couchdb to grafana as source, It is not availabe.

installing :
`sudo grafana-cli plugins install couchdb-datasource
Error: ✗ Failed to find requested plugin, check if the plugin_id is correct: 404 not found error
`

any hint maybe?
thanks

> ``

riker65, 2021-02-19:
did see a strange message
`
2021-02-19 19:41:09.729 [googleanalyticsjava-thread-1] WARN  com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl - Exception while sending the Google Analytics tracker request Request [parms={DOCUMENT_ENCODING=ANSI_X3.4-1968, APPLICATION_VERSION=V-UNKNOWN, EVENT_CATEGORY=startup, HIT_TYPE=event, PROTOCOL_VERSION=1, USER_LANGUAGE=en-US, EVENT_LABEL=solarthing-2021.3.0.jar, EVENT_ACTION=pvoutput-upload, TRACKING_ID=UA-70767765-2, APPLICATION_NAME=solarthing-program, CLIENT_ID=72d7f35b-b425-4dee-8f0a-08476ffdc020}, customDimensions={}, customMetrics={}]
java.lang.NullPointerException: null
`

riker65, 2021-02-19:
good new, pvputput is running now! systemkey needs to be updated. 
who to join your team?

next topic : some more to go

retrodaredevil, 2021-02-19:
> how to add couchdb to grafana as source, It is not availabe.

You add the `graphql-datasource`. You connect the GraphQL datasource to SolarThing GraphQL, which will query CouchDB.

> did see a strange message

Hmm. Weird. Hopefully that's a 1 time thing. In my Google Analytics I can see that there are some people reporting versions 2021.3.0, so I know it's at least working most of the time.

>  who to join your team?

If you set `join_team` to true, it should have automatically joined https://pvoutput.org/map.jsp?tid=1528

riker65, 2021-02-20:
Quick question? I am nor using slack or mattermost, what would be the best to start with?  using some telegram bots

retrodaredevil, 2021-02-20:
If you want to pick between slack and mattermost, I'd recommend going with slack unless you want to host it yourself.

I might consider supporting telegram in the future.

riker65, 2021-03-02:
Hi I am back:-)
had a systemcrash which needed to be recovered. Moved the renogy now into the mobile home. access is a little difficult.

having still issue with graphQL: 

> `Then make sure the URL in Grafana is set to http://localhost:8080/graphql or similar.`

how and where to check this for the plugin?

another questions: how to log to a remote chouchdb-> how should the base.json look like for database?

Thanks T

riker65, 2021-03-02:
> > how to add couchdb to grafana as source, It is not availabe.
> 
> You add the `graphql-datasource`. You connect the GraphQL datasource to SolarThing GraphQL, which will query CouchDB.

This is still unclear to me.

couchdb is running. Grafana also, installed the plugin. . how to connect 

> GraphQL datasource to SolarThing GraphQL, 

thanks

retrodaredevil, 2021-03-02:
It sounds like you have the datasource installed, you just have to configure it correctly. Instead of `http://127.0.0.1:8080`, you need to do `http://127.0.0.1:8080/graphql`. Also, do you have the SolarThing GraphQL program running?

If you aren't sure if SolarThing GraphQL is running, run `systemctl status solarthing-graphql`.

> another questions: how to log to a remote chouchdb-> how should the base.json look like for database?

It will look the same as a local couchdb database, but instead of 127.0.0.1 or localhost, you'll have the IP to your remote computer which hosts the database.

riker65, 2021-03-03:
Hi, 

I am getting closer, 

but another topic with graphql

`
pi@pi1pixel:/opt/solarthing/program/graphql $ ./run.sh

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.3.1.RELEASE)

2021-03-03 20:40:34.305  INFO 28984 --- [           main] m.r.s.g.SolarThingGraphQLApplication     : Starting SolarThingGraphQLApplication on pi1pixel with PID 28984 (/opt/solarthing/program/.downloads/solarthing-graphql-2021.3.0.jar started by pi in /opt/solarthing/program/graphql)
2021-03-03 20:40:34.330  INFO 28984 --- [           main] m.r.s.g.SolarThingGraphQLApplication     : No active profile set, falling back to default profiles: default
2021-03-03 20:40:45.283  INFO 28984 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat initialized with port(s): 8080 (http)
2021-03-03 20:40:45.415  INFO 28984 --- [           main] o.apache.catalina.core.StandardService   : Starting service [Tomcat]
2021-03-03 20:40:45.417  INFO 28984 --- [           main] org.apache.catalina.core.StandardEngine  : Starting Servlet engine: [Apache Tomcat/9.0.36]
2021-03-03 20:40:46.087  INFO 28984 --- [           main] o.a.c.c.C.[Tomcat].[localhost].[/]       : Initializing Spring embedded WebApplicationContext
2021-03-03 20:40:46.088  INFO 28984 --- [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 11233 ms
file: /opt/solarthing/program/config/couchdb.json absolute: /opt/solarthing/program/config/couchdb.json
No solcast config! Not using solcast!
Using defaultInstanceOptions=DefaultInstanceOptions(defaultSourceId='<UNUSED SOURCE ID THAT WILL NEVER BE IN A PACKET>', defaultFragmentId=0)
2021-03-03 20:41:01.713  INFO 28984 --- [           main] o.s.s.concurrent.ThreadPoolTaskExecutor  : Initializing ExecutorService 'applicationTaskExecutor'
2021-03-03 20:41:08.899  WARN 28984 --- [           main] ConfigServletWebServerApplicationContext : Exception encountered during context initialization - cancelling refresh attempt: org.springframework.context.ApplicationContextException: Failed to start bean 'webServerStartStop'; nested exception is org.springframework.boot.web.server.PortInUseException: Port 8080 is already in use
2021-03-03 20:41:08.903  INFO 28984 --- [           main] o.s.s.concurrent.ThreadPoolTaskExecutor  : Shutting down ExecutorService 'applicationTaskExecutor'
2021-03-03 20:41:08.931  INFO 28984 --- [           main] o.apache.catalina.core.StandardService   : Stopping service [Tomcat]
2021-03-03 20:41:09.094  INFO 28984 --- [           main] ConditionEvaluationReportLoggingListener :

Error starting ApplicationContext. To display the conditions report re-run your application with 'debug' enabled.
2021-03-03 20:41:09.131 ERROR 28984 --- [           main] o.s.b.d.LoggingFailureAnalysisReporter   :

***************************
APPLICATION FAILED TO START
***************************

Description:

Web server failed to start. Port 8080 was already in use.

Action:

Identify and stop the process that's listening on port 8080 or configure this application to listen on another port.
`

but checking with `sudo netstat -lptu | grep 8080
`
there is nothing running on the port.

Or what am I missing to check?

thanks T

retrodaredevil, 2021-03-03:
If I had to guess, I'd say that you're probably already running the GraphQL program. When you run it with `./run.sh`, you have to make sure that the systemctl service is stopped with `sudo systemctl stop solarthing-graphql`. If it's running, then you don't need to run it again with `./run.sh`. When you use `./run.sh`, it's for running solarthing-graphql temporarily (usually to test it).

If solarthing-graphql is not running and port 8080 is still in use, I don't have an idea what program is using it.

Also, I tried using the command you had, and I found that this one works better: `sudo netstat -tulpn | grep 8080`. If SolarThing GraphQL is the program using it, the output of that command will have something like `<PID>/java` at the end of it.

riker65, 2021-03-03:
perfect guess. parallel running service and manual.

so I assume setup is more or less ready, now trying to get some queries running.:-)

riker65, 2021-03-03:
Hi

do you have the queries for the 
https://github.com/wildmountainfarms/solarthing/blob/master/other/docs/grafana-example.png

retrodaredevil, 2021-03-04:
That screenshot is actually from back when I used InfluxDB. Since you aren't using InfluxDB, those queries aren't going to be useful to you.

>  so where to place the query?

That's an InfluxDB 1.X query. You don't want to use that since you aren't running InfluxDB.

You can find some GraphQL queries here: https://github.com/wildmountainfarms/solarthing/blob/master/other/grafana/graphql_queries.md just make sure to replace the to and from with "$__to" and "$__from".

You can find some example dashboards here: https://github.com/wildmountainfarms/solarthing/tree/master/other/grafana/dashboards . Now, I don't recommend copy pasting the entire json file, as something will break when you do that, but each file has some useful GraphQL queries in it (if you look at the files with graphql in the name).

riker65, 2021-03-05:
Hi 
wondering how to change the couchdb database
want to have the dummy/or other  database as own database for testing

how to redirect/configure  into database like
-solarthing_test
-solarthing_test_event
for example?
[https://github.com/wildmountainfarms/solarthing/blob/master/other/docs/couchdb_setup.md](url)
Thanks

riker65, 2021-03-05:
Hi Joshua,

quick question for solarthing android, 
which user should be set here?
is there a link to the documentation? 
do not understand the network switching topic?

Thanks T

riker65, 2021-03-05:
Hi Joshua, do you have the idea to add mqtt support?

retrodaredevil, 2021-03-05:
>  wondering how to change the couchdb database

Currently I don't support that. I used to, but it'd be a decent amount to change because I'd have to make the database name customizable for each program, so I'm not going to add that in the future.

You can always spin up a new couchdb on the same computer by using docker and docker-compose.

>  which user should be set here?

Leave username and password blank. If you set up your database correctly, you won't need a username and password to read it.

>  is there a link to the documentation?

Documentation for SolarThing Android is basically nothing right now. No one has actually asked about it except you.

>  do not understand the network switching topic?

If you just want to get something working, ignore network switching. It's optional. Basically what I use it for is for changing the IP address depending on what network I'm on. (So if I'm at home it uses 192.168.x.x, but if I'm away it could use xxx.duckdns.org as the IP instead).

retrodaredevil, 2021-03-05:
Also, I don't currently plan on adding MQTT support. It doesn't benefit me at all and when I looked into it, I wasn't exactly sure the best way to map SolarThing packets to channel names, or whatever the term is.

retrodaredevil, 2021-03-17:
I'm gonna go ahead and close this. Feel free to reopen this issue or open another issue if you need more help.

Also, for the potential addition of MQQT support, you can see this issue: https://github.com/wildmountainfarms/solarthing/issues/28

closed 2021-03-17 by retrodaredevil

retrodaredevil, 2021-12-21:
Hey @riker65 I remembered that you got SolarThing working with a device that uses the RS485 port.

I'm currently writing some documentation and I wanted to confirm that you did **not** have to install any additional drivers yourself. Sometimes for RS485 communication additional drivers need to be installed and I was just wondering if this was the case or not for communicating over the RS485 port on your DCC50S.