wildmountainfarms/solarthing #26: New Renogy 60amp Controller Modbus Exception

ssnowit, 2021-03-08:
It appears that the new Renogy 60amp Controller may have made some changes. I had solarthing and grafana up and working with my 30amp controller, I upgraded to the 60 amp and now I keep getting a Modbus Exception.  I will update if I make any progress but if anybody else has any ideas/suggestions I am open to hear them.

retrodaredevil, 2021-03-08:
Can you post the entire modbus exception?

ssnowit, 2021-03-08:
> Can you post the entire modbus exception?

I certainly can, If I get a chance to look at during my lunch I will post it otherwise I'll throw it up this evening.

ssnowit, 2021-03-08:
> > Can you post the entire modbus exception?
> 
> I certainly can, If I get a chance to look at during my lunch I will post it otherwise I'll throw it up this evening.
2021-03-08 13:22:44.482 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.ModbusTimeoutException: Timed out! startTime=1615231362480 currentTime=1615231364482 initialTimeout=2000
	at me.retrodaredevil.io.modbus.RtuDataEncoder.readBytes(RtuDataEncoder.java:113) ~[solarthing.jar:?]
	at me.retrodaredevil.io.modbus.RtuDataEncoder.readMessage(RtuDataEncoder.java:61) ~[solarthing.jar:?]
	at me.retrodaredevil.io.modbus.IOModbusSlaveBus.sendRequestMessage(IOModbusSlaveBus.java:26) ~[solarthing.jar:?]
	at me.retrodaredevil.io.modbus.ImmutableAddressModbusSlave.sendRequestMessage(ImmutableAddressModbusSlave.java:14) ~[solarthing.jar:?]
	at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:24) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:28) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheRangeInclusive(ModbusCacheSlave.java:36) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:60) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRoverProgram$5(RoverMain.java:118) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:40) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:94) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$1(RoverMain.java:54) ~[solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:124) [solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:51) [solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:83) [solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:189) [solarthing.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:207) [solarthing.jar:?]
2021-03-08 13:22:44.486 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 2993ms

retrodaredevil, 2021-03-08:
A modbus timeout exception usually means that something is either wrong with your cable, the rover doesn't have an address of 1 (which is the default), or something else.

Have you ever plugged your Rover 60A into a bluetooth module? I've talked with people who've done that and then couldn't get SolarThing to work with their Rover for some reason.

It's probably not the slave address, but it could be. I really need to give SolarThing a feature to scan for the rover on all the different possible addresses.

retrodaredevil, 2021-03-08:
Also, for the cable for the Rover 60A, you can either keep using your custom RJ12 cable that works for your 30amp controller, or you can buy the RJ45 cable here: https://www.renogy.com/rs485-to-usb-serial-cable/

The RJ12 port uses the RS232 protocol and the RJ45 port uses the RS485 protocol. Both should work, but if you think something may by wrong with your cable, then you could try buying the cable.

ssnowit, 2021-03-08:
I have confirmed the Bluetooth works when its plugged in so the controller is functioning. I am using a RS232 cable from rich solar that worked on my 30 AMP controller I haven't changed the addressing but I will confirm when I am able to get back out to the property where the solar setup is at. I have ordered a RS485 cable and will give it a go when it comes in as well.

retrodaredevil, 2021-03-08:
I don't know why the RS232 cable wouldn't work with the Rover 60A, but the RS485 is worth a try I guess. I also didn't realize that Rich Solar made a cable. I'll have to buy that instead of making a cable next time.

It's possible that for your particular Rover, it just doesn't work after having a BT module plugged into it. On [this post (link now doesn't work)](https://support.renogy.com/en/support/discussions/topics/35000015051/page/3?url_locale=en), the person said that they couldn't get communication to their Rover after plugging in the BT module. I don't know how common this problem is, but it's possible you're having it. (Or maybe not). The only thing I can think of being changed after the BT module is plugged in would be the slave address. (Which can be changed in SolarThing by setting `modbus` to something other than 1).

Also, while looking at the picture on [Rich Solar RS232 cable](https://richsolar.com/products/rs232-cable), I don't understand how it would work since it looks like it it's using pins 1, 3, and 4, when it should be using 1, 2, and (3 or 4). Maybe the picture is incorrect. If the picture isn't incorrect, then I'm surprised it worked for your 30A CC. Although I do think that if it worked for your 30A, then it should probably work for your 60A.

Anyway, I'll be interested to hear if you get the RS485 cable working. I've only tested this stuff with a Rover 40A and a custom cable, so it's not like I know everything about these charge controllers.

ssnowit, 2021-03-09:
Well I found my problem and it was 2 fold. I had some time to head to the property last night and do some additional troubleshooting. It just wasn't sitting right with me that the rs232 cable that was working on the old controller would have just failed at the same time I swapped to a new controller. 

So I hooked the old controller backup and it wouldn't connect. Swapped out Pi's and the old controller was alive again. Plugged back into the new controller with no connection and I double checked the addressing to find it had changed to 16 when I plugged in the bluetooth module as prior to plugging in the module it was set to 1. Changed it back and life is good. I am still going to swap over to the rs485 cable so I can use the rs232 cable for another project. Thanks for your assistance this project has been alot of fun and very useful. Now I need to figure out how to monitor my actual power consumption.

retrodaredevil, 2021-03-09:
Awesome. Glad to hear that you figured out the address got changed to 16. I'll have to document that somewhere in SolarThing.

As a note, if you wanted to keep the address as 16, you could do that, would just change `modbus` to 16 in base.json.

If you need more help, feel free to reopen this or preferably, create another issue.

closed 2021-03-09 by retrodaredevil