wildmountainfarms/solarthing #27: Rover 20A with Rasperry Pi Modbus exeption

Mouky66, 2021-03-09:
Hi,
I get the following error:

2021-03-09 14:32:57.681 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.handling.ParsedResponseDataLengthException: Expected a length of 35. Got a length of 4 instead!
        at me.retrodaredevil.io.modbus.handling.HandleResponseHelper.checkResponse(HandleResponseHelper.java:28) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:61) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:13) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:28) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheRangeInclusive(ModbusCacheSlave.java:36) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$3(RoverMain.java:85) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRoverProgram$8(RoverMain.java:138) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:66) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:111) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:83) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:144) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:51) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:103) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:204) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:244) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:272) [solarthing-2021.3.0.jar:?]
2021-03-09 14:32:57.685 [main] INFO  me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Communication with rover working well. Got this response back: function code=0x03 data='00 0A 00 11' feel free to open issue at https://github.com/wildmountainfarms/solarthing/issues/
2021-03-09 14:32:57.686 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4957ms

It is a brand new Rover, never connected to any BT Adapters.
Thank you

BR Chris

retrodaredevil, 2021-03-09:
That's a really weird error. Can you add `bulk_request: false` to your base.json and see if you still get that error?

retrodaredevil, 2021-03-09:
It should be a slightly different error if you added `"bulk_request": false`. Can you post the new error?

Mouky66, 2021-03-09:
You are right :

"2021-03-09 19:57:43.033 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.handling.ParsedResponseDataLengthException: Expected a length of 3. Got a length of 4 instead!
        at me.retrodaredevil.io.modbus.handling.HandleResponseHelper.checkResponse(HandleResponseHelper.java:28) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:61) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.io.modbus.handling.ReadHoldingRegisters.handleResponse(ReadHoldingRegisters.java:13) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead.get(RoverModbusSlaveRead.java:47) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead.oneRegister(RoverModbusSlaveRead.java:50) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead.getMaxVoltageValue(RoverModbusSlaveRead.java:58) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.solar.renogy.rover.RoverStatusPackets.createFromReadTable(RoverStatusPackets.java:13) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:67) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:111) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:83) ~[solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:144) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:51) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:103) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:204) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:244) [solarthing-2021.3.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:272) [solarthing-2021.3.0.jar:?]
2021-03-09 19:57:43.037 [main] INFO  me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Communication with rover working well. Got this response back: function code=0x03 data='00 0A 00 01' feel free to open issue at https://github.com/wildmountainfarms/solarthing/issues/
"

retrodaredevil, 2021-03-09:
I was hoping I'd never have to see an error like this. This is really weird, and I don't think it's a bug in SolarThing, maybe a bug in the Rover itself. Anyway, I'm probably gonna have you do some testing with Python and minimalmodbus, but before I do, can you tell me exactly what Rover model you have or give a link to the exact model? The model is probably exactly ` RNG-CTRL-RVR20 `, so a link to where you bought it may also be helpful.

My first guess as to why this is happening is that your particular rover doesn't support a number of things that other models support, and instead of giving a valid response, it gives a bad response for things that aren't supported. So, when I have you test with Python I'll confirm that the core functionality works (battery voltage, etc), then test other things to see if they work.

Mouky66, 2021-03-09:
I hope it is not an unsopported model ;) 
RNG-CTRL-RVR20
I bought it here: https://www.amazon.de/gp/product/B01MRWTAB5/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Please let me know what to test. I am really looking forward to this :)

Mouky66, 2021-03-09:
The Version is V030001 and the Date 2020.09

Here is the output:
root@testpi:/opt/solarthing/program/rover# apt install python3-pip
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
python3-pip ist schon die neueste Version (9.0.1-2+rpt2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
root@testpi:/opt/solarthing/program/rover# python3 -m pip install solarshed
Requirement already satisfied: solarshed in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: prometheus-client==0.1.0 in /usr/local/lib/python3.5/dist-packages (from solarshed)
Requirement already satisfied: MinimalModbus==0.7 in /usr/local/lib/python3.5/dist-packages (from solarshed)
Requirement already satisfied: pyserial==3.4 in /usr/local/lib/python3.5/dist-packages (from solarshed)
root@testpi:/opt/solarthing/program/rover# python3
Python 3.5.3 (default, Nov 18 2020, 21:09:16)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from solarshed.controllers.renogy_rover import RenogyRover
>>> controller = RenogyRover('/dev/ttyUSB0', 1)
>>> controller.battery_voltage()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/solarshed/controllers/renogy_rover.py", line 86, in battery_voltage
    return self.read_register(257, numberOfDecimals=1)
  File "/usr/local/lib/python3.5/dist-packages/minimalmodbus.py", line 258, in read_register
    return self._genericCommand(functioncode, registeraddress, numberOfDecimals=numberOfDecimals, signed=signed)
  File "/usr/local/lib/python3.5/dist-packages/minimalmodbus.py", line 697, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)
  File "/usr/local/lib/python3.5/dist-packages/minimalmodbus.py", line 798, in _performCommand
    payloadFromSlave = _extractPayload(response, self.address, self.mode, functioncode)
  File "/usr/local/lib/python3.5/dist-packages/minimalmodbus.py", line 1075, in _extractPayload
    raise ValueError(text)
ValueError: Checksum error in rtu mode: '\x01Ô' instead of 'IÔ' . The response is: '\x01\x03\x01\x01\x00\x01Ô' (plain response: '\x01\x03\x01\x01\x00\x01Ô')
>>> controller.system_voltage_current()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/solarshed/controllers/renogy_rover.py", line 48, in system_voltage_current
    register = self.read_register(10)
  File "/usr/local/lib/python3.5/dist-packages/minimalmodbus.py", line 258, in read_register
    return self._genericCommand(functioncode, registeraddress, numberOfDecimals=numberOfDecimals, signed=signed)
  File "/usr/local/lib/python3.5/dist-packages/minimalmodbus.py", line 697, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)
  File "/usr/local/lib/python3.5/dist-packages/minimalmodbus.py", line 798, in _performCommand
    payloadFromSlave = _extractPayload(response, self.address, self.mode, functioncode)
  File "/usr/local/lib/python3.5/dist-packages/minimalmodbus.py", line 1075, in _extractPayload
    raise ValueError(text)
ValueError: Checksum error in rtu mode: '\x00\x01' instead of '\x9dÿ' . The response is: '6\x01\x03\x00\n\x00\x01' (plain response: '6\x01\x03\x00\n\x00\x01')

retrodaredevil, 2021-03-09:
Ok, well it's good to know for sure that the bug isn't in SolarThing since these two modbus libraries handle things a bit differently.

It's extremely weird that the Rover sends back a mostly valid response. Hopefully the cause of this is the cable being used. What cable are you using to communicate with your Rover?

Can you try out the rover-setup program now?
You can use this as your base.json, or better yet, create a new file called base-setup.json and run SolarThing using `./run.sh config/base-setup.json`.

```
{ "type": "rover-setup", "io": "config/default_linux_serial.json", "modbus": 1} 
```
Now run SolarThing, and type stuff like `batteryvoltage` into the input. SolarThing should crash, then just run it again and try something different. The more errors the better for me.

List of things to type:
* `pvcurrent`
* `productmodel`
* `ratedcurrent`

retrodaredevil, 2021-03-09:
Ok, that's an interesting cable setup, and I don't think it's going to work. The good news is that I don't think you need to buy anything else, you just have to change how the cable is wired.

I can't read German, so I don't reference exactly what's going on in those Amazon links, but I'll do my best.

So you have an RS232-TTL converter. This is perfect for use with the Raspberry Pi's GPIO. You can use this if you want to not use USB. Basically, you'd wire it to the GPIO with the little jumper cables, then you'd take the DB9 output and connect that to some sort of RJ12 cable which goes into the Rover. Personally, I've never tried this solution, but it's possible.

Now, the solution I recommend, is just using your USB to RS232 cable. Plug the USB into the Pi, then you wire the outputs on the DB9 side to the Rover. (I actually recommend a breakout board, but it's optional).

I don't know how you got a TTL converter in the mix, but you need to either choose one or the other. My guess is that you probably mean to buy a breakout board for the DB9 connector, but ended up accidentally buying a TTL converter.

I mean, you did get something working, which is really amazing that you got a response that tricked SolarThing a couple of times to thinking it was something that it wasn't. I've never seen a response like that before.

Mouky66, 2021-03-09:
I will try both ways tomorrow and let you know the resultsts.
I feel relieved that it seems to be the cable.
Thank you for your time and your suggestions.

retrodaredevil, 2021-03-09:
If you decide to go the GPIO route, you won't be using `/dev/ttyUSB0` anymore, you'll be using something else.

Mouky66, 2021-03-11:
Hi Joshua,

it has been the cable. I have thrown out the ttl converter and now it is working like it should :)
Thanks again for your help.
BR Chris

closed 2021-03-11 by Mouky66