wildmountainfarms/solarthing #28: Feature request: Send Data to MQTT

Mouky66, 2021-03-11:
Hi,

is it possible to send the requested Data to MQTT instead of to a database?

Background: I have a home-assistant setup running and collect all Sensor data via MQTT.
I also want to integrate the solarthing-data into HA to create some automations for monitoring and alarming.

Thank you
BR Chris

retrodaredevil, 2021-03-11:
Currently I'm not interested in supporting MQTT, but you aren't the first person to ask, so I may consider considering it in the future.

There is some limited automation stuff you can do with solarthing, there's even an `automation` program, which isn't really that well documented. There's also slack notifications that can be sent based on certain events (low battery and some other stuff).

I also played around with uploading sensor data directly to home assistant a while ago. I haven't played with it much, and it's mostly just for FX devices, so it's not useful in its current state for most people.

I'll go ahead and leave this issue open, so there's a chance I'll come back to it later.

retrodaredevil, 2021-03-24:
I got some free time this week so I was thinking of implementing this. However, before I just go ahead and make something, I wanted your input on how this would work best for you.

When making adding a database to SolarThing, usually I have to decide how to squeeze packets into some new structure to work with said database. I was thinking that each packet could have it's own topic, and it would be stored in JSON.

So for instance, a rover could have the topic `solarthing/default/2/RoverIdentifier()`. The message in that topic would always be JSON data. Or, if you prefer, there could be topics such as `solarthing/default/2/RoverIdentifier()/packet/batteryVoltage` for battery voltage and `solarthing/default/2/RoverIdentifier()/json` for raw JSON.

I assume you'll be using Node Red (and others will probably want that too), so I'd like to do whatever makes it easier for people using it. I'll probably end up putting the raw json data in either way, so the question is whether or not you want to have a bunch of different topics for each datapoint.

riker65, 2021-03-24:
Hi 
thanks for picking this up adding MQTT support.

I would vote for a structure like 
`solarthing/default/2/RoverIdentifier()/packet/batteryVoltage`
best would be to have topics configurable 

also for parameter like **retain**, 

I assume you thought about this for:
default
packet

what is your intention of default and 2 in topic?

a raw JSON is usefull as well.

Thanks a lot   T

retrodaredevil, 2021-03-24:
> also for parameter like **retain**,

What do you mean by this?

> what is your intention of default and 2 in topic?

"default" is the default source ID that all the example base.json files have and 2 is the source ID that is present in the base.json file for the rover config.

So for now, the default topic could be like this (and could be configurable too) `solarthing/%source/%fragment/%identifier` and it would have subtopics such as `json` and `packet/*a bunch of fields*` and `dateMillis`.

People that only monitor a single rover don't really have to worry about source ID and fragment ID. Each device uploading to a database needs a different fragment ID, and the source ID is usually the same unless you are monitoring two different batteries. My setup: MatePi ("default", 1), RoverPi("default", 2), CabinPi("cabin", 11). MatePi and RoverPi monitor the same battery, but CabinPi monitors a different battery. Having the correct source ID and fragment ID are crucial in making sure SolarThing can tell devices apart and also group readings logically.

Once I get this added in, if you wanted to your topic configuration could leave out source ID and fragment ID altogether, but I recommend keeping them in case you ever need to monitor another device. SolarThing was built to handle many different devices.

riker65, 2021-03-24:
> "default" is the default source ID that all the example base.json files have and 2 is the source ID that is present in the base.json file for the rover config.

great, I was not using this at the moment, but perfect design.

Thanks for clarification

retrodaredevil, 2021-03-25:
The latest SolarThing release has MQTT support. You can see an example config here: https://github.com/wildmountainfarms/solarthing/blob/master/config_templates/databases/mqtt_template.json

There is also an advanced version of the config in the same directory with some more options.

I have it set to retain by default. Let me know if you think that or anything else should change. I probably won't end up using MQTT in my setup but I hope it benefits you.

Mouky66, 2021-03-25:
This works perfekt. Thank you very much :)

riker65, 2021-03-25:
> This works perfekt. Thank you very much :)

ok how did you implement it?
what is triggering the mqtt?
where does the config hat to be placed?

Thanks for clarification

BR T

Mouky66, 2021-03-25:
Just copy the mqtt.json example to your config folder and put in your data. Then reference to it via base.json.

cat /opt/solarthing/program/rover/config/base.json
{
  "type": "rover",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "databases": [
    "config/mqtt.json"
  ],
  "io": "config/default_linux_serial.json",
  "modbus": 1
}

riker65, 2021-03-25:
perfect understood, thanks

retrodaredevil, 2021-03-25:
It sounds like you both got it working. If either of you have trouble with MQTT stuff, feel free to reopen this issue.

riker65, 2021-03-26:
Hi Joshua,

great and fast job with MQTT as Database.

question: did you implement any living will handling?

thanks T

retrodaredevil, 2021-03-26:
> question: did you implement any living will handling?

I currently do not have the MQTT implementation set a "will" if it becomes disconnected for some reason. I could probably add that and publish to a certain topic if that's useful?

riker65, 2021-03-27:
Hi Joshua, yes a topic like 
solarthing/%source/%fragment/%identifier/LWT 
payload online / offline 

would be perfect 

Thanks T

retrodaredevil, 2021-03-27:
>  solarthing/%source/%fragment/%identifier/LWT

Yeah I can definitely do something like that. I probably won't name it LWT, but I'll let you know when it's done.

retrodaredevil, 2021-03-27:
Alright, I now have LWT implemented. For the default topic, it will go to `solarthing/%source/%fragment/status`. It will say either `offline` or `online`. Note that it uses `status` instead of the identifier. I have to do it this way because I designed it so there's a possibility that there can be multiple rovers (with multiple identifiers) and I can only have one LWT message.

Note that if SolarThing for some reason crashes or stops uploading while it is not connected, the LWT will still say `online`, even if it offline. You can and should use `solarthing/%source/%fragment/dateMillis` to check to make sure it has been updated recently.

Honestly I don't know how useful this LWT is gonna be since there's a chance that it will still be out of date even when it says `online`, so let me know if you think I can make any improvements.

riker65, 2021-03-30:
perfect, lwt working for me .

thanks

closed 2021-04-01 by retrodaredevil