wildmountainfarms/solarthing #29: renogy dcc50s bt-2  modbus exception

riker65, 2021-03-24:
Hi 
Iwas testing the renogy Bluetooth module bt-2

having a requestion,

once connected to the renogy.

I do not get any serial data when bt was disconnected again?
Modbus exception

have you experienced something like this?

Thanks T

retrodaredevil, 2021-03-24:
It is likely that the bt module changed the rovers modbus address. It commonly changes it to 16, so in your base.json, try setting `modbus` to 16. If that doesn't work, I can run the `rover-setup` program and run `scan 1 20`. It will take a while, but it should show you what address it has. 

You should also be able to change the address back to 1 in the app somehow. 

I assume your error is a modbus timeout exception? If it's not, then posting the error would be helpful.

riker65, 2021-03-24:
thanks for the hint
will check for new adress with the scan.

i thought something like this an tried address 2 already . will check for 16 and scan.

Perfect thanks

retrodaredevil, 2021-03-24:
It just occurred to me that the scan feature is not a part of SolarThing 2021.3.0, so you can try to find the address manually, compile SolarThing yourself and use scan, or wait for me to update it.

retrodaredevil, 2021-03-25:
SolarThing 2021.3.1 has the scan feature. You can update if you'd like, but there aren't any new features I'm aware of that will benefit you.

Close this if you get it working.

riker65, 2021-03-25:
Hi 
thanks just updated

but how to run `scan 1 20`"?

riker65, 2021-03-25:
Hi Joshua,

found the scan feature. but where do i get the result from?
`
Scan finished!

scan 1 20
Starting scan. (Will tell you when scan has stopped)
}`

thanks

riker65, 2021-03-25:
Hi Joshua,

I found new slave address at hex 0x60/dec 96  confirmed by renogy support
given by BT-2 connection

Thanks for your support

closed 2021-03-25 by riker65

retrodaredevil, 2021-03-25:
Huh. That's really weird that it decided it wanted an address of 96. If I have to recommend that people scan from 1 to 247, that's gonna take a long time.

>  Scan finished!

If there was nothing between this message and the "scan started" message, it means it didn't find a rover, which makes sense because it was on an address outside the range you were scanning.

I probably need to figure out a way to do a quicker scan. 2 seconds per address isn't really acceptable if it's address can be in a large range of addresses.

I guess I'll probably recommend try to find their address through their BT module from now on lol.