wildmountainfarms/solarthing #30: add more data from modbus read

riker65, 2021-03-27:
Hi Joshua,

is it possible to add field to be retrieved from device.

my model is having also alternator information.

[https://www.renogy.com/dcc50s-12v-50a-dc-dc-on-board-battery-charger-with-mppt/](url)

I did get those via a python program: for example:

`altVoltage  =  instrument.read_register(260,1)
altAmps  =  instrument.read_register(261,2)
altWatts  =  instrument.read_register(262)
accum_gen_power_wh = instrument.read_register(284,2)
# Error Codes
errorCodesLow = instrument.read_register(257)
errorCtrlOverTemp2 = ( errorCodesLow & 0b0000000000010000) > 0
errorAltInputOverCurrent = (errorCodesLow & 0b0000000000100000) > 0
errorAltInputOverVolt = (errorCodesLow & 0b0000000100000000) > 0
errorAltPolarityReverse = (errorCodesLow & 0b0000001000000000) > 0
errorBmsOverCharge = (errorCodesLow & 0b0000010000000000) > 0
errorAuxLowTemp = (errorCodesLow & 0b0000100000000000) > 0`

[renogy_DCC_Charger_Modbus_RS485_V1.7.pdf](https://github.com/wildmountainfarms/solarthing/files/6215379/renogy_DCC_Charger_Modbus_RS485_V1.7.pdf)

Thx for checking T

retrodaredevil, 2021-03-27:
For the alternator voltage, amps, and watts, you can currently use `loadVoltage`, `loadCurrent`, and `loadPower`. This applies only to the MQTT currently. For GraphQL, they are named correctly. The reason they are named incorrectly is because of some differences between the Rover and DCDC charger.

`accum_gen_power_wh` is stored in kWh, so you can multiply by 100. The field is named `cumulativeKWH`. Note that its value will not reset each day, that particular field just keeps accumulating. Personally I usually use `dailyKWH` if I just want the current day.

The raw error code value (in your example `errorCodesLow`) is stored as `errorMode`. It's very possible for me to add those fields, but I would prefer not to. The reason I'd prefer not to is that they are specific to the DCDC charger, and I'd rather keep as much logic the same for both the Rover and DCDC charger.

While not ideal, I think it should be possible for you to manually extract your desired error mode just by using the `errorMode` field. If it's not possible to do that easily, then I might consider adding a topic for each value.

I might also consider making the topics published to MQTT more similar to the GraphQL program. The GraphQL program adds many convenience fields instead of just giving spitting out raw JSON data like the MQTT database currently does.

riker65, 2021-03-28:
ok thanks 

will check this

closed 2021-03-28 by riker65