wildmountainfarms/solarthing #31: error in pvoutput

riker65, 2021-04-01:
Hi Joshua,

please have a lock:

thanks for checking 

suspect error I found:
`Going to do stuff now.
2021-04-01 11:15:27.788 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Got throwable
java.lang.ClassCastException: class com.fasterxml.jackson.databind.node.NullNode cannot be cast to class com.fasterxml.jackson.databind.node.ObjectNode (com.fasterxml.jackson.databind.node.NullNode and com.fasterxml.jackson.databind.node.ObjectNode are in unnamed module of loader 'app')
	at me.retrodaredevil.solarthing.couchdb.CouchDbQueryHandler.query(CouchDbQueryHandler.java:34) ~[solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain.startRealTimeProgram(PVOutputUploadMain.java:285) ~[solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain.startPVOutputUpload(PVOutputUploadMain.java:118) ~[solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:199) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:235) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:263) [solarthing-2021.3.2.jar:?]
2021-04-01 11:15:27.828 [main] DEBUG me.retrodaredevil.solarthing.program.SolarMain - Going to shutdown LogManager.`

full log :
`./run.sh
Using base config = config/base.json
2021-04-01 11:24:34,074 main ERROR Unable to create file logs/log_debug.log java.io.IOException: Permission denied
        at java.base/java.io.UnixFileSystem.createFileExclusively(Native Method)
        at java.base/java.io.File.createNewFile(File.java:1026)
        at org.apache.logging.log4j.core.appender.rolling.RollingFileManager$RollingFileManagerFactory.createManager(RollingFileManager.java:665)
        at org.apache.logging.log4j.core.appender.rolling.RollingFileManager$RollingFileManagerFactory.createManager(RollingFileManager.java:648)
        at org.apache.logging.log4j.core.appender.AbstractManager.getManager(AbstractManager.java:113)
        at org.apache.logging.log4j.core.appender.OutputStreamManager.getManager(OutputStreamManager.java:100)
        at org.apache.logging.log4j.core.appender.rolling.RollingFileManager.getFileManager(RollingFileManager.java:205)
        at org.apache.logging.log4j.core.appender.RollingFileAppender$Builder.build(RollingFileAppender.java:146)
        at org.apache.logging.log4j.core.appender.RollingFileAppender$Builder.build(RollingFileAppender.java:62)
        at org.apache.logging.log4j.core.config.plugins.util.PluginBuilder.build(PluginBuilder.java:122)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createPluginObject(AbstractConfiguration.java:1002)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createConfiguration(AbstractConfiguration.java:942)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createConfiguration(AbstractConfiguration.java:934)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.doConfigure(AbstractConfiguration.java:552)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.initialize(AbstractConfiguration.java:241)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.start(AbstractConfiguration.java:288)
        at org.apache.logging.log4j.core.LoggerContext.setConfiguration(LoggerContext.java:622)
        at org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:695)
        at org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:712)
        at org.apache.logging.log4j.core.LoggerContext.start(LoggerContext.java:267)
        at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:155)
        at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:47)
        at org.apache.logging.log4j.LogManager.getContext(LogManager.java:196)
        at org.apache.logging.log4j.spi.AbstractLoggerAdapter.getContext(AbstractLoggerAdapter.java:137)
        at org.apache.logging.slf4j.Log4jLoggerFactory.getContext(Log4jLoggerFactory.java:45)
        at org.apache.logging.log4j.spi.AbstractLoggerAdapter.getLogger(AbstractLoggerAdapter.java:47)
        at org.apache.logging.slf4j.Log4jLoggerFactory.getLogger(Log4jLoggerFactory.java:30)
        at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:363)
        at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:388)
        at me.retrodaredevil.solarthing.program.SolarMain.<clinit>(SolarMain.java:61)

2021-04-01 11:24:34,125 main ERROR Could not create plugin of type class org.apache.logging.log4j.core.appender.RollingFileAppender for element RollingFile: java.lang.IllegalStateException: ManagerFactory [org.apache.logging.log4j.core.appender.rolling.RollingFileManager$RollingFileManagerFactory@14646ef] unable to create manager for [logs/log_debug.log] with data [org.apache.logging.log4j.core.appender.rolling.RollingFileManager$FactoryData@5ffe9e[pattern=logs/log_debug_%d{yyyy.MM.dd}-%i.log.gz, append=true, bufferedIO=true, bufferSize=16384, policy=CompositeTriggeringPolicy(policies=[OnStartupTriggeringPolicy, TimeBasedTriggeringPolicy(nextRolloverMillis=0, interval=1, modulate=true), SizeBasedTriggeringPolicy(size=3221225472)]), strategy=DefaultRolloverStrategy(min=1, max=1, useMax=true), advertiseURI=null, layout=%d{yyyy-MM-dd HH:mm:ss.SSS} [%t] %-5level %logger{36} - %msg%n, filePermissions=rw-rw-r--, fileOwner=null]] java.lang.IllegalStateException: ManagerFactory [org.apache.logging.log4j.core.appender.rolling.RollingFileManager$RollingFileManagerFactory@14646ef] unable to create manager for [logs/log_debug.log] with data [org.apache.logging.log4j.core.appender.rolling.RollingFileManager$FactoryData@5ffe9e[pattern=logs/log_debug_%d{yyyy.MM.dd}-%i.log.gz, append=true, bufferedIO=true, bufferSize=16384, policy=CompositeTriggeringPolicy(policies=[OnStartupTriggeringPolicy, TimeBasedTriggeringPolicy(nextRolloverMillis=0, interval=1, modulate=true), SizeBasedTriggeringPolicy(size=3221225472)]), strategy=DefaultRolloverStrategy(min=1, max=1, useMax=true), advertiseURI=null, layout=%d{yyyy-MM-dd HH:mm:ss.SSS} [%t] %-5level %logger{36} - %msg%n, filePermissions=rw-rw-r--, fileOwner=null]]
        at org.apache.logging.log4j.core.appender.AbstractManager.getManager(AbstractManager.java:115)
        at org.apache.logging.log4j.core.appender.OutputStreamManager.getManager(OutputStreamManager.java:100)
        at org.apache.logging.log4j.core.appender.rolling.RollingFileManager.getFileManager(RollingFileManager.java:205)
        at org.apache.logging.log4j.core.appender.RollingFileAppender$Builder.build(RollingFileAppender.java:146)
        at org.apache.logging.log4j.core.appender.RollingFileAppender$Builder.build(RollingFileAppender.java:62)
        at org.apache.logging.log4j.core.config.plugins.util.PluginBuilder.build(PluginBuilder.java:122)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createPluginObject(AbstractConfiguration.java:1002)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createConfiguration(AbstractConfiguration.java:942)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createConfiguration(AbstractConfiguration.java:934)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.doConfigure(AbstractConfiguration.java:552)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.initialize(AbstractConfiguration.java:241)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.start(AbstractConfiguration.java:288)
        at org.apache.logging.log4j.core.LoggerContext.setConfiguration(LoggerContext.java:622)
        at org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:695)
        at org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:712)
        at org.apache.logging.log4j.core.LoggerContext.start(LoggerContext.java:267)
[... 222 lines cut ...]
2021-04-01 11:24:46.922 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2021-04-01 11:24:46.950 [main] INFO  me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Going to join SolarThing team...
2021-04-01 11:24:47.095 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Executing call
[... cut ...]
        at me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain.startRealTimeProgram(PVOutputUploadMain.java:285)
        at me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain.startPVOutputUpload(PVOutputUploadMain.java:118)
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:199)
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:235)
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:263)
`

retrodaredevil, 2021-04-01:
Yup, you're right that's the error I need. The other garbage is caused by wrong permissions. If you run `other/linux/update_perms.sh`, it should help you out. And also make sure to add yourself to the `solarthing` group with `sudo usermod -a -G solarthing,dialout,tty,video $USER`. Now, this won't fix that problem, but it should get rid of a bunch of warnings.

As for the error, that really shouldn't happen ever. I could try a quick fix for it, but I'd rather find the problem in the first place. 

I believe the problem has something to do with the way your CouchDB is set up. Have you ever gotten CouchDB to work with something like the Android app or GraphQL? I'm also assuming you went through the [CouchDB setup](https://github.com/wildmountainfarms/solarthing/blob/master/other/docs/couchdb_setup.md) correctly?

Can you give me the output of the link (replacing the ip with yours) `http://<YOUR IP>:5984/solarthing/_design/packets/_view/millis?startkey=9999999999999&descending=true&limit=10`. I'd prefer if you uploaded a file, but you could copy paste that if you'd like.

riker65, 2021-04-02:
Hi Joshua,

thanks again for the great and fast support.

did update/correct permissions via your script running with sudo.

Couchdb worked I assume .
and pvoutput and solarthing android also.

probably I messed around with permissions because I wanted to use **log4j**   to be configured no logging to extend ssd card's life.
renamed: **no-debug_log4j2.xml to log4j2.xml**, but still had logging.

another thing I tried was maybe also wrong.
I changed the view in couchdb. - unfortunately I am old school just working with sql and the whole mad/reduce topic is new to me.

```
function(doc) {
  emit(doc.dateMillis, doc.packetType,doc);
}
```

How should the initial map function for view millis should look like?

requested output:
- some data from dummy tests in and real data as well ( request program, temperature and rover)
`{
"total_rows": 36238,
"offset": 0,
"rows": [
{
"id": "2021,04,02,09,(13/30),[9ffe4789]",
"key": 1617348336048,
"value": null
},
{
"id": "2021,04,02,09,(13/30),[1783f0da]",
"key": 1617348330877,
"value": null
},
{
"id": "2021,04,02,09,(12/30),[9ffe4789]",
"key": 1617348231057,
"value": null
},
{
"id": "2021,04,02,09,(12/30),[1783f0da]",
"key": 1617348200866,
"value": null
},
{
"id": "2021,04,02,09,(11/30),[9ffe4789]",
"key": 1617348111042,
"value": null
},
{
"id": "2021,04,02,09,(11/30),[1783f0da]",
"key": 1617348085848,
"value": null
},
{
"id": "2021,04,02,09,(10/30),[9ffe4789]",
"key": 1617347991038,
"value": null
},
{
"id": "2021,04,02,09,(10/30),[1783f0da]",
"key": 1617347990816,
"value": null
},
{
"id": "2021,04,02,09,(09/30),[9ffe4789]",
"key": 1617347871034,
"value": null
},
{
"id": "2021,04,02,09,(09/30),[1783f0da]",
"key": 1617347855792,
"value": null
}
]
}`

riker65, 2021-04-02:
Hi Joshua, it is running again.

change db view back to 

> function(doc) {
  emit(doc.dateMillis,doc);
}

 only .

thanks

closed 2021-04-02 by riker65

retrodaredevil, 2021-04-02:
I was just typing up a response. Yup that's the correct view. Glad you got it working