wildmountainfarms/solarthing #32: Support consumed power upload to PVOutput with Rover

alx-k, 2021-04-10:
Would it be possible to make the Renogy Rover also output its used power to PVOutput? If I understand it right, [this](https://github.com/wildmountainfarms/solarthing/blob/69469d3c1e0d62c9f43f9907c3448ddcdcd18bf9/client/src/main/java/me/retrodaredevil/solarthing/program/pvoutput/PVOutputHandler.java#L100) is where energy generation and consumption are set but consumption is only set for fx inverter packets

retrodaredevil, 2021-04-10:
I just committed a fix for this. If you could run `git pull`, then compile SolarThing yourself by running `./compile_and_move.sh`, once you restart `solarthing-pvoutput`, it should upload the consumption.

alx-k, 2021-04-11:
OK, I pulled the changes and ran the compile script and everything seems to be working fine now, thanks for the quick fix!

alx-k, 2021-04-11:
I was probably a little bit fast on closing this issue, energy conumption is now uploaded, but power consumption (parameter v4) is still always 0.

alx-k, 2021-04-11:
I just wanted to add that I looked at the packets in couchdb and "loadPower" isn't actually 0. (So there's actually something wrong)

retrodaredevil, 2021-04-11:
Huh. Thanks for pointing that out. The cause of that was actually a bug I probably wouldn't have found. Try pulling and recompiling now.

alx-k, 2021-04-12:
Yep, everything is working perfectly fine now, thanks again!

closed 2021-04-12 by alx-k