wildmountainfarms/solarthing #33: usb issue raspberry usp port changing

riker65, 2021-04-13:
Hi 

I have an issue using more usb devices on the raspberry.
USB names are changing after rebooting.

I tried to define the port diffent in config

`{
  "type": "serial",
  "port": "/dev/serial/by-path/platform-3f980000.usb-usb-0:1.5:1.0-port",
  "serial_config": null
}`

but looks like it is not working .

`2021-04-13 07:39:16.498 [main] ERROR me.retrodaredevil.solarthing.io.ReloadableIOBundle - Could not create IOBundle
java.lang.RuntimeException: java.io.FileNotFoundException: usb_path.json (No such file or directory)
	at me.retrodaredevil.solarthing.program.SolarMain.createIOBundle(SolarMain.java:129) ~[solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRoverProgram$7(RoverMain.java:133) ~[solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.io.ReloadableIOBundle.reload(ReloadableIOBundle.java:54) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.io.ReloadableIOBundle.<init>(ReloadableIOBundle.java:29) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:133) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:54) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:106) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:195) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:235) [solarthing-2021.3.2.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:263) [solarthing-2021.3.2.jar:?]
Caused by: java.io.FileNotFoundException: usb_path.json (No such file or directory)
	at java.io.FileInputStream.open0(Native Method) ~[?:?]
	at java.io.FileInputStream.open(FileInputStream.java:219) ~[?:?]
	at java.io.FileInputStream.<init>(FileInputStream.java:157) ~[?:?]
	at me.retrodaredevil.solarthing.program.SolarMain.createIOBundle(SolarMain.java:127) ~[solarthing-2021.3.2.jar:?]
	... 9 more`

retrodaredevil, 2021-04-13:
>  USB names are changing after rebooting.

I can't really help you with USB device names changing after reboot. Usually you're able to use `/dev/ttyUSB0` for a simple USB to RS232 converter.

As for that error, you defined the path to your serial config as `usb_path.json`. If you did that, the file needs to be in `/program/rover/usb_path.json`. However, you likely put the file in `/program/rover/config/usb_path.json`. So, you just need to update your path to be `config/usb_path.json`.

I used to have the problem where my port changed from `/dev/ttyUSB0` to `/dev/ttyUSB1`, but I fixed it by replacing my micro USB to USB (I use a Raspberry Pi Zero to monitor my Rover). I could support multiple ports in the future, but if I were to guess, this is probably a hardware issue and not something I should try and fix with software *yet*.

riker65, 2021-04-13:
Hi 

ok sorry understood.

sorry my fault.!!! 

just also found that there was a path error.  

yes checking the by-id topic

using DIY cable for modbus and the other serial reading the Vitron Shunt

riker65, 2021-04-13:
thanks for the great and fast support

closed 2021-04-13 by riker65