wildmountainfarms/solarthing #34: modbus errors

boss569, 2021-04-28:
hi,
i have a renogy rover set up, it only gives one or two reads when running then throws modbus errors,
any ideas?
Thanks 

`Using base config = config/base.json
2021-04-27 21:11:01.604 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: solarthing-2021.3.3.jar Java version: 11.0.9.1
[stdout] Beginning main. Jar: solarthing-2021.3.3.jar Java version: 11.0.9.1
[stderr] Beginning main. Jar: solarthing-2021.3.3.jar Java version: 11.0.9.1
2021-04-27 21:11:02.065 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2021-04-27 21:11:02.729 [main] INFO  me.retrodaredevil.solarthing.program.RoverMain - Beginning rover program
2021-04-27 21:11:02.739 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2021-04-27 21:11:02.800 [main] DEBUG me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics UUID: cdb08e08-aa78-4b44-b916-7c216b604ea6
2021-04-27 21:11:04.799 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2021-04-27 21:11:05.386 [main] DEBUG me.retrodaredevil.solarthing.io.ReloadableIOBundle - Successfully reloaded IOBundle
2021-04-27 21:11:07.497 [main] DEBUG me.retrodaredevil.solarthing.program.RoverPacketListUpdater - took 1315ms to read from Rover
2021-04-27 21:11:07.500 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Debugging all packets
2021-04-27 21:11:07.875 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - [ {
  "packetType" : "RENOGY_ROVER_STATUS",
  "packetVersion" : 2,
  "maxVoltage" : 24,
  "ratedChargingCurrent" : 20,
  "ratedDischargingCurrent" : 20,
  "productType" : 0,
  "productModelEncoded" : "ICBSTkctQ1RSTC1SVlIyMA==",
  "softwareVersion" : 196609,
  "hardwareVersion" : 2,
  "productSerialNumber" : 302449205,
  "controllerDeviceAddress" : 1,
  "batteryCapacitySOC" : 58,
  "batteryVoltage" : 12.2,
  "chargingCurrent" : 0.0,
  "controllerTemperatureRaw" : 12,
  "batteryTemperatureRaw" : 13,
  "loadVoltage" : 12.2,
  "loadCurrent" : 1.49,
  "loadPower" : 18,
  "inputVoltage" : 0.0,
  "pvCurrent" : 0.0,
  "chargingPower" : 0,
  "dailyMinBatteryVoltage" : 12.1,
  "dailyMaxBatteryVoltage" : 14.6,
  "dailyMaxChargingCurrent" : 14.57,
  "dailyMaxDischargingCurrent" : 1.89,
  "dailyMaxChargingPower" : 169,
  "dailyMaxDischargingPower" : 21,
  "dailyAH" : 44,
  "dailyAHDischarging" : 5,
  "dailyKWH" : 0.561,
  "dailyKWHConsumption" : 0.06,
  "operatingDaysCount" : 4,
  "batteryOverDischargesCount" : 0,
  "batteryFullChargesCount" : 0,
  "chargingAmpHoursOfBatteryCount" : 137,
  "dischargingAmpHoursOfBatteryCount" : 50,
  "cumulativeKWH" : 1.729,
  "cumulativeKWHConsumption" : 0.607,
  "streetLightValue" : 128,
  "chargingState" : 0,
  "errorMode" : 0,
  "nominalBatteryCapacity" : 200,
  "systemVoltageSetting" : 255,
  "recognizedVoltage" : 0,
  "batteryType" : 3,
  "overVoltageThresholdRaw" : 160,
  "chargingVoltageLimitRaw" : 155,
  "equalizingChargingVoltageRaw" : 152,
  "boostChargingVoltageRaw" : 142,
  "floatingChargingVoltageRaw" : 138,
  "boostChargingRecoveryVoltageRaw" : 132,
[... 671 lines cut ...]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:239) [solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:267) [solarthing-2021.3.3.jar:?]
2021-04-27 21:13:01.830 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4966ms
2021-04-27 21:13:06.885 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.RedundancyException: Incorrect CRC checksum. Expected: 4610 but got: 59338 bytes: [1, 3, 34, 24, 20, 20, 0, 32, 32, 82, 78, 71, 45, 67, 84, 82, 76, 45, 82, 86, 82, 50, 48, 0, 3, 0, 1, 0, 0, 0, 2, 18]
        at me.retrodaredevil.io.modbus.RtuDataEncoder.fromBytes(RtuDataEncoder.java:82) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.io.modbus.RtuDataEncoder.readMessage(RtuDataEncoder.java:63) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.io.modbus.IOModbusSlaveBus.sendRequestMessage(IOModbusSlaveBus.java:26) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.MutableAddressModbusSlave.sendRequestMessage(MutableAddressModbusSlave.java:18) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:24) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:28) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheRangeInclusive(ModbusCacheSlave.java:36) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$3(RoverMain.java:91) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRoverProgram$7(RoverMain.java:143) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:66) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:21) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:113) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.lambda$doRover$2(RoverMain.java:87) ~[solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:149) [solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.doRover(RoverMain.java:55) [solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.RoverMain.connectRover(RoverMain.java:107) [solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:199) [solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:239) [solarthing-2021.3.3.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:267) [solarthing-2021.3.3.jar:?]
2021-04-27 21:13:06.893 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4906ms`

retrodaredevil, 2021-04-28:
I'm really temped to say this is a hardware issue with your serial adapter, but those errors are really interesting. It's weird the errors are alternating.

What serial adapter are you using?

I'll look further into those errors and see if they're recoverable, especially since you got some good data to start with.

boss569, 2021-04-29:
I tried a 3rd serial adapter off an already working project, that has fixed it.
strange that it was intermittent with the other adapters.
thanks for the program.

closed 2021-04-29 by boss569

retrodaredevil, 2021-04-29:
Thanks for confirming it was the serial adapter. Sometimes I see weird errors like this and wonder if I have to make a big change to SolarThing to try and make it work.