wildmountainfarms/solarthing #35: writing to influxdb  stops

riker65, 2021-05-08:
Hi 

I noticed that writing to influxdb stopped but rover process is still running .

how to monitor if writing to influx is still running?

For testing purposes I am writing parallel to influx , couch and mqtt.

retrodaredevil, 2021-05-08:
I can almost guarantee SolarThing is attempting to write to InfluxDB if you have it configured. It sounds like something is going wrong. If you look in `logs/log_info.log`, you can see what SolarThing is logging as errors, which will include any connection errors being made when trying to connect to InfluxDB. If you could post a few of those errors, I could probably help identify what the problem is.

Also, if you could post your `base.json` and your `influxdb.json` (or whatever you called it), that would be helpful.

riker65, 2021-05-12:
Hi

ok will check the log and respond

Thanks

retrodaredevil, 2021-05-18:
Were you able to figure out what the problem was? If you provide logs I can always try to explain what the seemingly gibberish error message means.

riker65, 2021-05-19:
will close it for now. I am on holdiay.:-) 
maybe reopen if required. 

thanks for following up

closed 2021-05-19 by riker65