wildmountainfarms/solarthing #38: Rover RS-232

j0777, 2021-07-07:
Hello,
Thank you for the solarthing!

I've been trying to connect Rover>>>RJ12Cable>>>FT3232RL>>>USBCable>>>RasPi, but no communication.

Could you please clarify the wire connections with the rover, I'm getting Tx-GND(-5.6v) Rx-GND(-0.07v) from its RJ12 port, see attached image for reference:

[image]

retrodaredevil, 2021-07-08:
I'm unfamiliar with FT3232RLs, but you do have the pinout correct. The only thing you need to make sure is that you are connecting Tx to Rx and Rx to Tx. This is so your converter can receive what the rover transmits and so the rover can receive what your converter transmits.

Make sure that your converter is using RS232 (I think the one you have can do more than just RS232).

I do have one question: Has your rover ever been plugged into a bt-module? Those are known to change the Modbus address of the rover, sometimes changing it to 10 or 16.

Since I know very little about your converter, another test you could do would be to connect Tx and Rx on your converter, then send some data through your serial port and make sure that you receive data. This is not something that SolarThing is capable of, but it should be pretty easy to do if you have a little Python experience. This isn't strictly necessary for you to test, but I do recommend it just so you know that your converter is capable of something.

retrodaredevil, 2021-07-08:
Also, looking back on your post, I have another question, what's going on with this "FT3232RL>>>USBCable"? What is the USB cable in this?

j0777, 2021-07-08:
Thank you for your kind reply, 
This Rover never been plug into the BT-module and the USB cable is just cable, please see [here](https://alexnld.com/product/ft232r-ft232rl-module-usb-to-serial-port-usb-to-ttl-adapter-module-with-1-5-m-cable-3-3v-or-5v/) for reference of what I have.
I'll be looking into what you mentioned above and get back you, I appreciate it.

j0777, 2021-07-15:
I was able to communicate with the Rover by getting rid of the FT232RL chip and used the MAX3232 instead, Thanks for the support.

[image]

closed 2021-07-15 by retrodaredevil