wildmountainfarms/solarthing #41: Installing CouchDB

nimratbrar, 2021-08-31:
I am using raspberry pi 0. I am unable to install couchdb using the solarthing.jar 
i am getting the following:

pi@raspberrypi:/opt/solarthing/program/request/config $ java -jar ../solarthing.jar --couchdb-setup config/couchdb.json
Error: Unable to access jarfile ../solarthing.jar

retrodaredevil, 2021-08-31:
So you'll notice that your working directory is `/opt/solarthing/program/request/config`. The relative path `../solarthing.jar` while in that working directory will give you `/opt/solarthing/program/request/solarthing.jar`, which doesn't exist. So, you can either stay in that working directory and use `../../solarthing.jar`, or you can change your working directory to `/opt/solarthing/program/request/`.

I recommend changing your working directory to `/opt/solarthing/program/request/`, so that paths for the configuration are correct.

nimratbrar, 2021-09-10:
Sorry for the late response. I am unable to find the .jar file in any of those directories

nimratbrar, 2021-09-10:
You will now setup your CouchDB instance! Some databases will be automatically created (enter)

Exception in thread "main" java.lang.RuntimeException: me.retrodaredevil.couchdbjava.exception.CouchDbException: java.net.ConnectException: Failed to connect to localhost/0:0:0:0:0:0:0:1:5984
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:196)
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:209)
Caused by: me.retrodaredevil.couchdbjava.exception.CouchDbException: java.net.ConnectException: Failed to connect to localhost/0:0:0:0:0:0:0:1:5984

I am getting the following error now

nimratbrar, 2021-09-10:
this is the first line of the error
2021-09-10 17:32:24,113 main ERROR RollingFileManager (logs/log_debug.log) java.io.FileNotFoundException: logs/log_debug.log (Permission denied) java.io.FileNotFoundException: logs/log_debug.log (Permission denied)

retrodaredevil, 2021-09-10:
> Exception in thread "main" java.lang.RuntimeException: me.retrodaredevil.couchdbjava.exception.CouchDbException: java.net.ConnectException: Failed to connect to localhost/0:0:0:0:0:0:0:1:5984

This means that SolarThing cannot access a CouchDB database on `localhost:5984`. Getting CouchDB running on your Raspberry Pi is pretty much up to you. `solarthing.jar` doesn't actually try to install CouchDB, it just sets it up. There's not exactly a walk through in the docs on how to do it. You can look at [this page](https://docs.couchdb.org/en/stable/install/unix.html) for instructions that will work on your Raspberry Pi or if you know how to use docker, you can follow [this](https://docs.couchdb.org/en/stable/install/docker.html). I only recommend going the docker route if you have experience with docker.

Once you have it installed, there will be a default username and password, which you can either change, or you can use to run the CouchDB setup program like you did when you got that error.

I will consider creating a script that can install it for you, since I'm guessing you aren't the only one who just wants to get SolarThing up and running without figuring out how to install CouchDB.

> 2021-09-10 17:32:24,113 main ERROR RollingFileManager (logs/log_debug.log) java.io.FileNotFoundException: logs/log_debug.log (Permission denied) java.io.FileNotFoundException: logs/log_debug.log (Permission denied)

This line of the error doesn't actually affect any functionality. This just means that permissions aren't set up correctly to get logging working. If you want to fix this, you can run this:

```shell
sudo /opt/solarthing/other/linux/update_perms.sh
```

nimratbrar, 2021-09-11:
The official support page of CouchDB says that it is not available for 32-bit systems. I am using a raspberry 3b . I guess i'll need to host it on a 64 bit device

retrodaredevil, 2021-09-11:
I haven't attempted to install CouchDB on an RPi before. Dang, I thought it was supported on there. I'm surprised no one has pointed this out to me sooner.

However, this is probably for the best, I don't recommend running CouchDB off a Raspberry Pi since it's not good to frequently write to the Pi's SD card. I've had two SD cards go bad from writing to them too much.

If you still want to try to install CouchDB on an RPi, you can look up how to do that, or you can try to use this: https://github.com/treehouses/rpi-couchdb (which uses docker). It looks like the docker option doesn't yet support CouchDB 3, but I believe that's OK. I don't recommend trying to install this from source (you'll find tutorials for this online) because that seems hard to maintain. I also probably wouldn't recommend using that docker image as it hasn't been updated in a while.

Also, although I tend not to recommend this option, but I believe that InfluxDB works on Raspberry Pis. If you just want to get Grafana working, InfluxDB isn't a bad choice. It is a bad choice if you want to do more complex stuff in the future that requires CouchDB, though.

nimratbrar, 2021-09-11:
Thanks, I got it working on my laptop and pi could create the databases. However, I had to change to bind address to 0.0.0.0 from 127.0.0.1 and as a result now i cant access the GUI for couchdb.

I am also encountering an issue with getting GraphQL working because i cannot find the directory /var/lib/grafana/plugins and i am not sure what is meant by "You will need to be a member of the grafana group to access that directory." I cant find that group anywhere on my pc

retrodaredevil, 2021-09-11:
> However, I had to change to bind address to 0.0.0.0 from 127.0.0.1 and as a result now i cant access the GUI for couchdb.

That shouldn't change whether or not if the GUI is accessible. It might be possible you flipped another setting that turned off the GUI. You are accessing it at `<your ip>:5984/_utils`, right?

> I am also encountering an issue with getting GraphQL working because i cannot find the directory /var/lib/grafana/plugins and i am not sure what is meant by "You will need to be a member of the grafana group to access that directory." I cant find that group anywhere on my pc

How did you install Grafana, and what operating system did you install it on?

nimratbrar, 2021-09-12:
I used homebrew on macos

retrodaredevil, 2021-09-12:
I believe that Mac OS usually names users and groups differently. So on Mac OS, I think the user would be named something like `_grafana` (or maybe it's not an underscore). As for that directory, on Mac OS, is should be something like `/usr/local/Cellar/grafana`.

Most of the documentation is for Linux systems (specifically Debian based systems), so some of the Grafana stuff you'll have to figure out yourself.

nimratbrar, 2021-09-13:
Thanks I'll try to figure it out.

retrodaredevil, 2021-09-30:
If you were able to get this working let me know (or let me know things I can improve). I'm debating making a video tutorial for setting up SolarThing if I find the time, so let me know if there's anything you want me to cover in particular.

nimratbrar, 2021-09-30:
> If you were able to get this working let me know (or let me know things I can improve). I'm debating making a video tutorial for setting up SolarThing if I find the time, so let me know if there's anything you want me to cover in particular.

i was not able to Set up GraphQl on my local grafana instance. However, i got it working on a grafana cloud. At the end best solution for me was to setup influxdb on a google cloud server and a local grafana instance that fetches the data.