wildmountainfarms/solarthing #50: phocos any-grid inverter supported?

test2a, 2022-01-16:
i own a phocos anygrid solar inverter that has a built in mppt solar charge controller. it has an rs-232 interface which i hope i can use woth a raspbery pi to get data into home assistant.

https://www.phocos.com/product/psw-h-hybrid-inverter-charger/

https://web.archive.org/web/20200919221625/https://www.phocos.com/wp-content/uploads/2019/10/Any-Grid_PSW-H_EN-manual_2020-07-29.pdf

i have tried to follow your instructions but i am getting 

`solarthing check --port /dev/ttyUSB0 --type tracer --modbus 1 --scan
Going to open serial port using default serial configuration...
Successfully opened serial port...
Checking on address: 1
Got timeout. This means that the modbus address is incorrect or that the cable is not functioning properly.
`

retrodaredevil, 2022-01-16:
SolarThing does not support any inverters except for Outback inverters connected to an Outback Mate unit. 

So no, that inverter is not supported by SolarThing.

test2a, 2022-01-16:
alright thanks

closed 2022-01-16 by test2a