wildmountainfarms/solarthing #51: some issues with socat maybe

tesau, 2022-02-23:
Heck of a project you whipped up.

I have a mikrotik RB750G  connected to the rover via 9pin serial then share the port over ip
And the I use socat to connect to connect a linux mint pc running your solarthing
Things seem to be working good when I was testing on the bench.
Now that I have the setup 60 miles away, I am getting some errors.

Seems if I request smaller amounts of data..  or change settings on the rover .. from my linux mint pc 
I  see almost no errors..   

I don't need to get all info everytime..   what would I need to send to the rover to get maybe half size requests

I maybe missing an obvious mistake.

If you would, please take a look at the following and steer me straight ...    

Thank-you
Terry 
Calgary, AB

/usr/bin/socat pty,link=/home/solarthing/dev/ttyV0,raw,mode=777 TCP:10.0.16.3:10000

{
  "type": "request",
  "source": "default",
  "fragment": 2,
  "unique": 30,
 "databases": [ 
        "config/couchdb.json" 
         ],
  "request": [
    {
      "type": "modbus",
      "io": "config/rover_serial.json",
      "devices": {
        "1": {
          "type": "rover",
	  "server": { "port": 5051 },
     "commands": [
            "LOAD ON", "LOAD OFF"
          ]
	}
      }
    }
     ],
  "commands": [
    {
      "name": "LOAD ON",
      "display_name": "Load On",
      "description": "Turns on the load",
      "action": "config/commands/load_on.json"
    },
    {
      "name": "LOAD OFF",
      "display_name": "Load Off",
      "description": "Turns off the load",
      "action": "config/commands/load_off.json"
    }
	]
         }

2022-02-22 17:31:21.794 [main] INFO  me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Got part of a response back. (Maybe timed out halfway through?) data='01 03 22 18 28 14 00 20 20 52 4E 47 2D 43 54 52 4C 2D 52 56 52 34 30 00 03 00 01 00 00 00 02 12 07 07 0B 00 01 4A 74 01 03 14 00 64 01 17 00 5E 84 8A 01 17 00 26 00 0A 01 97 00 42 00 1B A0 25' Feel free to open an issue at https://github.com/wildmountainfarms/solarthing/issues/
2022-02-22 17:31:21.795 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4.801 seconds
2022-02-22 17:31:28.722 [main] DEBUG me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - took 2.126 seconds to read from device
2022-02-22 17:31:28.722 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - Debugging all packets
2022-02-22 17:31:28.724 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - [ {
  "packetType" : "RENOGY_ROVER_STATUS",
  "packetVersion" : 7,
  "maxVoltage" : 24,
  "ratedChargingCurrent" : 40,
[... 148 lines cut ...]
	at me.retrodaredevil.io.modbus.RtuDataEncoder.fromBytes(RtuDataEncoder.java:95) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.io.modbus.RtuDataEncoder.fromBytes(RtuDataEncoder.java:72) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.io.modbus.RtuDataEncoder.parseMessage(RtuDataEncoder.java:63) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.io.modbus.IODataEncoder.readMessage(IODataEncoder.java:17) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.io.modbus.IOModbusSlaveBus.sendRequestMessage(IOModbusSlaveBus.java:26) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.io.modbus.ImmutableAddressModbusSlave.sendRequestMessage(ImmutableAddressModbusSlave.java:14) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:24) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:38) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheHolding(ModbusCacheSlave.java:47) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheHoldingRangeInclusive(ModbusCacheSlave.java:50) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.config.request.modbus.RoverModbusRequester.cacheRegisters(RoverModbusRequester.java:63) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.config.request.modbus.RoverModbusRequester.lambda$create$0(RoverModbusRequester.java:81) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper.receive(ModbusListUpdaterWrapper.java:57) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.doRequest(RequestMain.java:76) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:70) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:40) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:145) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:187) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:252) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.1.1.jar:?]
2022-02-22 17:31:36.638 [main] INFO  me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Got part of a response back. (Maybe timed out halfway through?) data='6F 84 8A 01 16 00 41 00 12 01 95 00 4F 00 20 BB BE' Feel free to open an issue at https://github.com/wildmountainfarms/solarthing/issues/
2022-02-22 17:31:36.639 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4.958 seconds
^C

retrodaredevil, 2022-02-23:
I added an option a while back to make requests to the rover in bulk. I made bulk requests default a while back too.

To make a bunch of small requests, you can set the `"bulk_request": false`. So:

```
{
  "type": "rover",
  "bulk_request": false,
  "server": {
    "port": 5051
  },
  "commands": [
    "LOAD ON",
    "LOAD OFF"
  ]
}
```

Also, I'm glad to see someone is using the server feature and the commands.

tesau, 2022-02-24:
Thank-You for your quick reply

adding    "bulk_request": false   gives me no expected replys just the following

2022-02-23 19:39:35.740 [main] INFO  me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Got part of a response back. (Maybe timed out halfway through?) data='01 03 10 20 20 52 4E 47' Feel free to open an issue at https://github.com/wildmountainfarms/solarthing/issues/
2022-02-23 19:39:35.740 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4.454 seconds
2022-02-23 19:39:40.236 [main] ERROR me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Modbus exception
me.retrodaredevil.io.modbus.RedundancyException: Incorrect CRC checksum. Expected: 55342 but got: 36282 bytes: [45, 67, 84, 82, 76, 45, 82, 86, 82, 52, 48, 46, -40]

When I send commands manually cut/paste,  wait for reply which is correct...  and send same commands again without pause   the process dies.But if I wait/pause 10 seconds I get correct reply.   
Then I try sending one command at a time, not wait time after reply.    I continue to get correct reply.

Makes one wonder if the new controllers have less capacity/memory.
Or even worse maybe the guy at the keyboard has less and less capacity/memory

Thanks 
Terry

{
  "type": "rover-setup",
  "io": "config/rover_serial.json",
  "modbus": 1
}

acer@acer:/opt/solarthing/program/rover$ sudo -u solarthing ./run.sh
Using base config = config/base.json
2022-02-23 21:00:38.448 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.1.1.jar Last Modified: 2022-01-03T03:31:49Z Java version: 11.0.13
[stdout] Beginning main. Jar: Jar: solarthing-2022.1.1.jar Last Modified: 2022-01-03T03:31:49Z Java version: 11.0.13
[stderr] Beginning main. Jar: Jar: solarthing-2022.1.1.jar Last Modified: 2022-01-03T03:31:49Z Java version: 11.0.13
2022-02-23 21:00:38.552 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2022-02-23 21:00:39.276 [main] DEBUG me.retrodaredevil.solarthing.io.ReloadableIOBundle - Successfully reloaded IOBundle
Starting rover setup! This is deprecated and will be removed in a future version!.
batterytype

undervoltagewarninglevelraw
discharginglimitvoltageraw
overdischargerecoveryvoltageraw
overdischargetimedelayseconds

boostchargingvoltageraw
boostchargingrecoveryvoltageraw
boostchargingtimeminutes

equalizingchargingvoltageraw
equalizingchargingtimeminutes
equalizingchargingintervaldays

floatingchargingvoltageraw

chargingvoltagelimitraw 
overvoltagethresholdraw
user-unlocked
0
112
110
112
120
149
120
110
151
130
0
136
154
batterytype

undervoltagewarninglevelraw
discharginglimitvoltageraw
overdischargerecoveryvoltageraw
overdischargetimedelayseconds

boostchargingvoltageraw
boostchargingrecoveryvoltageraw
boostchargingtimeminutes

equalizingchargingvoltageraw
equalizingchargingtimeminutes
equalizingchargingintervaldays

floatingchargingvoltageraw

chargingvoltagelimitraw 
overvoltagethresholdraw
2022-02-23 21:01:22.402 [main] ERROR me.retrodaredevil.solarthing.program.RoverMain - (Fatal)Got exception!
java.util.NoSuchElementException: valueCode: 156 not found in enum: class me.retrodaredevil.solarthing.solar.renogy.RoverBatteryType
	at me.retrodaredevil.solarthing.packets.Modes.getActiveMode(Modes.java:61) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.RoverReadTable.getBatteryType(RoverReadTable.java:563) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverSetupProgram.handleSplit(RoverSetupProgram.java:190) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverSetupProgram.startRoverSetup(RoverSetupProgram.java:30) ~[solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.doRoverProgram(RoverMain.java:83) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.RoverMain.connectRoverSetup(RoverMain.java:65) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:141) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:187) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:252) [solarthing-2022.1.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.1.1.jar:?]
acer@acer:/opt/solarthing/program/rover$

retrodaredevil, 2022-02-24:
How far away is the rover from the PC you are using to run SolarThing? I know you're using socat, just want to know if the rover is close to the computer or is very far (like 60 miles away?).

Those `Incorrect CRC checksum` errors are definitely problematic. My guess is that SolarThing receives some data, has a gap of time, then the remaining data doesn't get through. That remaining data could throw off future requests (I think, I need to look at the code).

You can probably remove the `bulk_request` thing if it's making things worse. In a future release, I will probably allow the configuration of some timeout values, which *might* fix your problem. Most of the timeout values are hardcoded and are tuned for a rover directly connected to the device running SolarThing.

tesau, 2022-02-25:
Serial Cable is 4ft long  Rover connects to a Mikrotik RB750G which has one serial port shared over TCP/IP
The pc is 60 miles from the Rover.

Thanks
Terry

closed 2022-02-25 by tesau

retrodaredevil, 2022-02-27:
Go ahead and [update](https://solarthing.readthedocs.io/en/latest/updating.html). Then make these changes to your configuration:

```json
{
  "type": "modbus",
  "io": "config/rover_serial.json",
  "end_timeout": "PT0.1S",
  "devices": {
    "1": {
      "type": "rover",
      "server": {
        "port": 5051
      },
      "commands": [
        "LOAD ON",
        "LOAD OFF"
      ]
    }
  }
}
```

Basically all you should need is the addition of `"end_timeout": "PT0.1S"`. You can keep increasing that 0.1, but I don't recommend going over 0.3. The higher that value is, the longer each request is going to take. As long as you keep bulk requests enabled (the default), it shouldn't be too bad.

Let me know if this helps or if there's some error you come across. (I haven't personally tested overriding the end_timeout value).

riker65, 2022-07-16:
Hi, who would 
"io": "config/rover_serial.json",

this log like? did not see file config/rover_serial.json

thanks for guiding

Thomas

retrodaredevil, 2022-07-19:
@riker65 in this case `config/rover_serial.json`'s contents are probably very similar to 

```
{
  "type": "serial",
  "port": "/dev/ttyUSB0",
  "serial_config": null
}
```
except that the port is a port created by `socat`. This is why SolarThing technically supports socat: it's because SolarThing recognizes what socat exposes as serial ports.

riker65, 2022-07-19:
Hi,
I am struggling with the right socat option

`sudo socat pty,link=/home/zwh100/dev/ttyVR0,raw,mode=777 TCP:192.168.1.184:2024
`

error when cheching with ./run.sh
`
2022-07-19 14:37:24.752 [main] ERROR me.retrodaredevil.solarthing.program.RoverPacketListUpdater - Modbus exception
me.retrodaredevil.io.modbus.ModbusRuntimeException: Got exception while writing

...
Caused by: me.retrodaredevil.solarthing.io.NotInitializedIOException: IO not initialized!

`

on remote IP I have 
ser2net.yaml:
`# Renogy
connection: &con0296
    accepter: tcp,2024
    enable: on
    options:
      banner: *banner
      kickolduser: true
      telnet-brk-on-sync: false
    connector: serialdev,
              /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0,
              9600n81,local
`

telnet ip 2024 is giving me 
`"((RCC40RVRE-G1
                   Ž"((RCC40RVRE-G1
                                    `

and testing socat TCP4:IP:2024 
`socat - TCP4:192.168.1.184:2024

ser2net port tcp,2024 device serialdev, /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0, 9600n81,local [] (Debian GNU/Linux)

"((RCC40RVRE-G1
                   Žd��d��
`

'{	
  "type": "serial",
  "port": "/home/zwh100/dev/ttyVR0",
  "serial_config": null
}'

thanks for hints  Thomas

retrodaredevil, 2022-07-20:
> Caused by: me.retrodaredevil.solarthing.io.NotInitializedIOException: IO not initialized!

Please give the full error. This particular error isn't that useful to me, but there is likely output above that which is useful.

retrodaredevil, 2022-07-21:
@riker65 I still need more info than that. There should be some errors right after the program starts. If you can capture those errors and the initial logs from when the program starts it should give a more helpful message.