wildmountainfarms/solarthing #52: Stuck in solarthing-couchdb interface?

bpehrson, 2022-03-16:
I thought I was almost there, but got stuck and cannot see why. Any feedback is welcome

Started solarthing on a raspberrypi connected to a Epever Triron which should be compatible with Tracer

sa0bxi@bull:~$ sudo systemctl start solarthing-custom_tracer.service 

checking status, which seems OK....

sa0bxi@bull:~$ sudo systemctl status solarthing-custom_tracer.service 
● solarthing-custom_tracer.service - The service for SolarThing custom_tracer
   Loaded: loaded (/etc/systemd/system/solarthing-custom_tracer.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2022-03-16 23:02:06 CET; 5s ago
 Main PID: 7210 (sh)
    Tasks: 15 (limit: 2059)
   CGroup: /system.slice/solarthing-custom_tracer.service
           ├─7210 sh /opt/solarthing/program/custom_tracer/run.sh
           ├─7212 sh ../.scripts/solarthing.sh
           ├─7214 bash ../.scripts/../.bin/local_log_solarthing --base config/base.json
           ├─7220 sh ../.scripts/../.bin/../.bin/raw_solarthing --base config/base.json
           └─7224 java --add-opens=java.base/java.lang.invoke=ALL-UNNAMED -jar ../.scripts/../.bin/../.bin/../solarthing.jar --base config/base.json

mar 16 23:02:06 bull systemd[1]: Started The service for SolarThing custom_tracer.

BUT failed after a while.....

sa0bxi@bull:~$ sudo systemctl status solarthing-custom_tracer.service 
● solarthing-custom_tracer.service - The service for SolarThing custom_tracer
   Loaded: loaded (/etc/systemd/system/solarthing-custom_tracer.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2022-03-16 23:02:23 CET; 1s ago
  Process: 7210 ExecStart=/opt/solarthing/program/custom_tracer/run.sh (code=exited, status=197)
 Main PID: 7210 (code=exited, status=197)

mar 16 23:02:06 bull systemd[1]: Started The service for SolarThing custom_tracer.
mar 16 23:02:23 bull systemd[1]: solarthing-custom_tracer.service: Main process exited, code=exited, status=197/n/a
mar 16 23:02:23 bull systemd[1]: solarthing-custom_tracer.service: Failed with result 'exit-code'.

Here is the log. and no data in /opt/solarthing/program/custom_tracer/config/triron.json

sa0bxi@bull:~$ tail /opt/solarthing/program/custom_tracer/logs/log_info.log 
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:252) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.2.0.jar:?]
Caused by: com.fasterxml.jackson.databind.exc.MismatchedInputException: No content to map due to end-of-input
 at [Source: (FileInputStream); line: 1, column: 0]
	at com.fasterxml.jackson.databind.exc.MismatchedInputException.from(MismatchedInputException.java:59) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:4765) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4667) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3666) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.ConfigUtil.getDatabaseConfig(ConfigUtil.java:46) ~[solarthing-2022.2.0.jar:?]
	... 8 more
sa0bxi@bull:~$

retrodaredevil, 2022-03-16:
You must name your config `base.json` or it will not work. I imagine the reason it was running for a little bit is because the program was booting up. It takes a few seconds to boot up the program on older Raspberry Pis.

bpehrson, 2022-03-16:
Thanks for a very quick respons!

I do have a base.json
sa0bxi@bull:~$ ls /opt/solarthing/program/custom_tracer/config/
base.json  base.json~  couchdb.json  couchdb.json~  tracer_serial.json	triron.json
sa0bxi@bull:~$ cat /opt/solarthing/program/custom_tracer/config/base.json
{
  "type": "request",
  "source": "default",
  "fragment": 5,
  "unique": 30,
  "databases": [
    "config/triron.json"
  ],
  "request": [
    {
      "type": "modbus",
      "io": "config/tracer_serial.json",
      "devices": {
        "1": {
          "type": "tracer",
          "clock": {
            "threshold": "PT5S"
          }
        }
      }
    }
  ]
}
sa0bxi@bull:~$ 

And here is tail -70 /opt/solarthing/program/custom_tracer/logs/log_info.log
sa0bxi@bull:~$ tail -70 /opt/solarthing/program/custom_tracer/logs/log_info.log
	at com.fasterxml.jackson.databind.ObjectMapper.createGenerator(ObjectMapper.java:1203) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper.writeValue(ObjectMapper.java:3763) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.analytics.AnalyticsManager.<init>(AnalyticsManager.java:42) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:38) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:145) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:187) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:252) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.2.0.jar:?]
2022-03-16 23:02:22.765 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2022-03-16 23:02:22.977 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - [LOG] Ending SolarThing. Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.12
2022-03-16 23:02:22.981 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Got throwable
java.lang.RuntimeException: Couldn't parse data!
	at me.retrodaredevil.solarthing.program.ConfigUtil.getDatabaseConfig(ConfigUtil.java:48) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.ConfigUtil.getDatabaseConfigs(ConfigUtil.java:34) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.PacketHandlerInit.initHandlers(PacketHandlerInit.java:156) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:45) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:40) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:145) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:187) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:252) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.2.0.jar:?]
Caused by: com.fasterxml.jackson.databind.exc.MismatchedInputException: No content to map due to end-of-input
 at [Source: (FileInputStream); line: 1, column: 0]
	at com.fasterxml.jackson.databind.exc.MismatchedInputException.from(MismatchedInputException.java:59) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:4765) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4667) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3666) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.ConfigUtil.getDatabaseConfig(ConfigUtil.java:46) ~[solarthing-2022.2.0.jar:?]
	... 8 more
2022-03-16 23:17:23.912 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.12
2022-03-16 23:17:25.129 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2022-03-16 23:17:29.234 [main] INFO  me.retrodaredevil.solarthing.program.RequestMain - Beginning request program
2022-03-16 23:17:29.254 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2022-03-16 23:17:29.523 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Generated a new Analytics UUID
2022-03-16 23:17:29.529 [main] WARN  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Couldn't save analytics data!
java.io.FileNotFoundException: .data/analytics_data.json (Permission denied)
	at java.io.FileOutputStream.open0(Native Method) ~[?:?]
[... 9 lines cut ...]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:187) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:252) [solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.2.0.jar:?]
[... cut ...]
	at com.fasterxml.jackson.databind.exc.MismatchedInputException.from(MismatchedInputException.java:59) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:4765) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4667) ~[solarthing-2022.2.0.jar:?]
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3666) ~[solarthing-2022.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.ConfigUtil.getDatabaseConfig(ConfigUtil.java:46) ~[solarthing-2022.2.0.jar:?]
	... 8 more
sa0bxi@bull:~$

retrodaredevil, 2022-03-17:
Ok, my bad. It looks like there's some sort of problem with your JSON in your `config/triron.json` file. If you could show me the contents of that I can help out.

bpehrson, 2022-03-17:
On 2022-03-17 01:23, Joshua Shannon wrote:

> Ok, my bad. It looks like there's some sort of problem with your JSON 
> in your |config/triron.json| file. If you could show me the contents 
> of that I can help out.
>

Oh, obviously my bad. The |config/triron.json| file is empty since I 
found no clue about its content.

I was stupid enough to assume that it would  magically be filled with 
the content that I have seen coming from the Triron regulator via 
solarthing at an intermediate test earlier in the installation process.

Even now, when I must assume that I should fill it with something, I 
cannot find out with what.

What should I do?

> Message ID: ***@***.***>
>

retrodaredevil, 2022-03-17:
So you won't need a file called `triron.json`, but you will eventually need a file containing configuration for whatever database you want to use. First, I recommend removing `"config/triron.json"` from your `base.json` so that it has no databases. Once you have done that, run SolarThing. Ideally there should be no errors and you'll eventually see data from your charge controller. If you don't get any data and just continuously get errors, you can send me those errors and I'll help out.

Once you have gotten data from your charge controller, then and only then should you try to configure a database. Most people will go for a CouchDB database: https://solarthing.readthedocs.io/en/latest/config/couchdb.html. Once CouchDB is configured, then most people will set up the GraphQL program and Grafana.

bpehrson, 2022-03-17:
OK, just to explain, the reason I put triron.json in the base.jason file was how I interpreted the page https://solarthing.readthedocs.io/en/latest/configuration-edit-base-json.html

I just undid the edit and tried to start again. Now it is running without failing:

sa0bxi@bull:/opt/solarthing/program/custom_tracer/config$ sudo systemctl status solarthing-custom_tracer.service
● solarthing-custom_tracer.service - The service for SolarThing custom_tracer
   Loaded: loaded (/etc/systemd/system/solarthing-custom_tracer.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2022-03-17 19:14:05 CET; 12min ago
 Main PID: 9393 (sh)
    Tasks: 15 (limit: 2059)
   CGroup: /system.slice/solarthing-custom_tracer.service
           ├─9393 sh /opt/solarthing/program/custom_tracer/run.sh
           ├─9395 sh ../.scripts/solarthing.sh
           ├─9397 bash ../.scripts/../.bin/local_log_solarthing --base config/base.json
           ├─9403 sh ../.scripts/../.bin/../.bin/raw_solarthing --base config/base.json
           └─9407 java --add-opens=java.base/java.lang.invoke=ALL-UNNAMED -jar ../.scripts/../.bin/../.bin/../solarthing.jar --base config/bas

mar 17 19:14:05 bull systemd[1]: Started The service for SolarThing custom_tracer.

---
I attach the last 70 lines of the logfile

[solarthing-test-2022-03-17-1920.txt](https://github.com/wildmountainfarms/solarthing/files/8289198/solarthing-test-2022-03-17-1920.txt)

bpehrson, 2022-03-17:
I should add that I do get data from the Epever Triron Charge Controller.
I just ran this to check that I am still in contact with it:
sa0bxi@bull:~$ solarthing check --port /dev/ttyXRUSB0 --type tracer
Going to open serial port using default serial configuration...
Successfully opened serial port...
Checking on address: 1
Success! Battery Voltage: 13.05
sa0bxi@bull:~$ 

I also believe that I installed and configured  couchdb and GraphQL. I have installed but not yet completed Grafana since I do not have a database yet.

bpehrson, 2022-03-17:
Here is my config/couchdb.json file:
sa0bxi@bull:~$ cat /opt/solarthing/program/custom_tracer/config/couchdb.json
{
  "type": "couchdb",
  "settings": {
    "packet_upload": {
      "throttle_factor": 3,
      "initial_skip": 1
    },
    "command_download": {
      "throttle_factor": 3,
      "initial_skip": 4
    }
  },
  "config": {
    "protocol": "http",
    "host": "192.168.1.254",
    "port": 5984,
    "username": "sa0bxi",
    "password": "********",
    "connection_timeout": 1.5,
    "call_timeout": 10
  }
}
sa0bxi@bull:~$

bpehrson, 2022-03-17:
I attach a piece of the DEBUG log. So there is where the data goes?!

I will try to proceed but I am not sure about how to configure the database. I can access the couchdb web-interface but  I guess that is where I went wrong, when creating the config/triron.json file. Any hint?

[solarthing-debug-log-2022-03-17.txt](https://github.com/wildmountainfarms/solarthing/files/8292574/solarthing-debug-log-2022-03-17.txt)

retrodaredevil, 2022-03-17:
Yup. Looks like SolarThing is working. Once you have access to the CouchDB web interface, you don't actually need to use the web interface anymore. You have a CouchDB config file that should point towards your CouchDB database (assuming your "host" is correct it's good).

Now you follow the instructions here: https://solarthing.readthedocs.io/en/latest/config/couchdb.html#running-the-setup-program. Running that will ask you questions. Some of those questions will ask you if you wish to configure users. For security, I recommend setting up the `uploader` user. (The `manager` user is optional and is for advanced users). Note that setting up these 1 or 2 additional CouchDB users is optional, so you really could skip the setup of both users. 

Once you do or don't set up additional users, the program will automatically set up some of the required databases that SolarThing needs to use. At that point, you can add your CouchDB configuration file to the `"databases"` in `base.json`.

Once you have CouchDB setup and you have it configured to upload to CouchDB, you should check the log file to make sure there aren't consistent CouchDB errors. If there are no CouchDB errors, you can double confirm that the uploading is working by navigating to the web interface and viewing the `solarthing` database.

If you get CouchDB uploading working, then you can start to setup your GraphQL progarm, and after that Grafana.

bpehrson, 2022-03-17:
OK, the first step seems to have worked out well:
sa0bxi@bull:/opt/solarthing/program/custom_tracer$ solarthing run --couchdb-setup config/couchdb.json
23:04:30.962 [main] INFO  m.r.s.p.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.12
[stdout] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.12
[stderr] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.12
You will now setup your CouchDB instance! Some databases will be automatically created (enter)

Already exists: solarthing
Already exists: solarthing_events
Already exists: solarthing_closed
Already exists: solarthing_open
Already exists: solarthing_cache
Already exists: solarthing_alter
All necessary databases have been created.

Now views and security will be configured for each database. Please enter the name of the user to be added as an admin to each database.
This user is commonly named 'uploader'. (Leave blank to not configure)
Name of user: uploader
User: uploader will be used. (Enter to confirm)

Please enter a password for 'uploader'.
********
You can also enter the name of the user to manage the solarthing_cache and solarthing_alter databases.
This user is commonly named 'manager'. (Leave blank to not configure) (Use 'uploader' to use same user to manage the cache database)

No user will be configured to manage the solarthing_cache and solarthing_alter database. (Enter to confirm)

Adding packets design to database solarthing
This database will have the millisNull view
This database will be readonly
updated _design/packets document on database: solarthing
Configuring security for database solarthing

Adding packets design to database solarthing_events
This database will have the millisNull view
This database will be readonly
updated _design/packets document on database: solarthing_events
Configuring security for database solarthing_events

Adding packets design to database solarthing_closed
This database will be readonly
updated _design/packets document on database: solarthing_closed
Configuring security for database solarthing_closed

Adding packets design to database solarthing_open
This database will have the millisNull view
updated _design/packets document on database: solarthing_open
Configuring security for database solarthing_open

Configuring security for database solarthing_cache

Adding packets design to database solarthing_alter
This database will have the simpleAllDocs view
This database will be readonly
updated _design/packets document on database: solarthing_alter
Configuring security for database solarthing_alter

Completed successfully!
sa0bxi@bull:/opt/solarthing/program/custom_tracer$

bpehrson, 2022-03-17:
Did you mean that the base.json now should look like this (the databases line)?
{
  "type": "request",
  "source": "default",
  "fragment": 5,
  "unique": 30,
  "databases": [ conf/couchdb.json ],
  "request": [
    {
      "type": "modbus",
      "io": "config/tracer_serial.json",
      "devices": {
        "1": {
          "type": "tracer",
          "clock": {
            "threshold": "PT5S"
          }
        }
      }
    }
  ]
}

retrodaredevil, 2022-03-17:
> Did you mean that the base.json now should look like this (the databases line)?

Yup. It should be `... "databases": [ "config/couchdb.json" ], ...`

bpehrson, 2022-03-17:
config/couchdb.json I mean, not conf/

bpehrson, 2022-03-19:
It seems as I am stumbling towards the goal. I have some sort of problem 
in the interface between graphql and grafana and I am a novice on both. 
I got this error related to $SourceId defining the dataPath for 
data.batteryVoltage.  I cannot find any guidance about how and where to 
set it in any of the related documents. Let me know if I am asking too 
much. Your support has been really great.

retrodaredevil, 2022-03-19:
The documentation for setting up GraphQL queries is lacking. What reference did you use for the query that you're trying to get working right now? 

$sourceId is a variable I use in many of my queries, and you have to add a variable or constant to your Grafana dashboard. Alternatively, don't use that variable at all and just set it to "default", which for a simple setup with a single source ID is fine. 

I might be able to give more detailed instructions later.

bpehrson, 2022-03-19:
I was using https://grafana.com/grafana/plugins/fifemon-graphql-datasource/?tab=installation to install the data source and https://solarthing.readthedocs.io/en/latest/data/graphql-grafana.html?highlight=localhost%3A8080#graphql-and-grafana 
to configure it. 
I enclose some illustrating screenshot
[image]
[image]
[image]
s

retrodaredevil, 2022-03-19:
Hmm, although it's saying your data path is incorrect, it looks good to me.

However like I said, `$sourceId` is the problem here. Go ahead and replace that with `default` and tell me if that fixes anything. I will have to update that part of the documentation to not recommend people use a variable there. Also, did you confirm that communication between the GraphQL plugin and SolarThing GraphQL is working? (When you set it up did you get the green success box saying that it connected correctly?)

bpehrson, 2022-03-20:
The change to default did not work (see this screenshot)

[image]
 
Regarding your second question, is this screenshot what you mean?
[image]

retrodaredevil, 2022-03-20:
That all looks good to me. Can you show me your query inspector? Should look something like this:

[image]

bpehrson, 2022-03-20:
Mine is very empty...
[image]

bpehrson, 2022-03-20:
sorry, wrong screenshot, here it i
[image]
s

retrodaredevil, 2022-03-20:
Go into the Query tab and take a screenshot there. You data tab will always be empty if any part of the query fails.

bpehrson, 2022-03-20:
Is this what you mean?
[image]

if you want to, I could make the RaspberryPi available on a public ip-address and create a user joshua for you.....

bpehrson, 2022-03-20:
as long as needed for your test the pi is at 44.5.4.37. I need an sms number to send passwd

retrodaredevil, 2022-03-20:
As I see why it's doing that. You don't have the right IP in your data source settings. You need to use `http://localhost:8080/graphql` not `http://localhost:3000/graphql`. 3000 is the port Grafana is running on. 8080 is the default port for SolarThing GraphQL. As for why your data source did a successful test in the first place, I don't know.

If that doesn't fix it, I can SSH into your Pi with a public key. I never recommend exposing SSH with password authentication.
```
ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDHATUtgfZELMgmia0AgplK4MMd3h9N1CCutopUf1+9gWAZ+A39Ss+p+3QPrnuwd0WAVdRjwFCZG9IjV3Mllx39v7oc5jsJHGzVtZkoUldX6V/VLIRZdnrSAiXW1qyi0vCnywMYHws7asBheJXyrea8FNspZBveJsr9KvKY4pX0nW8O7oOvdSox2lwzXlKtYRIVC+YdBQfDQvYx7n85wUIYUhuDPYB/Ddl5fPmNJlD+VC1VDbZZx/u8E/3QKUsPH1ISDktJ31fsd9JTXa3zkPezc/MfdGgxlcSjhH+V5dQbvmg+gZo0Fpy51HrpJtErxFmGNJudOH98yRb2Prn2KZ17 josh@josh-hp-linux
```
Or the commend directly.
```shell
echo "ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDHATUtgfZELMgmia0AgplK4MMd3h9N1CCutopUf1+9gWAZ+A39Ss+p+3QPrnuwd0WAVdRjwFCZG9IjV3Mllx39v7oc5jsJHGzVtZkoUldX6V/VLIRZdnrSAiXW1qyi0vCnywMYHws7asBheJXyrea8FNspZBveJsr9KvKY4pX0nW8O7oOvdSox2lwzXlKtYRIVC+YdBQfDQvYx7n85wUIYUhuDPYB/Ddl5fPmNJlD+VC1VDbZZx/u8E/3QKUsPH1ISDktJ31fsd9JTXa3zkPezc/MfdGgxlcSjhH+V5dQbvmg+gZo0Fpy51HrpJtErxFmGNJudOH98yRb2Prn2KZ17 josh@josh-hp-linux" >> /home/joshua/.ssh/authorized_keys
```
I don't normally SSH to do setup for other people, but if we still can't figure this out I'll ssh in.

bpehrson, 2022-03-20:
I changed the port. Some change happened but still not data
[image]
I will add the public key to your account

retrodaredevil, 2022-03-20:
Can you expand the error object in the bottom left of the screen? It should say exactly what the error is. Or hover over the red "!" and it might tell you there too.

bpehrson, 2022-03-20:
First I am hovering over the !
[image]

I do not see a error button in the low left of the screen, unless you mean clicking on the word **errors** in the Inspect/query-frame

bpehrson, 2022-03-20:
The last few lines expanded are
[hideFromInspector:false]()
[response:Object]()
[errors:Array[1]]()
[0:Object]()
[data:Object]()
[data:null]()

retrodaredevil, 2022-03-20:
Ok, that error is good. There should be more to that error in SolarThing GraphQL's log files. As long as you are running a recent version of SolarThing, you can view these logs in `/opt/solarthing/program/graphql/logs/log_warn.log`. If you could give me some of that log file (the last 100 lines would probably do), I bet that exact error will be there.

bpehrson, 2022-03-20:
unfortuntaey no log in that directory. Did I miss something in settings?
bjorn@bpx390:~$ ls  /opt/solarthing/program/graphql
config  README.md  run.sh
bjorn@bpx390:~$

bpehrson, 2022-03-20:
sorry, a typo, here it is
sa0bxi@bull:~$ tail /opt/solarthing/program/graphql/logs/log_warn.log
2022-03-20 22:44:43.398  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and @me.retrodaredevil.solarthing.annotations.NotNull() java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
2022-03-20 22:44:43.430  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
sa0bxi@bull:~$ tail -70 /opt/solarthing/program/graphql/logs/log_warn.log
2022-03-20 22:44:43.398  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and @me.retrodaredevil.solarthing.annotations.NotNull() java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
2022-03-20 22:44:43.430  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
sa0bxi@bull:~$ cat /opt/solarthing/program/graphql/logs/log_warn.log
2022-03-20 22:44:43.398  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and @me.retrodaredevil.solarthing.annotations.NotNull() java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
2022-03-20 22:44:43.430  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
sa0bxi@bull:~$

retrodaredevil, 2022-03-20:
Sorry, can I see the debug logs instead. I forgot that what I wanted to see actually goes to INFO logs. 

```
tail -100 /opt/solarthing/program/graphql/logs/log_debug.log
```
should do it.

Also, I am unable to ssh using my key. Make sure that my `.ssh` directory and `.ssh/authorized_keys` has correct permissions. Likely run the command: `sudo chown -R joshua:joshua /home/joshua/.ssh` and it will be fixed.

bpehrson, 2022-03-20:
Here is the debug log
sa0bxi@bull:~$ tail -70 /opt/solarthing/program/graphql/logs/log_debug.log 
	at org.apache.catalina.core.StandardWrapperValve.invoke(StandardWrapperValve.java:197) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.catalina.core.StandardContextValve.invoke(StandardContextValve.java:97) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.catalina.authenticator.AuthenticatorBase.invoke(AuthenticatorBase.java:540) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.catalina.core.StandardHostValve.invoke(StandardHostValve.java:135) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.catalina.valves.ErrorReportValve.invoke(ErrorReportValve.java:92) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.catalina.core.StandardEngineValve.invoke(StandardEngineValve.java:78) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.catalina.connector.CoyoteAdapter.service(CoyoteAdapter.java:357) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.coyote.http11.Http11Processor.service(Http11Processor.java:382) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.coyote.AbstractProcessorLight.process(AbstractProcessorLight.java:65) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.coyote.AbstractProtocol$ConnectionHandler.process(AbstractProtocol.java:895) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.tomcat.util.net.NioEndpoint$SocketProcessor.doRun(NioEndpoint.java:1732) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.tomcat.util.net.SocketProcessorBase.run(SocketProcessorBase.java:49) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.tomcat.util.threads.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1191) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.tomcat.util.threads.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:659) [tomcat-embed-core-9.0.56.jar!/:?]
	at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61) [tomcat-embed-core-9.0.56.jar!/:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]
Caused by: me.retrodaredevil.solarthing.database.exception.SolarThingDatabaseException
	at me.retrodaredevil.solarthing.database.couchdb.ExceptionUtil.createFromCouchDbException(ExceptionUtil.java:30) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.database.couchdb.ExceptionUtil.createFromCouchDbException(ExceptionUtil.java:18) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.database.couchdb.CouchDbMillisDatabase.query(CouchDbMillisDatabase.java:73) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:116) ~[classes!/:?]
	... 77 more
Caused by: me.retrodaredevil.couchdbjava.exception.CouchDbException: java.net.SocketTimeoutException: connect timed out
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.executeCall(OkHttpCouchDbInstance.java:92) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.auth.CookieAuthHandler.authSession(CookieAuthHandler.java:122) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.auth.CookieAuthHandler.preAuthorize(CookieAuthHandler.java:160) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.preAuthorize(OkHttpCouchDbInstance.java:161) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase$OkHttpCouchDbShared.queryView(OkHttpCouchDbDatabase.java:389) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.queryView(OkHttpCouchDbDatabase.java:242) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.CouchDbShared.queryView(CouchDbShared.java:23) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.solarthing.database.couchdb.CouchDbMillisDatabase.query(CouchDbMillisDatabase.java:71) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:116) ~[classes!/:?]
	... 77 more
Caused by: java.net.SocketTimeoutException: connect timed out
	at java.net.PlainSocketImpl.socketConnect(Native Method) ~[?:?]
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:399) ~[?:?]
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:242) ~[?:?]
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:224) ~[?:?]
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392) ~[?:?]
	at java.net.Socket.connect(Socket.java:609) ~[?:?]
	at okhttp3.internal.platform.Platform.connectSocket(Platform.kt:128) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.RealConnection.connectSocket(RealConnection.kt:295) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.RealConnection.connect(RealConnection.kt:207) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.ExchangeFinder.findConnection(ExchangeFinder.kt:226) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.ExchangeFinder.findHealthyConnection(ExchangeFinder.kt:106) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.ExchangeFinder.find(ExchangeFinder.kt:74) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.RealCall.initExchange$okhttp(RealCall.kt:255) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.ConnectInterceptor.intercept(ConnectInterceptor.kt:32) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109) ~[okhttp-4.9.3.jar!/:?]
[... cut ...]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.queryView(OkHttpCouchDbDatabase.java:242) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.CouchDbShared.queryView(CouchDbShared.java:23) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.solarthing.database.couchdb.CouchDbMillisDatabase.query(CouchDbMillisDatabase.java:71) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:116) ~[classes!/:?]
	... 77 more

sa0bxi@bull:~$

retrodaredevil, 2022-03-20:
Can I have 150 lines instead of 70? The error messages get pretty long

bpehrson, 2022-03-20:
Here is 150
sa0bxi@bull:~$ tail -150 /opt/solarthing/program/graphql/logs/log_debug.log 
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.lambda$new$0(OkHttpCouchDbInstance.java:49) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.RealCall.getResponseWithInterceptorChain$okhttp(RealCall.kt:201) ~[okhttp-4.9.3.jar!/:?]
	at okhttp3.internal.connection.RealCall.execute(RealCall.kt:154) ~[okhttp-4.9.3.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.executeCall(OkHttpCouchDbInstance.java:90) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.auth.CookieAuthHandler.authSession(CookieAuthHandler.java:122) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.auth.CookieAuthHandler.preAuthorize(CookieAuthHandler.java:160) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.preAuthorize(OkHttpCouchDbInstance.java:161) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase$OkHttpCouchDbShared.queryView(OkHttpCouchDbDatabase.java:389) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.queryView(OkHttpCouchDbDatabase.java:242) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.CouchDbShared.queryView(CouchDbShared.java:23) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.solarthing.database.couchdb.CouchDbMillisDatabase.query(CouchDbMillisDatabase.java:71) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:116) ~[classes!/:?]
	... 77 more

2022-03-20 22:52:33.374  INFO [nio-8080-exec-8] m.r.s.r.g.SolarThingExceptionHandler     : Got database exception

me.retrodaredevil.solarthing.rest.exceptions.DatabaseException: Exception querying from 1647790520773 to 1647812120774
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:118) ~[classes!/:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.queryPackets(SimpleQueryHandler.java:123) ~[classes!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.queryStatus(SimpleQueryHandler.java:162) ~[classes!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.service.SolarThingGraphQLService.queryStatus(SolarThingGraphQLService.java:60) ~[classes!/:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at io.leangen.graphql.metadata.execution.FixedMethodInvoker.execute(FixedMethodInvoker.java:22) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.metadata.Resolver.resolve(Resolver.java:102) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.lambda$execute$1(OperationExecutor.java:93) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:103) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:99) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:58) ~[spqr-0.11.2.jar!/:?]
	at graphql.execution.ExecutionStrategy.fetchField(ExecutionStrategy.java:270) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.ExecutionStrategy.resolveFieldWithInfo(ExecutionStrategy.java:203) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.AsyncExecutionStrategy.execute(AsyncExecutionStrategy.java:60) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.Execution.executeOperation(Execution.java:165) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.Execution.execute(Execution.java:104) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.execute(GraphQL.java:557) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.parseValidateAndExecute(GraphQL.java:482) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.executeAsync(GraphQL.java:446) ~[graphql-java-16.2.jar!/:?]
	at java.util.concurrent.CompletableFuture.uniComposeStage(CompletableFuture.java:1106) [?:?]
	at java.util.concurrent.CompletableFuture.thenCompose(CompletableFuture.java:2235) [?:?]
	at graphql.spring.web.servlet.components.DefaultGraphQLInvocation.invoke(DefaultGraphQLInvocation.java:41) [graphql-java-spring-webmvc-2.0.jar!/:?]
	at graphql.spring.web.servlet.components.GraphQLController.executeRequest(GraphQLController.java:150) [graphql-java-spring-webmvc-2.0.jar!/:?]
	at graphql.spring.web.servlet.components.GraphQLController.graphqlPOST(GraphQLController.java:84) [graphql-java-spring-webmvc-2.0.jar!/:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
[... cut ...]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.queryView(OkHttpCouchDbDatabase.java:242) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.couchdbjava.CouchDbShared.queryView(CouchDbShared.java:23) ~[couchdb-java-83d5c5bb78.jar!/:?]
	at me.retrodaredevil.solarthing.database.couchdb.CouchDbMillisDatabase.query(CouchDbMillisDatabase.java:71) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:116) ~[classes!/:?]
	... 77 more

sa0bxi@bull:~$ tail -150 /opt/solarthing/program/graphql/logs/log_debug.log

retrodaredevil, 2022-03-20:
So one of two things is happening here: You need to increase the timeout value so it can connect, or the IP you are trying to reach is not correct, and it is timing out just because of that. It's my understanding that you have the main SolarThing program for uploading tracer data and then the GraphQL program running on the Pi while CouchDB is on another computer and Grafana is on that Raspberry Pi?

Have you confirmed that your main SolarThing program is successfully uploading data to CouchDB by going into the CouchDB web interface and making sure there is data in the `solarthing` database?

If you think you have the ip correct in your `couchdb.json`, then inside your CouchDB config, increase that value that is `1.5` right now to something like `10`. That value is in seconds

bpehrson, 2022-03-20:
let me know if the ssh login woks now

retrodaredevil, 2022-03-20:
> let me know if the ssh login woks now

It does work now. I'll take a look at it in about 2 hours and will probably play with stuff then. Feel free to try out my suggestions above before then and get back to me.

bpehrson, 2022-03-20:
everything is running on the same pi
I am using the web version of couchdb

retrodaredevil, 2022-03-20:
Ah, change the ip you are using from `192.168.X.X` to `localhost` in both `custom_tracer/config` and `graphql/config`. I don't know where you got `192.168.1.73` from, but it doesn't seem to be resolving to your pi. If that isn't it, I'll take a closer look later.

bpehrson, 2022-03-20:
I installed the apache coachdb according to 
https://docs.couchdb.org/en/stable/install/unix.html#installation-using-the-apache-couchdb-convenience-binary-packages
it is running
sa0bxi@bull:~$ sudo service couchdb status
● couchdb.service - Couchdb service
   Loaded: loaded (/etc/systemd/system/couchdb.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2022-03-20 22:17:40 CET; 1h 21min ago
 Main PID: 528 (beam.smp)
    Tasks: 42 (limit: 2059)
   CGroup: /system.slice/couchdb.service
           ├─  528 /home/couchdb/bin/../erts-10.2.4/bin/beam.smp -K true -A 16 -Bd -- -root /home/couchdb/bin/.. -progname couchdb -- -home /home/couchdb -- -boot /home/couchdb/bin/../releases/3.2.1/cou
           ├─  574 erl_child_setup 1024
           ├─11016 inet_gethost 4
           └─11017 inet_gethost 4

bpehrson, 2022-03-20:
Yippie, that made the trick. the 192.168.1.73 was the address of the pi before moving it to the public address. That would have worked after I changed the url but before I changed the url,  I changed the IP to the public one. When I changed to localhst, it all worked. You can see that in the diagram.
Thanks a LOT. Bedtime here nowzzz.
[image]

retrodaredevil, 2022-03-21:
Glad we finally got this figured out. This should help me make the documentation better (if I ever get around to it :smile:).

Also, please take your pi off that public IP if you leave password authentication enabled in SSH. Bots will start hitting that ip with login attempts. No bot would probably get in if you use good passwords, but it's still a good idea. Personally I leave password authentication enabled locally, but then disable it for login attempts outside my local network.

closed 2022-03-21 by retrodaredevil