wildmountainfarms/solarthing #55: function code=0x83 data='02'

Pcpro745, 2022-05-27:
Getting this error when trying to run solarthing for the first time.

2022-05-26 19:19:18.301 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4.725 seconds
2022-05-26 19:19:23.279 [main] ERROR me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Modbus exception
me.retrodaredevil.io.modbus.handling.ErrorCodeException: Expected to get 3 as the function code, but actually got 131. With exception code 2
        at me.retrodaredevil.io.modbus.handling.HandleResponseHelper.checkResponse(HandleResponseHelper.java:17) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.io.modbus.handling.BaseReadRegisters.handleResponse(BaseReadRegisters.java:52) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.io.modbus.handling.BaseReadRegisters.handleResponse(BaseReadRegisters.java:11) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:38) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheHolding(ModbusCacheSlave.java:47) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheHoldingRangeInclusive(ModbusCacheSlave.java:50) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.config.request.modbus.RoverModbusRequester.cacheRegisters(RoverModbusRequester.java:66) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.config.request.modbus.RoverModbusRequester.lambda$create$0(RoverModbusRequester.java:81) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper.receive(ModbusListUpdaterWrapper.java:57) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.RequestMain.doRequest(RequestMain.java:76) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:70) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:40) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:145) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:187) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:246) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.2.0.jar:?]
2022-05-26 19:19:23.302 [main] INFO  me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Communication with rover working well. Got this response back: function code=0x83 data='02' feel free to open issue at https://github.com/wildmountainfarms/solarthing/issues/
2022-05-26 19:19:23.304 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4.723 seconds

retrodaredevil, 2022-05-27:
What model charge controller are you using?

Pcpro745, 2022-05-27:
Renogy 3500 charge controller inverter. Same as the SRNE.

retrodaredevil, 2022-05-27:
Gotcha. SolarThing has never been tested on that model. SolarThing does not support inverters yet, but I imagine I might be able to tweak it to get it working with an inverter charger. I will take a closer look next week. Wish I had the time to look sooner.

Pcpro745, 2022-05-27:
Sweet! No worries!

Pcpro745, 2022-05-27:
To be honest the only thing that would be cool to get working would be the solar input. What it's currently pushing to the batteries. If you point me in a direct I can do some digging. I'm not too good with java but I have done some stuff with it.

retrodaredevil, 2022-05-29:
So, there are a few older documents that likely relate to what data is available for this device:
for inverter: [Inverter Modbus Protocol V1.6.zh-CN.en.pdf](https://github.com/wildmountainfarms/solarthing/files/8792580/Inverter.Modbus.Protocol.V1.6.zh-CN.en.pdf)
for solar charge controller: [ROVER_MODBUS.pdf](https://github.com/wildmountainfarms/solarthing/files/8792582/ROVER_MODBUS.pdf)

SolarThing is set up to get ALL the data from that second document because SolraThing supports all charge controllers that use that SRNE modbus spec. That error is telling us that some (or maybe most) of the spec for SRNE charge controllers is unsupported on this model. My hope is that I can update SolarThing to make some of the spec that is unsupported optional data, but if the solar part of this new spec is wildly different, then not only would it take a lot of work, we would have to hope that Renogy would give out that spec (which they have done in the past if you email them).

So to start, you can set `"bulk_request": false` in your base.json. I will eventually make a page on the docs for it, but for now I explained how to do that here: https://github.com/wildmountainfarms/solarthing/issues/51#issuecomment-1048362731.

Once you add that to your `base.json`, it should spit out a new error. You can post that error here. This will likely be a long process  of figuring out which fields are not supported by this model. I will update SolarThing to remove one of the unsupported fields, you can look at that example, and replicate the process for other unsupported fields.

Example of an unsupported field here: https://github.com/wildmountainfarms/solarthing/blob/c94178e33f67f135566f33c6620342effc17848e/core/src/main/java/me/retrodaredevil/solarthing/solar/renogy/rover/modbus/RoverModbusSlaveRead.java#L396. If we find more unsupported fields in that model, I will have to update the caching mechanism to not grab the possibly unsupported fields so you can set `bulk_request` back to true (the default).

This is all assuming that the spec is mostly the same.

Pcpro745, 2022-05-30:
For some odd reason i get an error when adding the bulk request option.

 (Fatal)Error while parsing ProgramOptions.
com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "bulk_request" (class me.retrodaredevil.solarthing.config.request.modbus.ModbusDataRequester), not marked as ignorable (4 known properties: "devices", "io", "end_timeout", "initial_timeout"])
 at [Source: (FileReader); line: 17, column: 6] (through reference chain: me.retrodaredevil.solarthing.config.options.RequestProgramOptions["request"]->java.util.ArrayList[0]->me.retrodaredevil.solarthing.config.request.modbus.ModbusDataRequester["bulk_request"])

retrodaredevil, 2022-05-30:
Can you post your base.json? It's possible I didn't document how you are supposed to configure bulk_request, but if I got that right you probably just put it in the wrong spot.

Pcpro745, 2022-05-30:
{
  "type": "request",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "databases": [] ,
  "request": [
    {
      "type": "modbus",
      "bulk_request": false,
      "io": "config/rover_serial.json",
      "devices": {
        "33": {
          "type": "rover"
        }
      }
    }
  ]
}

retrodaredevil, 2022-05-31:
You need to get the bulk_request at the same level that type rover is at, not type modbus. 

https://solarthing.readthedocs.io/en/latest/rover/bulk-request.html#disabling

Pcpro745, 2022-05-31:
{
  "type": "request",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "databases": [] ,
  "request": [
    {
      "type": "modbus"
      "io": "config/rover_serial.json",
      "devices": {
        "33": {
          "type": "rover"
          "bulk_request": false,
        }
      }
    }
  ]
}

Now getting this error. Which seems like it things there is an extra " somewhere?

Using base config = config/base.json
2022-05-30 20:42:56.119 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.15
[stdout] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.15
[stderr] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.15
2022-05-30 20:42:56.640 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2022-05-30 20:42:58.187 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Error while parsing ProgramOptions.
com.fasterxml.jackson.databind.JsonMappingException: Unexpected character ('"' (code 34)): was expecting comma to separate Object entries
 at [Source: (FileReader); line: 10, column: 8] (through reference chain: me.retrodaredevil.solarthing.config.options.RequestProgramOptions["request"]->java.util.ArrayList[0])
        at com.fasterxml.jackson.databind.JsonMappingException.wrapWithPath(JsonMappingException.java:392) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.JsonMappingException.wrapWithPath(JsonMappingException.java:363) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer._deserializeFromArray(CollectionDeserializer.java:371) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:244) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:28) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.impl.FieldProperty.deserializeAndSet(FieldProperty.java:138) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:313) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.BeanDeserializer._deserializeOther(BeanDeserializer.java:214) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:186) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer._deserializeTypedForId(AsPropertyTypeDeserializer.java:144) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer.deserializeTypedFromObject(AsPropertyTypeDeserializer.java:110) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.AbstractDeserializer.deserializeWithType(AbstractDeserializer.java:263) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.impl.TypeWrappedDeserializer.deserialize(TypeWrappedDeserializer.java:74) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:322) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4674) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3642) ~[solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:121) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:187) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:246) [solarthing-2022.2.0.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.2.0.jar:?]
Caused by: com.fasterxml.jackson.core.JsonParseException: Unexpected character ('"' (code 34)): was expecting comma to separate Object entries
 at [Source: (FileReader); line: 10, column: 8]
        at com.fasterxml.jackson.core.JsonParser._constructError(JsonParser.java:2391) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.core.base.ParserMinimalBase._reportError(ParserMinimalBase.java:735) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.core.base.ParserMinimalBase._reportUnexpectedChar(ParserMinimalBase.java:659) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.core.json.ReaderBasedJsonParser._skipComma(ReaderBasedJsonParser.java:2382) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.core.json.ReaderBasedJsonParser.nextToken(ReaderBasedJsonParser.java:716) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer._deserializeTypedForId(AsPropertyTypeDeserializer.java:141) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer.deserializeTypedFromObject(AsPropertyTypeDeserializer.java:110) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.AbstractDeserializer.deserializeWithType(AbstractDeserializer.java:263) ~[solarthing-2022.2.0.jar:?]
        at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer._deserializeFromArray(CollectionDeserializer.java:357) ~[solarthing-2022.2.0.jar:?]
        ... 17 more

retrodaredevil, 2022-05-31:
You have a syntax error in your JSON. You can't have a trailing comma, and you need a comma after the type rover line.

Pcpro745, 2022-05-31:
Using base config = config/base.json
2022-05-30 20:54:19.687 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.15
[stdout] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.15
[stderr] Beginning main. Jar: Jar: solarthing-2022.2.0.jar Last Modified: 2022-02-27T07:18:51Z Java version: 11.0.15
2022-05-30 20:54:20.211 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2022-05-30 20:54:21.890 [main] INFO  me.retrodaredevil.solarthing.program.RequestMain - Beginning request program
2022-05-30 20:54:21.900 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2022-05-30 20:54:21.931 [main] DEBUG me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics UUID: b0988526-3c9e-4dcc-93da-493c94181d5e
2022-05-30 20:54:23.609 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2022-05-30 20:54:23.737 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Commands are disabled
2022-05-30 20:54:23.933 [googleanalyticsjava-thread-1] WARN  com.brsanthu.googleanalytics.httpclient.ApacheHttpClientImpl - Exception while sending the Google Analytics tracker request com.brsanthu.googleanalytics.httpclient.HttpRequest@1d59f04
org.apache.http.conn.HttpHostConnectException: Connect to www.google-analytics.com:443 [www.google-analytics.com/0.0.0.0, www.google-analytics.com/0:0:0:0:0:0:0:0] failed: Connection refused (Connection refused)
        at org.apache.http.impl.conn.DefaultHttpClientConnectionOperator.connect(DefaultHttpClientConnectionOperator.java:159) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.conn.PoolingHttpClientConnectionManager.connect(PoolingHttpClientConnectionManager.java:359) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.execchain.MainClientExec.establishRoute(MainClientExec.java:381) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.execchain.MainClientExec.execute(MainClientExec.java:237) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.execchain.ProtocolExec.execute(ProtocolExec.java:185) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.execchain.RetryExec.execute(RetryExec.java:89) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.execchain.RedirectExec.execute(RedirectExec.java:111) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.client.InternalHttpClient.doExecute(InternalHttpClient.java:185) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:83) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:108) ~[solarthing-2022.2.0.jar:?]
        at com.brsanthu.googleanalytics.httpclient.ApacheHttpClientImpl.execute(ApacheHttpClientImpl.java:86) ~[solarthing-2022.2.0.jar:?]
        at com.brsanthu.googleanalytics.httpclient.ApacheHttpClientImpl.post(ApacheHttpClientImpl.java:102) ~[solarthing-2022.2.0.jar:?]
        at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.postSingle(GoogleAnalyticsImpl.java:165) ~[solarthing-2022.2.0.jar:?]
        at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.post(GoogleAnalyticsImpl.java:105) ~[solarthing-2022.2.0.jar:?]
        at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.lambda$postAsync$0(GoogleAnalyticsImpl.java:91) ~[solarthing-2022.2.0.jar:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:829) [?:?]
Caused by: java.net.ConnectException: Connection refused (Connection refused)
        at java.net.PlainSocketImpl.socketConnect(Native Method) ~[?:?]
        at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:412) ~[?:?]
        at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:253) ~[?:?]
        at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:237) ~[?:?]
        at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392) ~[?:?]
        at java.net.Socket.connect(Socket.java:609) ~[?:?]
        at org.apache.http.conn.ssl.SSLConnectionSocketFactory.connectSocket(SSLConnectionSocketFactory.java:339) ~[solarthing-2022.2.0.jar:?]
        at org.apache.http.impl.conn.DefaultHttpClientConnectionOperator.connect(DefaultHttpClientConnectionOperator.java:142) ~[solarthing-2022.2.0.jar:?]
        ... 18 more
2022-05-30 20:54:23.988 [googleanalyticsjava-thread-1] WARN  com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl - Exception while sending the Google Analytics tracker request Request [parms={APPLICATION_NAME=solarthing-program, SCREEN_NAME=request, USER_LANGUAGE=en-GB, APPLICATION_VERSION=V-UNKNOWN, TRACKING_ID=UA-70767765-2, HIT_TYPE=screenview, DOCUMENT_ENCODING=UTF-8, PROTOCOL_VERSION=1, CLIENT_ID=b0988526-3c9e-4dcc-93da-493c94181d5e}, customDimensions={}, customMetrics={}]
java.lang.NullPointerException: null
[... cut ...]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:256) [solarthing-2022.2.0.jar:?]
2022-05-30 20:54:31.385 [main] INFO  me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Communication with rover working well. Got this response back: function code=0x83 data='02' feel free to open issue at https://github.com/wildmountainfarms/solarthing/issues/
2022-05-30 20:54:31.387 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 3.175 seconds

retrodaredevil, 2022-05-31:
That's interesting. It seems that the "daily max discharging power" (register `0x0110`) is unsupported, and it's possible that many other registers are unsupported. What is interesting is that it seems that "daily max discharging current" is supported.

I emailed renogy asking for a PDF document for all the modbus mappings. We'll see if they have it. If they don't have it, I may start to make fields such as "daily max charging power" optional so that we can continue to find the fields that are optional.

Pcpro745, 2022-05-31:
I wonder if SRNE can provide any documentation? Seems like the Renogy is the exact same as the SRNE. I downloaded SRNE's Windows software and it connects to the Renogy no problem.

retrodaredevil, 2022-05-31:
That would be awesome if they could do that. Renogy got back to me asking to sign an NDA, which probably wouldn't work out since I want to use it to write open source software. I haven't read it entirely yet, but still don't like the idea of signing it even if I could use it to write SolarThing code.

Pcpro745, 2022-06-02:
I havent heard anything from SRNE. I emailed PowMR also to see if they have mappings they are also a rebranded SRNE.

retrodaredevil, 2022-06-03:
Yeah my guess is that they don't want this info getting out for some reason (they want to sell their BT modules). If Renogy all of a sudden makes people sign NDAs for people to get these documents, I'd imagine other companies either do the same, or it isn't worth giving out these documents and dealing with SRNE being mad. Props to Renogy for having the option, though. 

I might make something in the solarthing check program to detect which registers are unsupported. I do have a feeling that because daily max discharging power is unsupported and other similar ones are supported, that some of those fields may not actually be a mapping to what I hoped they'd be. 

The rover-setup program would probably work great for detecting supported or unsupported fields, but I'm going to eventually remove it. (it's an older part of solarthing, so you probably haven't heard of it). I'll think of the best way to test this stuff. Really I just need a way to see the data of all the supported fields and have all the unsupported data show up as null, which shouldn't be too difficult.

retrodaredevil, 2022-06-03:
Alright I added a `--debug` option to the `solarthing check` command. If you want to compile SolarThing after running `git pull`, you can run `solarthing check` with the `--debug` flag. I'll add some more detailed instructions later tonight or tomorrow, but if you feel like figuring it out yourself, go for it.

Once it's working it should spit out a bunch of JSON with information on fields that are unsupported. Or, it might not work. I'll be honest I haven't tested it yet. I'll be testing it soon on my own system, though.

retrodaredevil, 2022-06-04:
My Pi didn't feel like compiling SolarThing for some reason, so I ended up creating a new release.

You can [update SolarThing](https://solarthing.readthedocs.io/en/latest/updating.html), then run something like `solarthing check --port /dev/ttyUSB0 --type rover --debug`. Just make sure that when you run `solarthing check` that you don't have SolarThing running.

Pcpro745, 2022-06-06:
Hey sorry i was out of cell service since Friday. We were trying to build the driveway for our Off Grid property. Ill check this out today!

retrodaredevil, 2022-06-14:
Let me know when you're able to test this or if it ends up not working.

retrodaredevil, 2022-06-23:
Will reopen if you respond again. I hope to hear from you!

closed 2022-06-23 by retrodaredevil