wildmountainfarms/solarthing #59: Error starting solarthing-graphql

gecko951, 2022-07-11:
After updating to solarthing-2022.3.1.jar I am seeing the following error when trying to startup solarthing-graphql.  Are there any other updates that my system needs?

java -version
openjdk version "11.0.15" 2022-04-19
OpenJDK Runtime Environment (build 11.0.15+10-Ubuntu-0ubuntu0.20.04.1)
OpenJDK 64-Bit Server VM (build 11.0.15+10-Ubuntu-0ubuntu0.20.04.1, mixed mode, sharing)

`/opt/solarthing/program/graphql/run.sh

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)

2022-07-11 22:25:47.900  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : Starting SolarThingGraphQLApplication using Java 11.0.15 on nanoiot with PID 1973 (/opt/solarthing/program/.downloads/solarthing-graphql-2022.3.1.jar started by gecko951 in /opt/solarthing/program/graphql)
2022-07-11 22:25:47.908 DEBUG [           main] m.r.s.r.SolarThingGraphQLApplication     : Running with Spring Boot v2.6.6, Spring v5.3.18
2022-07-11 22:25:47.910  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : No active profile set, falling back to 1 default profile: "default"
2022-07-11 22:25:48.965  INFO [           main] o.s.b.w.e.t.TomcatWebServer              : Tomcat initialized with port(s): 8080 (http)
2022-07-11 22:25:48.987  INFO [           main] o.a.c.h.Http11NioProtocol                : Initializing ProtocolHandler ["http-nio-8080"]
2022-07-11 22:25:48.988  INFO [           main] o.a.c.c.StandardService                  : Starting service [Tomcat]
2022-07-11 22:25:48.989  INFO [           main] o.a.c.c.StandardEngine                   : Starting Servlet engine: [Apache Tomcat/9.0.60]
2022-07-11 22:25:49.081  INFO [           main] o.a.c.c.C.[.[.[/]                        : Initializing Spring embedded WebApplicationContext
2022-07-11 22:25:49.081  INFO [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 1121 ms
2022-07-11 22:25:49.169 DEBUG [           main] m.r.s.r.CommonProvider                   : Using defaultInstanceOptions=DefaultInstanceOptions(defaultSourceId='<UNUSED SOURCE ID THAT WILL NEVER BE IN A PACKET>', defaultFragmentId=0)
2022-07-11 22:25:49.169 DEBUG [           main] m.r.s.r.CommonProvider                   : Database file: /opt/solarthing/program/rover/config/couchdb/couchdb.json
2022-07-11 22:25:49.170 DEBUG [           main] m.r.s.r.CommonProvider                   : Working directory: /opt/solarthing/program/graphql/.
2022-07-11 22:25:49.337  WARN [           main] ConfigServletWebServerApplicationContext : Exception encountered during context initialization - cancelling refresh attempt: org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'commandHandlerProvider' defined in URL [jar:file:/opt/solarthing/program/.downloads/solarthing-graphql-2022.3.1.jar!/BOOT-INF/classes!/me/retrodaredevil/solarthing/rest/command/CommandHandlerProvider.class]: Unsatisfied dependency expressed through constructor parameter 0; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'commonProvider': Invocation of init method failed; nested exception is java.lang.RuntimeException: Couldn't parse data!
2022-07-11 22:25:49.339  INFO [           main] o.a.c.c.StandardService                  : Stopping service [Tomcat]
2022-07-11 22:25:49.362  INFO [           main] ConditionEvaluationReportLoggingListener :

Error starting ApplicationContext. To display the conditions report re-run your application with 'debug' enabled.
2022-07-11 22:25:49.379 ERROR [           main] o.s.b.SpringApplication                  : Application run failed

org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'commandHandlerProvider' defined in URL [jar:file:/opt/solarthing/program/.downloads/solarthing-graphql-2022.3.1.jar!/BOOT-INF/classes!/me/retrodaredevil/solarthing/rest/command/CommandHandlerProvider.class]: Unsatisfied dependency expressed through constructor parameter 0; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'commonProvider': Invocation of init method failed; nested exception is java.lang.RuntimeException: Couldn't parse data!
        at org.springframework.beans.factory.support.ConstructorResolver.createArgumentArray(ConstructorResolver.java:800) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.ConstructorResolver.autowireConstructor(ConstructorResolver.java:229) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.autowireConstructor(AbstractAutowireCapableBeanFactory.java:1372) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBeanInstance(AbstractAutowireCapableBeanFactory.java:1222) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:582) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:542) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:335) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:234) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:333) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:208) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:953) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:918) ~[spring-context-5.3.18.jar!/:5.3.18]
        at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:583) ~[spring-context-5.3.18.jar!/:5.3.18]
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:145) ~[spring-boot-2.6.6.jar!/:2.6.6]
        at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:740) ~[spring-boot-2.6.6.jar!/:2.6.6]
        at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:415) ~[spring-boot-2.6.6.jar!/:2.6.6]
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:303) ~[spring-boot-2.6.6.jar!/:2.6.6]
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:1312) ~[spring-boot-2.6.6.jar!/:2.6.6]
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:1301) ~[spring-boot-2.6.6.jar!/:2.6.6]
        at me.retrodaredevil.solarthing.rest.SolarThingGraphQLApplication.main(SolarThingGraphQLApplication.java:10) ~[classes!/:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.springframework.boot.loader.MainMethodRunner.run(MainMethodRunner.java:49) ~[solarthing-graphql-2022.3.1.jar:?]
        at org.springframework.boot.loader.Launcher.launch(Launcher.java:108) ~[solarthing-graphql-2022.3.1.jar:?]
        at org.springframework.boot.loader.Launcher.launch(Launcher.java:58) ~[solarthing-graphql-2022.3.1.jar:?]
        at org.springframework.boot.loader.JarLauncher.main(JarLauncher.java:88) ~[solarthing-graphql-2022.3.1.jar:?]
[... cut ...]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.resolveDependency(DefaultListableBeanFactory.java:1309) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.ConstructorResolver.resolveAutowiredArgument(ConstructorResolver.java:887) ~[spring-beans-5.3.18.jar!/:5.3.18]
        at org.springframework.beans.factory.support.ConstructorResolver.createArgumentArray(ConstructorResolver.java:791) ~[spring-beans-5.3.18.jar!/:5.3.18]
        ... 27 more`

retrodaredevil, 2022-07-11:
You have an outdated `couchdb.json` file. Can you post it (without username and pass ofc) so I can yell you which fields have been removed/renamed? 

That's probably my bad on not putting that in the release notes somewhere, alright `socket_timeout` was removed many updates ago I believe.

retrodaredevil, 2022-07-11:
https://github.com/wildmountainfarms/solarthing/releases/tag/v2021.4.1

There's the release that this changed in. You don't need to send me your file if changing `socket_timeout` to `call_timeout`

gecko951, 2022-07-12:
Excellent, that fixed my issue and solarthing services are running. :)

closed 2022-07-12 by gecko951