wildmountainfarms/solarthing #60: modbus ser2net

riker65, 2022-07-14:
Hi,

I am wondering if you have used ser2net to send modbus data to another server?

My RPI is kind of weak for all, so I would migrate it to a more powerfull server.

Thanks for hints

Thomas

retrodaredevil, 2022-07-16:
I personally don't have any experience with ser2net. I also don't recommend doing something like this because SolarThing can run just fine on Raspberry Pis, but I do recommend that the database be put on another computer.

If ser2net exposes a serial port similar to what `socat` does, then I imagine you could treat whatever ser2net gives you as a serial port. If it does not do that, then I likely won't add support for ser2net integration, but if you still want to continue along this path, you could see this issue for reference on how this might be done: https://github.com/wildmountainfarms/solarthing/issues/51

riker65, 2022-07-16:
ok thanks will check

riker65, 2022-07-19:
closed socat should be used

closed 2022-07-19 by riker65