wildmountainfarms/solarthing #61: Question: local JSON file

githubtobi, 2022-07-19:
Hi there,
thank you for developing this awesome piece of software!
I just wanted to know how to export the data to a local JSON file (without a database). It is mentioned in the readme file, but not in the docs. :-)

retrodaredevil, 2022-07-20:
Yeah. I think the local JSON file used to exist a long time ago (I'll update the readme). The post feature does exist but is not documented. I likely won't add the JSON file feature back because it's just not useful except for debugging.

You can use the post feature by using this in place of a database:

```
{
  "type": "post",
  "url": "http://localhost:8080/"
}
```

That JSON data goes in its own file like a `couchdb.json` file or a `influxdb.json` file. You can name it whatever you'd like. Maybe `post.json`.

The API that consumes the data should expect one POST call per packet. I'll hopefully add some documentation on that at some point, but it's one of the lesser used features.

githubtobi, 2022-07-20:
thanks for the quick anwer, will try the post method, otherwise I'll got with mqtt :-).

closed 2022-07-20 by githubtobi