wildmountainfarms/solarthing #64: graphql json example?

cguay, 2022-08-05:
Hi there ,
I currently have solarthing working with couchdb and reporting up to pvoutput.
I would like to also use graphql , which I have installed but cantr seem to configure . All that gets reported is the battery voltage...

[image]

retrodaredevil, 2022-08-05:
Nice job getting it working this far. The verbiage used to describe what SolarThing GraphQL (soon to be renamed to SolarThing Server) can be a bit confusing. 

You have the graphql/server program up and running right now. By navigating to that URL, you have brought up the web interface for SolarThing. This web interface is very basic at the moment because it is mostly just a landing page to show if you have set up everything correctly, which you have. 

Since you have it running, the graphql endpoint is there, it's just under the /graphql path, so you can navigate to http://localhost:8080/graphql. Now, navigating to that endpoint in your web browser isn't useful because I haven't yet put a frontend page over the graphql API, but most people will attempt to call that endpoint from Grafana. There is limited documentation out there on how to construct a graphql query to use with the GraphQL data source, but I recommend looking into that if that is your end goal. 

Of you just want to tinker with GraphQL queries, I recommend downloading GraphQL Playground. 

Lemme know if I explained everything well enough and what your end goal is.

cguay, 2022-08-05:
This is what I get if i connect to http://x.x.x.x:8080/graphql - 
[image]

retrodaredevil, 2022-08-05:
Graphql will work if you use something besides your browser to make a request to it. Like I said I don't have a UI around graphql yet, so you need to use something else to make requests to it. 

Most people follow the tutorial for Grafana here: https://solarthing.readthedocs.io/en/latest/data/graphql-grafana.html#install-graphql-datasource-to-grafana

cguay, 2022-08-05:
Ok thank you!

closed 2022-08-05 by cguay