wildmountainfarms/solarthing #67: data used documentation

riker65, 2022-08-25:
Hi Joshua,

thanks again for your great software.

Having a understanding issue:
what is the base of those 2 fields and the meaning?

Please confirm my understanding:

pvCurrent  -> solar charging current
ChargingCurrent?- > derived from charging state   06H:current-limited   charging stage  (solar/alternator)

Also I have the values true for bulk charging and float charging at the same time.  How did you use those states, which from my understanding can only have 1 state? 

Do you have a mapping or guide me to the docu which I probably missed

Thanks a lot.

BR Thomas

retrodaredevil, 2022-08-26:
Please give me some details on which charge controller you are using.

PV current and charging current are two different things. Before I talk about those I want to just talk about PV power and charging power. PV power would be the power measured coming from the panels, and charging power would be the amount of power going into the batteries. Normally these power values are very close and sometimes equivalent. In general charging power will be a little less than PV power because of inefficiencies.

PV current * PV voltage = PV Power and Charging current * battery voltage = charging power. Charging current isn't exactly derived from the charging state, but the charging state may make the controller limit the charging power, which in turn would also limit the PV current being pulled.

Can you show me what query you are using that shows true for bulk and float at the same time? There's not really any good mappings or guides that I have. I've been meaning to improve the GraphQL documentation, but haven't gotten around to it. The best advice I can give you is to understand what each property means from one of the modbus documents that are out there for your given controller. SolarThing pretty much does a 1 to 1 mapping of the information on the modbus documents, so SolarThing usually isn't doing anything special with the data.

If you are referring to a `Bulk/Float` mode, that mode means that the controller is either in bulk or float mode, but I'm only aware of this being the case on Tracer controllers.

riker65, 2022-08-26:
Hi Yoshua,

thanks for the details.

I am using Rover Elite 40 A

riker65, 2022-08-26:
Hi

looking tinto the modbus docu from Reneogy I maybe was confused.

I am talking about 120H information: 0120H Charging state 
Which looks like is a part of historic data information.

similar info is found in : battery state also 120H

document is a little confusing to me

retrodaredevil, 2022-08-26:
Yeah Renogy's documentation is very confusing, which is why I made everything 1 to 1 so you can interpret the data later once the data is collected. The charging state is not part of the historic data information section. The charging state tells you the single state the charge controller is in. I believe that all Rover models should correctly report what state they are in. "Current limiting/overpower" is basically the same as Boost, but this state is only used for lithium batteries.

You can see a code snippet here of how I have these states defined. I have some comments in the code that are not documented anywhere else:

https://github.com/wildmountainfarms/solarthing/blob/8787d7ac0b7c58958da4de52f5580eecc73ea012/core/src/main/java/me/retrodaredevil/solarthing/solar/renogy/rover/ChargingState.java#L14-L27

Personally I have never seen the `ACTIVATED`, `CURRENT_LIMITING`, `DIRECT_CHARGE` states used on a Renogy Rover 40A. I have seen all of the other states.

retrodaredevil, 2022-08-26:
> Also I have the values true for bulk charging and float charging at the same time. How did you use those states, which from my understanding can only have 1 state?

I still don't understand what you mean here. Feel free to elaborate further

riker65, 2022-08-27:
> 

Hi Joshua,

looks like this is a misuse and understanding from my side. So looks like I made an error using the data and did not clear old values. so I this this is solved - user error.

Thanks Thomas

riker65, 2022-08-27:
Thanks for your explanations and snippit

closed 2022-08-27 by riker65