wildmountainfarms/solarthing #71: Running for the first time error

Joe13183, 2022-09-26:
Hi, i have followed the instructions very closely but i must be doing something wrong. This is the 2nd install ive done but both fail at this stage. here is the result when i enter the command
pi@raspberrypi:/opt/solarthing/program/custom_rover $ sudo -u solarthing ./run.sh
Using base config = config/base.json
2022-09-26 22:32:22,384 main ERROR Unable to create file logs/log_debug.log java.io.IOException: Could not create directory /opt/solarthing/program/custom_rover/logs
        at org.apache.logging.log4j.core.util.FileUtils.mkdir(FileUtils.java:120)
        at org.apache.logging.log4j.core.util.FileUtils.makeParentDirs(FileUtils.java:137)
        at org.apache.logging.log4j.core.appender.rolling.RollingFileManager$RollingFileManagerFactory.createManager(RollingFileManager.java:732)
        at org.apache.logging.log4j.core.appender.rolling.RollingFileManager$RollingFileManagerFactory.createManager(RollingFileManager.java:716)
        at org.apache.logging.log4j.core.appender.AbstractManager.getManager(AbstractManager.java:144)
        at org.apache.logging.log4j.core.appender.OutputStreamManager.getManager(OutputStreamManager.java:100)
        at org.apache.logging.log4j.core.appender.rolling.RollingFileManager.getFileManager(RollingFileManager.java:217)
        at org.apache.logging.log4j.core.appender.RollingFileAppender$Builder.build(RollingFileAppender.java:146)
        at org.apache.logging.log4j.core.appender.RollingFileAppender$Builder.build(RollingFileAppender.java:62)
        at org.apache.logging.log4j.core.config.plugins.util.PluginBuilder.build(PluginBuilder.java:124)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createPluginObject(AbstractConfiguration.java:1122)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createConfiguration(AbstractConfiguration.java:1047)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createConfiguration(AbstractConfiguration.java:1039)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.doConfigure(AbstractConfiguration.java:651)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.initialize(AbstractConfiguration.java:249)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.start(AbstractConfiguration.java:295)
        at org.apache.logging.log4j.core.LoggerContext.setConfiguration(LoggerContext.java:621)
        at org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:694)
        at org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:711)
        at org.apache.logging.log4j.core.LoggerContext.start(LoggerContext.java:253)
        at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:155)
        at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:47)
        at org.apache.logging.log4j.LogManager.getContext(LogManager.java:196)
        at org.apache.logging.log4j.spi.AbstractLoggerAdapter.getContext(AbstractLoggerAdapter.java:137)
        at org.apache.logging.slf4j.Log4jLoggerFactory.getContext(Log4jLoggerFactory.java:55)
        at org.apache.logging.log4j.spi.AbstractLoggerAdapter.getLogger(AbstractLoggerAdapter.java:47)
        at org.apache.logging.slf4j.Log4jLoggerFactory.getLogger(Log4jLoggerFactory.java:33)
        at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:363)
        at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:388)
        at me.retrodaredevil.solarthing.program.SolarMain.<clinit>(SolarMain.java:50)

2022-09-26 22:32:22,433 main ERROR Could not create plugin of type class org.apache.logging.log4j.core.appender.RollingFileAppender for element RollingFile: java.lang.IllegalStateException: ManagerFactory [org.apache.logging.log4j.core.appender.rolling.RollingFileManager$RollingFileManagerFactory@433defed] unable to create manager for [logs/log_debug.log] with data [org.apache.logging.log4j.core.appender.rolling.RollingFileManager$FactoryData@2a693f59[pattern=logs/log_debug_%d{yyyy.MM.dd}-%i.log.gz, append=true, bufferedIO=true, bufferSize=16384, policy=CompositeTriggeringPolicy(policies=[OnStartupTriggeringPolicy, TimeBasedTriggeringPolicy(nextRolloverMillis=0, interval=1, modulate=true), SizeBasedTriggeringPolicy(size=3221225472)]), strategy=DefaultRolloverStrategy(min=1, max=1, useMax=true), advertiseURI=null, layout=%d{yyyy-MM-dd HH:mm:ss.SSS} [%t] %-5level %logger{36} - %msg%n, filePermissions=rw-rw-r--, fileOwner=null]] java.lang.IllegalStateException: ManagerFactory [org.apache.logging.log4j.core.appender.rolling.RollingFileManager$RollingFileManagerFactory@433defed] unable to create manager for [logs/log_debug.log] with data [org.apache.logging.log4j.core.appender.rolling.RollingFileManager$FactoryData@2a693f59[pattern=logs/log_debug_%d{yyyy.MM.dd}-%i.log.gz, append=true, bufferedIO=true, bufferSize=16384, policy=CompositeTriggeringPolicy(policies=[OnStartupTriggeringPolicy, TimeBasedTriggeringPolicy(nextRolloverMillis=0, interval=1, modulate=true), SizeBasedTriggeringPolicy(size=3221225472)]), strategy=DefaultRolloverStrategy(min=1, max=1, useMax=true), advertiseURI=null, layout=%d{yyyy-MM-dd HH:mm:ss.SSS} [%t] %-5level %logger{36} - %msg%n, filePermissions=rw-rw-r--, fileOwner=null]]
        at org.apache.logging.log4j.core.appender.AbstractManager.getManager(AbstractManager.java:146)
        at org.apache.logging.log4j.core.appender.OutputStreamManager.getManager(OutputStreamManager.java:100)
        at org.apache.logging.log4j.core.appender.rolling.RollingFileManager.getFileManager(RollingFileManager.java:217)
        at org.apache.logging.log4j.core.appender.RollingFileAppender$Builder.build(RollingFileAppender.java:146)
        at org.apache.logging.log4j.core.appender.RollingFileAppender$Builder.build(RollingFileAppender.java:62)
        at org.apache.logging.log4j.core.config.plugins.util.PluginBuilder.build(PluginBuilder.java:124)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createPluginObject(AbstractConfiguration.java:1122)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createConfiguration(AbstractConfiguration.java:1047)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.createConfiguration(AbstractConfiguration.java:1039)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.doConfigure(AbstractConfiguration.java:651)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.initialize(AbstractConfiguration.java:249)
        at org.apache.logging.log4j.core.config.AbstractConfiguration.start(AbstractConfiguration.java:295)
        at org.apache.logging.log4j.core.LoggerContext.setConfiguration(LoggerContext.java:621)
        at org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:694)
        at org.apache.logging.log4j.core.LoggerContext.reconfigure(LoggerContext.java:711)
        at org.apache.logging.log4j.core.LoggerContext.start(LoggerContext.java:253)
        at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:155)
        at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:47)
        at org.apache.logging.log4j.LogManager.getContext(LogManager.java:196)
        at org.apache.logging.log4j.spi.AbstractLoggerAdapter.getContext(AbstractLoggerAdapter.java:137)
        at org.apache.logging.slf4j.Log4jLoggerFactory.getContext(Log4jLoggerFactory.java:55)
        at org.apache.logging.log4j.spi.AbstractLoggerAdapter.getLogger(AbstractLoggerAdapter.java:47)
        at org.apache.logging.slf4j.Log4jLoggerFactory.getLogger(Log4jLoggerFactory.java:33)
        at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:363)
        at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:388)
        at me.retrodaredevil.solarthing.program.SolarMain.<clinit>(SolarMain.java:50)

2022-09-26 22:32:22,463 main ERROR Unable to invoke factory method in class org.apache.logging.log4j.core.appender.RollingFileAppender for element RollingFile: java.lang.IllegalStateException: No factory method found for class org.apache.logging.log4j.core.appender.RollingFileAppender java.lang.IllegalStateException: No factory method found for class org.apache.logging.log4j.core.appender.RollingFileAppender
        at org.apache.logging.log4j.core.config.plugins.util.PluginBuilder.findFactoryMethod(PluginBuilder.java:238)
[... cut ...]
[stderr] Beginning main. Jar: Jar: solarthing-2022.3.1.jar Last Modified: 2022-06-23T04:33:38Z Java version: 11.0.16
2022-09-26 22:32:24.777 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2022-09-26 22:32:28.692 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Unable to create data directory! dataDirectory=.data absolute=/opt/solarthing/program/custom_rover/.data

my serial address is 16 and i receive voltage data from the check command

please help!!

retrodaredevil, 2022-09-26:
All of those errors are permission errors which are telling you that however you are running solarthing does not have enough permissions. I notice that you are using `sudo -u solarthing ./run.sh`, which is correct.

Assuming you installed solarthing without manually cloning it yourself, the installation process should have set the right permissions on the `/opt/solarthing` directory. Whether you installed it the official way or not, there's a pretty easy fix for this. Go ahead and run this:

```
sudo /opt/solarthing/other/linux/update_perms.sh
```

The above command is also referenced in the documentation for updating solarthing: https://solarthing.readthedocs.io/en/latest/updating.html#permission-issues

If you still have the same errors after running that command, please tell me the output of `ls -la /opt/solarthing`. Since you receive voltage data from the check command, I don't imagine you will have any more errors unless you did not set up your database configuration correctly. Feel free to reply back with any additional errors you get.

Joe13183, 2022-09-26:
Perfect. thankyou.
Ive tried probably every renogy monitoring program but some i couldnt get to work and some display info but not the same as what the controller says on the LCD. Been very frustrating. But i feel this is the one. Especially as you responded to my issue so quickly. Fingers crossed theres a fund u page in some way.

retrodaredevil, 2022-09-27:
> some i couldnt get to work and some display info but not the same as what the controller says on the LCD

Many of the other monitoring programs don't implement all of the data values of the Rover, but SolarThing should get every single value. I was very thorough. Let me know if there's any particular value that you need help figuring out which is which. Some of the names aren't as intuitive as they could be.

> Fingers crossed theres a fund u page in some way.

It's optional of course, but if you would like to donate, the only thing I have set up right now is GitHub sponsors: https://github.com/sponsors/retrodaredevil

Let me know if you need any more help or close this issue. I usually close issues are a little while of inactivity anyway.

Joe13183, 2022-09-28:
Hi again. Just done a quick test. Ran the program while looking at the controller itself. The voltages are always correct but the incoming (PV) amps are always off. Unless im reading it wrong. Always seems to be 3 times diffence. See code and pictures

2022-09-28 13:57:49.668 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 3.435 seconds
2022-09-28 13:57:54.640 [main] DEBUG me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - took 1.535 seconds to read from device
2022-09-28 13:57:54.642 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - Debugging all packets
2022-09-28 13:57:54.834 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - [ {
  "packetType" : "RENOGY_ROVER_STATUS",
  "packetVersion" : 7,
  "maxVoltage" : 24,
  "ratedChargingCurrent" : 20,
  "ratedDischargingCurrent" : 20,
  "productType" : 0,
  "productModelEncoded" : "ICBSTkctQ1RSTC1SVlIyMA==",
  "softwareVersion" : 196609,
  "hardwareVersion" : 2,
  "productSerialNumber" : 302450395,
  "controllerDeviceAddress" : 16,
  "batteryCapacitySOC" : 100,
  "batteryVoltage" : 13.9,
  "chargingCurrent" : 0.71,
  "controllerTemperatureRaw" : 19,
  "batteryTemperatureRaw" : 15,
  "loadVoltage" : 13.9,
  "loadCurrent" : 0.24,
  "loadPower" : 3,
  "inputVoltage" : 42.1,
  "pvCurrent" : 0.23,
  "chargingPower" : 10,
  "dailyMinBatteryVoltage" : 13.1,
  "dailyMaxBatteryVoltage" : 14.8,
  "dailyMaxChargingCurrent" : 3.57,
  "dailyMaxDischargingCurrent" : 0.53,
  "dailyMaxChargingPower" : 32,
  "dailyMaxDischargingPower" : 6,
  "dailyAH" : 4,
  "dailyAHDischarging" : 1,
  "dailyKWH" : 0.056,
  "dailyKWHConsumption" : 0.014,
  "operatingDaysCount" : 550,
  "batteryOverDischargesCount" : 0,
  "batteryFullChargesCount" : 623,
  "chargingAmpHoursOfBatteryCount" : 3829,
  "dischargingAmpHoursOfBatteryCount" : 2923,
  "cumulativeKWH" : 51.373,
  "cumulativeKWHConsumption" : 36.385,
  "streetLightValue" : 128,
  "chargingState" : 5,
  "errorMode" : 0,
  "nominalBatteryCapacity" : 200,
  "systemVoltageSetting" : 12,
  "recognizedVoltage" : 12,
  "batteryType" : 2,
  "overVoltageThresholdRaw" : 160,
  "chargingVoltageLimitRaw" : 155,
  "equalizingChargingVoltageRaw" : 146,
  "boostChargingVoltageRaw" : 144,
  "floatingChargingVoltageRaw" : 138,
  "boostChargingRecoveryVoltageRaw" : 132,
  "overDischargeRecoveryVoltageRaw" : 126,
  "underVoltageWarningLevelRaw" : 120,
  "overDischargeVoltageRaw" : 111,
  "dischargingLimitVoltageRaw" : 106,
  "endOfChargeSOC" : 100,
  "endOfDischargeSOC" : 50,
  "overDischargeTimeDelaySeconds" : 5,
  "equalizingChargingTimeRaw" : 120,
  "boostChargingTimeRaw" : 120,
  "equalizingChargingIntervalRaw" : 30,
  "temperatureCompensationFactorRaw" : 3,
  "operatingStage1" : {
    "durationHours" : 0,
    "operatingPowerPercentage" : 0
  },
  "operatingStage2" : {
    "durationHours" : 0,
    "operatingPowerPercentage" : 0
  },
  "operatingStage3" : {
    "durationHours" : 0,
    "operatingPowerPercentage" : 0
  },
  "operatingMorningOn" : {
    "durationHours" : 0,
    "operatingPowerPercentage" : 0
  },
  "loadWorkingMode" : 17,
  "lightControlDelayMinutes" : 10,
  "lightControlVoltage" : 5,
  "ledLoadCurrentSettingRaw" : 660,
  "specialPowerControlE021Raw" : 5,
  "sensed1" : {
    "workingHoursRaw" : 0,
    "powerWithPeopleSensedRaw" : 0,
    "powerWithNoPeopleSensedRaw" : 0
  },
  "sensed2" : {
    "workingHoursRaw" : 0,
    "powerWithPeopleSensedRaw" : 0,
    "powerWithNoPeopleSensedRaw" : 0
  },
  "sensed3" : {
    "workingHoursRaw" : 0,
    "powerWithPeopleSensedRaw" : 0,
    "powerWithNoPeopleSensedRaw" : 0
  },
  "sensingTimeDelayRaw" : 0,
  "ledLoadCurrentRaw" : 0,
  "specialPowerControlE02DRaw" : 0,
  "productModelString" : "RNG-CTRL-RVR20"
}, {
  "packetType" : "SOURCE",
  "sourceId" : "default"
}, {
  "packetType" : "FRAGMENT_INDICATOR",
  "fragmentId" : 2
} ]
2022-09-28 13:57:54.853 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 3.253 seconds
[image]
[image]
[image]

retrodaredevil, 2022-09-28:
The values you see are chargingCurrent, loadCurrent, and inputVoltage. 

Now, pvCurrent also exists, so you might be confused as to what the different is between that and chargingCurrent. Well, chargingCurrent goes alongside the battery voltage and pvCurrent goes alongside inputVoltage so that these are roughly equivalent:

``` 
Charging power ~= solar power 
chargingCurrent * batteryVoltage ~= pvCurrent * inputVoltage
```

Joe13183, 2022-09-28:
Oh I see.  I always thought that was solar current. I can see the corresponding read on the output now thankyou

closed 2022-09-28 by Joe13183