wildmountainfarms/solarthing #72: PV output issues

Joe13183, 2022-09-28:
Hi again
Ive installed and setup couchdb, (I think it is working) but now ive tried to output to PVoutput but on my login no new data is showing. In base.json in PVoutput/config folder i have amended the system id and added the API key produced from the settings page in pvoutput.org. Not sure if im missing something again :-(

pi@raspberrypi:/opt/solarthing/program/pvoutput $ sudo -u solarthing ./run.sh
Using base config = config/base.json
2022-09-28 19:58:45.807 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.3.1.jar Last Modified: 2022-06-23T04:33:38Z Java version: 11.0.16
[stdout] Beginning main. Jar: Jar: solarthing-2022.3.1.jar Last Modified: 2022-06-23T04:33:38Z Java version: 11.0.16
[stderr] Beginning main. Jar: Jar: solarthing-2022.3.1.jar Last Modified: 2022-06-23T04:33:38Z Java version: 11.0.16
2022-09-28 19:58:46.639 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2022-09-28 19:58:50.405 [main] INFO  me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Starting PV Output upload program
2022-09-28 19:58:50.560 [main] INFO  me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Using time zone: British Time
2022-09-28 19:58:50.565 [main] INFO  me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Using default instance options: DefaultInstanceOptions(defaultSourceId='default', defaultFragmentId=0)
2022-09-28 19:58:54.203 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2022-09-28 19:59:01.033 [main] DEBUG me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics UUID: 8b24005a-0b94-44e2-b079-021d03dab765
2022-09-28 19:59:01.942 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2022-09-28 19:59:01.973 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Going to do stuff now.
2022-09-28 19:59:02.571 [main] DEBUG me.retrodaredevil.couchdbjava.okhttp.auth.CookieAuthHandler - Got new authentication cookie that expires at 1664392142568. Persistent: true
2022-09-28 19:59:04.387 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Got packets
2022-09-28 19:59:04.428 [main] WARN  me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - The last packet is more than 5 minutes in the past! now=1664391541975 packet date=1664390029910
2022-09-28 19:59:08.003 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Packets: [{"packetType":"RENOGY_ROVER_STATUS","packetVersion":7,"maxVoltage":24,"ratedChargingCurrent":20,"ratedDischargingCurrent":20,"productType":0,"productModelEncoded":"ICBSTkctQ1RSTC1SVlIyMA==","softwareVersion":196609,"hardwareVersion":2,"productSerialNumber":302450395,"controllerDeviceAddress":16,"batteryCapacitySOC":83,"batteryVoltage":12.8,"chargingCurrent":0.0,"controllerTemperatureRaw":17,"batteryTemperatureRaw":16,"loadVoltage":12.8,"loadCurrent":0.46,"loadPower":5,"inputVoltage":0.0,"pvCurrent":0.0,"chargingPower":0,"dailyMinBatteryVoltage":12.6,"dailyMaxBatteryVoltage":14.8,"dailyMaxChargingCurrent":3.57,"dailyMaxDischargingCurrent":1.22,"dailyMaxChargingPower":32,"dailyMaxDischargingPower":10,"dailyAH":7,"dailyAHDischarging":1,"dailyKWH":0.097,"dailyKWHConsumption":0.013,"operatingDaysCount":551,"batteryOverDischargesCount":0,"batteryFullChargesCount":624,"chargingAmpHoursOfBatteryCount":3832,"dischargingAmpHoursOfBatteryCount":2925,"cumulativeKWH":51.414,"cumulativeKWHConsumption":36.412,"streetLightValue":128,"chargingState":0,"errorMode":0,"nominalBatteryCapacity":200,"systemVoltageSetting":12,"recognizedVoltage":12,"batteryType":2,"overVoltageThresholdRaw":160,"chargingVoltageLimitRaw":155,"equalizingChargingVoltageRaw":146,"boostChargingVoltageRaw":144,"floatingChargingVoltageRaw":138,"boostChargingRecoveryVoltageRaw":132,"overDischargeRecoveryVoltageRaw":126,"underVoltageWarningLevelRaw":120,"overDischargeVoltageRaw":111,"dischargingLimitVoltageRaw":106,"endOfChargeSOC":100,"endOfDischargeSOC":50,"overDischargeTimeDelaySeconds":5,"equalizingChargingTimeRaw":120,"boostChargingTimeRaw":120,"equalizingChargingIntervalRaw":30,"temperatureCompensationFactorRaw":3,"operatingStage1":{"durationHours":0,"operatingPowerPercentage":0},"operatingStage2":{"durationHours":0,"operatingPowerPercentage":0},"operatingStage3":{"durationHours":0,"operatingPowerPercentage":0},"operatingMorningOn":{"durationHours":0,"operatingPowerPercentage":0},"loadWorkingMode":17,"lightControlDelayMinutes":10,"lightControlVoltage":5,"ledLoadCurrentSettingRaw":660,"specialPowerControlE021Raw":5,"sensed1":{"workingHoursRaw":0,"powerWithPeopleSensedRaw":0,"powerWithNoPeopleSensedRaw":0},"sensed2":{"workingHoursRaw":0,"powerWithPeopleSensedRaw":0,"powerWithNoPeopleSensedRaw":0},"sensed3":{"workingHoursRaw":0,"powerWithPeopleSensedRaw":0,"powerWithNoPeopleSensedRaw":0},"sensingTimeDelayRaw":0,"ledLoadCurrentRaw":0,"specialPowerControlE02DRaw":0,"productModelString":"RNG-CTRL-RVR20"}]
2022-09-28 19:59:08.012 [main] DEBUG me.retrodaredevil.solarthing.program.pvoutput.PVOutputUploadMain - Going to sleep now
^C
pi@raspberrypi:/opt/solarthing/program/pvoutput $ cd
pi@raspberrypi:~ $ sudo /opt/solarthing/other/systemd/install.sh pvoutput
sudo systemctl start solarthing-pvoutput
sudo systemctl enable solarthing-pvoutput
SolarThing root is /opt/solarthing
Added service
Reloaded systemctl
Created symlink /etc/systemd/system/multi-user.target.wants/solarthing-pvoutput.service → /etc/systemd/system/solarthing-pvoutput.service.
pi@raspberrypi:~ $ systemctl status solarthing-pvoutput
● solarthing-pvoutput.service - The service for SolarThing pvoutput
     Loaded: loaded (/etc/systemd/system/solarthing-pvoutput.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-09-28 20:03:48 BST; 27s ago
   Main PID: 7533 (sh)
      Tasks: 18 (limit: 779)
        CPU: 42.265s
     CGroup: /system.slice/solarthing-pvoutput.service
             ├─7533 sh /opt/solarthing/program/pvoutput/run.sh
             ├─7536 sh ../.scripts/solarthing.sh
             ├─7538 bash ../.scripts/../.bin/local_log_solarthing run --base config/base.json
             ├─7545 sh ../.scripts/../.bin/../.bin/raw_solarthing run --base config/base.json
             └─7549 java --add-opens=java.base/java.lang.invoke=ALL-UNNAMED -jar ../.scripts/../.bin/../.bin/../solarthing.jar run --base config/base.json

Sep 28 20:03:48 raspberrypi systemd[1]: Started The service for SolarThing pvoutput.
pi@raspberrypi:~ $

retrodaredevil, 2022-09-29:
The output `The last packet is more than 5 minutes in the past! now=1664391541975 packet date=1664390029910` tells me that your program to upload rover data has not been running for 25 minutes. Can you make sure to enable and start that service?

retrodaredevil, 2022-10-09:
Let me know if you are still having issues or got it to work. I'll close this issue soon as inactive, but feel free to reopen and comment if I do.

closed 2022-10-17 by retrodaredevil