wildmountainfarms/solarthing #81: SolarThing almost running

rbroen3rd, 2023-01-08:
SolarThing Output

rbrown3rd@bobspi01:/opt/solarthing/program/mate$ solarthing check --port /dev/ttyUSB0 --type mate
Going to open serial port using MATE's default serial configuration...
Successfully opened serial port...
Got 4 packets
Got 4 packets
Got 4 packets
Got 4 packets
Got data from mate!

Running SolarThing outputs the following:

rbrown3rd@bobspi01:/opt/solarthing/program/mate$ sudo -u solarthing ./run.sh
Using base config = config/base.json
2023-01-08 17:41:22.415 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36:45Z Java version: 11.0.16
[stdout] Beginning main. Jar: Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36:45Z Java version: 11.0.16
[stderr] Beginning main. Jar: Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36:45Z Java version: 11.0.16
2023-01-08 17:41:22.720 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2023-01-08 17:41:23.632 [main] INFO  me.retrodaredevil.solarthing.program.OutbackMateMain - Beginning mate program
2023-01-08 17:41:23.646 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2023-01-08 17:41:23.669 [main] DEBUG me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics UUID: 625b28be-b975-42df-a9e2-0503f8cb7e71
2023-01-08 17:41:24.300 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2023-01-08 17:41:24.313 [main] DEBUG me.retrodaredevil.solarthing.program.OutbackMateMain - IO Bundle File: config/mate_serial.json
2023-01-08 17:41:24.609 [main] ERROR me.retrodaredevil.solarthing.io.ReloadableIOBundle - Could not create IOBundle
me.retrodaredevil.io.serial.SerialPortException: Was unsuccessful while trying to open port: ttyUS00 descriptive name: User-Specified Port description: User-Specified Port
	at me.retrodaredevil.io.serial.JSerialIOBundle.<init>(JSerialIOBundle.java:22) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.config.io.SerialIOConfig.createIOBundle(SerialIOConfig.java:46) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.io.ReloadableIOBundle.reload(ReloadableIOBundle.java:56) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.io.ReloadableIOBundle.<init>(ReloadableIOBundle.java:31) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.OutbackMateMain.connectMate(OutbackMateMain.java:55) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:145) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:202) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:262) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:276) ~[solarthing-2022.4.1.jar:?]
2023-01-08 17:41:24.681 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Commands are disabled
2023-01-08 17:41:25.105 [googleanalyticsjava-thread-1] WARN  com.brsanthu.googleanalytics.httpclient.ApacheHttpClientImpl - Exception while sending the Google Analytics tracker request com.brsanthu.googleanalytics.httpclient.HttpRequest@13e2596
javax.net.ssl.SSLException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
	at sun.security.ssl.Alert.createSSLException(Alert.java:133) ~[?:?]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:353) ~[?:?]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:296) ~[?:?]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:291) ~[?:?]
	at sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1689) ~[?:?]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:471) ~[?:?]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:427) ~[?:?]
	at org.apache.http.conn.ssl.SSLConnectionSocketFactory.createLayeredSocket(SSLConnectionSocketFactory.java:396) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.conn.ssl.SSLConnectionSocketFactory.connectSocket(SSLConnectionSocketFactory.java:355) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.conn.DefaultHttpClientConnectionOperator.connect(DefaultHttpClientConnectionOperator.java:142) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.conn.PoolingHttpClientConnectionManager.connect(PoolingHttpClientConnectionManager.java:359) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.execchain.MainClientExec.establishRoute(MainClientExec.java:381) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.execchain.MainClientExec.execute(MainClientExec.java:237) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.execchain.ProtocolExec.execute(ProtocolExec.java:185) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.execchain.RetryExec.execute(RetryExec.java:89) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.execchain.RedirectExec.execute(RedirectExec.java:111) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.client.InternalHttpClient.doExecute(InternalHttpClient.java:185) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:83) ~[solarthing-2022.4.1.jar:?]
	at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:108) ~[solarthing-2022.4.1.jar:?]
	at com.brsanthu.googleanalytics.httpclient.ApacheHttpClientImpl.execute(ApacheHttpClientImpl.java:86) ~[solarthing-2022.4.1.jar:?]
	at com.brsanthu.googleanalytics.httpclient.ApacheHttpClientImpl.post(ApacheHttpClientImpl.java:102) ~[solarthing-2022.4.1.jar:?]
	at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.postSingle(GoogleAnalyticsImpl.java:165) ~[solarthing-2022.4.1.jar:?]
	at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.post(GoogleAnalyticsImpl.java:105) ~[solarthing-2022.4.1.jar:?]
	at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.lambda$postAsync$0(GoogleAnalyticsImpl.java:91) ~[solarthing-2022.4.1.jar:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:829) ~[?:?]
Caused by: java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
	at sun.security.validator.PKIXValidator.<init>(PKIXValidator.java:102) ~[?:?]
	at sun.security.validator.Validator.getInstance(Validator.java:181) ~[?:?]
	at sun.security.ssl.X509TrustManagerImpl.getValidator(X509TrustManagerImpl.java:300) ~[?:?]
[... 119 lines cut ...]
java.lang.NullPointerException: null
	at com.brsanthu.googleanalytics.httpclient.ApacheHttpClientImpl.post(ApacheHttpClientImpl.java:113) ~[solarthing-2022.4.1.jar:?]
	at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.postSingle(GoogleAnalyticsImpl.java:165) ~[solarthing-2022.4.1.jar:?]
	at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.post(GoogleAnalyticsImpl.java:105) ~[solarthing-2022.4.1.jar:?]
	at com.brsanthu.googleanalytics.internal.GoogleAnalyticsImpl.lambda$postAsync$0(GoogleAnalyticsImpl.java:91) ~[solarthing-2022.4.1.jar:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:829) ~[?:?]

Program seems to be running here, but not output to screen after the above.

Contents of base.json file is as follows:

{
  "type": "mate",
  "source": "default",
  "fragment": 1,
  "unique": 60,
  "databases": [ ],
  "io": "config/mate_serial.json"
}
```

```

rbroen3rd, 2023-01-08:
log_debug.log contains:     rbrown3rd@bobspi01:/opt/solarthing/program/mate/logs$ less log_debug.log

2023-01-08 18:02:04.169 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36:45Z Java version: 11.0.16
2023-01-08 18:02:04.474 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2023-01-08 18:02:05.357 [main] INFO  me.retrodaredevil.solarthing.program.OutbackMateMain - Beginning mate program
2023-01-08 18:02:05.363 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2023-01-08 18:02:05.398 [main] DEBUG me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics UUID: 625b28be-b975-42df-a9e2-0503f8cb7e71
2023-01-08 18:02:06.033 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2023-01-08 18:02:06.046 [main] DEBUG me.retrodaredevil.solarthing.program.OutbackMateMain - IO Bundle File: config/mate_serial.json
2023-01-08 18:02:06.341 [main] ERROR me.retrodaredevil.solarthing.io.ReloadableIOBundle - Could not create IOBundle
me.retrodaredevil.io.serial.SerialPortException: Was unsuccessful while trying to open port: ttyUS00 descriptive name: User-Specified Port description: User-Specified Port
        at me.retrodaredevil.io.serial.JSerialIOBundle.<init>(JSerialIOBundle.java:22) ~[solarthing-2022.4.1.jar:?]
        at me.retrodaredevil.solarthing.config.io.SerialIOConfig.createIOBundle(SerialIOConfig.java:46) ~[solarthing-2022.4.1.jar:?]
        at me.retrodaredevil.solarthing.io.ReloadableIOBundle.reload(ReloadableIOBundle.java:56) ~[solarthing-2022.4.1.jar:?]
        at me.retrodaredevil.solarthing.io.ReloadableIOBundle.<init>(ReloadableIOBundle.java:31) ~[solarthing-2022.4.1.jar:?]
        at me.retrodaredevil.solarthing.program.OutbackMateMain.connectMate(OutbackMateMain.java:55) ~[solarthing-2022.4.1.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:145) ~[solarthing-2022.4.1.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:202) ~[solarthing-2022.4.1.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:262) ~[solarthing-2022.4.1.jar:?]
        at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:276) ~[solarthing-2022.4.1.jar:?]
2023-01-08 18:02:06.412 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Commands are disabled

rbroen3rd, 2023-01-08:
My Google Sheet shows my data after manually capturing it from the serial to USB cable from the Mate to my iMac.   https://docs.google.com/spreadsheets/d/1QEX0m8pC8CoDV7GOINe-g8fUpPDu2Ujw0IvWptkW8x4/edit?usp=sharing

retrodaredevil, 2023-01-08:
The first line that pops out at me is `Was unsuccessful while trying to open port: ttyUS00`. In your `config/mate_serial.json` file, make sure you have the path to your device set correctly.

As for why there are so many errors going to stdout in your original post, I have never seen those errors before, but they seem to be related to Google Analytics. You can ignore them, but I will look into those errors some other time to see if I can make them go away.

rbroen3rd, 2023-01-08:
Found a typo in mate_serial.json. ttyUS00 instead of ttyUSB0.  Corrected that, and seems to run better.  Sucess!  I have data in log_debug.log as follows:

 more log_debug.log
2023-01-08 19:00:24.916 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36
:45Z Java version: 11.0.16
2023-01-08 19:00:25.220 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2023-01-08 19:00:26.120 [main] INFO  me.retrodaredevil.solarthing.program.OutbackMateMain - Beginning mate program
2023-01-08 19:00:26.136 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2023-01-08 19:00:26.159 [main] DEBUG me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics UUID: 625b28be-b975-42df-a9e2-0503f8cb7e71
2023-01-08 19:00:27.054 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2023-01-08 19:00:27.068 [main] DEBUG me.retrodaredevil.solarthing.program.OutbackMateMain - IO Bundle File: config/mate_serial.json
2023-01-08 19:00:27.621 [main] DEBUG me.retrodaredevil.solarthing.io.ReloadableIOBundle - Successfully reloaded IOBundle
2023-01-08 19:00:27.687 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Commands are disabled
2023-01-08 19:00:27.836 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=1
2023-01-08 19:00:27.837 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=2
2023-01-08 19:00:27.838 [main] INFO  me.retrodaredevil.solarthing.program.MateCommandSender - Not going to (possibly) send commands now because stale=true!
2023-01-08 19:00:27.846 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPackets - We must be on old firmware. 41st char(index=40): ','
2023-01-08 19:00:27.858 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=3
2023-01-08 19:00:27.864 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPackets - We must be on old firmware. 41st char(index=40): ','
2023-01-08 19:00:27.865 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=4
2023-01-08 19:00:28.092 [main] WARN  me.retrodaredevil.solarthing.packets.handling.implementations.TimedPacketReceiver - Was going to send off 4 packets to packetListReceive
r, but stale=true!
2023-01-08 19:00:28.780 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=1
2023-01-08 19:00:28.806 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=2
2023-01-08 19:00:28.832 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPackets - We must be on old firmware. 41st char(index=40): ','
2023-01-08 19:00:28.832 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=3
2023-01-08 19:00:28.858 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPackets - We must be on old firmware. 41st char(index=40): ','
2023-01-08 19:00:28.859 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=4
2023-01-08 19:00:29.050 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - Debugging all packets
2023-01-08 19:00:29.276 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - [ {
  "packetType" : "FX_STATUS",
  "address" : 1,
  "packetVersion" : 2,
  "inverterCurrent" : 5.0,
  "inverterCurrentRaw" : 5,
  "chargerCurrent" : 0.0,
  "chargerCurrentRaw" : 0,
  "buyCurrent" : 0.0,
  "buyCurrentRaw" : 0,
  "inputVoltage" : 124,
  "inputVoltageRaw" : 124,
  "outputVoltage" : 124,
  "outputVoltageRaw" : 124,
  "sellCurrent" : 3.0,
  "sellCurrentRaw" : 3,
  "operatingMode" : 9,
  "errorMode" : 0,
  "acMode" : 2,
  "batteryVoltage" : 51.2,
  "misc" : 152,
  "warningMode" : 0,
  "chksum" : 50
}, {
  "packetType" : "FX_STATUS",
  "address" : 2,
  "packetVersion" : 2,
  "inverterCurrent" : 5.0,
  "inverterCurrentRaw" : 5,
  "chargerCurrent" : 0.0,
  "chargerCurrentRaw" : 0,
  "buyCurrent" : 0.0,
  "buyCurrentRaw" : 0,
  "inputVoltage" : 125,
  "inputVoltageRaw" : 125,
  "outputVoltage" : 126,
  "outputVoltageRaw" : 126,
  "sellCurrent" : 3.0,
  "sellCurrentRaw" : 3,
  "operatingMode" : 9,
  "errorMode" : 0,
[... 163 lines cut ...]
} ]
2023-01-08 19:00:29.383 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - [LOG] Ending SolarThing. Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36:4
5Z Java version: 11.0.16
2023-01-08 19:00:29.384 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Got throwable
java.util.NoSuchElementException: valueCode: 63 not found in enum: class me.retrodaredevil.solarthing.solar.outback.mx.AuxMode
	at me.retrodaredevil.solarthing.packets.Modes.getActiveMode(Modes.java:61) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPacket.getAuxMode(MXStatusPacket.java:147) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPacket.isFlexMax(MXStatusPacket.java:237) ~[solarthing-2022.4.1.jar:?]
[... cut ...]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:145) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:202) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:262) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:276) ~[solarthing-2022.4.1.jar:?]
2023-01-08 19:00:29.397 [main] DEBUG me.retrodaredevil.solarthing.program.SolarMain - Going to shutdown LogManager.

rbroen3rd, 2023-01-08:
I have no experience with online databases.  What do you recommend for me to use to get web access to my data?  Thank you so much for your patience with me.  I used to work on Unix workstations a long time ago, but have been retired for many years.  I'm now 82 and a lot of that has come back while working on this project.

retrodaredevil, 2023-01-08:
Alright so first off, glad you got it somewhat working. Second off, thank you for helping me test the Outback stuff. I run SolarThing on a pretty old Outback system as well and have never gotten that error.

It looks like your MX4 is in a particular aux mode. Do you know what mode that is? It seems to be some sort of mode that SolarThing is unfamiliar with and I would love help documenting it if you are able.

rbroen3rd, 2023-01-08:
I don't know what mode that is.  I saw that when I was editing my spreadsheet.  I may have to get my installer out to diagnose the system.  This is one reason I wanted to get SolarThing running with a web database.  My output has dropped and I suspect a bad panel or something.  My system is fifteen years old.

rbroen3rd, 2023-01-08:
I'm happy to help you test the Outback stuff.

retrodaredevil, 2023-01-08:
> I have no experience with online databases. What do you recommend for me to use to get web access to my data? Thank you so much for your patience with me. I used to work on Unix workstations a long [...]

This is usually an exercise for the reader because most people will want to set up Grafana to see graphs and stuff, but if you don't want to mess with that, I recommend this:

* SolarThing mate running on a raspberry pi (you have this setup)
* Install CouchDB on a separate computer, preferably with a hard drive
* SolarThing graphql (soon to be renamed to SolarThing server) running on that same separate computer.

If you don't have a separate computer to install CouchDB on, you can install it on your Raspberry Pi, I just don't know how long the SD card will last. SolarThing GraphQL will provide a web interface for you to view data.

rbroen3rd, 2023-01-08:
I'm running a Raspberry Pi 4 B model, so I am okay with that.  The only other computers I have are my iMac which has a form of Linux for its OS, and an older Raspberry Pi 1.  I want to get web-based real-time access even if it only updates every fifteen minutes or so.

retrodaredevil, 2023-01-08:
Depending on how much you want to help out, I would love to know the firmware of your Mate 2 and your MX 4 device. Also, is your MX 4 actually an MX, or is it a FlexMax (FM)? If it's a FlexMax, I suspect that the FlexMax is trying to report a specific aux mode to the mate, but the mate does not have new enough firmware to understand that mode. That's just a guess though.

If you're able to look at your MX 4's screen, it might also have the aux mode on it somewhere. I suspect that your Mate cannot interpret your MX 4's aux mode if it's a FlexMax.

No matter what the problem is, I'm gonna have to release an update to SolarThing at some point to make "63" a valid aux mode for MX/FM devices.

rbroen3rd, 2023-01-08:
I'm happy to help out as much as you need me to.  My MXs are not Flexmax, they are MXs.  I'll get the firmware version for you.  Also, my Mate is not even a 2.  I think it might be a Mate 1.  It's fifteen years old.  I'm going out to look at the MX screen for the Aux mode.

retrodaredevil, 2023-01-08:
> I'm running a Raspberry Pi 4 B model, so I am okay with that. The only other computers I have are my iMac which has a form of Linux for its OS, and an older Raspberry Pi 1. I want to get web-based real-time access even if it only updates every fifteen minutes or so.

It's definitely possible to get it running on your Mac OS device, but I don't have the time or Mac OS knowledge to help out with that. Best I can do is recommend to either just run CouchDB on your Pi and hope for the best, or plug a USB hard drive into the pi and have CouchDB's data stored on there (which I might be able to help with, but I'm not entirely sure).

I imagine that this would be easiest for you if SolarThing could upload directly to PVOutput from the mate program, but that's not currently a feature, but it is one that I would consider adding after this.

rbroen3rd, 2023-01-08:
I need to think about my options a bit before proceeding.  Just being able to capture the data log is a big plus for me.  Even if I only display it locally, that's a big step up over the Mate and the MX buttons to navigate the data.   If I can access the data file into a spreadsheet that would be a big help.  

Both MXs are showing Aux=off.   So, the 63 mode code is a strange one.

rbroen3rd, 2023-01-08:
I have CouchDB running on my Mac.  No problem.  Very easy.

rbroen3rd, 2023-01-08:
Mate version 32
Code 3.50 0xD787
Serial # MA020020

retrodaredevil, 2023-01-08:
Oh nice. I forgot how easy it can be to configure CouchDB on mac OS. You can configure SolarThing mate to upload data to there, then you have a couple of options.

First, you are either going to configure SolarThing GraphQL on your Pi, or you are going to configure SolarThing PVOutput on your pi. If you configure GraphQL, you can view many aspects of your SolarThing data on a web page. This options contains no graphs. If you choose to upload to PVOutput, you will only see energy generated from solar panels and energy used by the FXs.

So the decision on which one you want to go with is up to you.

> If I can access the data file into a spreadsheet that would be a big help.

SolarThing currently does not have a feature like this, but if all you care about is daily kWh (solar generation) and energy usage data, then I believe PVOutput may allow you to download data in a CSV format.

Alternatively, there is a feature that I haven't used in forever that still exists in SolarThing. Instead of a CouchDB database configuration, you can have a configuration like so:

```json
{
  "type": "latest",
  "file": "/path/to/store/json/data.json"
}
```

rbroen3rd, 2023-01-08:
Okay, I have some options to choose from.  I have a social obligation coming up shortly and will have to get back to this project later tonight or tomorrow.  The json format data output sounds easy to get going.  I usually get the most straightforward solution running and then tackle the more complex ones in order of complexity.  I may choose the json configuration to get familiar with it.

retrodaredevil, 2023-01-08:
The particular bug that detecting a FLEXmax while using old mate firmware has been fixed and the fix is present in this release: https://github.com/wildmountainfarms/solarthing/releases/tag/v2023.1.0.

Be sure to update. Alternatively, this bug is actually caused by having analytics enabled, as whether or not a charge controller is an MX or FLEXmax is one of the few things being reported. You can always disable analytics instead.

Link to updating here: https://solarthing.readthedocs.io/en/latest/updating.html I'm going to leave this issue open to answer questions however I can, but the problem is effectively fixed now.

rbroen3rd, 2023-01-08:
Thanks for keeping the issue open.   I have a few more questions.  I will be shutting down for the day.  Thanks so much for your wonderful support and responsiveness.

rbroen3rd, 2023-01-08:
Updated and ran SolarThing with the following output:
rbrown3rd@bobspi01:/opt/solarthing/program/mate$ sudo -u solarthing ./run.sh
Using base config = config/base.json
2023-01-08 23:42:19.901 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36:45Z Java version: 11.0.16
[stdout] Beginning main. Jar: Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36:45Z Java version: 11.0.16
[stderr] Beginning main. Jar: Jar: solarthing-2022.4.1.jar Last Modified: 2022-11-09T05:36:45Z Java version: 11.0.16
2023-01-08 23:42:20.207 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
2023-01-08 23:42:21.091 [main] INFO  me.retrodaredevil.solarthing.program.OutbackMateMain - Beginning mate program
2023-01-08 23:42:21.098 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Google Analytics is ENABLED!
2023-01-08 23:42:21.132 [main] DEBUG me.retrodaredevil.solarthing.analytics.AnalyticsManager - Using Analytics UUID: 625b28be-b975-42df-a9e2-0503f8cb7e71
2023-01-08 23:42:22.024 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending program type to Google Analytics
2023-01-08 23:42:22.046 [main] DEBUG me.retrodaredevil.solarthing.program.OutbackMateMain - IO Bundle File: config/mate_serial.json
2023-01-08 23:42:22.595 [main] DEBUG me.retrodaredevil.solarthing.io.ReloadableIOBundle - Successfully reloaded IOBundle
2023-01-08 23:42:22.644 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Commands are disabled
2023-01-08 23:42:23.146 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=1
2023-01-08 23:42:23.152 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=2
2023-01-08 23:42:23.180 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPackets - We must be on old firmware. 41st char(index=40): ','
2023-01-08 23:42:23.188 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=3
2023-01-08 23:42:23.204 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPackets - We must be on old firmware. 41st char(index=40): ','
2023-01-08 23:42:23.205 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=4
2023-01-08 23:42:23.396 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - Debugging all packets
2023-01-08 23:42:23.638 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - [ {
  "packetType" : "FX_STATUS",
  "address" : 1,
  "packetVersion" : 2,
  "inverterCurrent" : 0.0,
  "inverterCurrentRaw" : 0,
  "chargerCurrent" : 1.0,
  "chargerCurrentRaw" : 1,
  "buyCurrent" : 3.0,
  "buyCurrentRaw" : 3,
  "inputVoltage" : 122,
  "inputVoltageRaw" : 122,
  "outputVoltage" : 123,
  "outputVoltageRaw" : 123,
  "sellCurrent" : 0.0,
  "sellCurrentRaw" : 0,
  "operatingMode" : 3,
  "errorMode" : 0,
  "acMode" : 2,
  "batteryVoltage" : 52.0,
  "misc" : 24,
  "warningMode" : 0,
  "chksum" : 34
}, {
  "packetType" : "FX_STATUS",
  "address" : 2,
  "packetVersion" : 2,
  "inverterCurrent" : 0.0,
  "inverterCurrentRaw" : 0,
  "chargerCurrent" : 0.0,
  "chargerCurrentRaw" : 0,
  "buyCurrent" : 1.0,
  "buyCurrentRaw" : 1,
  "inputVoltage" : 124,
  "inputVoltageRaw" : 124,
  "outputVoltage" : 126,
  "outputVoltageRaw" : 126,
  "sellCurrent" : 0.0,
  "sellCurrentRaw" : 0,
  "operatingMode" : 10,
  "errorMode" : 0,
  "acMode" : 2,
  "batteryVoltage" : 52.0,
  "misc" : 24,
  "warningMode" : 0,
  "chksum" : 35
}, {
  "packetType" : "MXFM_STATUS",
  "address" : 3,
[... 87 lines cut ...]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:202) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:262) ~[solarthing-2022.4.1.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:276) ~[solarthing-2022.4.1.jar:?]
2023-01-08 23:42:23.740 [main] DEBUG me.retrodaredevil.solarthing.program.SolarMain - Going to shutdown LogManager.

java.util.NoSuchElementException: valueCode: 63 not found in enum: class me.retrodaredevil.solarthing.solar.outback.mx.AuxMode
	at me.retrodaredevil.solarthing.packets.Modes.getActiveMode(Modes.java:61)
	at me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPacket.getAuxMode(MXStatusPacket.java:147)
	at me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPacket.isFlexMax(MXStatusPacket.java:237)
	at me.retrodaredevil.solarthing.analytics.MateAnalyticsHandler.send(MateAnalyticsHandler.java:42)
	at me.retrodaredevil.solarthing.analytics.MateAnalyticsHandler.handle(MateAnalyticsHandler.java:25)
	at me.retrodaredevil.solarthing.packets.handling.PacketHandlerMultiplexer.handle(PacketHandlerMultiplexer.java:29)
	at me.retrodaredevil.solarthing.program.PacketListReceiverHandler.handleSinglePacketCollection(PacketListReceiverHandler.java:88)
	at me.retrodaredevil.solarthing.program.PacketListReceiverHandler.handle(PacketListReceiverHandler.java:82)
	at me.retrodaredevil.solarthing.program.PacketListReceiverHandler.lambda$new$2(PacketListReceiverHandler.java:67)
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20)
[... cut ...]
	at me.retrodaredevil.solarthing.program.SolarMain.initReader(SolarMain.java:62)
	at me.retrodaredevil.solarthing.program.OutbackMateMain.connectMate(OutbackMateMain.java:106)
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:145)
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:202)
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:262)
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:276)

retrodaredevil, 2023-01-09:
Sorry, I forgot to bump the version. Give the update a try now after pulling.

rbroen3rd, 2023-01-09:
Running now.  Output below:
2023-01-09 11:13:25.582 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=1
2023-01-09 11:13:25.609 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=2
2023-01-09 11:13:25.635 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPackets - We must be on old firmware. 41st char(inde
x=40): ','
2023-01-09 11:13:25.636 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=3
2023-01-09 11:13:25.658 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.mx.MXStatusPackets - We must be on old firmware. 41st char(inde
x=40): ','
2023-01-09 11:13:25.658 [main] DEBUG me.retrodaredevil.solarthing.solar.outback.MatePacketCreator49 - Created outback packet address=4
2023-01-09 11:13:25.808 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - Debugging all packets
2023-01-09 11:13:25.814 [main] DEBUG me.retrodaredevil.solarthing.program.PacketListReceiverHandlerBundle - [ {
  "packetType" : "FX_STATUS",
  "address" : 1,
  "packetVersion" : 2,
  "inverterCurrent" : 0.0,
  "inverterCurrentRaw" : 0,
  "chargerCurrent" : 0.0,
  "chargerCurrentRaw" : 0,
  "buyCurrent" : 1.0,
  "buyCurrentRaw" : 1,
  "inputVoltage" : 122,
  "inputVoltageRaw" : 122,
  "outputVoltage" : 124,
  "outputVoltageRaw" : 124,
  "sellCurrent" : 0.0,
  "sellCurrentRaw" : 0,
  "operatingMode" : 10,
  "errorMode" : 0,
  "acMode" : 2,
  "batteryVoltage" : 50.8,
  "misc" : 24,
  "warningMode" : 0,
  "chksum" : 36
}, {
  "packetType" : "FX_STATUS",
  "address" : 2,
  "packetVersion" : 2,
  "inverterCurrent" : 0.0,
  "inverterCurrentRaw" : 0,
  "chargerCurrent" : 0.0,
  "chargerCurrentRaw" : 0,
  "buyCurrent" : 1.0,
  "buyCurrentRaw" : 1,
  "inputVoltage" : 124,
  "inputVoltageRaw" : 124,
  "outputVoltage" : 126,
  "outputVoltageRaw" : 126,
  "sellCurrent" : 0.0,
  "sellCurrentRaw" : 0,
  "operatingMode" : 10,
  "errorMode" : 0,
  "acMode" : 2,
  "batteryVoltage" : 50.4,
  "misc" : 24,
  "warningMode" : 0,
  "chksum" : 37
}, {
  "packetType" : "MXFM_STATUS",
  "address" : 3,
  "packetVersion" : 2,
  "chargerCurrent" : 0,
  "pvCurrent" : 0,
  "inputVoltage" : 30,
  "dailyKWH" : 3.8,
  "ampChargerCurrent" : 0.0,
  "auxMode" : 0,
  "errorMode" : 0,
  "chargerMode" : 0,
  "batteryVoltage" : 50.4,
  "dailyAH" : 0,
  "dailyAHSupport" : "NOT_SUPPORTED",
  "chksum" : 43
}, {
  "packetType" : "MXFM_STATUS",
  "address" : 4,
  "packetVersion" : 2,
  "chargerCurrent" : 0,
  "pvCurrent" : 0,
  "inputVoltage" : 29,
rbrown3rd@bobspi01:/opt/solarthing/program/mate/logs$ tail  log_debug.log
  "warningModeValue" : 0,
  "miscValue" : 24,
  "acModeValues" : [ 2 ]
}, {
  "packetType" : "SOURCE",
  "sourceId" : "default"
}, {
  "packetType" : "FRAGMENT_INDICATOR",
  "fragmentId" : 1
} ]

rbroen3rd, 2023-01-09:
To run the "latest" config, which creates a file representing the latest data do I edit the base.json config file?   Do I add
{
  "type": "latest",
  "file": "/path/to/store/json/data.json"
}

In there?

retrodaredevil, 2023-01-09:
> data do I edit the base.json config file

You are going to edit the `base.json` file, but first you need to create a separate file in the same directory as `base.json`. You can edit `config/latest.json` (full path should be `/opt/solarthing/program/mate/config/latest.json`).
Inside of `config/latest.json`, you can then paste

```json
{
"type": "latest",
"file": "/path/to/store/json/data.json"
}
```
Make sure to edit the path to wherever you want to be able to access the file. If you want to just have the file in `/opt/solarthing/program/mate`, you can either use `/opt/solarthing/program/mate/latest_data.json` or more simply `latest_data.json`. As a note, the `solarthing` user needs to be able to edit files in the directory that you provide here, so sticking with my recommendation is probably the easiest option unless you have a specific place you want the file to be.

Once you have that configuration done, you can now reference `config/latest.json` in your `base.json` configuration. The documentation for that is here: https://solarthing.readthedocs.io/en/latest/configuration-edit-base-json.html

rbroen3rd, 2023-01-09:
Right now my base.json file contains the reference to the Mate as:
{
  "type": "mate",
  "source": "default",
  "fragment": 1,
  "unique": 60,
  "databases": [ ],
  "io": "config/mate_serial.json"
}
Do I add the reference to the latest below that or replace it?

rbroen3rd, 2023-01-09:
Couldn't parse data from file: /opt/solarthing/program/mate/config/latest.json Please make sure your JSON is correct.
latest.json
`{
"type": "latest",
"file": "/opt/solarthing/program/mate/output/data.json"
}

rbroen3rd, 2023-01-10:
2023-01-10 01:26:25.074 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - SolarThing is shutting down. This is caused by an error in your configuration or in how your environment is set up. Detailed message below:

Couldn't parse data from file: config/latest.json Please make sure your JSON is correct.

rbroen3rd, 2023-01-10:
The AUX mode on my second MX reads "Vent Fan" on the MX display.   That must be the mode 63 that I  keep seeing.

retrodaredevil, 2023-01-10:
Sorry, that's my bad. Seems I forgot how I'm supposed to format configuration files. Try this in your latest.json

```
{
  "type": "latest",
  "config": {
    "file": "/opt/solarthing/program/mate/output/data.json"
  }
}
```

For more advanced configuration, use something like this:
```
{
  "type": "latest",
  "settings": {
    "packet_upload": {
      "throttle_factor": 10,
      "initial_skip": 0
    }
  },
  "config": {
    "file": "/opt/solarthing/program/mate/output/data.json"
  }
}
```
That last example will be a little nicer on your pi's SD card, writing every 10 seconds rather than every one second.

rbroen3rd, 2023-01-10:
My Pi is humming away logging data.  I used the ten-second config file you provided.  Thanks for that. I would be happy to figure out a way to import the json files into a Google Sheets file.  I still am interested in something like Grafana using CouchDB.  I need to take a little break from this project, but I will work towards that now that I am capturing data successfully. 

I have two charge controllers.  I had to replace one of them two or three years ago. The newer one is a FlexMax, but as you suggest, the Mate will always call it an MX.  
FLEXmax 60
FM60-150VDC
Maximum Power Point Tracking

closed 2023-01-11 by retrodaredevil