wildmountainfarms/solarthing #84: CouchDB setup failed

rbroen3rd, 2023-01-12:
I started to setup a CouchDB database running on my Mac on the same network as my Pi.  Setup the couchdb.json config file as on the website and got the following errors.  

```
rbrown3rd@bobspi01:/opt/solarthing/program/mate$ solarthing run --couchdb-setup config/couchdb.json
16:02:06.194 [main] INFO  m.r.s.p.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
[stdout] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
[stderr] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
com.fasterxml.jackson.core.JsonParseException: Unexpected character (':' (code 58)): was expecting comma to separate Object entries
 at [Source: (File); line: 15, column: 20]
	at com.fasterxml.jackson.core.JsonParser._constructError(JsonParser.java:2391)
	at com.fasterxml.jackson.core.base.ParserMinimalBase._reportError(ParserMinimalBase.java:735)
	at com.fasterxml.jackson.core.base.ParserMinimalBase._reportUnexpectedChar(ParserMinimalBase.java:659)
	at com.fasterxml.jackson.core.json.UTF8StreamJsonParser.nextToken(UTF8StreamJsonParser.java:778)
	at com.fasterxml.jackson.databind.util.TokenBuffer._copyBufferContents(TokenBuffer.java:1166)
	at com.fasterxml.jackson.databind.util.TokenBuffer.copyCurrentStructure(TokenBuffer.java:1148)
	at com.fasterxml.jackson.databind.DeserializationContext.bufferAsCopyOfValue(DeserializationContext.java:565)
	at com.fasterxml.jackson.databind.deser.impl.ExternalTypeHandler.handlePropertyValue(ExternalTypeHandler.java:180)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer._deserializeWithExternalTypeId(BeanDeserializer.java:983)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeWithExternalTypeId(BeanDeserializer.java:948)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeWithExternalTypeId(BeanDeserializer.java:941)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:349)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:184)
	at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:323)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4674)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3494)
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:217)
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:272)
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:286)
Problem reading CouchDB database settings file.```

retrodaredevil, 2023-01-12:
Your JSON is formatted incorrectly. Send me your couchdb config file.

rbroen3rd, 2023-01-12:
```{
  "type": "couchdb",
  "settings": {
    "packet_upload": {
      "throttle_factor": 3,
      "initial_skip": 1
    },
    "command_download": {
      "throttle_factor": 3,
      "initial_skip": 4
    }
  },
  "config": {
    "protocol": "http",
    "host": "host": "192.168.1.64",
    "port": 5984,
    "username": "admin",
    "password": "xxxxxx",
    "connection_timeout": 1.5,
    "call_timeout": 10
  }
}
```

retrodaredevil, 2023-01-12:
> "host": "host": "192.168.1.64",

Might want to fix this line.

Should be `"host": "192.168.1.64",`

rbroen3rd, 2023-01-12:
```Created solarthing
Created solarthing_events
Created solarthing_closed
Created solarthing_open
Created solarthing_cache
Created solarthing_alter
All necessary databases have been created.
```

rbroen3rd, 2023-01-12:
I restarted SolarThing with the new base.json containing the database config reference.  It is running fine.  Now I need to figure out how to view my database.

rbroen3rd, 2023-01-12:
I'm overwhelmed.  I have CouchDB installed, and SolarThing is writing to a database.  What's the easiest way to access and display the data?  Grafana looks neat, but it seems like overkill for me right now.  I'm open to suggestions for the best way to display the data in a simple format and be accessible remotely.

retrodaredevil, 2023-01-12:
Here's the official documentation for your exact question: https://solarthing.readthedocs.io/en/latest/view-data.html

Since you have a CouchDB database up and running, I would normally recommend SolarThing Android, but I need to update it so it won't crash when it reads an invalid aux mode from your FM2 in port 4. If this is something that interests you, I can have it updated within a few days.

SolarThing GraphQL is what I would recommend. It provides a web interface out of the box that is mostly tailored for my usage, but will still provide you with a lot of good data points. Here's what it looks like:

[image]

Not the prettiest thing in the world, but it'll display your information. I'm uncertain what "Generator Status" would display for you as your FXs are programmed differently than mine.

rbroen3rd, 2023-01-12:
That looks perfect.  I'll check it out.

rbroen3rd, 2023-01-12:
What are the requirements to run the android app?

retrodaredevil, 2023-01-12:
You have CouchDB database running and a somewhat up to date android phone.

rbroen3rd, 2023-01-12:
Okay.  I'll work on getting GraphQL running.  I no longer have any android devices.

rbroen3rd, 2023-01-13:
I have Solarthing running on my Raspberry Pi 4B with two options for output using different versions of base.json referencing different config files.  One writes output to a data.json file every five minutes, while the other writes to a CouchDB database on my office Mac on the same network as my Pi.  Thank you so much for your exceptional support in getting me to this point.  

I've tried to use what I have but constructing CouchDB queries to get the few parameters I want to display has proved challenging. What you propose with GraphQL with your questions will be what I need in the long run.  I want to monitor daily production from my two PV strings and their current and voltage to detect bad panels because of lower-than-expected voltage.  Monitoring buying and selling kWh would be helpful too.

I need some "big picture" concepts to help me move on with this effort.  Is GraphQL a query system for CouchDB?  Does it generate HTML pages with graphic output?  Where is the web server running?

Sorry for the plethora of questions.  I retired from IT 19 years ago.  I'm a little rusty on my skills.

retrodaredevil, 2023-01-13:
If all you care about is production, then you might want SolarThing PVOutput. 

SolarThing GraphQL (AKA SolarThing Server) is a web server containing a web interface (in my last picture) and a GraphQL endpoint. 

GraphQL is a query language that is an alternative to REST style queries. I have implemented many GraphQL queries for SolarThing which will then query CouchDB based on the provided parameters (such as time period to get data for). 

I recommended this for you only because it would be trivial to get up and running. (Follow part of the GraphQL and Grafana tutorial). Once it is up and running, you just navigate to where it is hosted (http://piip:8080). From there, you should see data. 

Having this program running is a requirement for getting Grafana working, but the web interface is right there without any other configuring. You have the ability to create your own GraphQL queries with this running, but I figured you probably either wouldn't be doing that, or would only do it once you want a more advanced way to see raw data. I can give you an example query to later today to start you off with a simple way to see the input voltages or any other parameter off one of your devices.

rbroen3rd, 2023-01-13:
Your SolarThing GraphQL is the thing for what I am trying to accomplish.   I think your implementation will do what I am looking for.  Do I just follow the steps on the readthedocs.io or is there something special I have to do?

retrodaredevil, 2023-01-13:
Yup just follow the docs and stop when you get to the part about Grafana. The main thing is making sure you refer to your CouchDB configuration correctly. The working directory is `/opt/solarthing/program/graphql`, so to refer to your current CouchDB config, use `../mate/config/couchdb.json`.

rbroen3rd, 2023-01-13:
Thanks.  I'll tackle this tonight or over the weekend and will ask for help if I need to.

rbroen3rd, 2023-01-13:
```Error starting ApplicationContext. To display the conditions report re-run your application with 'debug' enabled.
2023-01-13 20:45:20.871 ERROR [           main] o.s.b.SpringApplication                  : Application run failed
```

retrodaredevil, 2023-01-13:
Did you define `solarthing.config.database` in `application.properties`? It's the only configuration required https://solarthing.readthedocs.io/en/latest/data/graphql-grafana.html

What does your `application.properties` look like?

rbroen3rd, 2023-01-13:
```../mate/config/couchdb.json
```

retrodaredevil, 2023-01-13:
The documentation shows that you need more than that. 

`solarthing.config.database=../mate/config/couchdb.json`

When I comment on here I'm usually commenting on my phone so the documentation will usually be more thorough than whatever I comment here.

rbroen3rd, 2023-01-13:
Also I think it needs to move up twice in directories before it's in the right place as in
```solarthing.config.database=../../mate/config/couchdb.json
```

retrodaredevil, 2023-01-13:
Nope! Keep in mind that although you would be correct if it was relative to the config directory, but the working directory is not the config directory.

rbroen3rd, 2023-01-13:
Oh, right.  Of course.  Thanks.

rbroen3rd, 2023-01-13:
```
2023-01-13 23:37:37.756  INFO [           main] o.s.b.a.w.s.WelcomePageHandlerMapping    : Adding welcome page: class path resource [static/index.html]
2023-01-13 23:37:38.168  INFO [           main] o.a.c.h.Http11NioProtocol                : Starting ProtocolHandler ["http-nio-8080"]
2023-01-13 23:37:38.235  INFO [           main] o.s.b.w.e.t.TomcatWebServer              : Tomcat started on port(s): 8080 (http) with context path ''
2023-01-13 23:37:38.281  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : Started SolarThingGraphQLApplication in 20.071 seconds (JVM running for 24.277)

```

rbroen3rd, 2023-01-14:
Partial Success.  I have the graphics, but all categories say "loading." I should shut everything down and restart.   I think this will satisfy my needs when I get it fully working.

retrodaredevil, 2023-01-14:
Are there errors in the logs or in your web browser's console logs?

rbroen3rd, 2023-01-14:
I'll have to look tomorrow.  Thank you for being so helpful.

rbroen3rd, 2023-01-14:
I restarted everything this morning.  The CouchDB is updating.  GraphQL is showing the graphics, but all values are stuck on loading.  I suspect some permissions problem.   Could it be a user incompatibility?  
[image]

rbroen3rd, 2023-01-14:
Here is the debug log from GraphQL:
```2023-01-14 14:25:29.795  INFO [nio-8080-exec-3] m.r.s.r.g.SolarThingExceptionHandler     : Got database exception

me.retrodaredevil.solarthing.rest.exceptions.DatabaseException: Exception querying from 1673705610000 to 1673706330000
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:118) ~[classes!/:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.queryPackets(SimpleQueryHandler.java:123) ~[classes!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.queryStatus(SimpleQueryHandler.java:162) ~[classes!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.service.SolarThingGraphQLService.queryStatusLast(SolarThingGraphQLService.java:85) ~[classes!/:?]
	at jdk.internal.reflect.GeneratedMethodAccessor35.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at io.leangen.graphql.metadata.execution.FixedMethodInvoker.execute(FixedMethodInvoker.java:22) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.metadata.Resolver.resolve(Resolver.java:102) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.lambda$execute$1(OperationExecutor.java:93) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:103) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:99) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:58) ~[spqr-0.11.2.jar!/:?]
	at graphql.execution.ExecutionStrategy.fetchField(ExecutionStrategy.java:270) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.ExecutionStrategy.resolveFieldWithInfo(ExecutionStrategy.java:203) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.AsyncExecutionStrategy.execute(AsyncExecutionStrategy.java:60) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.Execution.executeOperation(Execution.java:165) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.Execution.execute(Execution.java:104) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.execute(GraphQL.java:557) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.parseValidateAndExecute(GraphQL.java:482) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.executeAsync(GraphQL.java:446) ~[graphql-java-16.2.jar!/:?]
	at java.util.concurrent.CompletableFuture.uniComposeStage(CompletableFuture.java:1106) ~[?:?]
	at java.util.concurrent.CompletableFuture.thenCompose(CompletableFuture.java:2235) ~[?:?]
	at graphql.spring.web.servlet.components.DefaultGraphQLInvocation.invoke(DefaultGraphQLInvocation.java:41) ~[graphql-java-spring-webmvc-2.0.jar!/:?]
	at graphql.spring.web.servlet.components.GraphQLController.executeRequest(GraphQLController.java:150) ~[graphql-java-spring-webmvc-2.0.jar!/:?]
	at graphql.spring.web.servlet.components.GraphQLController.graphqlPOST(GraphQLController.java:84) ~[graphql-java-spring-webmvc-2.0.jar!/:?]
	at jdk.internal.reflect.GeneratedMethodAccessor34.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:205) ~[spring-web-5.3.18.jar!/:5.3.18]
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:150) ~[spring-web-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:117) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:895) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:808) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1067) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:963) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1006) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:909) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:681) ~[tomcat-embed-core-9.0.60.jar!/:?]
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:764) ~[tomcat-embed-core-9.0.60.jar!/:?]
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:227) ~[tomcat-embed-core-9.0.60.jar!/:?]
[... cut ...]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.queryView(OkHttpCouchDbDatabase.java:242) ~[couchdb-java-8ff00d2c49.jar!/:?]
	at me.retrodaredevil.couchdbjava.CouchDbShared.queryView(CouchDbShared.java:23) ~[couchdb-java-8ff00d2c49.jar!/:?]
	at me.retrodaredevil.solarthing.database.couchdb.CouchDbMillisDatabase.query(CouchDbMillisDatabase.java:74) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:116) ~[classes!/:?]
	... 75 more
```

retrodaredevil, 2023-01-14:
Database might not be set up correctly. Did you fully run through the couchdb setup without exiting before it completed? I'll look at this more in depth later, but I think the _design/packets document is missing in the solarthing database.

rbroen3rd, 2023-01-14:
Just did a curl query as admin and got a list of all databases.  I think they are all owned by the admin account, but that is the one I used in the CouchDB config file.
```
["_replicator","_users","solarthing","solarthing_alter","solarthing_cache","solarthing_closed","solarthing_events","solarthing_open"]
```

rbroen3rd, 2023-01-14:
> Database might not be set up correctly. Did you fully run through the couchdb setup without exiting before it completed? I'll look at this more in depth later, but I think the _design/packets document is missing in the solarthing database.

I don't think I quit before installation was complete.  I'll look in the solarthing database for the _designpackets database.  It might be this afternoon before I can get back to this.  We are so close.  Is there a way to manually add the _design/packets document manually?

retrodaredevil, 2023-01-14:
Yeah there's a way to add it manually, but the setup should have done that. What version of CouchDB do you have installed? If you ran through the setup program I can try to reproduce this bug. 

I'll give you more details later about adding it manually.

rbroen3rd, 2023-01-14:
CouchDB is v. 3.3.0.

retrodaredevil, 2023-01-14:
Gotcha. I'm gonna look into this more tomorrow. I likely won't have any recommendations before then. This does give me an idea for a "CouchDB diagnostics" type program, though.

retrodaredevil, 2023-01-16:
If you could go through the setup program again, I think it should add it for you. When you go through the setup again, you can either press enter or type the same name you used last time for the uploader and manager users (it should not reprompt for password). The setup program should correct any incorrect permissions as those two users have permissions that are automatically given to them by the setup program.

Here's an example output of myself going through the setup program:

```
lavender@lavender-hp:~/programming/SolarThing/solarthing/program$ .bin/solarthing run --couchdb-setup config/develop_couchdb.json
18:01:56.872 [main] INFO  m.r.s.p.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 17.0.1
[stdout] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 17.0.1
[stderr] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 17.0.1
You will now setup your CouchDB instance! Some databases will be automatically created (enter)

Created solarthing
Created solarthing_events
Created solarthing_closed
Created solarthing_open
Created solarthing_cache
Created solarthing_alter
All necessary databases have been created.

Now views and security will be configured for each database. Please enter the name of the user to be added as an admin to each database.
This user is commonly named 'uploader'. (Leave blank to not configure)
Name of user: 
No user will be added as an admin, but members will still be cleared. (Enter to confirm)

You can also enter the name of the user to manage the solarthing_cache and solarthing_alter databases.
This user is commonly named 'manager'. (Leave blank to not configure)

No user will be configured to manage the solarthing_cache and solarthing_alter database. (Enter to confirm)

Adding packets design to database solarthing
This database will have the millisNull view
This database will be readonly
Configuring security for database solarthing

Adding packets design to database solarthing_events
This database will have the millisNull view
This database will be readonly
Configuring security for database solarthing_events

Adding packets design to database solarthing_closed
This database will be readonly
Configuring security for database solarthing_closed

Adding packets design to database solarthing_open
This database will have the millisNull view
Configuring security for database solarthing_open

Configuring security for database solarthing_cache

Adding packets design to database solarthing_alter
This database will have the simpleAllDocs view
This database will be readonly
Configuring security for database solarthing_alter

Completed successfully!
```

I did not configure the `uploader` or `manager` user, but that shouldn't affect whether or not it works. If you go through the program again and it still doesn't work, send me the output.

If you want to manually add it, here it is for the `solarthing` database. It may be slightly different for other databases. (I don't recommend manually adding it, so use this as a last resort).

```
{
  "_id": "_design/packets",
  "views": {
    "millisNull": {
      "map": "function(doc) {\n  emit(doc.dateMillis, null);\n}"
    }
  },
  "validate_doc_update": "function(newDoc, oldDoc, userCtx, secObj) {\n\n  secObj.admins = secObj.admins || {};\n  secObj.admins.names = secObj.admins.names || [];\n  secObj.admins.roles = secObj.admins.roles || [];\n\n  var isAdmin = false;\n  if(userCtx.roles.indexOf('_admin') !== -1) {\n    isAdmin = true;\n  }\n  if(secObj.admins.names.indexOf(userCtx.name) !== -1) {\n    isAdmin = true;\n  }\n  for(var i = 0; i < userCtx.roles; i++) {\n    if(secObj.admins.roles.indexOf(userCtx.roles[i]) !== -1) {\n      isAdmin = true;\n    }\n  }\n\n  if(!isAdmin) {\n    throw {'unauthorized':'This is read only when unauthorized'};\n  }\n}",
  "language": "javascript"
}
```

retrodaredevil, 2023-01-16:
Also, can you go to your web interface you have running using SolarThing GraphQL and head to the status page and give me output similar to the following?

```
Status DB: COMPLETE
Event DB: COMPLETE
Open DB: COMPLETE
Closed DB: COMPLETE
Alter DB: BAD_PERMISSIONS
Cache DB: BAD_PERMISSIONS
```

You might not have to do that if you get it working with my above steps.

rbroen3rd, 2023-01-16:
CouchDB Status

Error Loading Status

rbroen3rd, 2023-01-16:
It hangs here for a very long time and never returns a system prompt:
```
  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)

2023-01-16 01:23:16.907  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : Starting SolarThingGraphQLApplication using Java 11.0.16 on bobspi01 with PID 12752 (/opt/solarthing/program/.downloads/solarthing-graphql-2023.1.0.jar started by solarthing in /opt/solarthing/program/graphql)
2023-01-16 01:23:16.953 DEBUG [           main] m.r.s.r.SolarThingGraphQLApplication     : Running with Spring Boot v2.6.6, Spring v5.3.18
2023-01-16 01:23:16.958  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : No active profile set, falling back to 1 default profile: "default"
2023-01-16 01:23:21.942  INFO [           main] o.s.b.w.e.t.TomcatWebServer              : Tomcat initialized with port(s): 8080 (http)
2023-01-16 01:23:22.021  INFO [           main] o.a.c.h.Http11NioProtocol                : Initializing ProtocolHandler ["http-nio-8080"]
2023-01-16 01:23:22.023  INFO [           main] o.a.c.c.StandardService                  : Starting service [Tomcat]
2023-01-16 01:23:22.024  INFO [           main] o.a.c.c.StandardEngine                   : Starting Servlet engine: [Apache Tomcat/9.0.60]
2023-01-16 01:23:22.364  INFO [           main] o.a.c.c.C.[.[.[/]                        : Initializing Spring embedded WebApplicationContext
2023-01-16 01:23:22.364  INFO [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 5157 ms
2023-01-16 01:23:22.707 DEBUG [           main] m.r.s.r.CommonProvider                   : Using defaultInstanceOptions=DefaultInstanceOptions(defaultSourceId='<UNUSED SOURCE ID THAT WILL NEVER BE IN A PACKET>', defaultFragmentId=0)
2023-01-16 01:23:22.708 DEBUG [           main] m.r.s.r.CommonProvider                   : Database file: /opt/solarthing/program/graphql/../mate/config/couchdb.json
2023-01-16 01:23:22.710 DEBUG [           main] m.r.s.r.CommonProvider                   : Working directory: /opt/solarthing/program/graphql/.
2023-01-16 01:23:24.346 DEBUG [           main] m.r.s.r.c.CommandHandlerProvider         : No command configuration file. No one will be authorized to send commands.
2023-01-16 01:23:24.390 DEBUG [           main] m.r.s.r.g.GraphQLProvider                : No solcast config! Not using solcast!
2023-01-16 01:23:24.457 DEBUG [           main] m.r.s.r.g.GraphQLProvider                : Using timezone: Europe/London
2023-01-16 01:23:26.339  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and @me.retrodaredevil.solarthing.annotations.NotNull() java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
2023-01-16 01:23:26.351  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
2023-01-16 01:23:33.900  INFO [           main] o.s.b.a.w.s.WelcomePageHandlerMapping    : Adding welcome page: class path resource [static/index.html]
2023-01-16 01:23:34.340  INFO [           main] o.a.c.h.Http11NioProtocol                : Starting ProtocolHandler ["http-nio-8080"]
2023-01-16 01:23:34.398  INFO [           main] o.s.b.w.e.t.TomcatWebServer              : Tomcat started on port(s): 8080 (http) with context path ''
2023-01-16 01:23:34.440  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : Started SolarThingGraphQLApplication in 19.989 seconds (JVM running for 24.222)

```

rbroen3rd, 2023-01-16:
Status page from web app:
``CouchDB Status

Status DB: COMPLETE
Event DB: COMPLETE
Open DB: COMPLETE
Closed DB: COMPLETE
Alter DB: COMPLETE
Cache DB: BAD_PERMISSIONS
```

rbroen3rd, 2023-01-16:
Getting this output looping over and over:

```
2023-01-16 01:31:09.062  INFO [nio-8080-exec-4] m.r.s.r.g.SolarThingExceptionHandler     : Got database exception

me.retrodaredevil.solarthing.rest.exceptions.DatabaseException: Exception querying from 1673831950000 to 1673832670000
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:118) ~[classes!/:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.queryPackets(SimpleQueryHandler.java:123) ~[classes!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.queryStatus(SimpleQueryHandler.java:162) ~[classes!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.service.SolarThingGraphQLService.queryStatusLast(SolarThingGraphQLService.java:85) ~[classes!/:?]
	at jdk.internal.reflect.GeneratedMethodAccessor36.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at io.leangen.graphql.metadata.execution.FixedMethodInvoker.execute(FixedMethodInvoker.java:22) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.metadata.Resolver.resolve(Resolver.java:102) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.lambda$execute$1(OperationExecutor.java:93) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:103) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:99) ~[spqr-0.11.2.jar!/:?]
	at io.leangen.graphql.execution.OperationExecutor.execute(OperationExecutor.java:58) ~[spqr-0.11.2.jar!/:?]
	at graphql.execution.ExecutionStrategy.fetchField(ExecutionStrategy.java:270) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.ExecutionStrategy.resolveFieldWithInfo(ExecutionStrategy.java:203) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.AsyncExecutionStrategy.execute(AsyncExecutionStrategy.java:60) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.Execution.executeOperation(Execution.java:165) ~[graphql-java-16.2.jar!/:?]
	at graphql.execution.Execution.execute(Execution.java:104) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.execute(GraphQL.java:557) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.parseValidateAndExecute(GraphQL.java:482) ~[graphql-java-16.2.jar!/:?]
	at graphql.GraphQL.executeAsync(GraphQL.java:446) ~[graphql-java-16.2.jar!/:?]
	at java.util.concurrent.CompletableFuture.uniComposeStage(CompletableFuture.java:1106) ~[?:?]
	at java.util.concurrent.CompletableFuture.thenCompose(CompletableFuture.java:2235) ~[?:?]
	at graphql.spring.web.servlet.components.DefaultGraphQLInvocation.invoke(DefaultGraphQLInvocation.java:41) ~[graphql-java-spring-webmvc-2.0.jar!/:?]
	at graphql.spring.web.servlet.components.GraphQLController.executeRequest(GraphQLController.java:150) ~[graphql-java-spring-webmvc-2.0.jar!/:?]
	at graphql.spring.web.servlet.components.GraphQLController.graphqlPOST(GraphQLController.java:84) ~[graphql-java-spring-webmvc-2.0.jar!/:?]
	at jdk.internal.reflect.GeneratedMethodAccessor33.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:205) ~[spring-web-5.3.18.jar!/:5.3.18]
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:150) ~[spring-web-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:117) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:895) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:808) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1067) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:963) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1006) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:909) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:681) ~[tomcat-embed-core-9.0.60.jar!/:?]
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883) ~[spring-webmvc-5.3.18.jar!/:5.3.18]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:764) ~[tomcat-embed-core-9.0.60.jar!/:?]
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:227) ~[tomcat-embed-core-9.0.60.jar!/:?]
[... cut ...]
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.queryView(OkHttpCouchDbDatabase.java:242) ~[couchdb-java-8ff00d2c49.jar!/:?]
	at me.retrodaredevil.couchdbjava.CouchDbShared.queryView(CouchDbShared.java:23) ~[couchdb-java-8ff00d2c49.jar!/:?]
	at me.retrodaredevil.solarthing.database.couchdb.CouchDbMillisDatabase.query(CouchDbMillisDatabase.java:74) ~[core-0.0.1-SNAPSHOT.jar!/:?]
	at me.retrodaredevil.solarthing.rest.graphql.SimpleQueryHandler.lambda$queryPackets$0(SimpleQueryHandler.java:116) ~[classes!/:?]
	... 75 more
```

rbroen3rd, 2023-01-16:
I'm going to have to call it a night.  It's been a long day.  I can work on it some more tomorrow afternoon.

retrodaredevil, 2023-01-16:
Are you saying the setup program hangs for a while? Can you run it again and give me its output? 

I'll have a diagnostics program done by the end of the week. Should make this a lot easier for both of us.

rbroen3rd, 2023-01-16:
Yes.  The setup program hangs in what seems to be an endless loop giving the feedback posted three posts above this one over and over with no return to system prompt forever.  I'm giving up for the night.  Thank you for sticking with me on this.  I'll be able to  try again tomorrow afternoon.

rbroen3rd, 2023-01-16:
Before going out the door, I restarted my Pi and my Mac and tried the installation again.  These installation messages appeared and then the installation stopped as you see it below with no response for many minutes.  ```
  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)

2023-01-16 12:33:52.829  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : Starting SolarThingGraphQLApplication using Java 11.0.16 on bobspi01 with PID 917 (/opt/solarthing/program/.downloads/solarthing-graphql-2023.1.0.jar started by solarthing in /opt/solarthing/program/graphql)
2023-01-16 12:33:52.864 DEBUG [           main] m.r.s.r.SolarThingGraphQLApplication     : Running with Spring Boot v2.6.6, Spring v5.3.18
2023-01-16 12:33:52.868  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : No active profile set, falling back to 1 default profile: "default"
2023-01-16 12:33:58.224  INFO [           main] o.s.b.w.e.t.TomcatWebServer              : Tomcat initialized with port(s): 8080 (http)
2023-01-16 12:33:58.317  INFO [           main] o.a.c.h.Http11NioProtocol                : Initializing ProtocolHandler ["http-nio-8080"]
2023-01-16 12:33:58.319  INFO [           main] o.a.c.c.StandardService                  : Starting service [Tomcat]
2023-01-16 12:33:58.321  INFO [           main] o.a.c.c.StandardEngine                   : Starting Servlet engine: [Apache Tomcat/9.0.60]
2023-01-16 12:33:58.692  INFO [           main] o.a.c.c.C.[.[.[/]                        : Initializing Spring embedded WebApplicationContext
2023-01-16 12:33:58.693  INFO [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 5535 ms
2023-01-16 12:33:59.045 DEBUG [           main] m.r.s.r.CommonProvider                   : Using defaultInstanceOptions=DefaultInstanceOptions(defaultSourceId='<UNUSED SOURCE ID THAT WILL NEVER BE IN A PACKET>', defaultFragmentId=0)
2023-01-16 12:33:59.046 DEBUG [           main] m.r.s.r.CommonProvider                   : Database file: /opt/solarthing/program/graphql/../mate/config/couchdb.json
2023-01-16 12:33:59.047 DEBUG [           main] m.r.s.r.CommonProvider                   : Working directory: /opt/solarthing/program/graphql/.
2023-01-16 12:34:00.937 DEBUG [           main] m.r.s.r.c.CommandHandlerProvider         : No command configuration file. No one will be authorized to send commands.
2023-01-16 12:34:00.983 DEBUG [           main] m.r.s.r.g.GraphQLProvider                : No solcast config! Not using solcast!
2023-01-16 12:34:01.064 DEBUG [           main] m.r.s.r.g.GraphQLProvider                : Using timezone: Europe/London
2023-01-16 12:34:03.041  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and @me.retrodaredevil.solarthing.annotations.NotNull() java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
2023-01-16 12:34:03.054  WARN [           main] i.l.g.g.OperationMapper                  : Potential type name collision detected: 'Float' bound to multiple types: java.lang.Float (loaded by the bootstrap class loader) and java.lang.Double (loaded by the bootstrap class loader). Assign unique names using the appropriate annotations or override the TypeInfoGenerator. For details and solutions see https://github.com/leangen/graphql-spqr/wiki/Errors#non-unique-type-name. If this warning is a false positive, please report it: https://github.com/leangen/graphql-spqr/issues.
2023-01-16 12:34:10.973  INFO [           main] o.s.b.a.w.s.WelcomePageHandlerMapping    : Adding welcome page: class path resource [static/index.html]
2023-01-16 12:34:11.380  INFO [           main] o.a.c.h.Http11NioProtocol                : Starting ProtocolHandler ["http-nio-8080"]
2023-01-16 12:34:11.469  INFO [           main] o.s.b.w.e.t.TomcatWebServer              : Tomcat started on port(s): 8080 (http) with context path ''
2023-01-16 12:34:11.511  INFO [           main] m.r.s.r.SolarThingGraphQLApplication     : Started SolarThingGraphQLApplication in 21.317 seconds (JVM running for 26.312)
```

retrodaredevil, 2023-01-16:
> Yes.  The setup program hangs in what seems to be an endless loop giving the feedback posted three posts above this one over and over with no return to system prompt forever.  I'm giving up for the night.  Thank you for sticking with me on this.  I'll be able to  try again tomorrow afternoon.  

So I don't care about any of the output except for the setup program for the time being. Can you run through that and send me all the output of it? I don't want anymore of your graphql program output and that's all I see from your responses. 

The only output I see that is concerning from your GraphQL program is some of the errors that say it could not connect to your CouchDB instance, but we'll deal with those once we get the setup program run through successfully.

rbroen3rd, 2023-01-16:
Okay.  I'll run through that and post the output.   Just to read back to make sure I am doing this correctly.  CouchDB is installed on my Mac. You want to run the setup on CouchDB again.  I am drawing a blank in my memory as to how I did that.  Are you asking me to install it again?

rbroen3rd, 2023-01-16:
```
solarthing run --couchdb-setup config/couchdb.json
22:26:42.955 [main] INFO  m.r.s.p.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
[stdout] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
[stderr] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
java.io.FileNotFoundException: config/couchdb.json (No such file or directory)
	at java.base/java.io.FileInputStream.open0(Native Method)
	at java.base/java.io.FileInputStream.open(FileInputStream.java:219)
	at java.base/java.io.FileInputStream.<init>(FileInputStream.java:157)
	at com.fasterxml.jackson.core.JsonFactory.createParser(JsonFactory.java:1029)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3494)
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:217)
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:272)
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:286)
Problem reading CouchDB database settings file.
```

rbroen3rd, 2023-01-16:
```
 run --couchdb-setup config/couchdb.json
22:29:24.775 [main] INFO  m.r.s.p.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
[stdout] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
[stderr] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
You will now setup your CouchDB instance! Some databases will be automatically created (enter)

Exception in thread "main" java.lang.RuntimeException: me.retrodaredevil.couchdbjava.exception.CouchDbException: java.net.ConnectException: Failed to connect to /192.168.1.64:5984
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:238)
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:272)
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:286)
Caused by: me.retrodaredevil.couchdbjava.exception.CouchDbException: java.net.ConnectException: Failed to connect to /192.168.1.64:5984
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.executeCall(OkHttpCouchDbInstance.java:92)
	at me.retrodaredevil.couchdbjava.okhttp.auth.CookieAuthHandler.authSession(CookieAuthHandler.java:122)
	at me.retrodaredevil.couchdbjava.okhttp.auth.CookieAuthHandler.preAuthorize(CookieAuthHandler.java:161)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.preAuthorize(OkHttpCouchDbInstance.java:161)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.create(OkHttpCouchDbDatabase.java:97)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.createIfNotExists(OkHttpCouchDbDatabase.java:114)
	at me.retrodaredevil.couchdbjava.CouchDbDatabase.createIfNotExists(CouchDbDatabase.java:28)
	at me.retrodaredevil.solarthing.program.CouchDbSetupMain.createDatabase(CouchDbSetupMain.java:55)
	at me.retrodaredevil.solarthing.program.CouchDbSetupMain.doCouchDbSetupMain(CouchDbSetupMain.java:103)
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:229)
	... 2 more
Caused by: java.net.ConnectException: Failed to connect to /192.168.1.64:5984
	at okhttp3.internal.connection.RealConnection.connectSocket(RealConnection.kt:297)
	at okhttp3.internal.connection.RealConnection.connect(RealConnection.kt:207)
	at okhttp3.internal.connection.ExchangeFinder.findConnection(ExchangeFinder.kt:226)
	at okhttp3.internal.connection.ExchangeFinder.findHealthyConnection(ExchangeFinder.kt:106)
	at okhttp3.internal.connection.ExchangeFinder.find(ExchangeFinder.kt:74)
	at okhttp3.internal.connection.RealCall.initExchange$okhttp(RealCall.kt:255)
	at okhttp3.internal.connection.ConnectInterceptor.intercept(ConnectInterceptor.kt:32)
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109)
	at okhttp3.internal.cache.CacheInterceptor.intercept(CacheInterceptor.kt:95)
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109)
	at okhttp3.internal.http.BridgeInterceptor.intercept(BridgeInterceptor.kt:83)
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109)
	at okhttp3.internal.http.RetryAndFollowUpInterceptor.intercept(RetryAndFollowUpInterceptor.kt:76)
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.lambda$new$0(OkHttpCouchDbInstance.java:49)
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109)
	at okhttp3.internal.connection.RealCall.getResponseWithInterceptorChain$okhttp(RealCall.kt:201)
	at okhttp3.internal.connection.RealCall.execute(RealCall.kt:154)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.executeCall(OkHttpCouchDbInstance.java:90)
	... 11 more
Caused by: java.net.ConnectException: Connection refused (Connection refused)
	at java.base/java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.base/java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:412)
	at java.base/java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:255)
	at java.base/java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:237)
	at java.base/java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.base/java.net.Socket.connect(Socket.java:609)
	at okhttp3.internal.platform.Platform.connectSocket(Platform.kt:128)
	at okhttp3.internal.connection.RealConnection.connectSocket(RealConnection.kt:295)
	... 29 more

```

rbroen3rd, 2023-01-16:
```
rbrown3rd@bobspi01:/opt/solarthing/program/mate$ solarthing run --couchdb-setup config/couchdb.json
22:48:57.963 [main] INFO  m.r.s.p.SolarMain - [LOG] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
[stdout] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
[stderr] Beginning main. Jar: Jar: solarthing-2023.1.0.jar Last Modified: 2023-01-08T21:59:21Z Java version: 11.0.16
You will now setup your CouchDB instance! Some databases will be automatically created (enter)

Already exists: solarthing
Already exists: solarthing_events
Already exists: solarthing_closed
Already exists: solarthing_open
Already exists: solarthing_cache
Already exists: solarthing_alter
All necessary databases have been created.

Now views and security will be configured for each database. Please enter the name of the user to be added as an admin to each database.
This user is commonly named 'uploader'. (Leave blank to not configure)
Name of user: uploader
User: uploader will be used. (Enter to confirm)

Please enter a password for 'uploader'.
xxxxxxx
error
internal_server_error
Exception in thread "main" java.lang.RuntimeException: me.retrodaredevil.couchdbjava.exception.CouchDbCodeException: Unknown status code! code: 500 error: error reason: internal_server_error
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:238)
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:272)
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:286)
Caused by: me.retrodaredevil.couchdbjava.exception.CouchDbCodeException: Unknown status code! code: 500 error: error reason: internal_server_error
	at me.retrodaredevil.couchdbjava.okhttp.util.OkHttpUtil.createException(OkHttpUtil.java:88)
	at me.retrodaredevil.couchdbjava.okhttp.util.OkHttpUtil.createExceptionFromResponse(OkHttpUtil.java:69)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.executeAndHandle(OkHttpCouchDbInstance.java:101)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.putDocument(OkHttpCouchDbDatabase.java:145)
	at me.retrodaredevil.solarthing.program.CouchDbSetupMain.createUserIfNotExists(CouchDbSetupMain.java:93)
	at me.retrodaredevil.solarthing.program.CouchDbSetupMain.doCouchDbSetupMain(CouchDbSetupMain.java:118)
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:229)
	... 2 more
```

rbroen3rd, 2023-01-16:
Okay, I got the setup to the point where it asks for users.  I didn't know what to enter there so I just hit enter.  I'm guessing I shouldn't have done that.  Let me know if I should run it again.

retrodaredevil, 2023-01-16:
Interesting. I've never seen that error before. Alright couple of questions

1. Your password for your user was not blank, I assume?
2. Does the user defined in config/couchdb.json have admin permissions for all of your CouchDB instance? 

I want to explain what's going on a bit here because it's not documented the best.
The setup program aims to create stuff like `_design/packets`, but it can also create users. The `uploader` user should be the user used in a `couchdb.json` for any SolarThing program that only uploads to the database. The `manager` user should be used only for the `couchdb.json` file used by SolarThing GraphQL. **Configuring these users is completely optional**. Having multiple users with different permissions increases the security of the database in case a password is compromised for some reason. Honestly, it's probably not worth worrying about. You can use a single `couchdb.json` file as long as the user in that file has admin permissions.

I do plan on making this setup a lot easier eventually. At some point SolarThing GraphQL (SolarThing Server) is going to be able to make changes to CouchDB instead of the setup program. Successful completion of the setup program will make everything run smoothly without any restarts required. At no point should you *have* to restart any SolarThing instance.

Thanks for sticking with it. Sounds like I can improve the documentation. I will do that at some point to make it a bit easier to follow.

retrodaredevil, 2023-01-16:
As a follow up, if you get stuff like this again:

> Caused by: java.net.ConnectException: Failed to connect to /192.168.1.64:5984

It may be worth changing your `couchdb.json` file a tiny bit. You can increase your `connection_timeout` to something like 5 seconds. That should seriously help out with those connection issues.

rbroen3rd, 2023-01-17:
Thanks for the big picture explanation. That helps a lot.

rbroen3rd, 2023-01-17:
> Interesting. I've never seen that error before. Alright couple of questions
> 
> 1. Your password for your user was not blank, I assume?
> 2. Does the user defined in config/couchdb.json have admin permissions for all of your CouchDB instance?

I changed the timeout in the CouchDB.json as you suggested.  

Re: question 1, which user are you referring to?  The last time I ran the setup I hit return for all users, not entering a login or a password at all prompts.  It might be helpful to explain the users required.  

Re: Question 2....I have no idea.  Here is the file:  ```
{
  "type": "couchdb",
  "settings": {
    "packet_upload": {
      "throttle_factor": 3,
      "initial_skip": 1
    },
    "command_download": {
      "throttle_factor": 3,
      "initial_skip": 4
    }
  },
  "config": {
    "protocol": "http",
    "host": "192.168.1.64",
    "port": 5984,
    "username": "admin",
    "password": "xxxxxxx",
    "connection_timeout": 5,
    "call_timeout": 10
  }
}
```

retrodaredevil, 2023-01-17:
Alright. You are using the `admin` user, so you should be good no matter what.

> Now views and security will be configured for each database. Please enter the name of the user to be added as an admin to each database.
> This user is commonly named 'uploader'. (Leave blank to not [...]

So the fact that you typed `uploader` when it prompted for "Name of user" means that you had to provide a password. I just wanted to make sure that you did not just press enter when it asked for a password, as CouchDB might not like an empty password.

Here's my recommendation for right now. Run through the setup program again, and for each prompt, only press enter. That will make it so none of the users are configured. I expect the output to print `No user will be added as an admin, but members will still be cleared.`. If it does not print that output, you are specifying a user, which you do not need to do unless you want to make your database more secure.

rbroen3rd, 2023-01-17:
I don't think I entered Uploader as a user, but will run again with no users specified.

rbroen3rd, 2023-01-17:
```
Now views and security will be configured for each database. Please enter the name of the user to be added as an admin to each database.
This user is commonly named 'uploader'. (Leave blank to not configure)
Name of user: 
No user will be added as an admin, but members will still be cleared. (Enter to confirm)

You can also enter the name of the user to manage the solarthing_cache and solarthing_alter databases.
This user is commonly named 'manager'. (Leave blank to not configure)

No user will be configured to manage the solarthing_cache and solarthing_alter database. (Enter to confirm)

Adding packets design to database solarthing
This database will have the millisNull view
This database will be readonly
Exception in thread "main" java.lang.RuntimeException: me.retrodaredevil.couchdbjava.exception.CouchDbException: Connection failed!
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:238)
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:272)
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:286)
Caused by: me.retrodaredevil.couchdbjava.exception.CouchDbException: Connection failed!
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.executeCall(OkHttpCouchDbInstance.java:115)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.executeAndHandle(OkHttpCouchDbInstance.java:96)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbDatabase.putDocument(OkHttpCouchDbDatabase.java:145)
	at me.retrodaredevil.solarthing.program.CouchDbSetupMain.doCouchDbSetupMain(CouchDbSetupMain.java:160)
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:229)
	... 2 more
Caused by: java.io.InterruptedIOException: timeout
	at okhttp3.internal.connection.RealCall.timeoutExit(RealCall.kt:398)
	at okhttp3.internal.connection.RealCall.callDone(RealCall.kt:360)
	at okhttp3.internal.connection.RealCall.noMoreExchanges$okhttp(RealCall.kt:325)
	at okhttp3.internal.connection.RealCall.getResponseWithInterceptorChain$okhttp(RealCall.kt:209)
	at okhttp3.internal.connection.RealCall.execute(RealCall.kt:154)
	at retrofit2.OkHttpCall.execute(OkHttpCall.java:204)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.executeCall(OkHttpCouchDbInstance.java:105)
	... 6 more
Caused by: java.io.IOException: Canceled
	at okhttp3.internal.http.RetryAndFollowUpInterceptor.intercept(RetryAndFollowUpInterceptor.kt:72)
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109)
	at me.retrodaredevil.couchdbjava.okhttp.OkHttpCouchDbInstance.lambda$new$0(OkHttpCouchDbInstance.java:49)
	at okhttp3.internal.http.RealInterceptorChain.proceed(RealInterceptorChain.kt:109)
	at okhttp3.internal.connection.RealCall.getResponseWithInterceptorChain$okhttp(RealCall.kt:201)
	... 9 more
```

retrodaredevil, 2023-01-17:
Alright, well that's just a timeout error, so you can give it a try again. How is the connection between your pi and your mac hosting CouchDB?

rbroen3rd, 2023-01-17:
The connection between my Pi and Mac seems solid so far.  Is there any certain order things have to be started?  Do I have to open the web interface to CouchDB for it to receive data?

[51 more replies]

closed 2023-01-26 by retrodaredevil