wildmountainfarms/solarthing #85: PVoutput does not have voltage and temperature

egri-nagy, 2023-01-19:
I installed solarthing on a RPi3 connected to an Epever Tracer AN charge controller and started to upload to pvoutput.org. The system works fine and reliably. Thank you for this excellent software package!

One thing I miss is the voltage and temperature readings on pvoutput. The Tracer provides these, and they would be useful on PVoutput too. Could you please give some hints where to look for enabling reporting voltage and temperature? 

[image]

retrodaredevil, 2023-01-19:
Looks like you found something I need to document more. Here is the `base.json` I am using. I'll go into detail below.

```json
{
  "type": "pvoutput-upload",
  "database": "../config/couchdb.json",
  "system_id": 72206,
  "api_key": "xxx",
  "time_zone": "US/Mountain",
  "source": "default",
  "default_fragment": 1,
  "include_undefined_sources": true,
  "join_teams": true,
  "required": {
    "1": ["OutbackIdentifier(address=3)", "OutbackIdentifier(address=4)"],
    "2": ["RoverIdentifier()"],
    "5": ["TracerIdentifier()"]
  },
  "voltage_identifier": {
    "fragment": 1, "identifier": "OutbackIdentifier(address=3)"
  },
  "temperature_identifier": {
    "fragment": 102, "identifier": "DataIdentifier(dataId=1)"
  },
  "include_import": true
}
```
The main things to look at are `voltage_identifier` and `temperature_identifier`. I'm going to assume that your SolarThing program is using fragment 1 and you have a single tracer. You can have something like this:

```json
  "voltage_identifier": {
    "fragment": 1, "identifier": "TracerIdentifier()"
  },
```

This configuration is only this advanced because if you have multiple charge controllers you can only choose one charge controller to use as the source for the voltage that appears on PVOutput.

Now for temperature. SolarThing only supports uploading the temperature of a DS18B20 temperature sensor. I have no plans to support some sort of weather API retrieval, but may be open to suggestions for other temperature sensors to support. Here is the link to the documentation for setting up a DS18B20 temperature sensor with SolarThing: https://solarthing.readthedocs.io/en/latest/config/w1-temperature.html

egri-nagy, 2023-01-19:
Thank you for the prompt answer!

With  fragment ID 5 I could get the solar voltage uploaded. I was aiming for the battery voltage, but maybe that's not what pvoutput expects.

Regarding the temperature, I also meant the battery temperature. I have the Epever temperature sensor connected https://www.epever.com/product/accessories-remote-temperature-sensor/ , and the custom_tracer program reports the values, so I thought it was just the matter of plugging that value in the pvoutput upload. Maybe it is just a different fragment number?

When you refer to the tracer documentation, what document do you refer to?

retrodaredevil, 2023-01-19:
This is the documentation I was referring to https://solarthing.readthedocs.io/en/latest/tracer/config.html#tracer-configuration 

You make good points about using the battery voltage and battery temperature. I think I'd like to make those features to configure, at least the battery temperature. I'm not sure if PVOutput voltage will keep what's after the decimal if I upload the battery voltage, though.

retrodaredevil, 2023-01-21:
I have an idea for an eventual feature to support the voltage coming from different places. some possibilities of what a configuration could look like:

```json
{
  ...
  "voltage": {
    "type": "average-battery-voltage"
  }
}
```

```json
{
  ...
  "voltage": {
    "type": "battery-voltage",
    "identifier": "TracerIdentifier()"
  }
}
```
or just PV voltage like it is now

```json
{
  ...
  "voltage": {
    "type": "pv",
    "identifier": "TracerIdentifier()"
  }
}
```

I could do something similar with the temperature configuration. Maybe with types such as `battery-temperature`, `controller-temperature`, or just `w1`.

I'll keep this issue open as something I'll eventually get around to doing.

closed 2023-03-11 by retrodaredevil

retrodaredevil, 2023-03-11:
Hello! I have completed this feature to allow the customization I described above, but not with the exact same format as above. The [v2023.2.0](https://github.com/wildmountainfarms/solarthing/releases/tag/v2023.2.0) has these features, so make sure to [update](https://solarthing.readthedocs.io/en/latest/updating.html).

```json
{
  "type": "pvoutput-upload",
  ...
  "voltage_from": {
    "type": "packet",
    "identifier": { "fragment": 1, "identifier": "OutbackIdentifier(address=3)" }
  },
  "temperature_from": {
    "type": "packet",
    "identifier": { "fragment": 102, "identifier": "DataIdentifier(dataId=1)" }
  }
}

```

The above configuration will get the voltage from the PV Input of the specified packet and will get the temperature from the specified temperature sensor. @egri-nagy you talked about wanting to use the battery voltage and battery temperature. To do that, you can use this:

```json
{
  "type": "pvoutput-upload",
  ...
  "voltage_from": {
    "type": "packet",
    "from": "BATTERY",
    "identifier": { "fragment": 5, "identifier": "TracerIdentifier()" }
  },
  "temperature_from": {
    "type": "packet",
    "from": "BATTERY",
    "identifier": { "fragment": 5, "identifier": "TracerIdentifier()" }
  }
}
```

For the "packet" type for "voltage_from", you can use [`PV` (default), `BATTERY`]. For the "packet" type for "temperature_from" you can use [`PACKET` (default), `BATTERY`, `CONTROLLER`].

I hope this can be useful to you, but if you already have a setup you're happy with, I appreciate the feature request either way. I'm sure I'll have others use this feature. Now I just have to put better documentation around it.

commit fdec4447 (2023-03-11): Closes #85. Can now configure how temperature and voltage are retrieved