wildmountainfarms/solarthing #87: https://solarthing.readthedocs.io/en/latest/config/config-server.html?highlight=modbus%20id%20

riker65, 2023-02-16:
Hi

how to define modbus id for 

server connection in request program?

in rover there is a modbus id which can be changed

Thanks T

retrodaredevil, 2023-02-17:
The rover's modbus ID does not have anything to do with the configuration server. Let's take this `base.json`:

```json
{
  "type": "request",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "databases": [ ],
  "request": [
    {
      "type": "modbus",
      "io": "config/rover_serial.json",
      "devices": {
        "1": {
          "type": "rover"
        }
      }
    }
  ]
}
```
In the above configuration, the Modbus address of the rover is 1, as defined by `"1"`. You can view the documentation here  for [finding the modbus address](https://solarthing.readthedocs.io/en/latest/solarthing-check.html#devices-not-detected) and here for [rover configuration](https://solarthing.readthedocs.io/en/latest/rover/config.html).

Once you have that working, you can utilize the configuration server like described in its documentation. You can have a single configuration server for each device. In no circumstance that I know of should you have multiple devices per serial configuration unless you are using some sort of multiplexer, which is unlikely.

TDLR: You don't need to specify a modbus ID for the connection server, as it is implied by the already configured modbus ID that you are configuring the server for.

Hopefully that answered your question. If you intend to monitor multiple rover devices, then that's a separate answer, but it didn't sound like that's something you were doing.

riker65, 2023-02-17:
Hi 
thanks 
if I understand correct, I need this config
as modbus ID is now 17 changed after BT-2 was connected to the controller. Identified by your marvelous scanning program.

I only have request program running at the moment.

`TDLR: You don't need to specify a modbus ID for the connection server, as it is implied by the already configured modbus ID that you are configuring the server for.`
do not understand this part as there is nothing already defined, only in separate rover program, which I stopped using . What do you mean by **already** defined so?

so I think I need to adjust modbus id here? 
`      "devices": {
        "**17**": {
          "type": "rover"
        }`

Looks like this is working, nc is answering correct.

Thanks

retrodaredevil, 2023-02-17:
Yup. That 17 you put there defines the modbus address for requesting data and for the configuration server. 

The way you worded your question made it sound like you thought you had to define a modbus ID specifically for the configuration server feature, but it sounds like you understand it now and have it working.

retrodaredevil, 2023-02-21:
I'm closing this issue now. Feel free to reopen if you still need help.

closed 2023-02-21 by retrodaredevil