wildmountainfarms/solarthing #101: Error installing latest fatal: --local can only be used inside a git repository

mgrouch, 2023-03-24:
On Raspberry Pi4

```
This updates the permission of all the files in the solarthing repository. Modifications will be made to the directory printed below:
/opt/solarthing
continuing
Updating...
fatal: --local can only be used inside a git repository
```

retrodaredevil, 2023-03-24:
Can you run `cat /opt/solarthing/other/linux/update_perms.sh` and show me the output? Have you pulled/cloned SolarThing recently? I too had this error and I believe I fixed it recently, but I may not have

retrodaredevil, 2023-03-24:
Also worth noting that if we can't figure out how to fix this error, you can continue on configuring SolarThing (assuming you have an up to date solarthing repo). That error indicates that it was unable to set a flag on the repo, which really isn't that important.

In other words, you *could* ignore it, but I wouldn't recommend it since I would like to figure out what the problem is.

mgrouch, 2023-03-24:
Might be this:Is there a way for git to identify the git directory? It says- fatal: --local can only be used inside a git repositorystackoverflow.comSent from my iPhone

retrodaredevil, 2023-03-24:
> Might be this:Is there a way for git to identify the git directory? It says- fatal: --local can only be used inside a git

So I know that the error message sounds like something is being done incorrectly and that it's not a git repository, but when I fixed the problem I found it was actually because of a permission error.

I asked you to run  `cat /opt/solarthing/other/linux/update_perms.sh` because I had recently updated that file and wanted to confirm that you were using the new version. You have to give me a little more context of what commands you ran before that one and when you cloned the solarthing repo for me to help.

I'd like to replicate your issue but I'm uncertain if this occurred right after cloning or if you ran the `update_perms.sh` script directly. Unless you messed up the solarthing repo somehow (unlikely), my guess is that this error is being caused by a permission error.

retrodaredevil, 2023-03-29:
Hello again. Sorry for not understanding your question earlier. I was trying to understand how you got that error, but I recently tried running the clone install instructions and was able to easily reproduce the error you showed above.

First off, you can run these commands:
```shell
sudo usermod -a -G solarthing,dialout,tty,video $USER
sudo git config --global --add safe.directory /opt/solarthing
git config --global --add safe.directory /opt/solarthing
```

The first command is just to make sure your user is a part of the solarthing group. The second command makes sure that the root user has `/opt/solarthing` configured as a safe directory. The third command makes sure that your user has `/opt/solarthing` configured as a safe directory.

I have modified the [installation instructions](https://solarthing.readthedocs.io/en/latest/installation.html) and I have also modified the `update_perms.sh` script, which is run when running the `clone_install.sh` script. This should mean that users that are installing Solarthing for the first time shouldn't have this problem.

I imagine other users have had this problem for a while now. The main problem is that I am trying to allow anyone part of the `solarthing` group access `/opt/solarthing`, but configuring it correctly has proven difficult. I plan on creating a better way to deal with this in the future as the current setup is something I'm trying to migrate away from.

If you need additional help with permission errors or weird git errors, please respond to this issue. I will close this issue after a little while, but feel free to reopen it or create another issue.

closed 2023-04-03 by retrodaredevil