wildmountainfarms/solarthing #121: Talk to the charger

Mouky66, 2023-05-04:
Hi, is it possible to talk to the charger (Renogy Rover)?
For example to turn the Load on or off, or do some other configuration changes?

Thanks
Chris

retrodaredevil, 2023-05-04:
Yeah check out the documentation here: https://solarthing.readthedocs.io/en/latest/config/config-server.html.
The configuration server allows you to change parameters on your charge controller while SolarThing is running (I'm assuming you have SolarThing fully setup).

If you want to be able to turn the load on or off from SolarThing android or from a slack channel (slack takes quite a bit of setup, though), then you can look into SolarThing actions: https://solarthing.readthedocs.io/en/latest/actions/index.html. SolarThing actions are a fairly advanced part of SolarThing. If you want me to elaborate on this more, let me know. There's no good documentation for this at the moment.

Mouky66, 2023-05-05:
Solarthing is running for about two years now and I really appreciate your work. :)

I tried the Server Option and the communication is working, but the Rover won't "save" the configuration.

```shell
root@rpi-rover-01:/opt/solarthing/program/custom_rover# solarthing version
SolarThing made by Lavender Shannon
Jar: solarthing-2023.2.0.jar
Jar last modified: 2023-03-11T23:11:53Z
Java version: 1.8.0_312
root@rpi-rover-01:/opt/solarthing/program/custom_rover# cat config/base.json
{
  "type": "request",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "databases": [
    "config/mqtt.json"
  ],
  "request": [
    {
      "type": "modbus",
      "io": "config/rover_serial.json",
      "devices": {
        "1": {
          "type": "rover",
          "server": { "port": 5051 }
        }
      }
    }
  ]
}
root@rpi-rover-01:/opt/solarthing/program/custom_rover# nc localhost 5051
overDischargeVoltageRaw
111
overDischargeVoltageRaw 112
success
overDischargeVoltageRaw
111
``` 
Am I doing something wrong?

retrodaredevil, 2023-05-05:
I answered this question a while back (weirdly it ended up in solarthing-docs issues): https://github.com/wildmountainfarms/solarthing-docs/issues/4#issuecomment-1421378331.

Specifically for changing the charging parameters, you will have to first get the charge controller into user mode. From what I understand you cannot get the controller into user mode remotely (correct me if I'm wrong), so you will have to physically go to the charge controller and change the battery type to user or self customized or whatever it is called. Once you have done that, you should be able to edit the charging parameters remotely. From my testing I believe you can edit most of them, but some of them won't budge.

closed 2023-05-12 by retrodaredevil