wildmountainfarms/solarthing #125: Feature Request -Add Renogy BT1 support

screech24, 2023-05-27:
It would be incredible if you could somehow add support for Renogy BT1 Bluetooth modules. See this github link which someone has created a python script to read data from these modules. 
[](https://github.com/cyrils/renogy-bt1)

retrodaredevil, 2023-05-27:
This was talked about a little here: https://github.com/wildmountainfarms/solarthing/issues/44. I'm less opposed to trying this out now that others have had success, but this is something that I won't be testing or adding support for.

I'm not opposed to having someone else contribute to make the feature work, but I expect that Java support for Bluetooth is not as good as Python support for Bluetooth. Should someone decide to contribute to this there would have to be a lot of requirements to make this feature useable:
* Configuration should require a Bluetooth ID (or whatever the Bluetooth terminology is for a device's ID)
  * (Eventually) Configuration should also be as similar to current configuration options, meaning it should require the specification of a Modbus ID
* `solarthing check` command needs to be able to scan for Bluetooth devices so people can figure out their device's Bluetooth ID

Sorry that this feature is not a priority for me right now. I do think that implementing this feature would get more people to use SolarThing.

screech24, 2023-05-27:
Thanks for the reply.

 Unfortunately I don't have the coding knowledge required to do the work to
implement this. But I do however have a raspberry pi and the BT-1 module so
if someone does feel so inclined to add support for Bluetooth I would be
happy to test it out to insure everything works as expected.

 It would be a much better way of getting the data from these rover series
controllers since the USB cable isn't available anymore..(I know I could
make one but I already have the Bluetooth module) and Bluetooth means the
pi doesn't have to be in the same area as the charge controller.

retrodaredevil, 2024-03-13:
Going to close this issue for now. Anyone is welcome to reopen this in the future if anyone has anything else to add or would like to contribute to this feature.

closed 2024-03-13 by retrodaredevil