wildmountainfarms/solarthing #137: Modbus exception on PowMR (Renogy Rover)

voriskaeo, 2023-06-26:
Good afternoon,

Attempting to get connected to my PowMr solar inverter (POW-LVM3.0K-24V) that appears to be a clone of a Renogy unit. I was able to complete the configuration and the RS485 device is recognized and communicating. It even read the battery voltage using the test command.

In the error log, I am seeing some mode exception errors. No clue if this is normal or if my device isn't fully supported/implemented in this tool.

I've been trying to come up with a way to remotely monitor and graph the device's performance. Since I am dumb when it comes to programming languages and coding, I need an assist.

Thanks in advance. Included all log files from custom_rover/logs/

```
2023-05-23 19:56:58.600 [main] INFO  me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Communication with device working well. Got this response back: function code=0x83 data='02' feel free to open issue at https://github.com/wildmountainfarms/solarthing/issues/
2023-05-23 19:56:58.606 [main] INFO  me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - This error is common on models of charge controllers that are unsupported by SolarThing. When you create your issue, please include the model of your charge controller.
2023-05-23 19:56:58.608 [main] ERROR me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Modbus exception
me.retrodaredevil.io.modbus.handling.ErrorCodeException: Expected to get 3 as the function code, but actually got 131. With exception code 2
	at me.retrodaredevil.io.modbus.handling.HandleResponseHelper.checkResponse(HandleResponseHelper.java:17) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.io.modbus.handling.BaseReadRegisters.handleResponse(BaseReadRegisters.java:52) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.io.modbus.handling.BaseReadRegisters.handleResponse(BaseReadRegisters.java:11) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cache(ModbusCacheSlave.java:38) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheHolding(ModbusCacheSlave.java:47) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.modbus.ModbusCacheSlave.cacheHoldingRangeInclusive(ModbusCacheSlave.java:50) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.config.request.modbus.RoverModbusRequester.cacheRegisters(RoverModbusRequester.java:65) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.config.request.modbus.RoverModbusRequester.lambda$create$0(RoverModbusRequester.java:80) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper.receive(ModbusListUpdaterWrapper.java:66) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.doRequest(RequestMain.java:80) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:74) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:40) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:151) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:212) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:272) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:286) ~[solarthing-2023.2.0.jar:?]
2023-05-23 19:56:58.653 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 4.695 seconds

[log_debug.log](https://github.com/wildmountainfarms/solarthing/files/11862100/log_debug.log)
[log_info.log](https://github.com/wildmountainfarms/solarthing/files/11862108/log_info.log)
[log_summary.log](https://github.com/wildmountainfarms/solarthing/files/11862115/log_summary.log)

```

retrodaredevil, 2023-06-26:
Your device likely isn't fully supported at the moment, but the changes I have to do should be minimal. SolarThing supports collecting lots of data, but that data isn't always supported by all models of these charge controllers.

In order to figure out what piece of data is unsupported, I will need you to disable a feature that makes SolarThing more performant. You need to disable bulk requests. Information on how to do that can be found here: https://solarthing.readthedocs.io/en/latest/rover/bulk-request.html. If you are having trouble disabling bulk requests, you can paste your `base.json` and I can help out more.

Once you disable bulk requests, restart SolarThing and paste your slightly different error below. (feel free to not include the full logs, just the error like you gave above).

retrodaredevil, 2023-06-26:
Looks like you missed a comma somewhere in your base.json. Feel free to post it if you need help. 

I now realize that you have an inverter, which is not supported by solarthing (I thought you were talking about a charge controller when I first read your post). I likely won't have time to add support for this device, but I would still be interested in seeing the new error after you disable bulk requests (it will tell me how many of these registers are common between charge controllers and inverters). 

You aren't the first person to ask about supporting an inverter similar to this, so I'm tempted to eventually add support for it, or at least add support for reading the battery voltage and other common registers from it. But for the time being, I won't be able to support it. 

If you would like I can point to towards some Python libraries that already have support for this. I will also link the other issue similar to this later when I'm at my computer.

voriskaeo, 2023-06-26:
Oh dear. I was missing a comma at the end of line 13. The error was pointing to the middle of the disable bulk requests line so I didn't catch it.

I appreciate your help with this.

```
me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper - Modbus exception
me.retrodaredevil.io.modbus.handling.ErrorCodeException: Expected to get 3 as the function code, but actually got 131. With exception code 2
	at me.retrodaredevil.io.modbus.handling.HandleResponseHelper.checkResponse(HandleResponseHelper.java:17) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.io.modbus.handling.BaseReadRegisters.handleResponse(BaseReadRegisters.java:52) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.io.modbus.handling.BaseReadRegisters.handleResponse(BaseReadRegisters.java:11) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.io.modbus.ModbusSlave.sendRequestMessage(ModbusSlave.java:25) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.solar.util.AbstractModbusRead.get(AbstractModbusRead.java:20) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.solar.util.AbstractModbusRead.oneRegister(AbstractModbusRead.java:23) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.modbus.RoverModbusSlaveRead.getDailyMaxDischargingPower(RoverModbusSlaveRead.java:188) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.solar.renogy.rover.RoverStatusPackets.createFromReadTable(RoverStatusPackets.java:25) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.receiver.RoverPacketListUpdater.receive(RoverPacketListUpdater.java:32) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.receiver.ModbusListUpdaterWrapper.receive(ModbusListUpdaterWrapper.java:67) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.packets.handling.PacketListReceiverMultiplexer.receive(PacketListReceiverMultiplexer.java:20) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.doRequest(RequestMain.java:80) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:74) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.RequestMain.startRequestProgram(RequestMain.java:40) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:151) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:212) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:272) ~[solarthing-2023.2.0.jar:?]
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:286) ~[solarthing-2023.2.0.jar:?]
```

retrodaredevil, 2023-06-26:
#55 is actually the most similar to this (kind of funny that that issue is the one that caused me to write the documentation for `bulk_request`). I remember that much of the data supported by the Renogy Rover should be available on a device like this, so it is possible that I can change a few things to get ONLY the data from the charge controller side of things.

I am interested in getting this working at some point, but I want to stress that fixing this issue is not my priority at the moment. And, when or if I do get around to doing this, I would only be focused on the data from the charge controller side of things, not the inverter side of things.

If you are interested in getting something working soon, I recommend finding a Python library that supports your device. I will actively answer any questions, though. I'm always happy to help out, just don't have much free time on my hands at the moment.

voriskaeo, 2023-06-27:
No worries at all. You've been more then helpful with my monitoring project. My code skills are severely lacking. I'm more of a hardware person. I was able to adapt a tracer cable to use with the RS485 port on this device. That was as far as i could get on my own. Your software helped immensely during testing my adaptations.

I do understand that you want to primarily focus on the solar aspect along with this is a project. Not a commercial release.
Your documentation for setting up and operating SolarThing are top notch. I'll read through #55 and start looking around at other tools that may better suit my requirements.

Have a fantastic day and thank you once again! I don't know if I need to close this as fixed or not so I'll let you do that.

retrodaredevil, 2023-06-27:
I'll go ahead and keep this open, as there's a good chance I'll come back and at least support the charge controller functions. In an ideal world I would also implement the inverter side of things, but not sure if that will happen.

Maikeleg, 2023-06-28:
I had almost the same error and changed my base.json as follow;
`
...
          "type": "tracer",
          "clock": {
            "threshold": "PT10S"
...`
added comma behind "type": "tracer", 
"threshold" was spelled incorrectly
This wasnt an issue before the update
It's solved now and thanks @voriskaeo  for submitting this issue and @retrodaredevil for keeping it open and all your great work