wildmountainfarms/solarthing #207: "java.util.NoSuchElementException: No line found" while configuring couchDB

andraii, 2024-03-16:
After creating the couch.json and while attempting to run the setup program through the Docker Install option, I get the following error:

```
[stderr] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
2024-03-16 07:21:05.101 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
[stdout] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
You will now setup your CouchDB instance! Some databases will be automatically created (enter)
Exception in thread "main" java.util.NoSuchElementException: No line found
	at java.base/java.util.Scanner.nextLine(Unknown Source)
	at me.retrodaredevil.solarthing.program.CouchDbSetupMain$ScannerPrompt.promptContinue(CouchDbSetupMain.java:249)
	at me.retrodaredevil.solarthing.program.CouchDbSetupMain.doCouchDbSetupMain(CouchDbSetupMain.java:147)
	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:220)
	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:268)
	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:284)
```

Have I done something wrong? Sorry I can't give a more detailed summary, I don't know where to start!

retrodaredevil, 2024-03-16:
No, you didn't do anything wrong. I realize the documentation is slightly wrong. 

Currently the documentation reads this 

``` 
sudo docker run --rm -v ./config:/app/config ghcr.io/wildmountainfarms/solarthing run --couchdb-setup config/couchdb.json
```

I believe what will fix it is adding `-it` like so:

``` 
sudo docker run --rm -it -v ./config:/app/config ghcr.io/wildmountainfarms/solarthing run --couchdb-setup config/couchdb.json
```

That should allow the container to take input, which means that the setup program should be able to ask you things and you can answer them. 

I haven't done a walkthrough of making sure the documentation is error free when setting SolarThing up via docker, so let me know if any other errors pop up.

andraii, 2024-03-17:
Thank you very much for helping with this. I found two more things to fix while continuing with the instructions. These might be obvious but tripped me up as a novice.

1. The location "./config" does not work from within docker:
`sudo docker run --rm -it -v ./config:/app/config ghcr.io/wildmountainfarms/solarthing run --couchdb-setup config/couchdb.json`
You must use the whole location ie "/home/xxxxx/Documents/solarthing-config/config"

2. In my case I have both the DB and Solarthing within a Docker container. The DB connection is refused when connecting to "localhost" from within Docker. I had to use my host machine's IP address:
{
  "type": "couchdb",
  "settings": {
    "packet_upload": {
      "throttle_factor": 3,
      "initial_skip": 1
    },
    "command_download": {
      "throttle_factor": 3,
      "initial_skip": 4
    }
  },
  "config": {
    "url": "http://my_host_machine_IP_address:5984",
    "username": "xxxxxxxxx",
    "password": "xxxxxxxxxx",
    "connection_timeout": 1.5,
    "call_timeout": 10
  }
}

andraii, 2024-03-17:
Geez I didn't get far before running into another issue. I've updated my base.json but there appears to be something wrong with it. Can you please tell me what I've done wrong?

```
Attaching to solarthing
solarthing  | 2024-03-16 10:02:24.416 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
solarthing  | [stdout] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
solarthing  | [stderr] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
solarthing  | 2024-03-16 10:02:24.498 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
solarthing  | 2024-03-16 10:02:24.791 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Error while parsing ProgramOptions.
solarthing  | me.retrodaredevil.solarthing.config.ConfigException: Couldn't parse data from file: config/base.json Please make sure your JSON is correct.
solarthing  | 	at me.retrodaredevil.solarthing.config.CommonConfigUtil.createExceptionFromJackson(CommonConfigUtil.java:84) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.config.CommonConfigUtil.readConfig(CommonConfigUtil.java:92) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.config.ConfigUtil.readConfig(ConfigUtil.java:51) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:125) [solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:203) [solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:268) [solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:284) [solarthing.jar:?]
solarthing  | Caused by: com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "databases_config" (class me.retrodaredevil.solarthing.config.options.MateProgramOptions), not marked as ignorable (16 known properties: "fx_warning_ignore", "io", "ignore_check_sum", "commands", "action_config", "request", "fragment", "unique", "databases", "source", "short", "actions", "time_zone", "correct_check_sum", "database_config", "analytics_enabled"])
solarthing  |  at [Source: UNKNOWN; byte offset: #UNKNOWN] (through reference chain: me.retrodaredevil.solarthing.config.options.MateProgramOptions["databases_config"])
solarthing  | 	at com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException.from(UnrecognizedPropertyException.java:61) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownProperty(DeserializationContext.java:1153) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.std.StdDeserializer.handleUnknownProperty(StdDeserializer.java:2224) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownProperty(BeanDeserializerBase.java:1793) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownProperties(BeanDeserializerBase.java:1743) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.BeanDeserializer._deserializeUsingPropertyBased(BeanDeserializer.java:546) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.deserializeFromObjectUsingNonDefault(BeanDeserializerBase.java:1493) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:348) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.BeanDeserializer._deserializeOther(BeanDeserializer.java:220) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:187) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer._deserializeTypedForId(AsPropertyTypeDeserializer.java:170) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.jsontype.impl.AsPropertyTypeDeserializer.deserializeTypedFromObject(AsPropertyTypeDeserializer.java:136) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.AbstractDeserializer.deserializeWithType(AbstractDeserializer.java:263) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.impl.TypeWrappedDeserializer.deserialize(TypeWrappedDeserializer.java:74) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:342) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.ObjectMapper._readValue(ObjectMapper.java:4875) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3033) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.ObjectMapper.treeToValue(ObjectMapper.java:3497) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.config.CommonConfigUtil.readConfig(CommonConfigUtil.java:90) ~[solarthing.jar:?]
solarthing  | 	... 5 more
```

This is what my base.json looks like:

{
  "type": "mate",
  "source": "default",
  "fragment": 1,
  "unique": 60,
  "databases_config": {
    "databases": [
        {
        "external": "config/couchdb.json"
        }
        ]
  }
  "io": "config/mate_serial.json"
}

retrodaredevil, 2024-03-17:
I'll give you a more detailed response later, but right now the thing that sticks out is that it should be `database_config` rather than `databases_config`.

andraii, 2024-03-17:
Thank you, changing that gets me this error now

`Attaching to solarthing
solarthing  | 2024-03-17 05:18:42.090 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
solarthing  | [stdout] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
solarthing  | [stderr] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
solarthing  | 2024-03-17 05:18:42.181 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
solarthing  | 2024-03-17 05:18:42.343 [main] ERROR me.retrodaredevil.solarthing.program.SolarMain - (Fatal)Error while parsing ProgramOptions.
solarthing  | me.retrodaredevil.solarthing.config.ConfigException: Couldn't parse data from file: config/base.json Please make sure your JSON is correct.
solarthing  | 	at me.retrodaredevil.solarthing.config.CommonConfigUtil.createExceptionFromJackson(CommonConfigUtil.java:84) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.config.CommonConfigUtil.readAndInterpolate(CommonConfigUtil.java:67) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.config.CommonConfigUtil.readConfig(CommonConfigUtil.java:88) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.config.ConfigUtil.readConfig(ConfigUtil.java:51) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.program.SolarMain.doMainCommand(SolarMain.java:125) [solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.program.SolarMain.doMain(SolarMain.java:203) [solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.program.SolarMain.determineMainSubprogram(SolarMain.java:268) [solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.program.SolarMain.main(SolarMain.java:284) [solarthing.jar:?]
solarthing  | Caused by: com.fasterxml.jackson.core.JsonParseException: Unexpected character ('"' (code 34)): was expecting comma to separate Object entries
solarthing  |  at [Source: REDACTED (`StreamReadFeature.INCLUDE_SOURCE_IN_LOCATION` disabled); line: 13, column: 4]
solarthing  | 	at com.fasterxml.jackson.core.JsonParser._constructError(JsonParser.java:2481) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.core.base.ParserMinimalBase._reportError(ParserMinimalBase.java:752) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.core.base.ParserMinimalBase._reportUnexpectedChar(ParserMinimalBase.java:676) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.core.json.UTF8StreamJsonParser.nextFieldName(UTF8StreamJsonParser.java:1065) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.std.BaseNodeDeserializer._deserializeContainerNoRecursion(JsonNodeDeserializer.java:536) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.std.JsonNodeDeserializer.deserialize(JsonNodeDeserializer.java:100) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.std.JsonNodeDeserializer.deserialize(JsonNodeDeserializer.java:25) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.deser.DefaultDeserializationContext.readRootValue(DefaultDeserializationContext.java:342) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.ObjectMapper._readTreeAndClose(ObjectMapper.java:4941) ~[solarthing.jar:?]
solarthing  | 	at com.fasterxml.jackson.databind.ObjectMapper.readTree(ObjectMapper.java:3258) ~[solarthing.jar:?]
solarthing  | 	at me.retrodaredevil.solarthing.config.CommonConfigUtil.readAndInterpolate(CommonConfigUtil.java:65) ~[solarthing.jar:?]
solarthing  | 	... 6 more
solarthing exited with code 0
`

retrodaredevil, 2024-03-17:
That error means that you don't have valid JSON in base.json. You could post your base.json again or try to put it in an online JSON checker or something that can help you recognize the syntax error.

andraii, 2024-03-17:
All good I didn't realise there was such a thing as a JSON checker. I got it working. FYI your documentation states `databases_config` in the base.json

retrodaredevil, 2024-03-17:
> 1. The location "./config" does not work from within docker:

Yeah I need to fix that. I forgot you cannot use relative paths with `docker run`. You are able to use relative paths when using a docker compose file, I just forgot you can't for `docker run`.

> 2\. In my case I have both the DB and Solarthing within a Docker container. The DB connection is refused when connecting to "localhost" from within Docker. I had to use my host machine's IP address in the couchdb.json:

I will try to make the documentation more clear on this part. If you are running SolarThing and CouchDB from the same docker compose file, you are also able to refer to the database using the name of the service, for instance, let's say you have this:

```yaml
# ...
services:
  solarthing-main:
    # ...
  couchdb:
    # ...
```

You would be allowed to refer to `http://couchdb:5984` and you wouldn't have to include a `ports:` section unless you needed other things to access the database (which you most likely do anyway). Additionally, another docker related networking thing to keep in mind is that if you would prefer to not refer to the IP address of your device on your LAN network, you could instead refer to `http://172.17.0.1:5984`. The `172.17.0.1` IP address is fairly standard with most docker installs and it allows you to refer to your host machine. If you decided to refer to CouchDB in that way, you would have to have a `ports:` section in your docker compose file.

I mention all this mostly for completeness sake. If you have it working how you want it, it's not like you need to change anything. I hope to improve documentation on all that in the future, but to do that I have to figure out how to convey all the information above in a way that's digestible for newcomers.

> I got it working. FYI your documentation states `databases_config` in the base.json

Yeah I'll be sure to update the documentation.

Additionally, as a warning that I think is included in the documentation, be careful running any sort of database on a Raspberry Pi. Doing something that constantly writes to the Pi's SD card is very bad for the SD card, and will likely result in the SD card failing in less than a year.

Thanks for providing helpful error messages. All this will help me improve the documentation. I'm going to close this issue for now. Feel free to respond to this if you need more help or open another issue if you find something else wrong with the docs or SolarThing itself.

closed 2024-03-17 by retrodaredevil