wildmountainfarms/solarthing #212: Cant connect to CouchDB (despite initial config completing)

TheAlchemistFLT, 2024-04-22:
I have configured CouchDB (in Docker) and successfully created the Solarthing databases using the --couchdb-setup command as per the configuration instructions. However when I run Solarthing to upload the Rover data I get the following error messages:

```
SolarThing  | 2024-04-22 16:53:47.442 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - Using base configuration file: config/base.json
SolarThing  | 2024-04-22 16:53:47.819 [main] INFO  me.retrodaredevil.solarthing.program.RequestMain - Beginning request program
SolarThing  | 2024-04-22 16:53:47.820 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Analytics are ENABLED! (Note) For this SolarThing version, analytics are not sent. This will be changed in a future version. If you see this log message, a future version of SolarThing may send analytics data (if you decide to update).
SolarThing  | 2024-04-22 16:53:47.972 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Packets will be uploaded to database: **CouchDB http://admin@172.17.0.3:5984/**
SolarThing  | 2024-04-22 16:53:48.250 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Commands are disabled
SolarThing  | 2024-04-22 16:53:48.723 [main] DEBUG me.retrodaredevil.solarthing.io.ReloadableIOBundle - Successfully reloaded IOBundle

2024-04-22 16:53:50.454 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Sending Rover status to analytics. data=(0) RNG-CTRL-RVR40,40 uptime hours=0 (Note) For this SolarThing version, analytics are not sent. This will be changed in a future version. If you see this log message, a future version of SolarThing may send analytics data (if you decide to update).
SolarThing  | 2024-04-22 16:53:50.457 [main] DEBUG me.retrodaredevil.solarthing.program.RequestMain - Going to sleep for 3.286 seconds
SolarThing  | 2024-04-22 16:53:52.027 [pool-3-thread-1] ERROR me.retrodaredevil.solarthing.packets.handling.AsyncRetryingPacketHandler - Error while uploading packet collection. ID: 2024,4,22,16,53,50,0.7955947831015174 dateMillis: 1713804830423. Will retry using me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@5f9b4285

2024-04-22 16:53:56.810 [pool-2-thread-1] **DEBUG me.retrodaredevil.solarthing.packets.handling.PrintPacketHandleExceptionWrapper - Caught a common packet handle exception with message: We got a DbAccessException probably meaning we couldn't reach the database.**
SolarThing  | 2024-04-22 16:53:58.251 [pool-4-thread-1] DEBUG me.retrodaredevil.solarthing.packets.handling.AsyncRetryingPacketHandler - Going to try to reupload 1 packets using me.retrodaredevil.solarthing.couchdb.CouchDbPacketSaver@5f9b4285
SolarThing  | 2024-04-22 16:53:59.754 [pool-4-thread-1] ERROR me.retrodaredevil.solarthing.packets.handling.AsyncRetryingPacketHandler - Reuploaded 0 / 1 packets before getting exception
**SolarThing  | me.retrodaredevil.solarthing.packets.handling.CommonPacketHandleException: We got a DbAccessException probably meaning we couldn't reach the database.**

```
What am I doing wrong?

retrodaredevil, 2024-04-23:
It looks like you have both SolarThing and CouchDB running in Docker.

Are you running SolarThing and CouchDB on the same device? Do you have a single `docker-compose.yml` file or multiple? Could you post your docker compose file(s)?

This line sticks out to me:

```
SolarThing  | 2024-04-22 16:53:47.972 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Packets will be uploaded to database: CouchDB http://admin@172.17.0.3:5984/
```

It seems that you your URL configured as `http://172.17.0.3:5984`, correct? My guess is that you'll need to change that IP address to either a different IP address to correctly refer to CouchDB. I'll help you figure out what host you should change it to after I understand your docker compose structure.

TheAlchemistFLT, 2024-04-23:
Thanks - yes both SolarThing and CouchDB are running on Docker (CouchDB installed using Portainer).  When running the SolarThing container is using IP 172.18.0.2 and CouchDB is using 172.17.0.3.

What I dont understand is why the couchdb setup works to create all the databases and users - and then not when it is run. If there was a conflict with IP or installation/config then the setup wouldnt work either?

SolarThing docker-compose.yml is:

....
version: '3.7'

services:
  solarthing-main:
    image: 'ghcr.io/wildmountainfarms/solarthing:latest'
    container_name: SolarThing
    restart: 'unless-stopped'
    command: run --base config/base.json
#    group_add: # this is only necessary if you are using a user other than root
#      - dialout
    cap_add:
      - SYS_RAWIO
    devices:
      - '/dev/ttyUSB0:/dev/ttyUSB0'
    volumes:
      - './main/config:/app/config:ro'
      - './main/logs:/app/logs'
...

retrodaredevil, 2024-04-23:
I'm not familiar with Portainer, but my understanding is that unless you explicitly put two containers defined in two separate compose files on the same network, they won't be able to communicate with each other. If you were to get both containers on the same network using docker networks (this is done by default if both containers are defined in a single compose file), then you could change the URL to `http://couchdb:5984` because docker lets you refer to containers on the same network via their hostname.

With the way you have it set up now, you should change the URL to `http://172.17.0.1:5984`. The reason for this is that 172.17.0.1 is a special IP, that refers to the docker host. This means that 172.17.0.1 is referring to the actual host of the machine, and because I assume you have a ports section in your CouchDB's docker compose file, you have exposed the 5984 port on the host machine of the computer, so other docker containers can access it as long as they refer to the docker host.

This is not specified in the documentation because most people don't run the SolarThing uploader and CouchDB on the same machine, and if they do, they usually have them both defined in a single compose file.

TheAlchemistFLT, 2024-04-23:
Thanks - made the change to 172.17.0.1 and its all working like a dream... tried the using the couchdb alias before but noticed that it url field is parsed so it didnt work as I named my container CouchDB... haha that will teach me!!

Your help much appreciated - now to look into data visualisation!

retrodaredevil, 2024-04-23:
Awesome! Let me know if you run into any more issues. I updated the docs recently to describe how to set up Wild GraphQL Data Source with Grafana. So if you look into that it's all gonna be very new somewhat untested stuff. 

I am also going to release a dashboard soon for Renogy Rover devices so people don't have to make a dashboard if they don't want to fiddle with it.

TheAlchemistFLT, 2024-04-24:
Hi - got solrthing-server and Grafana installed and the graphql datasource (all validated) - however when I start the server dashboard I dont get any data - it sticks on "loading data". I checked the IP:8080/status and I get "COMPLETE" against all DBs other than  StatusDB which gives "BAD_PERMISSIONS"...

retrodaredevil, 2024-04-24:
My best guess is probably that the communication between SolarThing server and CouchDB is not functioning correctly.

Let's first try to see if those permissions are an issue. For reference, here's what the permissions tab of my status database looks like (the database named `solarthing`
[image]

The important thing here is that there are no members defined so the database can be public. This is generally what I recommend because the SolarThing setup will prevent unauthorized writes to the database, but will allow anyone to read it when it is configured without any members. You can get to this web interface that I have shown by navigating to `http://<your IP>:5984>/_utils`. Login, then navigate to the solarthing database, then the permissions tab of it.

However, if your `couchdb.json` used by SolarThing server defines a username and password, you can ignore this last paragraph about permissions.

The thing that I actually suspect is the problem is that SolarThing server is not communicating with your database correctly. First, I want you to post or at least look at the log files generated by SolarThing Server. The way I do this is to cd to the directory that is housing my docker-compose.yml file, and then run `docker compose logs solarthing-server` where solarthing-server is the name of the service as defined under the `services:` directive in your YAML. Alternatively, to view the logs, you can do `docker logs <container name>` where the container name is the name of the container. I would also guess that portainer has a way for you to view logs.

Once you are viewing the logs, I suspect that you have a bunch of log messages that start with `Got database exception`. The text that comes after that is going to help us understand if you're getting another connection error.

TheAlchemistFLT, 2024-04-24:
Hi - checked the permissions and there was something in the members fields so I removed those and everything seems to be working now. Also got Solarthing Android working for good measure! thanks for all your help...

now for Grafana mastery!

closed 2024-04-24 by TheAlchemistFLT

retrodaredevil, 2024-04-24:
Glad you got it working. I'm planning to create a repository to fill with pre-made SolarThing dashboards soon. If you want to try out one of the ones I use, you can import this dashboard: https://github.com/wildmountainfarms/wild-graphql-datasource/blob/main/provisioned-dashboards/solarthing-dashboard.json

That's the only dashboard I have public right now, I'm just not advertising that because I'm going to put it in a different repository sometime soon.