wildmountainfarms/solarthing #214: Running MQTT and CouchDB uploader in parallel?

TheAlchemistFLT, 2024-04-25:
Just wondering if it is possible to run both the MQTT and CouchDB uploaders in parallel? when I added a second database config to the base.json config file it seemed to parse only the second one (in this case MQTT) and ignored the first (CouchDB).
My use-case is posting the messages and reading in Home Assistant... while also saving to the database for prosperity/visualisation in Grafana (ps. your shared dashboard worked great).

TheAlchemistFLT, 2024-04-25:
Yeah - if the MQTT works then i dont mind, gives me something else to play with haha.

my base.json file is:

...

  "type": "request",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "database_config": {
    "databases": [
        {
        "external": "config/couchdb.json",
        "external": "config/mqtt.json"
       }
    ]
  },
  "request": [
    {
      "type": "modbus",
      "io": "config/rover_serial.json",
      "devices": {
        "16": {
          "type": "rover"
        }
      }
    }
  ]
}

...

retrodaredevil, 2024-04-25:
Change it to this:

```
{
  "type": "request",
  "source": "default",
  "fragment": 2,
  "unique": 30,
  "database_config": {
    "databases": [
       {
         "external": "config/couchdb.json",
       },
       {
         "external": "config/mqtt.json"
       }
    ]
  },
  "request": [
    {
      "type": "modbus",
      "io": "config/rover_serial.json",
      "devices": {
        "16": {
          "type": "rover"
        }
      }
    }
  ]
}

```

In your base.json you defined `external` twice in the same object, so only one of them was read.

TheAlchemistFLT, 2024-04-25:
woohoo! after removing the spurious "," after the initial couchdb.json reference it is now working posting to CouchDB and MQTT broker in parallel! :)

cat /home/SolarThing/main/logs/log_summary.log
2024-04-25 15:57:52.309 [main] INFO  me.retrodaredevil.solarthing.program.SolarMain - [LOG] Beginning main. Jar: solarthing.jar Last Modified: 2024-02-25T23:27:42Z Java version: 19.0.2
2024-04-25 15:57:53.270 [main] INFO  me.retrodaredevil.solarthing.program.RequestMain - Beginning request program
2024-04-25 15:57:53.272 [main] INFO  me.retrodaredevil.solarthing.analytics.AnalyticsManager - Analytics are ENABLED! (Note) For this SolarThing version, analytics are not sent. This will be changed in a future version. If you see this log message, a future version of SolarThing may send analytics data (if you decide to update).
2024-04-25 15:57:53.591 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Packets will be uploaded to database: CouchDB http://admin@172.17.0.1:5984/
2024-04-25 15:57:54.072 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Packets will be uploaded to database: MQTT tcp://172.17.0.1:1883
2024-04-25 15:57:54.133 [main] INFO  me.retrodaredevil.solarthing.program.PacketHandlerInit - Commands are disabled
admin@37srpi:~ $

TheAlchemistFLT, 2024-04-25:
ps. possibly worth noting in both cases that I had to use the "172.17.0.1:port" instead of "localhost:port" to get either to work... and if one fails, both fail...

retrodaredevil, 2024-04-25:
Glad you got it working. If pasting the configuration files fail, yeah SolarThing will crash. However, if you ever run into a situation where an incorrect IP for one of the databases causes the other uploading to the database to fail, I'd like to see the log messages because that isn't supposed to happen.

closed 2024-04-26 by TheAlchemistFLT