wildmountainfarms/solarthing #215: Grafana Datasource - fieldname translation?

TheAlchemistFLT, 2024-04-26:
Hello again... looking at the Grafana dashboards it looks like the Grafana GraphQL datasource transposes the names of each of the datafields as stored in couchDB or published on the MQTT broker. Is there anywhere a list of the relevant datafields (I cant seem to find anywhere in Grafana to show the full list of valid fields (and also it wouldnt provide the mapping to couchdb etc.)

Being a newbie at this - I may of course just be being thick and missing something quite basic...

retrodaredevil, 2024-04-26:
This has been a problem until recently when I released Wild GraphQL Data Source. Within the query editor, you should be able to view the documentation of the GraphQL API. I'll show you some screenshots of how to navigate it:

First, click the documentation button:

[image]

Within that, look for the root query:

[image]

Now you are here after clicking on "Query":

[image]

Search for or scroll to `roverStatus`:

[image]

Then click on PacketNode_RoverStatusPacket, then click on RoverStatusPacket and you should get here:

[image]

From this, you have many different fields to choose from. For instance, if you wanted to graph the battery temperature Celsius and the controller temperature Celsius, you could do this:

```graphql
query ($from: Long!, $to: Long!) {
  data: queryStatus(
    from: $from
    to: $to
  ) {
    roverStatus {
      dateMillis
      packet {
        batteryTemperatureFahrenheit
        controllerTemperatureFahrenheit
        identityInfo {
          displayName
        }
      }
    }
  }
}
```

If you pay close enough attention to the documentation, you'll notice that `packet` corresponds with everything documented in RoverStatusPacket. I frequently will place my cursor inside that block and press CTRL+Space. That will bring up a list of things that can be autocompleted for.

Alternatively, you can also hover over something like "roverStatus" or "packet" and it should bring up a dialog that allows you to jump straight to the documentation of that thing.

There is definitely a lack of documentation, and I should probably create a more formal page describing what I just showed you here, so I appreciate the feedback. Additionally, I made a repo yesterday which I plan to maintain to house dashboards that anyone can use: https://github.com/wildmountainfarms/solarthing-dashboards/

TheAlchemistFLT, 2024-04-27:
Thanks - very, very helpful.... and now I'm starting to get the hang of it... I've tweaked a few things and my own dashboard is starting to come together!

closed 2024-04-27 by TheAlchemistFLT