wildmountainfarms/solarthing #223: docker and nc netcat

riker65, 2024-12-15:
Hi 
wondering how to invoke the changes using netcat in docker environment?

nc localhost 5051

does not do anything

in base.json

```

..
  "request": [
	    { "type": "cpu-temp", "processors": 1 },
	   {
      "type": "modbus",
      "io": "config/usb_id.json",
      "devices": {
        "17": 	{
          "type": "rover",
          "server": { "port": 5051 }
..
```

retrodaredevil, 2024-12-15:
When using any piece of software with docker, you have to manually expose the ports you want accessible in your docker compose file with a ports section:

``` 
services:
  solarthing:
    # ...
    ports
      - "127.0.0.1:5051:5051"
```

Notice that I use the `127.0.0.1` to bind only to localhost, so that other devices on your network cannot access it. This is a security feature that is automatically set up for the native install, but requires that manual configuration for the docker install. 

Now, once you do that, it's not going to work yet. This is because of that security feature the native install has, that doesn't work well with the docker install and that port mapping. 

We need to tell the SolarThing docker container to listen on all interfaces for 5051. Instead of `{ "port": 5051 }`, we need to change it to `{ "port": 5051, "bind": "0.0.0.0" }`. 

Once you do that and restart/recreate the container, it should work just like a native install.

retrodaredevil, 2024-12-15:
Alternatively, you can ignore everything I said in the last comment and run this whenever you need to access the configuration server:

```shell 
# Instead of this 
nc localhost 5051
# do this 
docker compose exec solarthing-service-name-here nc localhost 5051
```

The reason that will work without any of the configuration I mentioned before is that you're running the netcat command inside the container, so no port mapping is needed and you're accessing it through localhost, so the bind address doesn't need to change. 

I'm not at my computer now and cannot confirm that the `nc` command is actually available through the `docker compose exec` command, but honestly the `docker compose exec` command is probably the easier way of accessing the configuration server. I'll try to make sure that's possible in a future version of SolarThing. 

Thanks for bringing this to my attention. This clearly needs some updated documentation to accommodate docker installs.

riker65, 2024-12-15:
Thanks for the fast response 
Will check

riker65, 2024-12-15:
looks like I need more deails.
unclear for me - docker newbee - which service is running.

sudo docker compose exec solarthing-request nc localhost 5051
-> service "solarthing-request" is not running

docker ps-all shows:
35f6c7b5e2ef   ghcr.io/wildmountainfarms/solarthing:latest   "java -XX:ErrorFile=…"   27 hours ago   Up 3 minutes             solarthing-main

following the manual, I used the standard compose

`services:
  main:
    image: 'ghcr.io/wildmountainfarms/solarthing:latest'
    container_name: solarthing-main
    restart: 'unless-stopped'
    command: run --base config/base.json
#    group_add: # this is only necessary if you are using a user other than root
#      - dialout
#    cap_add:
#      - SYS_RAWIO
#    devices:
#      - '/dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0:/dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0'
    volumes:
      - './main/config:/app/config:ro'
      - './main/logs:/app/logs'
    ports:
      - "127.0.0.1:5051:5051" `

do I need to start solarting-service here?
 I thought in base.json 

type request 
defines the service and rover device 
`
{
  "type": "request",
  "source": "rpi3solar",
  "fragment": 3,
  "unique": 30,
  "analytics_enabled": false,
  "databases": [ 
	"config/mqtt_lh_request.json"
	],
  "request": [
	    { "type": "cpu-temp", "processors": 1 },
	   {
      "type": "modbus",
      "io": "config/usb_id.json",
      "devices": {
        "17": 	{
          "type": "rover",
          "server": { "port": 5051 }
				}
			}
		}
	]
	}`

thanks a lot Thomas

riker65, 2024-12-15:
using 
**sudo docker compose exec main  nc localhost 5051**
error: 
OCI runtime exec failed: exec failed: unable to start container process: exec: "nc": executable file not found in $PATH: unknown

retrodaredevil, 2024-12-15:
Yeah it looks like the `docker compose exec` isn't working to run the command `nc` because that command is not installed inside the container. I will need to update the docker container myself so that `nc` is supported out of the box. 

In the meantime, you can refer to my first comment if you want to be able to access the configuration server for a docker install.

retrodaredevil, 2024-12-18:
ffdbfed87f9a5afdf74400d7514d0c53ae0b284e addresses this. I will release SolarThing v2025.1.1 at some point, and that docker image will allow you to be able to access `nc` with `docker compose exec service-name nc`.

For a technical reasons ([jitpack is down today](https://statusgator.com/services/jitpack)), I cannot make a release yet.

retrodaredevil, 2024-12-22:
Feel free to try out `docker compose exec main nc localhost 5051` with the latest Docker image.

[v2025.1.1](https://github.com/wildmountainfarms/solarthing/releases/tag/v2025.1.1) fixes this.

closed 2024-12-22 by retrodaredevil